Koenkk/zigbee2mqtt #22485: [New device support]: Generic Zigbee 3.0 "Mini Smart Switch"

Shlokp, 2024-05-06:
### Link

https://www.aliexpress.com/item/1005006379052047.html?spm=a2g0o.order_list.order_list_main.4.1d331802SE1mgA

### Database entry

{"id":7,"type":"Unknown","ieeeAddr":"0x70b3d52b600aa863","nwkAddr":36884,"epList":[1],"endpoints":{"1":{"epId":1,"inClusterList":[],"outClusterList":[],"clusters":{"genOnOff":{"attributes":{"onOff":1}},"manuSpecificTuya_3":{"attributes":{"powerOnBehavior":2,"switchType":0}},"genBasic":{"attributes":{"appVersion":100}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":100,"interviewCompleted":false,"meta":{},"lastSeen":1714999281469}

### Comments

This device appears to be an no-brand version of Aubess TMZ02 however I am unable to get it to be identified.

On trying to pair the device, the following error comes up "interview failed because could not get node descriptor"
Then the device shows up 

[image]

Ive had a read of similar issues for the official Aubess Mini Smart Switch (https://github.com/Koenkk/zigbee2mqtt/issues/13929), however in my instance, there is no model number or manufacturer. 

I have tried adding a custom external definition but I cannot get it to assign.

I appreciate any help provided.

### External definition

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
    fingerprint: [{ieee_address: '0x70b3d52b600aa863'}],   // WARNING! Change this ieee_address for yours ieee_address!!!!
    description: 'Wall switch module',
    toZigbee: extend.switch().toZigbee.concat([tz.moes_power_on_behavior, tz.tuya_switch_type]),
    fromZigbee: extend.switch().fromZigbee.concat([fz.moes_power_on_behavior, fz.tuya_switch_type]),
    exposes: extend.switch().exposes.concat([e.power_on_behavior(), e.switch_type_2()]),
    configure: async (device, coordinatorEndpoint, logger) => {
        await reporting.bind(device.getEndpoint(1), coordinatorEndpoint, ['genOnOff']);},
};

module.exports = definition;
```

Shlokp, 2024-05-14:
Managed to get it working!

For those with similar issues, despite the manufactuer name and model not showing, when using the Dev console, it did indeed verify the product - "modelId":"TS0001","hwVersion":2,"manufacturerName":"_TZ3000_fdxihpp7"

Managed to fix it by making a custom external definition with the following code - https://gist.github.com/Koenkk/550c7c2b486173a8c019bc8ba764fad8

Even though this version doesnt support the power monitoring, the rest of the settings work as expected

closed 2024-12-28 by github-actions[bot]

mahnunchik, 2026-09-19:
Related https://github.com/romasku/tuya-zigbee-switch/issues/517