Koenkk/zigbee2mqtt #22486: SNZB-06P Failed to configure on Z2M 1.37.0

idogrf, 2024-05-06:
### What happened?

Trying to configure the SNZB-06P presence sensor on Zigbee2MQTT version 1.37.0 fails with the following error - 
```
attempt 2 (TypeError: func is not a function at Object.configure (/app/node_modules/zigbee-herdsman-converters/src/index.ts:160:27) at Configure.configure (/app/lib/extension/configure.ts:121:13) at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:174:17))
```

Reverting to Z2M 1.35.1 seems to solve the issue (haven't tried 1.36.0)

### How to reproduce it (minimal and precise)

Try to pair the device with Z2M

### Zigbee2MQTT version

1.37.0

### Adapter firmware version

1.0.6

### Adapter

SNZB-06P

### Setup

Dockerized Z2M on a raspberry pi

darkfuncat, 2024-05-07:
Hi, new to zigbee.
I have just one coordinator sonoff dongle-e, recently flashed (>7.4) and have the same problem than you.

darkfuncat, 2024-05-07:
Update : just tried 1.36.1 and there is no bug.

Koenkk, 2024-05-08:
Could it be that you are using an external converter for this device? (check your `configuration.yaml`)

idogrf, 2024-05-09:
I don't think I'm using an external converted either.
This is my configuration.yaml - 

```
homeassistant: true
permit_join: false
mqtt:
  base_topic: zigbee2mqtt
  server: mqtt://192.168.XXX:XXX
serial:
  port: /dev/ttyUSB0
advanced:
  network_key:
    - ...
  homeassistant_legacy_entity_attributes: false
  legacy_api: false
  log_level: debug
device_options:
  legacy: false
devices:
  '0x00158d0007005864':
    friendly_name: contact_sensor_1
  '0x00158d000775300c':
    friendly_name: motion_sensor
    occupancy_timeout: 90
    no_occupancy_since: []
  '0x04cf8cdf3c7c6b7a':
    friendly_name: single_switch
  '0x0ceff6fffe3b0481':
    friendly_name: presence_sensor_1
frontend:
  port: 8011
availability: false
```

where device `0x0ceff6fffe3b0481` is the problematic one

Koenkk, 2024-05-10:
Can you provide the command through which you run the Docker container?

jstaro, 2024-05-12:
I am also seeing this. No external converters, pretty vanilla install via the HA addon v1.37.1-1.

aslabsalbeh, 2024-05-29:
I am also seeing this. No external converters, pretty vanilla install via the HA addon v1.37.1-1. Coordinator is SNZB-06P

irata-coder, 2024-05-29:
Same here, this can be recreated on a fresh install. 

I am using the new ember adapter.  Zigbee2MQTT version 1.37.1.

>[2024-05-29 14:56:17] error:    z2m: Failed to configure 'Living Room Occupancy', attempt 3 (TypeError: func is not a function
    at Object.configure (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/src/index.ts:172:27)
    at processTicksAndRejections (node:internal/process/task_queues:95:5)
    at Configure.configure (/opt/zigbee2mqtt/lib/extension/configure.ts:121:13)
    at Configure.onMQTTMessage (/opt/zigbee2mqtt/lib/extension/configure.ts:55:21)
    at EventEmitter.wrappedCallback (/opt/zigbee2mqtt/lib/eventBus.ts:174:17))
[2024-05-29 14:56:17] debug:    z2m: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/configure', payload '{"data":{"id":"Living Room Occupancy"},"error":"Failed to configure (func is not a function)","status":"error","transaction":"ee5ye-3"}'

Koenkk, 2024-05-30:
This bug will be fixed in the next release (Saturday)

closed 2024-05-30 by Koenkk

Erotemic, 2024-12-03:
I'm not sure if I'm getting this issue or if it is slightly different. 

I'm setting up a new Z2M system. I have a "Zigbee 3.0 USB Dongle Plus ZBDongle-E" plugged into a desktop and I'm running  zigbee2mqtt and mqtt via docker compose. When I press the pairing button on the occupancy sensor, it seems to pair, but it does throw an error when I pair it and if I restart the docker containers. 

```
zigbee2mqtt  | [2024-12-03 11:14:16] error: 	z2m: Failed to configure 'OccupancySensor1', attempt 1 (Error: ZCL command 0x0ceff6fffe16c576/1 msOccupancySensing.configReport([{"minimumReportInterval":10,"maximumReportInterval":60,"reportableChange":0,"attribute":"occupancy"}], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed ({"target":41682,"apsFrame":{"profileId":260,"clusterId":1030,"sourceEndpoint":1,"destinationEndpoint":1,"options":4416,"groupId":0,"sequence":150},"zclSequence":3,"commandIdentifier":7} timed out after 10000ms)
zigbee2mqtt  |     at Object.start (/app/node_modules/zigbee-herdsman/src/adapter/ember/adapter/oneWaitress.ts:195:23)
zigbee2mqtt  |     at EmberOneWaitress.startWaitingFor (/app/node_modules/zigbee-herdsman/src/adapter/ember/adapter/oneWaitress.ts:219:50)
zigbee2mqtt  |     at /app/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:2131:55
zigbee2mqtt  |     at processTicksAndRejections (node:internal/process/task_queues:95:5)
zigbee2mqtt  |     at runNextTicks (node:internal/process/task_queues:64:3)
zigbee2mqtt  |     at processImmediate (node:internal/timers:447:9)
zigbee2mqtt  |     at Queue.execute (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:36:20)
zigbee2mqtt  |     at EmberAdapter.sendZclFrameToEndpoint (/app/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:2070:16)
zigbee2mqtt  |     at Request.send (/app/node_modules/zigbee-herdsman/src/controller/helpers/request.ts:82:20)
zigbee2mqtt  |     at Endpoint.sendRequest (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:325:20))
```

Otherwise it does seem to be working. So far I'm just ignoring the issue, but I wanted to report it in case it is causing some other problem. I'm happy to provide any more information, make a new issue if this is sufficiently different, or go through any other debugging steps.

jonbulica99, 2024-12-26:
I am facing the exact same problem as @Erotemic:

```
[2024-12-26 02:46:07] error: 	z2m: Failed to configure 'Human Presence Sensor', attempt 3 (Error: ZCL command 0x0ceff6fffe5977e7/1 msOccupancySensing.configReport([{"minimumReportInterval":10,"maximumReportInterval":60,"reportableChange":0,"attribute":"occupancy"}], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed ({"target":20609,"apsFrame":{"profileId":260,"clusterId":1030,"sourceEndpoint":1,"destinationEndpoint":1,"options":4416,"groupId":0,"sequence":253},"zclSequence":10,"commandIdentifier":7} timed out after 10000ms)
    at Object.start (/app/node_modules/zigbee-herdsman/src/adapter/ember/adapter/oneWaitress.ts:195:23)
    at EmberOneWaitress.startWaitingFor (/app/node_modules/zigbee-herdsman/src/adapter/ember/adapter/oneWaitress.ts:219:50)
    at /app/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:2131:55
    at Queue.execute (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:36:20)
    at EmberAdapter.sendZclFrameToEndpoint (/app/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:2070:16)
    at Request.send (/app/node_modules/zigbee-herdsman/src/controller/helpers/request.ts:82:20)
    at Endpoint.sendRequest (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:325:20)
    at Endpoint.zclCommand (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:951:28)
    at Endpoint.configureReporting (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:688:9)
    at setupAttributes (/app/node_modules/zigbee-herdsman-converters/src/lib/modernExtend.ts:162:17))
```

tracetechnical, 2025-01-14:
Same issue here