Koenkk/zigbee2mqtt #22491: latest update no longer supports a device

homeyjoenka, 2024-05-06:
### What happened?

Iluminize  Model | 511.10 is no longer supported after last update of zigbee2mqtt it worked for 2years how can i fix this?

### Zigbee2MQTT version

1.36.1

### Adapter firmware version

7.3.1.0 build 176

### Adapter

_SONOFF_Zigbee_3.0_USB_Dongle_Plus_V2

### Setup

os

Koenkk, 2024-05-06:
Could you provide the data/database.db entry of this device?

benjaminBoehme, 2024-05-07:
I'm having the same issue with this device since updating to 1.37.0. "Support status" field in device details is now "Null", the db entry is the following:

<code>{"id":11,"type":"Router","ieeeAddr":"0x000d6f0010682104","nwkAddr":65249,"manufId":4107,"manufName":"Sunricher","powerSource":"Mains (single phase)","modelId":"DIM Lighting","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":257,"inClusterList":[0,3,4,5,6,8,2821,4096],"outClusterList":[25,4096],"clusters":{"genBasic":{"attributes":{"appVersion":0,"stackVersion":0,"hwVersion":0}},"genOnOff":{"attributes":{"onOff":1}},"genLevelCtrl":{"attributes":{"currentLevel":254}}},"binds":[],"configuredReportings":[],"meta":{"scenes":{"1_20":{"name":"Hell-warm","state":{"state":"ON","brightness":254,"color_mode":"color_temp","color":{"x":0.4599,"y":0.4106},"color_temp":370}},"2_20":{"name":"Hell-neutral","state":{"state":"ON","brightness":254,"color_mode":"color_temp","color":{"x":0.4599,"y":0.4106},"color_temp":370}},"3_20":{"name":"Sonnenuntergang","state":{"state":"ON","brightness":173,"color_mode":"color_temp","color":{"x":0.4599,"y":0.4106},"color_temp":370}},"4_20":{"name":"Aufwachen","state":{"state":"ON","brightness":254,"color_mode":"color_temp","color":{"x":0.4599,"y":0.4106},"color_temp":370}},"5_20":{"name":"gedimmt","state":{"state":"ON","brightness":87,"color_mode":"color_temp","color":{"x":0.4599,"y":0.4106},"color_temp":370}},"1_23":{"name":"Hell","state":{"state":"ON","brightness":254,"color_mode":"color_temp","color":{"x":0.4599,"y":0.4106},"color_temp":370}}}}},"242":{"profId":41440,"epId":242,"devId":102,"inClusterList":[33],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":0,"stackVersion":0,"hwVersion":0,"dateCode":"NULL","swBuildId":"2.4.1_r30","zclVersion":3,"interviewCompleted":true,"meta":{"configured":1324213189},"lastSeen":1715089240881}
</code>

Reverting back to 1.36.1 immediately solves the issue for me.

homeyjoenka, 2024-05-07:
this?

{"id":1,"type":"Coordinator","ieeeAddr":"0xe0798dfffe9cb41a","nwkAddr":0,"manufId":0,"epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":48879,"inClusterList":[0,3,6,10,25,26,768],"outClusterList":[0,3,4,5,6,8,32,768,1024,1026,1029,1030,1280,2817,2819,2820,1794,4096,64513,64514],"clusters":{},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0xe0798dfffe9cb41a","endpointID":1}],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"interviewCompleted":true,"meta":{},"lastSeen":1715007116859}
{"id":2,"type":"Router","ieeeAddr":"0xa4c138577b9c6bce","nwkAddr":63586,"manufId":4417,"manufName":"_TZ3000_amdymr7l","powerSource":"Mains (single phase)","modelId":"TS011F","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":266,"inClusterList":[3,4,5,6,1794,2820,57344,57345,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"�iG-\u0012","65506":54,"65508":0,"65534":0,"modelId":"TS011F","manufacturerName":"_TZ3000_amdymr7l","powerSource":1,"zclVersion":3,"appVersion":192,"stackVersion":0,"hwVersion":1,"dateCode":""}},"haElectricalMeasurement":{"attributes":{"acCurrentDivisor":1000,"acCurrentMultiplier":1,"rmsVoltage":239,"rmsCurrent":0,"activePower":0}},"seMetering":{"attributes":{"divisor":100,"multiplier":1,"currentSummDelivered":[0,20062],"status":0}},"genOnOff":{"attributes":{"32768":0,"onOff":0,"onTime":0,"offWaitTime":0,"tuyaBacklightMode":1,"moesStartUpOnOff":1,"tuyaBacklightSwitch":1}},"manuSpecificTuya_3":{"attributes":{"53248":1,"53249":0,"53250":2474305,"53251":143,"53252":10522,"53253":1454,"powerOnBehavior":1,"switchType":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0xe0798dfffe9cb41a","endpointID":1},{"cluster":2820,"type":"endpoint","deviceIeeeAddress":"0xe0798dfffe9cb41a","endpointID":1},{"cluster":1794,"type":"endpoint","deviceIeeeAddress":"0xe0798dfffe9cb41a","endpointID":1}],"configuredReportings":[{"cluster":2820,"attrId":1285,"minRepIntval":5,"maxRepIntval":3600,"repChange":5,"manufacturerCode":null},{"cluster":2820,"attrId":1288,"minRepIntval":5,"maxRepIntval":3600,"repChange":50,"manufacturerCode":null},{"cluster":2820,"attrId":1291,"minRepIntval":5,"maxRepIntval":3600,"repChange":10,"manufacturerCode":null},{"cluster":1794,"attrId":0,"minRepIntval":5,"maxRepIntval":3600,"repChange":[1,1],"manufacturerCode":null}],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":192,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":1324213189},"lastSeen":1715095961914}
{"id":3,"type":"Router","ieeeAddr":"0xa4c13884ad9cdbcf","nwkAddr":9239,"manufId":4417,"manufName":"_TZ3000_gjnozsaz","powerSource":"Mains (single phase)","modelId":"TS011F","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":266,"inClusterList":[3,4,5,6,1794,2820,57344,57345,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"�iG-\u0012","65506":54,"65508":0,"65534":0,"modelId":"TS011F","manufacturerName":"_TZ3000_gjnozsaz","powerSource":1,"zclVersion":3,"appVersion":192,"stackVersion":0,"hwVersion":1,"dateCode":""}},"haElectricalMeasurement":{"attributes":{"acCurrentDivisor":1000,"acCurrentMultiplier":1,"rmsVoltage":239,"activePower":0,"rmsCurrent":0}},"seMetering":{"attributes":{"divisor":100,"multiplier":1,"currentSummDelivered":[0,2225]}},"genOnOff":{"attributes":{"32768":0,"onOff":1,"onTime":0,"offWaitTime":0,"tuyaBacklightMode":2,"moesStartUpOnOff":2,"tuyaBacklightSwitch":1}},"manuSpecificTuya_3":{"attributes":{"53248":0,"53249":0,"53250":0,"53251":0,"53252":0,"53253":0,"powerOnBehavior":2,"switchType":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0xe0798dfffe9cb41a","endpointID":1},{"cluster":2820,"type":"endpoint","deviceIeeeAddress":"0xe0798dfffe9cb41a","endpointID":1},{"cluster":1794,"type":"endpoint","deviceIeeeAddress":"0xe0798dfffe9cb41a","endpointID":1}],"configuredReportings":[{"cluster":2820,"attrId":1285,"minRepIntval":5,"maxRepIntval":3600,"repChange":5,"manufacturerCode":null},{"cluster":2820,"attrId":1288,"minRepIntval":5,"maxRepIntval":3600,"repChange":50,"manufacturerCode":null},{"cluster":2820,"attrId":1291,"minRepIntval":5,"maxRepIntval":3600,"repChange":10,"manufacturerCode":null},{"cluster":1794,"attrId":0,"minRepIntval":5,"maxRepIntval":3600,"repChange":[1,1],"manufacturerCode":null}],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":192,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":1324213189},"lastSeen":1715095964464}
{"id":4,"type":"Router","ieeeAddr":"0xa4c138c439c39204","nwkAddr":20216,"manufId":4417,"manufName":"_TZ3000_gjnozsaz","powerSource":"Mains (single phase)","modelId":"TS011F","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":266,"inClusterList":[3,4,5,6,1794,2820,57344,57345,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"�iG-\u0012","65506":54,"65508":0,"65534":0,"modelId":"TS011F","manufacturerName":"_TZ3000_gjnozsaz","powerSource":1,"zclVersion":3,"appVersion":192,"stackVersion":0,"hwVersion":1,"dateCode":""}},"haElectricalMeas [... cut ...] RepIntval":5,"maxRepIntval":3600,"repChange":10,"manufacturerCode":null},{"cluster":1794,"attrId":0,"minRepIntval":5,"maxRepIntval":3600,"repChange":[1,1],"manufacturerCode":null}],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":72,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":-892627439},"lastSeen":1715095802946}

homeyjoenka, 2024-05-07:
I hope I understood you correctly is the above what you need?

homeyjoenka, 2024-05-07:
[image]
 this is the product

Koenkk, 2024-05-07:
Fixed in https://github.com/Koenkk/zigbee2mqtt/issues/22468

closed 2024-05-07 by Koenkk