Koenkk/zigbee2mqtt #22492: problem wih ember driver and joining develco devices

mamrai1, 2024-05-06:
### What happened?

can'y join frient heat alarm with ember driver. Tried directly on coordinator and multiple routers.

### What did you expect to happen?

joining works with ezsp driver

### Zigbee2MQTT version

1.37.0-dev commit: 358e2ee

### Adapter firmware version

7.4.3.0 build 0

### Adapter

Ezsp or ember (sonoff plus e)

### Setup

z2n in docker 

### Debug log

[2024-05-06 07:15:25] error: 	z2m: Failed to configure 'Kitchen Smoke Sensor', attempt 1 (Error: Bind 0x0015bc00340008dd/35 ssIasZone from '0xe0798dfffe7d7366/1' failed ([ZDO] Failed response by NCP for "42235" cluster "32801" with status=ZDP_TABLE_FULL.)
    at EmberOneWaitress.resolveZDO (/app/node_modules/zigbee-herdsman/src/adapter/ember/adapter/oneWaitress.ts:138:35)
    at EmberAdapter.onZDOResponse (/app/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:573:26)
    at Ezsp.emit (node:events:517:28)
    at Ezsp.ezspIncomingMessageHandler (/app/node_modules/zigbee-herdsman/src/adapter/ember/ezsp/ezsp.ts:4524:22)
    at Ezsp.callbackDispatch (/app/node_modules/zigbee-herdsman/src/adapter/ember/ezsp/ezsp.ts:823:18)
    at Ezsp.tick (/app/node_modules/zigbee-herdsman/src/adapter/ember/ezsp/ezsp.ts:449:22)
    at listOnTimeout (node:internal/timers:569:17)
    at processTimers (node:internal/timers:512:7))

[2024-05-06 07:20:28] error: 	z2m: Failed to configure 'Kitchen Smoke Sensor', attempt 3 (Error: Bind 0x0015bc00340008dd/35 ssIasZone from '0xe0798dfffe7d7366/1' failed ([ZDO] Failed response by NCP for "42235" cluster "32801" with status=ZDP_TABLE_FULL.)
    at EmberOneWaitress.resolveZDO (/app/node_modules/zigbee-herdsman/src/adapter/ember/adapter/oneWaitress.ts:138:35)
    at EmberAdapter.onZDOResponse (/app/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:573:26)
    at Ezsp.emit (node:events:517:28)
    at Ezsp.ezspIncomingMessageHandler (/app/node_modules/zigbee-herdsman/src/adapter/ember/ezsp/ezsp.ts:4524:22)
    at Ezsp.callbackDispatch (/app/node_modules/zigbee-herdsman/src/adapter/ember/ezsp/ezsp.ts:823:18)
    at Ezsp.tick (/app/node_modules/zigbee-herdsman/src/adapter/ember/ezsp/ezsp.ts:449:22)
    at listOnTimeout (node:internal/timers:569:17)
    at processTimers (node:internal/timers:512:7))
[2024-05-06 18:02:57] error: 	zh:controller: Interview failed for '0x0015bc00340008dd with error 'Error: {"target":42235,"apsFrame":{"profileId":0,"clusterId":4,"sourceEndpoint":0,"destinationEndpoint":0,"options":4416,"groupId":0,"sequence":160},"responseClusterId":32772} timed out after 15000ms'
[2024-05-06 18:02:57] error: 	z2m: Failed to interview '0x0015bc00340008dd', device has not successfully been paired
[2024-05-06 18:03:44] error: 	z2m: Failed to configure '0x0015bc00340008dd', attempt 4 (Error: Bind 0x0015bc00340008dd/35 genPowerCfg from '0xe0798dfffe7d7366/1' failed (Delivery failed for {"profileId":0,"clusterId":33,"sourceEndpoint":0,"destinationEndpoint":0,"options":4416,"groupId":0,"sequence":178})
    at EmberOneWaitress.deliveryFailedFor (/app/node_modules/zigbee-herdsman/src/adapter/ember/adapter/oneWaitress.ts:96:31)
    at EmberAdapter.onMessageSentDeliveryFailed (/app/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:558:30)
    at Ezsp.emit (node:events:517:28)
    at Ezsp.ezspMessageSentHandler (/app/node_modules/zigbee-herdsman/src/adapter/ember/ezsp/ezsp.ts:3958:18)
    at Ezsp.callbackDispatch (/app/node_modules/zigbee-herdsman/src/adapter/ember/ezsp/ezsp.ts:795:18)
    at Ezsp.tick (/app/node_modules/zigbee-herdsman/src/adapter/ember/ezsp/ezsp.ts:449:22)
    at listOnTimeout (node:internal/timers:569:17)
    at processTimers (node:internal/timers:512:7))
[2024-05-06 18:03:47] warning: 	zh:ember:ezsp: Received network/route error ROUTE_ERROR_INDIRECT_TRANSACTION_EXPIRY for "42235".
[2024-05-06 18:03:49] warning: 	zh:ember:ezsp: Received network/route error ROUTE_ERROR_INDIRECT_TRANSACTION_EXPIRY for "42235".
[2024-05-06 18:03:50] warning: 	zh:ember:ezsp: Received network/route error ROUTE_ERROR_INDIRECT_TRANSACTION_EXPIRY for "42235".
[2024-05-06 18:04:19] warning: 	zh:ember: [TRUST CENTER] Device 42235:0x0015bc00340008dd was denied joining via 40337.
[2024-05-06 18:04:19] warning: 	zh:ember: [TRUST CENTER] Device 42235:0x0015bc00340008dd was denied joining via 40337.
[2024-05-06 18:04:19] warning: 	z2m: Device '0x0015bc00340008dd' left the network
[2024-05-06 18:04:42] warning: 	zh:ember: [TRUST CENTER] Device 29742:0x0015bc00340008dd was denied joining via 40337.
[2024-05-06 18:04:42] warning: 	zh:ember: [TRUST CENTER] Device 29742:0x0015bc00340008dd was denied joining via 40337.
[2024-05-06 18:05:20] error: 	zh:controller: Interview failed for '0x0015bc00340008dd with error 'Error: [ZDO] Failed response by NCP for "61782" cluster "32772" with status=ZDP_INVALID_REQUEST_TYPE.'
[2024-05-06 18:05:20] error: 	z2m: Failed to interview '0x0015bc00340008dd', device has not successfully been paired
[2024-05-06 18:13:47] warning: 	zh:ember:ezsp: Received network/route error ROUTE_ERROR_NO_ROUTE_AVAILABLE for "54868".
[2024-05-06 18:16:40] warning: 	z2m: Device 'Kitchen Smoke Sensor' left the network
[2024-05-06 18:17:04] warning: 	zh:ember: [TRUST CENTER] Device 33160:0x0015bc00340008dd was denied joining via 60665.
[2024-05-06 18:17:05] warning: 	zh:ember: [TRUST CENTER] Device 33160:0x0015bc00340008dd was denied joining via 60665.
[2024-05-06 18:17:28] warning: 	zh:ember: [TRUST CENTER] Device 28037:0x0015bc00340008dd was denied joining via 5159.
[2024-05-06 18:17:28] warning: 	zh:ember: [TRUST CENTER] Device 28037:0x0015bc00340008dd was denied joining via 5159.
[2024-05-06 18:17:51] warning: 	zh:ember: [TRUST CENTER] Device 48604:0x0015bc00340008dd was denied joining via 5159.
[2024-05-06 18:17:51] warning: 	zh:ember: [TRUST CENTER] Device 48604:0x0015bc00340008dd was denied joining via 5159.
[2024-05-06 18:18:14] warning: 	zh:ember: [TRUST CENTER] Device 61138:0x0015bc00340008dd was denied joining via 5159.
[2024-05-06 18:18:14] warning: 	zh:ember: [TRUST CENTER] Device 61138:0x0015bc00340008dd was denied joining via 5159.
[2024-05-06 18:19:15] warning: 	zh:ember:ezsp: Received network/route error ROUTE_ERROR_SOURCE_ROUTE_FAILURE for "50845".
[2024-05-06 18:19:15] warning: 	zh:ember:ezsp: Received network/route error ROUTE_ERROR_MANY_TO_ONE_ROUTE_FAILURE for "50845".
[2024-05-06 18:21:17] error: 	zh:controller: Interview failed for '0x0015bc00340008dd with error 'Error: [ZDO] Failed response by NCP for "26597" cluster "32772" with status=ZDP_INVALID_REQUEST_TYPE.'
[2024-05-06 18:21:17] error: 	z2m: Failed to interview 'Kitchen Smoke Sensor', device has not successfully been paired

mamrai1, 2024-05-06:
joining is ok with ezsp driver.

mamrai1, 2024-05-07:
Also after joining the smoke detectors with ezsp and  then using ember there is battery drainage issue on device connected directly on coordinator.

Nerivec, 2024-05-07:
Can you provide a log for `ezsp` that's roughly the same actions as the ones in the `ember` log above?
Your device is replying `ZDP_TABLE_FULL` at first, which means it can't bind anymore stuff.
Did you try to do a full reset of the device (see if there is a procedure for it)?

mamrai1, 2024-05-07:
[ember.log](https://github.com/Koenkk/zigbee2mqtt/files/15239736/ember.log)
[ezsp.log](https://github.com/Koenkk/zigbee2mqtt/files/15239738/ezsp.log)
attached 2 debug logs when resetting and joining office smoke sensor(develco) with ezsp and ember

Nerivec, 2024-05-07:
The denied join issue should be fixed after this is merged/available https://github.com/Koenkk/zigbee-herdsman/pull/1052
_Still looking into the ZDO issue._

roon7, 2024-05-09:
Tried to join to the network today: Remote Control IKEA E1524/E1810 and Aeotec Range extender WG001.
Many attempts without success, after changing in config ember to ezsp, devices joined in first try.
Back to ember, devices works.

CrazyCoder, 2024-05-09:
Had the same issue with adding HUE bulbs. Switched to Z2M Edge version and it helped.

Nerivec, 2024-05-10:
@mamrai1 Can you confirm if that device is still sending back ZDO errors with 1.37.1?

mamrai1, 2024-05-10:
No stil can't join the sensor. logs3 with latest commit.
[ezsp3.log](https://github.com/Koenkk/zigbee2mqtt/files/15271599/ezsp3.log)
[ember3.log](https://github.com/Koenkk/zigbee2mqtt/files/15271600/ember3.log)
[ezsp2.log](https://github.com/Koenkk/zigbee2mqtt/files/15271601/ezsp2.log)
[ember2.log](https://github.com/Koenkk/zigbee2mqtt/files/15271602/ember2.log)

juanchula, 2024-05-11:
I had the same issue with a Gynoid (Orvibo OEM) brand motion sensor, and after updating Z2M to version 1.37.1 and re-pairing the sensor, the problem was resolved.

Koenkk, 2024-05-14:
@mamrai1 do you by any chance have hardware to sniff while joining this device? https://www.zigbee2mqtt.io/advanced/zigbee/04_sniff_zigbee_traffic.html . If yes, could you make a sniff of `ember` (should fail) and `ezsp` (should work)

mamrai1, 2024-06-01:
> @mamrai1 do you by any chance have hardware to sniff while joining this device? https://www.zigbee2mqtt.io/advanced/zigbee/04_sniff_zigbee_traffic.html . If yes, could you make a sniff of `ember` (should fail) and `ezsp` (should work)

[ezsp_ember.zip](https://github.com/user-attachments/files/15523384/ezsp_ember.zip)

Koenkk, 2024-06-02:
The joins fails due to the simple descriptor request failing. This is an issue on the device side, the same happens on ezsp but the difference is that the device immediately joins again after which the simple descriptor request succeeds:

EZSP:

[image]

Ember:

[image]

With ember, could you try to re-pair the device immediately after interviewing fails?

mamrai1, 2024-06-02:
[ember.zip](https://github.com/user-attachments/files/15526860/ember.zip)
no it fails at configuring

mamrai1, 2024-06-23:
@Nerivec @Koenkk any feedback ?

Koenkk, 2024-06-24:
Could you try to configure it again via the frontend? It's the yellow configure button. Before doing that make sure to wakeup the device right before clicking the button.

mamrai1, 2024-06-24:
No, it fails with ember driver. (status=ZDP_TABLE_FULL. )
At least after joining the device with ezsp and changing to ember it works..Also the interview button works with no errors.

mamrai1, 2024-06-24:
2024-06-24 12:40:53z2m: Failed to configure 'Office Smoke Sensor', attempt 1 (Error: Bind 0x0015bc00340009e6/35 genBasic from '0xe0798dfffe7d7366/1' failed ([ZDO] Failed response by NCP for "16827" cluster "32801" with status=ZDP_TABLE_FULL.) at EmberOneWaitress.resolveZDO (/app/node_modules/zigbee-herdsman/src/adapter/ember/adapter/oneWaitress.ts:138:35) at EmberAdapter.onZDOResponse (/app/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:663:26) at Ezsp.emit (node:events:517:28) at Ezsp.ezspIncomingMessageHandler (/app/node_modules/zigbee-herdsman/src/adapter/ember/ezsp/ezsp.ts:4517:22) at Ezsp.callbackDispatch (/app/node_modules/zigbee-herdsman/src/adapter/ember/ezsp/ezsp.ts:820:18) at Ezsp.tick (/app/node_modules/zigbee-herdsman/src/adapter/ember/ezsp/ezsp.ts:446:22) at listOnTimeout (node:internal/timers:569:17) at processTimers (node:internal/timers:512:7))

At least the situation is better now. Switching to ember and intervewing the device works (blue button) with no errors. Also changing reporting values works as expected.. .
Zigbee2MQTT version:
1.38.0-dev commit: ee42d68
EmberZNet
Coordinator revision
7.4.3 [GA]

Nerivec, 2024-06-24:
Even though it sometimes works, I suspect this is more about luck, since the driver has zero control over how the device responds to a command being sent to it (`ZDP_TABLE_FULL` is reported directly from the device, means the binding table of that device is full, the only thing the z2m driver does is read said status when it receives it). Combining your previous logs, errors are also not consistent, since it failed at different steps during interviews with different errors.
Looks like a firmware mishap of some kind on that device.

Nerivec, 2024-07-17:
Relating to `ZDP_TABLE_FULL`:
I went over the `bind` part of the code. It seems other drivers (`ezsp`, `zstack`) are not checking the status (ZDO) of the bind response.
Leaves me thinking either:
- That develco device is not reporting the proper status even if successful, so it works if you just ignore it
- The bind always fails on that develco device but other drivers let Z2M think it didn't

_I glanced over the rest of the ZDO requests, seems the same applies for a few of them (most have a status in response)._
@Koenkk can you confirm this for zstack (at least the part about the status not being checked against)?

Koenkk, 2024-07-18:
@Nerivec zstack indeed doesn't check the return code.

@mamrai1 

Could you check if the issue is fixed with the following [external converter](https://gist.github.com/Koenkk/6d8476b9116b7fbf96e2be7179249f83): 

- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m, check if issue is fixed

mamrai1, 2024-07-18:
2024-07-18 22:06:55] error: 	z2m: Failed to load external converter file 'ext_converter.js' (Cannot find module '../lib/develco'
Require stack:
- /app/dist/util/utils.js
- /app/dist/util/settings.js
- /app/index.js)

Nerivec, 2024-07-18:
Change a couple of the top lines, from:

```javascript
const {develcoModernExtend} = require('../lib/develco');
const {battery} = require('../lib/modernExtend');
```
to:
```javascript
const {develcoModernExtend} = require('zigbee-herdsman-converters/lib/develco');
const {battery} = require('zigbee-herdsman-converters/lib/modernExtend');
```

mamrai1, 2024-07-18:
Warning 2024-07-19 00:17:55zh:ember: [ZDO] Node descriptor for '11734' reports device is only compliant to revision '21' of the ZigBee specification (current revision: 22).
Error 2024-07-19 00:18:00z2m: Request 'zigbee2mqtt/bridge/request/device/interview' failed with error: 'interview of 'Kitchen Smoke Sensor' (0x0015bc00340008dd) failed: Error: Delivery failed for {"profileId":0,"clusterId":4,"sourceEndpoint":0,"destinationEndpoint":0,"options":4416,"groupId":0,"sequence":223}'
Error 2024-07-19 00:18:23z2m: Failed to configure 'Kitchen Smoke Sensor', attempt 1 (ReferenceError: logger is not defined at configure (/app/data/extension/externally-loaded.js:42:17) at runNextTicks (node:internal/process/task_queues:60:5) at processImmediate (node:internal/timers:447:9) at Object.configure (/app/node_modules/zigbee-herdsman-converters/src/index.ts:190:21) at Configure.configure (/app/lib/extension/configure.ts:126:13) at Configure.onMQTTMessage

Nerivec, 2024-07-19:
Argh, missing an import. Went over the entire thing, made a few more tweaks. Try this:

```javascript
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const e = exposes.presets;
const ea = exposes.access;
const {develcoModernExtend} = require('zigbee-herdsman-converters/lib/develco');
const {battery} = require('zigbee-herdsman-converters/lib/modernExtend');
const {logger} = require('zigbee-herdsman-converters/lib/logger');

const definition =  {
    zigbeeModel: ['HESZB-120'],
    model: 'HESZB-120',
    vendor: 'Develco',
    description: 'Fire detector with siren',
    whiteLabel: [{vendor: 'Frient', model: '94431', description: 'Smart Intelligent Heat Alarm'}],
    fromZigbee: [fz.ias_smoke_alarm_1_develco, fz.ignore_basic_report, fz.ias_enroll, fz.ias_wd], //, develco.fz.fault_status],
    toZigbee: [tz.warning, tz.ias_max_duration, tz.warning_simple],
    // ota: ota.zigbeeOTA,
    extend: [
        develcoModernExtend.addCustomClusterManuSpecificDevelcoGenBasic(),
        develcoModernExtend.readGenBasicPrimaryVersions(),
        develcoModernExtend.temperature(), // TODO: ep 38
        battery({
            voltageToPercentage: '3V_2500',
            percentage: true,
            voltage: true,
            lowStatus: false,
            voltageReporting: true,
            percentageReporting: false,
        }),
    ],
    configure: async (device, coordinatorEndpoint) => {
        const endpoint = device.getEndpoint(35);

        for (const cluster of ['ssIasZone', 'ssIasWd', 'genBasic', 'genBinaryInput']) {
            try {
                await reporting.bind(endpoint, coordinatorEndpoint, [cluster]);
            } catch (error) {
                logger.debug(`Failed to bind '${cluster}' ${error}`, 'extconv:develco:heszb120');
            }
        }

        await endpoint.read('ssIasZone', ['iasCieAddr', 'zoneState', 'zoneId']);
        await endpoint.read('genBinaryInput', ['reliability', 'statusFlags']);
        await endpoint.read('ssIasWd', ['maxDuration']);
    },
    endpoint: (device) => {
        return {default: 35};
    },
    exposes: [
        e.smoke(),
        e.battery_low(),
        e.test(),
        e.numeric('max_duration', ea.ALL).withUnit('s').withValueMin(0).withValueMax(600).withDescription('Duration of Siren'),
        e.binary('alarm', ea.SET, 'START', 'OFF').withDescription('Manual Start of Siren'),
        e
            .enum('reliability', ea.STATE, ['no_fault_detected', 'unreliable_other', 'process_error'])
            .withDescription('Indicates reason if any fault'),
        e.binary('fault', ea.STATE, true, false).withDescription('Indicates whether the device are in fault state'),
    ],
};

module.exports = definition;
```

mamrai1, 2024-07-19:
Seems now that reconfiguring the device works!
Thanks

mamrai1, 2024-07-19:
@Nerivec how do you proceed now? Fixing the converter for the Develco heat alarm or a more generic solution ignoring the ZDO response?

Nerivec, 2024-07-19:
We shouldn't ignore the ZDO response, since it is supposed to tell us if the bind was successful or not. Z2M should not assume it worked since that can quickly cascade into other issues.

As for this device, now we need to figure out which scenario is effective here, either the bind worked, but the device reports a wrong status (in that case, we can override the converter to bypass it), or the bind just always fails (in that case, remains to see how "working" the device really is).

Koenkk, 2024-07-21:
- I've integrated the fixes into the Develco converter
- I've created a fix to check for the status code with ZStack adapters (https://github.com/Koenkk/zigbee-herdsman/pull/1125)

closed 2024-07-21 by Koenkk

Koenkk, 2024-07-22:
@mamrai1 to double check, does the device fully work with the external converter?

mamrai1, 2024-07-22:
Yes, already checked re configuring, re interviewing and also changing reporting intervals…seems ok

mamrai1, 2024-07-22:
So you want to fully reset the device and re pair it?

mamrai1, 2024-07-22:
No, this is not possible..if I fully reset the device it fails again at configuring with ember.. With Ezsp it pairs perfectly with the external converter..

mamrai1, 2024-07-22:
I tried everything : pairing near the coordinator, re configuring from webgui..only re interview is successful

Nerivec, 2024-07-22:
Is it the error with the node descriptor randomly failing like before?
It can't be failing `configure` with the external converter ignoring errors, so it has to be before that.

mamrai1, 2024-07-22:
Also the device is missing from OTA page after applying the external converter

mamrai1, 2024-07-23:
2024-07-23 01:11:02] warning: 	zh:ember: [ZDO] Device '40035' is only compliant to revision '21' of the ZigBee specification (current revision: 22).
[2024-07-23 01:12:24] error: 	z2m: Failed to configure 'Kitchen Smoke Sensor', attempt 1 (Error: Bind 0x0015bc00340008dd/35 genPowerCfg from '0xe0798dfffe7d7366/1' failed ([ZDO] Failed response for '40035' cluster '32801' Status 'TABLE_FULL'.)
    at EmberOneWaitress.resolveZDO (/app/node_modules/zigbee-herdsman/src/adapter/ember/adapter/oneWaitress.ts:136:35)
    at EmberAdapter.onZDOResponse (/app/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:530:30)
    at Ezsp.emit (node:events:517:28)
    at Ezsp.ezspIncomingMessageHandler (/app/node_modules/zigbee-herdsman/src/adapter/ember/ezsp/ezsp.ts:5283:18)
    at Ezsp.callbackDispatch (/app/node_modules/zigbee-herdsman/src/adapter/ember/ezsp/ezsp.ts:814:26)
    at Ezsp.onAshFrame (/app/node_modules/zigbee-herdsman/src/adapter/ember/ezsp/ezsp.ts:430:22)
    at UartAsh.emit (node:events:517:28)
    at Immediate.<anonymous> (/app/node_modules/zigbee-herdsman/src/adapter/ember/uart/ash.ts:1187:45)
    at processImmediate (node:internal/timers:476:21))
[2024-07-23 01:13:54] error: 	z2m: Failed to configure 'Kitchen Smoke Sensor', attempt 2 (Error: Bind 0x0015bc00340008dd/35 genPowerCfg from '0xe0798dfffe7d7366/1' failed ([ZDO] Failed response for '40035' cluster '32801' Status 'TABLE_FULL'.)
    at EmberOneWaitress.resolveZDO (/app/node_modules/zigbee-herdsman/src/adapter/ember/adapter/oneWaitress.ts:136:35)
    at EmberAdapter.onZDOResponse (/app/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:530:30)
    at Ezsp.emit (node:events:517:28)
    at Ezsp.ezspIncomingMessageHandler (/app/node_modules/zigbee-herdsman/src/adapter/ember/ezsp/ezsp.ts:5283:18)
    at Ezsp.callbackDispatch (/app/node_modules/zigbee-herdsman/src/adapter/ember/ezsp/ezsp.ts:814:26)
    at Ezsp.onAshFrame (/app/node_modules/zigbee-herdsman/src/adapter/ember/ezsp/ezsp.ts:430:22)
    at UartAsh.emit (node:events:517:28)
    at Immediate.<anonymous> (/app/node_modules/zigbee-herdsman/src/adapter/ember/uart/ash.ts:1187:45)
    at processImmediate (node:internal/timers:476:21))
[2024-07-23 01:14:24] error: 	z2m: Failed to bind cluster 'genPollCtrl' from 'Kitchen Smoke Sensor' to 'Coordinator' (Error: Bind 0x0015bc00340008dd/35 genPollCtrl from '0xe0798dfffe7d7366/1' failed (Delivery failed for {"profileId":0,"clusterId":33,"sourceEndpoint":0,"destinationEndpoint":0,"options":4416,"groupId":0,"sequence":192}))
[2024-07-23 01:14:33] error: 	z2m: Failed to bind cluster 'genPowerCfg' from 'Kitchen Smoke Sensor' to 'Coordinator' (Error: Bind 0x0015bc00340008dd/35 genPowerCfg from '0xe0798dfffe7d7366/1' failed (Delivery failed for {"profileId":0,"clusterId":33,"sourceEndpoint":0,"destinationEndpoint":0,"options":4416,"groupId":0,"sequence":201}))
[2024-07-23 01:19:24] error: 	z2m: Failed to bind cluster 'ssIasWd' from 'Kitchen Smoke Sensor' to 'Coordinator' (Error: Bind 0x0015bc00340008dd/35 ssIasWd from '0xe0798dfffe7d7366/1' failed (Delivery failed for {"profileId":0,"clusterId":33,"sourceEndpoint":0,"destinationEndpoint":0,"options":4416,"groupId":0,"sequence":31}))
[2024-07-23 01:19:27] error: 	z2m: Failed to bind cluster 'genPowerCfg' from 'Kitchen Smoke Sensor' to 'Coordinator' (Error: Bind 0x0015bc00340008dd/35 genPowerCfg from '0xe0798dfffe7d7366/1' failed ([ZDO] Failed response for '40035' cluster '32801' Status 'TABLE_FULL'.))
[2024-07-23 05:54:40] error: 	zh:ember: Delivery of BROADCAST failed for '65532' [apsFrame={"profileId":0,"clusterId":31,"sourceEndpoint":0,"destinationEndpoint":0,"options":0,"groupId":0,"sequence":0} messageTag=255]
[2024-07-23 05:56:03] warning: 	z2m: Failed to ping 'Left Bedside' (attempt 1/2, ZCL command 0x6c5cb1fffe60c97f/1 genBasic.read(["zclVersion"], {"timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Delivery failed for {"profileId":260,"clusterId":0,"sourceEndpoint":1,"destinationEndpoint":1,"options":4416,"groupId":0,"sequence":85}))
[2024-07-23 05:56:10] warning: 	z2m: Failed to ping 'Left Bedside' (attempt 2/2, ZCL command 0x6c5cb1fffe60c97f/1 genBasic.read(["zclVersion"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Delivery failed for {"profileId":260,"clusterId":0,"sourceEndpoint":1,"destinationEndpoint":1,"options":4416,"groupId":0,"sequence":86}))
[2024-07-23 06:14:31] warning: 	zh:ember: [ZDO] Device '40035' is only compliant to revision '21' of the ZigBee specification (current revision: 22).
[2024-07-23 06:16:33] error: 	z2m: Failed to configure 'Kitchen Smoke Sensor', attempt 3 (Error: Bind 0x0015bc00340008dd/35 genPowerCfg from '0xe0798dfffe7d7366/1' failed ([ZDO] Failed response for '40035' cluster '32801' Status 'TABLE_FULL'.)
    at EmberOneWaitress.resolveZDO (/app/node_modules/zigbee-herdsman/src/adapter/ember/adapter/oneWaitress.ts:136:35)
    at EmberAdapter.onZDOResponse (/app/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:530:30)
    at Ezsp.emit (node:events:517:28)
    at Ezsp.ezspIncomingMessageHandler (/app/node_modules/zigbee-herdsman/src/adapter/ember/ezsp/ezsp.ts:5283:18)
    at Ezsp.callbackDispatch (/app/node_modules/zigbee-herdsman/src/adapter/ember/ezsp/ezsp.ts:814:26)
    at Ezsp.onAshFrame (/app/node_modules/zigbee-herdsman/src/adapter/ember/ezsp/ezsp.ts:430:22)
    at UartAsh.emit (node:events:517:28)
    at Immediate.<anonymous> (/app/node_modules/zigbee-herdsman/src/adapter/ember/uart/ash.ts:1187:45)
    at processImmediate (node:internal/timers:476:21))

mamrai1, 2024-07-23:
Please find attached some logs regarding 2 devices. The 'Kitchen Smoke semsor' fails at reconfiguring while the other succeeds!
Both on map are shown connected on the same router (sonoff-e)
[develco_ember.txt](https://github.com/user-attachments/files/16346018/develco_ember.txt)
Iwill reset both devices in the afternoon and re pair them one by one and see how it goes....
I

mamrai1, 2024-07-23:
No, it fails when fully reset. Is it beacuse the external converter is not complete?

Nerivec, 2024-07-23:
I see you updated to latest dev. Did you remove the external converter, and did a fresh re-pair since Koenkk's fix would already be in there? And can you confirm these two are the same model devices (normally identified the same by Z2M)?

mamrai1, 2024-07-23:
[image]
Can you confirm the fix is already implemented?

Nerivec, 2024-07-23:
https://github.com/Koenkk/zigbee-herdsman-converters/commits/master/
Nevermind, there was no release built after the fix. So you'd still need the external converter for now.

mamrai1, 2024-07-23:
i have removed the external converter. The devices are exactly the same. Both at reconfigure give different errors. Interview completes for both of them ok.
[develco_ember2.txt](https://github.com/user-attachments/files/16352865/develco_ember2.txt)

2024-07-23 21:15:30z2m: Failed to configure 'Office Smoke Sensor', attempt 4 (Error: Bind 0x0015bc00340009e6/35 genBasic from '0xe0798dfffe7d7366/1' failed ([ZDO] Failed response for '16827' cluster '32801' Status 'TABLE_FULL'.) at EmberOneWaitress.resolveZDO (/app/node_modules/zigbee-herdsman/src/adapter/ember/adapter/oneWaitress.ts:136:35) at EmberAdapter.onZDOResponse (/app/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:530:30) at Ezsp.emit (node:events:517:28) at Ezsp.ezspIncomingMessageHandler (/app/node_modules/zigbee-herdsman/src/adapter/ember/ezsp/ezsp.ts:5283:18) at Ezsp.callbackDispatch (/app/node_modules/zigbee-herdsman/src/adapter/ember/ezsp/ezsp.ts:814:26) at Ezsp.onAshFrame (/app/node_modules/zigbee-herdsman/src/adapter/ember/ezsp/ezsp.ts:430:22) at UartAsh.emit (node:events:517:28) at Immediate.<anonymous> (/app/node_modules/zigbee-herdsman/src/adapter/ember/uart/ash.ts:1187:45) at processImmediate (node:internal/timers:476:21))
2024-07-23 21:15:34z2m: Failed to configure 'Kitchen Smoke Sensor', attempt 5 (Error: Bind 0x0015bc00340008dd/35 ssIasZone from '0xe0798dfffe7d7366/1' failed (Delivery failed for {"profileId":0,"clusterId":33,"sourceEndpoint":0,"destinationEndpoint":0,"options":4416,"groupId":0,"sequence":37}) at EmberOneWaitress.deliveryFailedFor (/app/node_modules/zigbee-herdsman/src/adapter/ember/adapter/oneWaitress.ts:98:31) at EmberAdapter.onMessageSent (/app/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:454:42) at Ezsp.emit (node:events:517:28) at Ezsp.ezspMessageSentHandler (/app/node_modules/zigbee-herdsman/src/adapter/ember/ezsp/ezsp.ts:5016:14) at Ezsp.callbackDispatch (/app/node_modules/zigbee-herdsman/src/adapter/ember/ezsp/ezsp.ts:770:26) at Ezsp.onAshFrame (/app/node_modules/zigbee-herdsman/src/adapter/ember/ezsp/ezsp.ts:430:22) at UartAsh.emit (node:events:517:28) at Immediate.<anonymous> (/app/node_modules/zigbee-herdsman/src/adapter/ember/uart/ash.ts:1187:45) at processImmediate (node:internal/timers:476:21))

mamrai1, 2024-07-23:
I will add again the converter and retry...

Nerivec, 2024-07-23:
Wait a bit for it to be available in dev, Koenkk is publishing a new release for converters as we speak.

mamrai1, 2024-07-23:
Double checked. Everything ok now with latest commit!! Thanks

Nerivec, 2024-07-23:
Good!
Let us know if you find some feature of the device is not working as it should (in case the device is indeed failing the bind internally -and not just reporting a wrong status-).

n1mda, 2024-08-06:
This is still an issue when I attempt to pair a [SMSZB-120](https://www.zigbee2mqtt.io/devices/SMSZB-120.html#develco-smszb-120) using zigbee2mqtt [1.39.1](https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.39.1) with `ember` adapter (Sonoff ZBDonlge-E with 7.0 firmware)

Logs:
```
Info 2024-08-06 08:11:15z2m: Zigbee: allowing new devices to join.
Info 2024-08-06 08:11:15zh:ember: [STACK STATUS] Network opened.
Info 2024-08-06 08:11:16z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/response/permit_join', payload '{"data":{"time":300,"value":true},"status":"ok"}'
Info 2024-08-06 08:11:27zh:controller: Interview for '0x0015bc0031010443' started
Info 2024-08-06 08:11:27z2m: Device '0x0015bc0031010443' joined
Info 2024-08-06 08:11:27z2m: Starting interview of '0x0015bc0031010443'
Info 2024-08-06 08:11:27z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x0015bc0031010443","ieee_address":"0x0015bc0031010443"},"type":"device_joined"}'
Info 2024-08-06 08:11:27z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x0015bc0031010443","ieee_address":"0x0015bc0031010443","status":"started"},"type":"device_interview"}'
Info 2024-08-06 08:11:28z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x0015bc0031010443","ieee_address":"0x0015bc0031010443"},"type":"device_announce"}'
Error 2024-08-06 08:11:29zh:controller: Interview failed for '0x0015bc0031010443 with error 'Error: [ZDO] Failed response for '61846' cluster '32772' Status 'INV_REQUESTTYPE'.'
Error 2024-08-06 08:11:29z2m: Failed to interview '0x0015bc0031010443', device has not successfully been paired
Info 2024-08-06 08:11:30z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x0015bc0031010443","ieee_address":"0x0015bc0031010443","status":"failed"},"type":"device_interview"}'
Info 2024-08-06 08:11:31z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0x0015bc0031010443', payload '{"battery_low":false,"linkquality":184,"restore_reports":true,"smoke":false,"supervision_reports":true,"test":false}'
Info 2024-08-06 08:11:46zh:controller: Interview for '0x0015bc0031010443' started
Info 2024-08-06 08:11:46z2m: Starting interview of '0x0015bc0031010443'
Info 2024-08-06 08:11:46z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x0015bc0031010443","ieee_address":"0x0015bc0031010443","status":"started"},"type":"device_interview"}'
Info 2024-08-06 08:11:47z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x0015bc0031010443","ieee_address":"0x0015bc0031010443"},"type":"device_announce"}'
Error 2024-08-06 08:11:47zh:controller: Interview failed for '0x0015bc0031010443 with error 'Error: [ZDO] Failed response for '61846' cluster '32772' Status 'INV_REQUESTTYPE'.'
Error 2024-08-06 08:11:47z2m: Failed to interview '0x0015bc0031010443', device has not successfully been paired
Info 2024-08-06 08:11:47z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x0015bc0031010443","ieee_address":"0x0015bc0031010443","status":"failed"},"type":"device_interview"}'
Info 2024-08-06 08:11:49z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0x0015bc0031010443', payload '{"battery_low":false,"linkquality":216,"restore_reports":true,"smoke":false,"supervision_reports":true,"test":false}'
Info 2024-08-06 08:12:35z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0x0015bc001e013ebc', payload '{"battery":0,"battery_low":false,"contact":false,"linkquality":120,"tamper":false,"temperature":24.75,"update":{"installed_version":197651,"latest_version":262146,"state":"available"},"voltage":2500}'
```

mamrai1, 2024-08-06:
First start with upgrading your firmware to 7.4.x

n1mda, 2024-08-06:
> First, start with upgrading your sonoff’s firmware to 7.4.x. (https://github.com/itead/Sonoff_Zigbee_Dongle_Firmware/tree/master/Dongle-E) Secondly, check if your smoke sensor is upgraded to 4.0.8 firmware. Also try to pair initially near the coordinator so that to exclude any potential problems with a router

Sorry I mis-typed, I am using 7.4.3 firmware on the dongle.
The smoke sensor is running 4.0.8 firmware.
The pairing was done in close proximity to the coordinator with no other router in the network

mamrai1, 2024-08-06:
The problem fixed was related to HESZB-120.. Maybe also the converter regarding SMSZB-120 needs adjustments..

Subito, 2024-08-06:
I tried this converter https://github.com/Koenkk/zigbee2mqtt/issues/22492#issuecomment-2237856933 with small modifications for the SMSZB-120 and had "success" in the sense temperature etc start working. Configuring seems to work now.

Koenkk, 2024-08-06:
@n1mda does the interview always fail? (give it multiple tries)

n1mda, 2024-08-08:
> @n1mda does the interview always fail? (give it multiple tries)

The device was paired with the network but interview had failed. I tried pressing the Interview button in the frontend, but it failed (I however lost the logs from this).
I forcefully removed the device from the network via the frontend.
I paired the device again. Interview was once more unsuccessful.
I pressed the Interview button, and this time interview was successful. The logs however contain a bunch of errors.
The device is now visible in the frontend, but the attributes like battery and temperature are missing.

Logs from pairing:
```
Info 2024-08-08 13:55:03z2m: Zigbee: allowing new devices to join via Coordinator.
Info 2024-08-08 13:55:03zh:ember: [STACK STATUS] Network opened.
Info 2024-08-08 13:55:03z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/response/permit_join', payload '{"data":{"device":"Coordinator","time":254,"value":true},"status":"ok","transaction":"vnjyo-1"}'
Info 2024-08-08 13:55:32zh:controller: Interview for '0x0015bc0031010443' started
Info 2024-08-08 13:55:32z2m: Device '0x0015bc0031010443' joined
Info 2024-08-08 13:55:32z2m: Starting interview of '0x0015bc0031010443'
Info 2024-08-08 13:55:32z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x0015bc0031010443","ieee_address":"0x0015bc0031010443"},"type":"device_joined"}'
Info 2024-08-08 13:55:32z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x0015bc0031010443","ieee_address":"0x0015bc0031010443","status":"started"},"type":"device_interview"}'
Info 2024-08-08 13:55:32z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x0015bc0031010443","ieee_address":"0x0015bc0031010443"},"type":"device_announce"}'
Error 2024-08-08 13:55:33zh:controller: Interview failed for '0x0015bc0031010443 with error 'Error: [ZDO] Failed response for '61846' cluster '32772' Status 'INV_REQUESTTYPE'.'
Error 2024-08-08 13:55:33z2m: Failed to interview '0x0015bc0031010443', device has not successfully been paired
Info 2024-08-08 13:55:33z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x0015bc0031010443","ieee_address":"0x0015bc0031010443","status":"failed"},"type":"device_interview"}'
Error 2024-08-08 13:55:33zh:ember: Delivery of BROADCAST failed for '65533' [apsFrame={"profileId":0,"clusterId":19,"sourceEndpoint":0,"destinationEndpoint":0,"options":1024,"groupId":0,"sequence":0} messageTag=255]
Error 2024-08-08 13:55:34zh:ember: Delivery of BROADCAST failed for '65533' [apsFrame={"profileId":0,"clusterId":6,"sourceEndpoint":0,"destinationEndpoint":0,"options":0,"groupId":0,"sequence":16} messageTag=255]
Error 2024-08-08 13:55:34zh:ember: Delivery of BROADCAST failed for '65532' [apsFrame={"profileId":0,"clusterId":54,"sourceEndpoint":0,"destinationEndpoint":0,"options":0,"groupId":0,"sequence":12} messageTag=255]
Info 2024-08-08 13:55:34z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0x0015bc0031010443', payload '{"battery_low":false,"linkquality":172,"restore_reports":true,"smoke":false,"supervision_reports":true,"test":false}'
Info 2024-08-08 13:55:43z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0x0015bc0031010443', payload '{"battery_low":false,"linkquality":192,"restore_reports":true,"smoke":false,"supervision_reports":true,"test":false,"update":{"installed_version":262152,"latest_version":262152,"state":"idle"}}'
Error 2024-08-08 13:55:44zh:ember: Delivery of BROADCAST failed for '65533' [apsFrame={"profileId":0,"clusterId":6,"sourceEndpoint":0,"destinationEndpoint":0,"options":0,"groupId":0,"sequence":24} messageTag=255]
```

Logs from interviewing via button in frontend:
```

Info 2024-08-08 13:58:29z2m: Interviewing '0x0015bc0031010443'
Warning 2024-08-08 13:58:32zh:ember: [ZDO] Device '61846' is only compliant to revision '21' of the ZigBee specification (current revision: 22).
Info 2024-08-08 13:58:48z2m: Successfully interviewed '0x0015bc0031010443'
Info 2024-08-08 13:58:48z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/interview', payload '{"data":{"id":"0x0015bc0031010443"},"status":"ok","transaction":"vnjyo-2"}'
Info 2024-08-08 13:59:18zh:ember: [STACK STATUS] Network closed.
Info 2024-08-08 14:03:47z2m: Configuring '0x0015bc0031010443'
Info 2024-08-08 14:03:47z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0x0015bc0031010443', payload '{"battery_low":false,"linkquality":216,"restore_reports":true,"smoke":false,"supervision_reports":true,"test":false,"update":{"installed_version":262152,"latest_version":262152,"state":"idle"}}'
Error 2024-08-08 14:03:49z2m: Failed to configure '0x0015bc0031010443', attempt 1 (Error: Bind 0x0015bc0031010443/35 genBinaryInput from '0xe0798dfffec311fd/1' failed ([ZDO] Failed response for '61846' cluster '32801' Status 'TABLE_FULL'.) at EmberOneWaitress.resolveZDO (/app/node_modules/zigbee-herdsman/src/adapter/ember/adapter/oneWaitress.ts:136:35) at EmberAdapter.onZDOResponse (/app/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:530:30) at Ezsp.emit (node:events:517:28) at Ezsp.ezspIncomingMessageHandler (/app/node_modules/zigbee-herdsman/src/adapter/ember/ezsp/ezsp.ts:5283:18) at Ezsp.callbackDispatch (/app/node_modules/zigbee-herdsman/src/adapter/ember/ezsp/ezsp.ts:814:26) at Ezsp.onAshFrame (/app/node_modules/zigbee-herdsman/src/adapter/ember/ezsp/ezsp.ts:430:22) at UartAsh.emit (node:events:517:28) at Immediate.<anonymous> (/app/node [... cut ...] 517:28) at Ezsp.ezspIncomingMessageHandler (/app/node_modules/zigbee-herdsman/src/adapter/ember/ezsp/ezsp.ts:5283:18) at Ezsp.callbackDispatch (/app/node_modules/zigbee-herdsman/src/adapter/ember/ezsp/ezsp.ts:814:26) at Ezsp.onAshFrame (/app/node_modules/zigbee-herdsman/src/adapter/ember/ezsp/ezsp.ts:430:22) at UartAsh.emit (node:events:517:28) at Immediate.<anonymous> (/app/node_modules/zigbee-herdsman/src/adapter/ember/uart/ash.ts:1187:45) at processImmediate (node:internal/timers:476:21))
```

Subito, 2024-08-08:
The Problem with the SMSZB-120 seems to be exactly the same as with the HESZB-120. This commit https://github.com/Koenkk/zigbee-herdsman-converters/commit/9ede78a950d703dac4e60271e08dabf7808661e6 only changes the behaviour for HESZB-120, though, and not the SMSZB-120. So I tried to adapt the converter for the SMSZB-120 and had success.

Unfortunately I don't feel confident in turning that converter into a PR, as I'm very inexperienced with this project. Hopefully somebody else can pick this up.

This converter works for me:

``` javascript
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const e = exposes.presets;
const ea = exposes.access;
const {develcoModernExtend} = require('zigbee-herdsman-converters/lib/develco');
const {battery} = require('zigbee-herdsman-converters/lib/modernExtend');
const {logger} = require('zigbee-herdsman-converters/lib/logger');

const definition =  {
    zigbeeModel: ['SMSZB-120'],
    model: 'SMSZB-120',
    vendor: 'Develco',
    description: 'Smoke detector with siren',
    whiteLabel: [{vendor: 'Frient', model: '94430', description: 'Smoke detector with siren'}],
    fromZigbee: [fz.ias_smoke_alarm_1_develco, fz.ignore_basic_report, fz.ias_enroll, fz.ias_wd], //, develco.fz.fault_status],
    toZigbee: [tz.warning, tz.ias_max_duration, tz.warning_simple],
    // ota: ota.zigbeeOTA,
    extend: [
        develcoModernExtend.addCustomClusterManuSpecificDevelcoGenBasic(),
        develcoModernExtend.readGenBasicPrimaryVersions(),
        develcoModernExtend.temperature(), // TODO: ep 38
        battery({
            voltageToPercentage: '3V_2500',
            percentage: true,
            voltage: true,
            lowStatus: false,
            voltageReporting: true,
            percentageReporting: false,
        }),
    ],
    configure: async (device, coordinatorEndpoint) => {
        const endpoint = device.getEndpoint(35);

        for (const cluster of ['ssIasZone', 'ssIasWd', 'genBasic', 'genBinaryInput']) {
            try {
                await reporting.bind(endpoint, coordinatorEndpoint, [cluster]);
            } catch (error) {
                logger.debug(`Failed to bind '${cluster}' ${error}`, 'extconv:develco:smszb120');
            }
        }

        await endpoint.read('ssIasZone', ['iasCieAddr', 'zoneState', 'zoneId']);
        await endpoint.read('genBinaryInput', ['reliability', 'statusFlags']);
        await endpoint.read('ssIasWd', ['maxDuration']);
    },
    endpoint: (device) => {
        return {default: 35};
    },
    exposes: [
        e.smoke(),
        e.battery_low(),
        e.test(),
        e.numeric('max_duration', ea.ALL).withUnit('s').withValueMin(0).withValueMax(600).withDescription('Duration of Siren'),
        e.binary('alarm', ea.SET, 'START', 'OFF').withDescription('Manual Start of Siren'),
        e
            .enum('reliability', ea.STATE, ['no_fault_detected', 'unreliable_other', 'process_error'])
            .withDescription('Indicates reason if any fault'),
        e.binary('fault', ea.STATE, true, false).withDescription('Indicates whether the device are in fault state'),
    ],
};

module.exports = definition;
```

Exelord, 2024-08-08:
I have similar issue:
```
z2m: Failed to configure '0x003c84fffed56339', attempt 2 (Error: Unbind 0x003c84fffed56339/1 genPollCtrl from '0xe0798dfffe9ee486/1' failed ([ZDO] Failed response for '19448' cluster '32802' Status 'NO_ENTRY'.) at EmberOneWaitress.resolveZDO (/app/node_modules/zigbee-herdsman/src/adapter/ember/adapter/oneWaitress.ts:136:35) at EmberAdapter.onZDOResponse (/app/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:530:30) at Ezsp.emit (node:events:517:28) at Ezsp.ezspIncomingMessageHandler (/app/node_modules/zigbee-herdsman/src/adapter/ember/ezsp/ezsp.ts:5283:18) at Ezsp.callbackDispatch (/app/node_modules/zigbee-herdsman/src/adapter/ember/ezsp/ezsp.ts:814:26) at Ezsp.onAshFrame (/app/node_modules/zigbee-herdsman/src/adapter/ember/ezsp/ezsp.ts:430:22) at UartAsh.emit (node:events:517:28) at Immediate.<anonymous> (/app/node_modules/zigbee-herdsman/src/adapter/ember/uart/ash.ts:1187:45) at processImmediate (node:internal/timers:476:21))
```

When pairing Sonoff ZBMiniL2

Nerivec, 2024-08-08:
@Exelord Not quite the same (device is not misbehaving in your case). Should be fixed after above PR is merged.

mamrai1, 2024-10-21:
After this time it seems that we should remove the bind on genpollctrl and gen.

mamrai1, 2024-10-24:
[log.log](https://github.com/user-attachments/files/17511689/log.log)
Seems that there is some kind of limit on 4.0.8 when binding clusters. if i remove the bind on genpollctrl and genBinaryInput i can bind powercfg and apply reporting of voltage. no table full error. if i try to rebind the genpollctrl or genbinaryinput the 'table full error' reappears. only 4 clusters can be bind on the device.

Koenkk, 2024-10-24:
Could you check if the issue is fixed with the following [external converter](https://gist.github.com/Koenkk/07a47c0c6678627c2547a902d50d2884) (this should limit the amount of binds to 4): 

- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m, 
- in the frontend, check if the description of this device now ends with `CUSTOM` (this indicates the external converter has been loaded correctly)
- check if issue is fixed

[1 more reply]