Koenkk/zigbee2mqtt #22496: Can not start Zigbee2MQTT as deamon

1Joe1, 2024-05-06:
### What happened?

Hello,

Tried to update Z2M with the update script from the Zigbee2MQTT directory:

cd /opt/zigbee2mqtt
./update.sh

This produced some errors due to the old node version that I was using

In the following I updated node using nvm. Tried different ways but failed several times as node was what it seems not installed globally.

Ran the update script again.

Now Z2M is up and running again as long as I start manually.

But I cannot start it as a demon anymore. Tried to update my original zigbee2mqtt.service file to the version mentioned in the new installation guide. Also tried to change the path from

ExecStart=/usr/bin/npm

To

ExecStart=/usr/local/bin/npm

Or 

ExecStart=npm start

Or

The path given by "which npm" (=/home/openhabian/.nvm/versions/node/v21.7.3/bin/npm)

and so on as these were suggestions I found for this problem. Sadly nothing helped.

This is my original service file:

[Unit]
Description=zigbee2mqtt
After=network.target

[Service]
ExecStart=/usr/bin/npm start
WorkingDirectory=/opt/zigbee2mqtt
StandardOutput=inherit
StandardError=inherit
Restart=always
User=openhabian
                                                 
[Install]
WantedBy=multi-user.target

I fear I messed up something with node.
Any help or advice for troubleshooting is highly appreciated. 

### What did you expect to happen?

Being Abe to start as a demon after updating without any changes to the service file.

### Zigbee2MQTT version

1.37.0

### Adapter firmware version

20211217

### Adapter

CC2562P

### Setup

Intel NUC with Ubuntu 

### Debug log

[image]

homeyjoenka, 2024-05-06:
{"id":1,"type":"Coordinator","ieeeAddr":"0xe0798dfffe9cb41a","nwkAddr":0,"manufId":0,"epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":48879,"inClusterList":[0,3,6,10,25,26,768],"outClusterList":[0,3,4,5,6,8,32,768,1024,1026,1029,1030,1280,2817,2819,2820,1794,4096,64513,64514],"clusters":{},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0xe0798dfffe9cb41a","endpointID":1}],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"interviewCompleted":true,"meta":{},"lastSeen":1715007116859}
{"id":2,"type":"Router","ieeeAddr":"0xa4c138577b9c6bce","nwkAddr":63586,"manufId":4417,"manufName":"_TZ3000_amdymr7l","powerSource":"Mains (single phase)","modelId":"TS011F","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":266,"inClusterList":[3,4,5,6,1794,2820,57344,57345,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"�iG-\u0012","65506":54,"65508":0,"65534":0,"modelId":"TS011F","manufacturerName":"_TZ3000_amdymr7l","powerSource":1,"zclVersion":3,"appVersion":192,"stackVersion":0,"hwVersion":1,"dateCode":""}},"haElectricalMeasurement":{"attributes":{"acCurrentDivisor":1000,"acCurrentMultiplier":1,"rmsVoltage":233,"rmsCurrent":0,"activePower":0}},"seMetering":{"attributes":{"divisor":100,"multiplier":1,"currentSummDelivered":[0,20062],"status":0}},"genOnOff":{"attributes":{"32768":0,"onOff":0,"onTime":0,"offWaitTime":0,"tuyaBacklightMode":1,"moesStartUpOnOff":1,"tuyaBacklightSwitch":1}},"manuSpecificTuya_3":{"attributes":{"53248":1,"53249":0,"53250":2474305,"53251":143,"53252":10522,"53253":1454,"powerOnBehavior":1,"switchType":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0xe0798dfffe9cb41a","endpointID":1},{"cluster":2820,"type":"endpoint","deviceIeeeAddress":"0xe0798dfffe9cb41a","endpointID":1},{"cluster":1794,"type":"endpoint","deviceIeeeAddress":"0xe0798dfffe9cb41a","endpointID":1}],"configuredReportings":[{"cluster":2820,"attrId":1285,"minRepIntval":5,"maxRepIntval":3600,"repChange":5,"manufacturerCode":null},{"cluster":2820,"attrId":1288,"minRepIntval":5,"maxRepIntval":3600,"repChange":50,"manufacturerCode":null},{"cluster":2820,"attrId":1291,"minRepIntval":5,"maxRepIntval":3600,"repChange":10,"manufacturerCode":null},{"cluster":1794,"attrId":0,"minRepIntval":5,"maxRepIntval":3600,"repChange":[1,1],"manufacturerCode":null}],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":192,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":1324213189},"lastSeen":1715020305170}
{"id":3,"type":"Router","ieeeAddr":"0xa4c13884ad9cdbcf","nwkAddr":9239,"manufId":4417,"manufName":"_TZ3000_gjnozsaz","powerSource":"Mains (single phase)","modelId":"TS011F","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":266,"inClusterList":[3,4,5,6,1794,2820,57344,57345,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"�iG-\u0012","65506":54,"65508":0,"65534":0,"modelId":"TS011F","manufacturerName":"_TZ3000_gjnozsaz","powerSource":1,"zclVersion":3,"appVersion":192,"stackVersion":0,"hwVersion":1,"dateCode":""}},"haElectricalMeasurement":{"attributes":{"acCurrentDivisor":1000,"acCurrentMultiplier":1,"rmsVoltage":234,"activePower":0,"rmsCurrent":81}},"seMetering":{"attributes":{"divisor":100,"multiplier":1,"currentSummDelivered":[0,2223]}},"genOnOff":{"attributes":{"32768":0,"onOff":1,"onTime":0,"offWaitTime":0,"tuyaBacklightMode":2,"moesStartUpOnOff":2,"tuyaBacklightSwitch":1}},"manuSpecificTuya_3":{"attributes":{"53248":0,"53249":0,"53250":0,"53251":0,"53252":0,"53253":0,"powerOnBehavior":2,"switchType":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0xe0798dfffe9cb41a","endpointID":1},{"cluster":2820,"type":"endpoint","deviceIeeeAddress":"0xe0798dfffe9cb41a","endpointID":1},{"cluster":1794,"type":"endpoint","deviceIeeeAddress":"0xe0798dfffe9cb41a","endpointID":1}],"configuredReportings":[{"cluster":2820,"attrId":1285,"minRepIntval":5,"maxRepIntval":3600,"repChange":5,"manufacturerCode":null},{"cluster":2820,"attrId":1288,"minRepIntval":5,"maxRepIntval":3600,"repChange":50,"manufacturerCode":null},{"cluster":2820,"attrId":1291,"minRepIntval":5,"maxRepIntval":3600,"repChange":10,"manufacturerCode":null},{"cluster":1794,"attrId":0,"minRepIntval":5,"maxRepIntval":3600,"repChange":[1,1],"manufacturerCode":null}],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":192,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":1324213189},"lastSeen":1715020357573}
{"id":4,"type":"Router","ieeeAddr":"0xa4c138c439c39204","nwkAddr":20216,"manufId":4417,"manufName":"_TZ3000_gjnozsaz","powerSource":"Mains (single phase)","modelId":"TS011F","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":266,"inClusterList":[3,4,5,6,1794,2820,57344,57345,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"�iG-\u0012","65506":54,"65508":0,"65534":0,"modelId":"TS011F","manufacturerName":"_TZ3000_gjnozsaz","powerSource":1,"zclVersion":3,"appVersion":192,"stackVersion":0,"hwVersion":1,"dateCode":""}},"haElectricalMeasuremen [... cut ...] RepIntval":5,"maxRepIntval":3600,"repChange":10,"manufacturerCode":null},{"cluster":1794,"attrId":0,"minRepIntval":5,"maxRepIntval":3600,"repChange":[1,1],"manufacturerCode":null}],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":72,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":-892627439},"lastSeen":1715020349974}

lux73, 2024-05-07:
@1Joe1 

did your z2m installation works if you start manually?

pls post the output from

`find / -name npm`

1Joe1, 2024-05-07:
Hello Lux73 and thanks for your help.

Yes everything works just fine if I start Z2M manually. Before updating it worked both manually and when started as a service.

Here is the output you asked for:

`openhabian@openhab:~$ find / -name npm
find: ‘/proc/tty/driver’: Permission denied
find: ‘/proc/1/task/1/fd’: Permission denied
find: ‘/proc/1/task/1/fdinfo’: Permission denied
find: ‘/proc/1/task/1/ns’: Permission denied
find: ‘/proc/1/fd’: Permission denied
find: ‘/proc/1/map_files’: Permission denied
find: ‘/proc/1/fdinfo’: Permission denied
find: ‘/proc/1/ns’: Permission denied
find: ‘/proc/2/task/2/fd’: Permission denied
find: ‘/proc/2/task/2/fdinfo’: Permission denied
find: ‘/proc/2/task/2/ns’: Permission denied
find: ‘/proc/2/fd’: Permission denied
find: ‘/proc/2/map_files’: Permission denied
find: ‘/proc/2/fdinfo’: Permission denied
find: ‘/proc/2/ns’: Permission denied
find: ‘/proc/3/task/3/fd’: Permission denied
find: ‘/proc/3/task/3/fdinfo’: Permission denied
find: ‘/proc/3/task/3/ns’: Permission denied
find: ‘/proc/3/fd’: Permission denied
find: ‘/proc/3/map_files’: Permission denied
find: ‘/proc/3/fdinfo’: Permission denied
find: ‘/proc/3/ns’: Permission denied
find: ‘/proc/4/task/4/fd’: Permission denied
find: ‘/proc/4/task/4/fdinfo’: Permission denied
find: ‘/proc/4/task/4/ns’: Permission denied
find: ‘/proc/4/fd’: Permission denied
find: ‘/proc/4/map_files’: Permission denied
find: ‘/proc/4/fdinfo’: Permission denied
find: ‘/proc/4/ns’: Permission denied
find: ‘/proc/6/task/6/fd’: Permission denied
find: ‘/proc/6/task/6/fdinfo’: Permission denied
find: ‘/proc/6/task/6/ns’: Permission denied
find: ‘/proc/6/fd’: Permission denied
find: ‘/proc/6/map_files’: Permission denied
find: ‘/proc/6/fdinfo’: Permission denied
find: ‘/proc/6/ns’: Permission denied
find: ‘/proc/8/task/8/fd’: Permission denied
find: ‘/proc/8/task/8/fdinfo’: Permission denied
find: ‘/proc/8/task/8/ns’: Permission denied
find: ‘/proc/8/fd’: Permission denied
find: ‘/proc/8/map_files’: Permission denied
find: ‘/proc/8/fdinfo’: Permission denied
find: ‘/proc/8/ns’: Permission denied
find: ‘/proc/9/task/9/fd’: Permission denied
find: ‘/proc/9/task/9/fdinfo’: Permission denied
find: ‘/proc/9/task/9/ns’: Permission denied
find: ‘/proc/9/fd’: Permission denied
find: ‘/proc/9/map_files’: Permission denied
find: ‘/proc/9/fdinfo’: Permission denied
find: ‘/proc/9/ns’: Permission denied
find: ‘/proc/10/task/10/fd’: Permission denied
find: ‘/proc/10/task/10/fdinfo’: Permission denied
find: ‘/proc/10/task/10/ns’: Permission denied
find: ‘/proc/10/fd’: Permission denied
find: ‘/proc/10/map_files’: Permission denied
find: ‘/proc/10/fdinfo’: Permission denied
find: ‘/proc/10/ns’: Permission denied
find: ‘/proc/11/task/11/fd’: Permission denied
find: ‘/proc/11/task/11/fdinfo’: Permission denied
find: ‘/proc/11/task/11/ns’: Permission denied
find: ‘/proc/11/fd’: Permission denied
find: ‘/proc/11/map_files’: Permission denied
find: ‘/proc/11/fdinfo’: Permission denied
[... 358 lines cut ...]
find: ‘/proc/233/task/233/fd’: Permission denied
find: ‘/proc/233/task/233/fdinfo’: Permission denied
find: ‘/proc/233/task/233/ns’: Permission denied
find: ‘/proc/233/fd’: Permission denied
find: ‘/proc/233/map_files’: Permission denied
find: ‘/proc/233/fdinfo’: Permission denied
find: ‘/proc/233/ns’: Permission denied
find: ‘/proc/243/task/243/fd’: Permission denied
find: ‘/proc/243/task/243/fdinfo’: Permission denied
find: ‘/proc/243/task/243/ns’: Permission denied
find: ‘/proc/243/fd’: Permission denied
find: ‘/proc/243/map_files’: Permission denied
find: ‘/proc/243/fdinfo’: Permission denied
find: ‘/proc/243/ns’: Permission denied
find: ‘/proc/245/task/245/fd’: Permission denied
find: ‘/proc/245/task/245/fdinfo’: Permission denied
find: ‘/proc/245/task/245/ns’: Permission denied
find: ‘/proc/245/fd’: Permission denied
find: ‘/proc/245/map_files’: Permission denied
find: ‘/proc/245/fdinfo’: Permission denied
find: ‘/proc/245/ns’: Permission denied
find: ‘/proc/275/task/275/fd’: Permission denied
find: ‘/proc/275/task/275/fdinfo’: Permission denied
find: ‘/proc/275/task/275/ns’: Permission denied
`

lux73, 2024-05-07:
hm, sorry i forget you are logged in as limited user ^^

lets try another command to find the `npm` Binary

`whereis npm`

pls post output

is should be anyting like tihis `npm: /usr/bin/npm /usr/share/man/man1/npm.1`

if not try `sudo whereis npm` to run the cmd as superuser

1Joe1, 2024-05-07:
`npm: /usr/bin/npm /home/openhabian/.nvm/versions/node/v21.7.3/bin/npm /usr/share/man/man1/npm.1`

lux73, 2024-05-07:
so `/usr/bin/npm` is correct in your Service File

really strange - all i can say i would delete your node installation & install **nodjs 20.x (LTS)**

https://kinsta.com/blog/how-to-install-node-js/ - scroll down, theres is a "How to install Node.js on Linux" manually

this works flawlessy for my Debian System

closed 2024-12-04 by github-actions[bot]