Koenkk/zigbee2mqtt #22498: Legrand OTA update

cocapic, 2024-05-06:
### What happened?

OTA failed

### What did you expect to happen?

OTA update

### How to reproduce it (minimal and precise)

many legrand device cannot be updated
[067646](https://www.zigbee2mqtt.io/devices/067646.html#legrand-067646)
[067774](https://www.zigbee2mqtt.io/devices/067774.html#legrand-067774)
[412170](https://www.zigbee2mqtt.io/devices/412170.html#legrand-412170)
[067771](https://www.zigbee2mqtt.io/devices/067771.html#legrand-067771)

### Zigbee2MQTT version

1.37.0

### Adapter firmware version

20230507

### Adapter

sonoff

### Setup

jeedom, pi5

### Debug log

Failed to check if update available for '***' (OTA: Device didn't respond to OTA request)
when checking for an update (update is available) z2m button is blue : check for...
AND
Update of 'LU ERWAN' failed (OTA: Device didn't respond to OTA request)
when trying to update (z2m button is the red one : update...

Any way to try manual update ?

many thanks

Loic691, 2024-05-13:
Hi
There is another post specially for one of Legrand device
https://github.com/Koenkk/zigbee2mqtt/issues/22455

cocapic, 2024-07-04:
still not working in 1.39.0

closed 2025-01-05 by github-actions[bot]

Loic691, 2025-05-04:
Hi with last version of Z2M 2.3.0 I was finally able to update home away 064873.
I have to test 2 or 3 times before update starts...