Koenkk/zigbee2mqtt #22511: SSWRM-ZB - stuck in dimming mode

reg98643t, 2024-05-07:
### What happened?

The buttons seem to be stuck in dimming mode. The button can be held for a few seconds to change dimming direction, but it will always be dimming.

### What did you expect to happen?

The lights should stay at one brightness, and only dim if the rotary dial is twisted or the button is held down.

### How to reproduce it (minimal and precise)

Don't have to do anything, they are always dimming up or down.

### Zigbee2MQTT version

1.37.0

### Adapter firmware version

0x26660700

### Adapter

ConBee2/RaspBee2 - Issue also on SONOFF Zigbee USB.

### Setup

HAOS Addon.

### Debug log

Logs don't seem to be generating, not sure what I've done wrong.

serot23, 2024-05-09:
I have 2 of these and one works like it should and the other has the same problem as yours. But i am on z2m version  1.36.0-1

reg98643t, 2024-05-09:
Do you know if yours are wired differently to each other or anything?

-------- Original Message --------

serot23, 2024-05-09:
I just got mine 2 weeks ago. They are wired exactly the same. Only neutral and live wires are connected. I don't use a physical switch i use the wireless rotary dimmer which was sold with it. When ysing the rf wireless rotary dimmer I don't have issues. It works like it should, but when i turn the one on from HA it immediately shoots to 100% even when i press half way the dimming bar. But when i press around 20% it works it stays there but everything above 25% it doesn't matter where i press it shoots to 100% brightness. But the other one works just fine

reg98643t, 2024-05-09:
That's interesting, I wonder if they are running different firmware. Might have to buy the official dongle to try and update them (assuming an update isn't what broke it)

-------- Original Message --------

serot23, 2024-05-09:
> That's interesting, I wonder if they are running different firmware. Might have to buy the official dongle to try and update them (assuming an update isn't what broke it)
> […](#)
> -------- Original Message --------

I use a sonoff dongle. I don't think that is the problem.

reg98643t, 2024-05-09:
No, I mean the Mercator Ikuu dongle, so I can use their official app to (presumably) update the firmware.

-------- Original Message --------

serot23, 2024-05-09:
> No, I mean the Mercator Ikuu dongle, so I can use their official app to (presumably) update the firmware. Sent from Proton Mail Android
> […](#)
> -------- Original Message --------

Mine is a tuya device maybe im wrong. Thought it was the same device 
[image]

stuartkirkASM, 2024-05-22:
I have a different issue where sometimes the physical controls don't work (pressing to turn on etc), but can operate via HA by setting a dimmer level. Once controlled from HA, it works fine from knob. I think it is somthing to do with the min/max setting which seems to reset itself.

serot23, 2024-05-22:
> I have a different issue where sometimes the physical controls don't work (pressing to turn on etc), but can operate via HA by setting a dimmer level. Once controlled from HA, it works fine from knob. I think it is somthing to do with the min/max setting which seems to reset itself.

Yeah sometimes i need to press it twice to turn it on. And there is sometimes a delay in the dimming or not working but when rotating multiple times it works and than with a small delay. They are great but not always smooth in operating unfortunately

stuartkirkASM, 2024-05-22:
Thanks, but no amount of pressing works for me when this happens. My issue is exactly as decribed in the crossed out text here.   https://community.home-assistant.io/t/mercator-ikuu-dimmers/694678/3?
I might try his solution (once I figure out how to).
It seems to be a common issue that they are slow to respond physically. But are still OK.
I am really hoping these work well as way better to turn a knob than hold a button to dim up/down like others I have.

closed 2024-12-28 by github-actions[bot]