Koenkk/zigbee2mqtt #22514: [Feature request] Last Update time on Dashboard

Ltek, 2024-05-07:
### Is your feature request related to a problem? Please describe

We need to know when each device has last communicated. Z2M frequently disconnects and we need to know when devices have stopped communicating without debugging everything.

### Describe the solution you'd like

Please time stamp each device property on the dashboard, and make visible 'last update' on each dashboard widget for each device

### Describe alternatives you've considered

no reasonable alternative exists

### Additional context

[image]

Great-Chart, 2024-05-07:
This can be activated through Z2M Settings, under Advanced options.  Scroll down (some way) to locate. Mine is set to ISO_8601.
See if this is what you're after here.

Ltek, 2024-05-07:
> This can be activated through Z2M Settings, under Advanced options. Scroll down (some way) to locate. Mine is set to ISO_8601. See if this is what you're after here.
> 
> Here's an example for context.
> 
> ![image](https://private-user-images.githubusercontent.com/101117351/328606020-4d2a7e76-9005-4f6e-bdff-866f5326c7d5.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTUxMTkwOTksIm5iZiI6MTcxNTExODc5OSwicGF0aCI6Ii8xMDExMTczNTEvMzI4NjA2MDIwLTRkMmE3ZTc2LTkwMDUtNGY2ZS1iZGZmLTg2NmY1MzI2YzdkNS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNTA3JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDUwN1QyMTUzMTlaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1kOTRmM2NkYTM3ZTQ3MjExNzFmMGMyMmYyZTE1Mzg4Y2MyNTIxZmMwN2RjODZmZDEzMzFhY2ZkOTViYTRhYzFlJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.8Fbfe_JjLyHsc4R0h05txmANlKc5l89dlevx_7Llc_U)

thank you. -- that setting is very hidden (and not an obvious name) ... it satisfies the overall, check in but not the each parameter last-update-time.

yes, the overall

Great-Chart, 2024-05-08:
If you check Devive Info you'll likely now have additional (hidden) entities that could add value for monitoring selected problem devices.

I'm sure I came across this setting (last seen) through a YouTube channel or similar as inevitably so many things end up slightly hard to find or comprehend.

closed 2024-12-05 by github-actions[bot]

Ltek, 2024-12-06:
still needed