Koenkk/zigbee2mqtt #22515: [New device support]: Tuya ZigBee Smart Door Lock

cloudbr34k84, 2024-05-07:
### Link

https://www.aliexpress.com/item/1005005810135795.html?spm=a2g0o.order_list.order_list_main.49.17fc1802mGzuAG

### Database entry

{"id":120,"type":"EndDevice","ieeeAddr":"0x1c34f1fffe1343f4","nwkAddr":25908,"manufId":0,"manufName":"_TYST12_bfejtdig","powerSource":"Battery","modelId":"TY0A01","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,10,4,5,61184],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"65503":"\u0000\u0000\u0000\u0000\u0012","65506":28,"appVersion":88,"modelId":"TY0A01","manufacturerName":"_TYST12_bfejtdig","powerSource":3,"zclVersion":3,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":88,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1715048650733}

### Comments

There is a product request, but it seems all over the place. I have added this to my tuya gateway. Can provide all the DPs.

Dp 1: Finger Print Unlock
DP 2: Password
DP 3: Temporary Password
DP 4: Dynamic Password
DP 68: Unlock by Offline Password
DP 5: Unlock by card
DP 8: Open or Close
DP 9: Alarm
DP 10: Battery
DP 12: Reverse Lock
DP 14: Doorbell
DP 15: Chose Ring Tone
DP 21: APP Unlock Without Password
DP 22: Answer APP
DP 23: Forbidden APP Unlock
DP 24: Create a Temporary Password
DP 25: Delete Temporary Password
DP 27: Frozen Temporary Password
DP 28: Unfreeze Temporary Password
DP 35: Hijack Alarm
DP 39: Reset temporary password
DP 41: APP Unlock
DP 47: Door Status
DP 48: New remote unlock without password - set key
Dp 49: New remote unlock without password - Safe Solution
DP 54: Create unlock methods from App
DP 55: Delete unlock methods from App
DP 57: Lock motor state
DP 61: Single Use Password
DP 62: Unlock by remote Voice

### External definition

```shell
TBA
```

cloudbr34k84, 2024-05-07:
@Koenkk is there any chance of some help on this?

errellion, 2024-06-09:
Interested in this as well, I just installed very smiliar one - https://pl.aliexpress.com/item/1005005020480943.html?spm=a2g0o.order_list.order_list_main.80.7e8d1c24GhtaHF&gatewayAdapt=glo2pol

It is addedd as unsupoprted:

Type:     EndDevice
Model Zigbee    TY0A01
Producent Zigbee    _TYST12_srptwvak

I've generated generated_external_definition from dev console, but it's very basic, if I could help in testing or make some additional setup (how?) let me know.

[image]

cloudbr34k84, 2024-06-11:
> Interested in this as well, I just installed very smiliar one - https://pl.aliexpress.com/item/1005005020480943.html?spm=a2g0o.order_list.order_list_main.80.7e8d1c24GhtaHF&gatewayAdapt=glo2pol
> 
> It is addedd as unsupoprted:
> 
> Type: EndDevice Model Zigbee TY0A01 Producent Zigbee _TYST12_srptwvak
> 
> I've generated generated_external_definition from dev console, but it's very basic, if I could help in testing or make some additional setup (how?) let me know.
> 
> ![obraz](https://private-user-images.githubusercontent.com/55739518/337970030-f900681b-a8f8-42b6-8688-f1de38aad5bc.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTgxNDU1OTUsIm5iZiI6MTcxODE0NTI5NSwicGF0aCI6Ii81NTczOTUxOC8zMzc5NzAwMzAtZjkwMDY4MWItYThmOC00MmI2LTg2ODgtZjFkZTM4YWFkNWJjLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA2MTElMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNjExVDIyMzQ1NVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTg4NmY3ZjcwZDA3MDJkYmNhMmM2NmU4YzZjYzZmNGVkM2E5NDhkZjQ3ZjhkNzYxMTA0ZmRjNjhhNDc2ZmE3ODQmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.DPlyBWv-PSNYPTEXtHvd2pyc_CIJtn_2-xpPb8i513M)

I dont have experience in this part of Z2M unfortunately

cloudbr34k84, 2024-07-16:
@Koenkk hey mate would  you be able to help with seeing if we can get this type of lock into Z2M. IT would open up more options from the Tuya range?

Koenkk, 2024-07-17:
Since you have the datapoints already, you can start making an external converter, you can use https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/src/devices/tuya.ts for inspiration

cloudbr34k84, 2024-07-18:
@Koenkk 

this is what i have so 

```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const e = exposes.presets;
const ea = exposes.access;

const tuyaLock = {
    fingerprint: [{ modelID: 'TY0A01', manufacturerName: '_TYST12_bfejtdig' }],
    model: 'TY0A01',
    vendor: 'Tuya',
    description: 'Tuya based lock with various unlock methods',
    fromZigbee: [fz.tuya_lock],
    toZigbee: [tz.tuya_lock],
    exposes: [
        e.lock(),
        e.battery(),
        e.boolean('fingerprint_unlock', ea.ALL),
        e.boolean('password_unlock', ea.ALL),
        e.boolean('temporary_password_unlock', ea.ALL),
        e.boolean('dynamic_password_unlock', ea.ALL),
        e.boolean('offline_password_unlock', ea.ALL),
        e.boolean('card_unlock', ea.ALL),
        e.binary('open_close', ea.ALL),
        e.binary('alarm', ea.ALL),
        e.enum('reverse_lock', ea.ALL, ['ON', 'OFF']),
        e.boolean('doorbell', ea.ALL),
        e.text('ring_tone', ea.ALL),
        e.boolean('app_unlock_without_password', ea.ALL),
        e.boolean('answer_app', ea.ALL),
        e.boolean('forbidden_app_unlock', ea.ALL),
        e.binary('create_temporary_password', ea.ALL),
        e.binary('delete_temporary_password', ea.ALL),
        e.binary('freeze_temporary_password', ea.ALL),
        e.binary('unfreeze_temporary_password', ea.ALL),
        e.binary('hijack_alarm', ea.ALL),
        e.binary('reset_temporary_password', ea.ALL),
        e.boolean('app_unlock', ea.ALL),
        e.binary('door_status', ea.ALL),
        e.binary('new_remote_unlock_without_password_set_key', ea.ALL),
        e.binary('new_remote_unlock_without_password_safe_solution', ea.ALL),
        e.binary('create_unlock_methods_from_app', ea.ALL),
        e.binary('delete_unlock_methods_from_app', ea.ALL),
        e.binary('lock_motor_state', ea.ALL),
        e.binary('single_use_password', ea.ALL),
        e.binary('unlock_by_remote_voice', ea.ALL)
    ],
};

module.exports = tuyaLock;

```

cloudbr34k84, 2024-08-07:
@Koenkk any feedback mate, what should i bee looking for when testing this?

Koenkk, 2024-08-07:
Does your external converter work?

cloudbr34k84, 2024-08-09:
> Does your external converter work?
@Koenkk 
Trying now,. i also look all through the tuya.js file and couldnt find oither examples of actual door locks.
{"id":174,"type":"EndDevice","ieeeAddr":"0x1c34f1fffe1343f4","nwkAddr":49413,"manufId":0,"manufName":"_TYST12_bfejtdig","powerSource":"Battery","modelId":"TY0A01","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,10,4,5,61184],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"65503":"�2I.\u0012�2I.\u0012�2I.\u0012�2I.\u0013�2I.\u0012�2I.\u0012�2I.\u0012�2I.\u0011","65506":28,"appVersion":88,"modelId":"TY0A01","manufacturerName":"_TYST12_bfejtdig","powerSource":3,"zclVersion":3,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":88,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1723198299091}

cloudbr34k84, 2024-08-09:
@Koenkk I ahve even stripped it right back and im getting an error ```[2024-08-09 20:47:28] error: 	z2m: Failed to load external converter file 'tuya_lock_converter.js' (Cannot read properties of undefined (reading 'options'))```

```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['TY0A01'],
    model: 'TY0A01',
    vendor: '_TYST12_bfejtdig',
    description: 'Tuya based lock with various unlock methods',
    fromZigbee: [fz.tuya_lock],
    toZigbee: [tz.tuya_lock],
    exposes: [
        e.lock(),
        e.numeric('battery', ea.STATE).withUnit('%').withDescription('Battery level'),
    ],
    meta: {},
};

module.exports = definition;
```

Koenkk, 2024-08-09:
That because fz and tz tuya lock doesnt exist

cloudbr34k84, 2024-08-09:
> That because fz and tz tuya lock doesnt exist

Ok so now what??

Koenkk, 2024-08-11:
Try with tuya.tz.datapoints and with the converter from https://github.com/Koenkk/zigbee2mqtt/issues/22515#issuecomment-2235313134

cloudbr34k84, 2024-08-11:
> Try with tuya.tz.datapoints and with the converter from [#22515 (comment)](https://github.com/Koenkk/zigbee2mqtt/issues/22515#issuecomment-2235313134)

Tried that, 

```
z2m: Failed to load external converter file 'tuya_lock_converter.js' (Cannot find module 'zigbee-herdsman converters/lib/extend'
Require stack:
- /app/dist/util/utils.js
- /app/dist/util/settings.js
- /app/index.js)
[2024-08-12 07:52:10] error: 	z2m: Probably there is a syntax error in the file or the external converter is not compatible with the current Zigbee2MQTT version ```

Koenkk, 2024-08-12:
With https://gist.github.com/Koenkk/8a6844676eb93872707aa7d79bcadd03 it should start, make sure that you are on the latest z2m dev.

cloudbr34k84, 2024-08-13:
OK that's concerter has worked and applied to the lock. Now how do I figure out how test? is there events to listen to when I use the lock?

Koenkk, 2024-08-13:
Try to control it via the exposes tab, also check if the exposes tab updates if you control the lock physically

kingp0dd, 2024-08-13:
Hold on, this is working now? Been following multiple threads of TY0A01
here in GitHub.

jhbcabral, 2024-08-18:
Hi guys,

I don’t know if that helps, but I have a P6EL panel from smatek which comes with Tuya zigbee hub on it. I have a full root access to the panel and could get any information from the logs and files you might need. I can see several keys from Tuya on the device, and I happen to have one of those TY01 locks as well. If such combination can help in any form let me know why should I look for and I am glad to help.

thanks!

cloudbr34k84, 2024-08-24:
> Try to control it via the exposes tab, also check if the exposes tab updates if you control the lock physically

Hey @Koenkk sorry its been awhile. Im unable to control or make any changes via the exposes tab. So i dont keep asking you what to do, is there a process of sorts?

cloudbr34k84, 2024-08-24:
> Hi guys,
>
> I don’t know if that helps, but I have a P6EL panel from smatek which comes with Tuya zigbee hub on it. I have a full root access to the panel and could get any information from the logs [...]

What are the keys for?

KodieM, 2024-08-25:
I've punch this into my config and can see everything exposed, but nothing in the state tab. I did some fiddling to dump the data from msg to give me a better idea of whats going on, which im guessing encouraged and update and all the states showed but with null values. 
I tried cutting it down to the following datapoints which I've found myself on the Tuya Developer site with no luck
1 - Fingerprint unlock
8 - Open or close
9 - Alarm
10 - Battery
14 - Doorbell

cloudbr34k84, 2024-08-25:
> I've punched this into my config and can see everything exposed, but nothing in the state tab. I did some fiddling to dump the data from msg to give me a better idea of whats going on, which im guessing encouraged and update and all the states showed but with null values. 
> I tried cutting it down to the following datapoints which I've found myself on the Tuya Developer site with no luck
> 1 - Fingerprint unlock
> 8 - Open or close
> 9 - Alarm
> 10 - Battery
> 14 - Doorbell
> 

Do you want to reshare that external converter you created?

KodieM, 2024-08-25:
Here it is. It's basically just adding the msg data and dumping it into a state.

```const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const e = exposes.presets;
const ea = exposes.access;

const getMsg = {
    cluster: 'genBasic',
    type: ['attributeReport', 'readResponse'],
    convert: (model, msg, publish, options, meta) => {
        return {
            msgData: msg,
        };
    },
};

const tuyaLock = {
    fingerprint: [{modelID: 'TY0A01', manufacturerName: '_TYST12_bfejtdig'}],
    model: 'TY0A01',
    vendor: 'Tuya',
    description: 'Tuya based lock with various unlock methods',
    fromZigbee: [tuya.fz.datapoints, getMsg],
    toZigbee: [tuya.tz.datapoints],
    exposes: [
        e.lock(),
        e.numeric('battery', ea.STATE).withUnit('%').withDescription('Battery level'),
        e.text('alarm', ea.STATE).withDescription('Alarm'),
        e.binary('fingerprint_unlock', ea.STATE_SET, 'ON', 'OFF').withDescription('Fingerprint Unlock'),
        e.binary('open_close', ea.STATE_SET, 'ON', 'OFF').withDescription('Open or Close'),
        e.binary('doorbell', ea.STATE_SET, 'ON', 'OFF').withDescription('Doorbell'),
    ],
    meta: {
        tuyaDatapoints: [
            [1, 'fingerprint_unlock', tuya.valueConverter.trueFalse],
            [8, 'open_close', tuya.valueConverter.trueFalse],
            [9, 'alarm', tuya.valueConverter.trueFalse],
            [10, 'battery', tuya.valueConverter.batteryPercentage],
            [14, 'doorbell', tuya.valueConverter.trueFalse],
        ],
    },
};

module.exports = tuyaLock;

cloudbr34k84, 2024-08-31:
> const tz = require('zigbee-herdsman-converters/converters/toZigbee');
> const exposes = require('zigbee-herdsman-converters/lib/exposes');
> const tuya = require('zigbee-herdsman-converters/lib/tuya'); [...]

Cheers, 

in my State tabe

i now see this
```
{
    "last_seen": "2024-08-31T12:45:11+10:00",
    "linkquality": 69,
    "alarm": null,
    "battery": null,
    "doorbell": null,
    "fingerprint_unlock": null,
    "open_close": null,
    "state": null,
    "msgData": {
        "cluster": "genBasic",
        "data": {
            "65503": "��e.\u0013��e.\u0012��e.\u0012��e.\u0012",
            "65506": 28,
            "appVersion": 88
        },
        "device": {
            "ID": 174,
            "_applicationVersion": 88,
            "_customClusters": {},
            "_dateCode": "",
            "_endpoints": [
                {
                    "ID": 1,
                    "_binds": [],
                    "_configuredReportings": [],
                    "_events": {},
                    "_eventsCount": 0,
                    "clusters": {
                        "genBasic": {
                            "attributes": {
                                "65503": "��e.\u0013��e.\u0012��e.\u0012��e.\u0012",
                                "65506": 28,
                                "appVersion": 88,
                                "dateCode": "",
                                "hwVersion": 1,
                                "manufacturerName": "_TYST12_bfejtdig",
                                "modelId": "TY0A01",
                                "powerSource": 3,
                                "stackVersion": 0,
                                "zclVersion": 3
                            }
                        }
                    },
                    "deviceID": 81,
                    "deviceIeeeAddress": "0x1c34f1fffe1343f4",
                    "deviceNetworkAddress": 1410,
                    "inputClusters": [
                        0,
                        10,
                        4,
                        5,
                        61184
                    ],
                    "meta": {},
                    "outputClusters": [
                        25
                    ],
                    "pendingRequests": {
                        "ID": 1,
                        "deviceIeeeAddress": "0x1c34f1fffe1343f4",
                        "sendInProgress": false
                    },
                    "profileID": 260
                }
            ],
            "_events": {},
            "_eventsCount": 0,
            "_hardwareVersion": 1,
            "_ieeeAddr": "0x1c34f1fffe1343f4",
            "_interviewCompleted": true,
            "_interviewing": false,
            "_lastDefaultResponseSequenceNumber": 52,
            "_lastSeen": 1725072308039,
            "_linkquality": 72,
            "_manufacturerID": 0,
            "_manufacturerName": "_TYST12_bfejtdig",
            "_modelID": "TY0A01",
            "_networkAddress": 1410,
            "_pendingRequestTimeout": 0,
            "_powerSource": "Battery",
            "_skipDefaultResponse": false,
            "_stackVersion": 0,
            "_type": "EndDevice",
            "_zclVersion": 3,
            "meta": {}
        },
        "endpoint": {
            "ID": 1,
            "_binds": [],
            "_configuredReportings": [],
            "_events": {},
            "_eventsCount": 0,
            "clusters": {
                "genBasic": {
                    "attributes": {
                        "65503": "��e.\u0013��e.\u0012��e.\u0012��e.\u0012",
                        "65506": 28,
                        "appVersion": 88,
                        "dateCode": "",
                        "hwVersion": 1,
                        "manufacturerName": "_TYST12_bfejtdig",
                        "modelId": "TY0A01",
                        "powerSource": 3,
                        "stackVersion": 0,
                        "zclVersion": 3
                    }
                }
            },
            "deviceID": 81,
            "deviceIeeeAddress": "0x1c34f1fffe1343f4",
            "deviceNetworkAddress": 1410,
            "inputClusters": [
                0,
                10,
                4,
                5,
                61184
            ],
            "meta": {},
            "outputClusters": [
                25
            ],
            "pendingRequests": {
                "ID": 1,
                "deviceIeeeAddress": "0x1c34f1fffe1343f4",
                "sendInProgress": false
            },
            "profileID": 260
        },
        "groupID": 0,
        "linkquality": 72,
        "meta": {
            "frameControl": {
                "direction": 1,
                "disableDefaultResponse": false,
                "frameType": 0,
                "manufacturerSpecific": false,
                "reservedBits": 0
            },
            "zclTransactionSequenceNumber": 52
        },
        "type": "attributeReport"
    }
}
```

jhbcabral, 2024-09-04:
Did it work? were you able to actually control the lock?

cloudbr34k84, 2024-09-04:
> Did it work? were you able to actually control the lock?

Na nothing, not even the battery state is showing

KodieM, 2024-09-04:
> Did it work? were you able to actually control the lock?

Unfortunately not. After a day the states from the datapoints just vanish from the state tab, probably because they never actually get updated and just fall off. I'm guessing that the datapoints aren't marrying up correctly and this door lock is just doing something weird.

kingp0dd, 2024-10-04:
> Here it is. It's basically just adding the msg data and dumping it into a state.
> 
> ```
> const tz = require('zigbee-herdsman-converters/converters/toZigbee');
> const exposes = require('zigbee-herdsman-converters/lib/exposes');
> const tuya = require('zigbee-herdsman-converters/lib/tuya');
> const e = exposes.presets;
> const ea = exposes.access;
> 
> const getMsg = {
>     cluster: 'genBasic',
>     type: ['attributeReport', 'readResponse'],
>     convert: (model, msg, publish, options, meta) => {
>         return {
>             msgData: msg,
>         };
>     },
> };
> 
> 
> const tuyaLock = {
>     fingerprint: [{modelID: 'TY0A01', manufacturerName: '_TYST12_bfejtdig'}],
>     model: 'TY0A01',
>     vendor: 'Tuya',
>     description: 'Tuya based lock with various unlock methods',
>     fromZigbee: [tuya.fz.datapoints, getMsg],
>     toZigbee: [tuya.tz.datapoints],
>     exposes: [
>         e.lock(),
>         e.numeric('battery', ea.STATE).withUnit('%').withDescription('Battery level'),
>         e.text('alarm', ea.STATE).withDescription('Alarm'),
>         e.binary('fingerprint_unlock', ea.STATE_SET, 'ON', 'OFF').withDescription('Fingerprint Unlock'),
>         e.binary('open_close', ea.STATE_SET, 'ON', 'OFF').withDescription('Open or Close'),
>         e.binary('doorbell', ea.STATE_SET, 'ON', 'OFF').withDescription('Doorbell'),
>     ],
>     meta: {
>         tuyaDatapoints: [
>             [1, 'fingerprint_unlock', tuya.valueConverter.trueFalse],
>             [8, 'open_close', tuya.valueConverter.trueFalse],
>             [9, 'alarm', tuya.valueConverter.trueFalse],
>             [10, 'battery', tuya.valueConverter.batteryPercentage],
>             [14, 'doorbell', tuya.valueConverter.trueFalse],
>         ],
>     },
> };
> 
> module.exports = tuyaLock;
> ```

May i ask if you were able to at least see the states (if door is locked or not) using this?

KodieM, 2024-10-04:
> May i ask if you were able to at least see the states (if door is locked or not) using this?

Na, nothing at all unfortunately.

cloudbr34k84, 2024-10-04:
We really need some help on this

KodieM, 2024-10-04:
> We really need some help on this

I really just hoping someone smarter will jump in at this point and figure it out. I'm at a loss on this one.

dimacoupe, 2024-10-30:
I'm bought and install moes smart door lock https://a.aliexpress.com/_EID9FPB and it same manufacturer _TYST12_bfejtdig .

cloudbr34k84, 2024-10-30:
Are you saying it works?

dimacoupe, 2024-10-30:
> Are you saying it works?

in tuya only. z2m not supported(

cloudbr34k84, 2024-10-30:
oh yeah we all got that

dandek01, 2024-10-31:
I bought smart lock _TYST12_j4xsdikm. It can be added to z2m but it is also unsupported like all the others. As i bought it directly from factory I asked them if they can, maybe, provide external converter as this is not working in z2m. Will see what will they say.

DarkAztek680, 2024-11-23:
[2024-11-22 22:33:48] debug: 	zh:controller:endpoint: Error: ZCL command 0xa46dd4fffe967ef4/1 genBasic.defaultRsp({"cmdId":10,"statusCode":0}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"reservedBits":0,"transactionSequenceNumber":51,"writeUndiv":false}) failed (Data request failed with error: 'No network route' (205))
    at ZStackAdapter.sendZclFrameToEndpointInternal (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:524:23)
    at ZStackAdapter.sendZclFrameToEndpointInternal (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:619:24)
    at ZStackAdapter.sendZclFrameToEndpointInternal (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:619:24)
    at ZStackAdapter.sendZclFrameToEndpointInternal (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:619:24)
    at ZStackAdapter.sendZclFrameToEndpointInternal (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:619:24)
    at /app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:422:20
    at Queue.execute (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:36:20)
    at ZStackAdapter.sendZclFrameToEndpoint (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:420:16)
    at Request.send (/app/node_modules/zigbee-herdsman/src/controller/helpers/request.ts:82:20)
    at Endpoint.sendRequest (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:325:20)

DarkAztek680, 2024-11-23:
[2024-11-22 22:34:30] debug: 	z2m:mqtt: Received MQTT message on 'zigbee2mqtt/0xa46dd4fffe967ef4/set' with data '{"state":"UNLOCK"}'
[2024-11-22 22:34:30] debug: 	z2m: Publishing 'set' 'state' to '0xa46dd4fffe967ef4'
[2024-11-22 22:34:30] error: 	z2m: Publish 'set' 'state' to '0xa46dd4fffe967ef4' failed: 'Error: No datapoint defined for 'state''
[2024-11-22 22:34:30] debug: 	z2m: Error: No datapoint defined for 'state'
    at Object.convertSet (/app/node_modules/zigbee-herdsman-converters/src/lib/tuya.ts:1474:27)
    at Publish.onMQTTMessage (/app/lib/extension/publish.ts:306:52)
    at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:206:23)
    at EventEmitter.emit (node:events:531:35)
    at EventBus.emitMQTTMessage (/app/lib/eventBus.ts:144:22)
    at MQTT.onMessage (/app/lib/mqtt.ts:150:27)
    at WebSocket.<anonymous> (/app/lib/extension/frontend.ts:163:27)
    at WebSocket.emit (node:events:519:28)
    at Receiver.receiverOnMessage (/app/node_modules/ws/lib/websocket.js:1220:20)
    at Receiver.emit (node:events:519:28)
[2024-11-22 22:34:32] debug: 	z2m:mqtt: Received MQTT message on 'zigbee2mqtt/0xa46dd4fffe967ef4/set' with data '{"state":"LOCK"}'
[2024-11-22 22:34:32] debug: 	z2m: Publishing 'set' 'state' to '0xa46dd4fffe967ef4'
[2024-11-22 22:34:32] error: 	z2m: Publish 'set' 'state' to '0xa46dd4fffe967ef4' failed: 'Error: No datapoint defined for 'state''
[2024-11-22 22:34:32] debug: 	z2m: Error: No datapoint defined for 'state'
    at Object.convertSet (/app/node_modules/zigbee-herdsman-converters/src/lib/tuya.ts:1474:27)
    at Publish.onMQTTMessage (/app/lib/extension/publish.ts:306:52)
    at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:206:23)
    at EventEmitter.emit (node:events:531:35)
    at EventBus.emitMQTTMessage (/app/lib/eventBus.ts:144:22)
    at MQTT.onMessage (/app/lib/mqtt.ts:150:27)
    at WebSocket.<anonymous> (/app/lib/extension/frontend.ts:163:27)
    at WebSocket.emit (node:events:519:28)
    at Receiver.receiverOnMessage (/app/node_modules/ws/lib/websocket.js:1220:20)
    at Receiver.emit (node:events:519:28)

almeidafb, 2024-11-30:
Hello Everyone,
I've just got a Smart Door Lock TY0A01 (NovaDigital brand).
Trying to setup it in the Home Assistant, I found this topic, now I'll join you in this journey.
I've never configured a unsupported device before, but I will try to help with that.

I've started to create a new device, based in this model at Tuya's Developer Plataform, just trying to get some information about it... and got a configuration file that describes the datapoints like that:

`{
            "attr": 1280,
            "code": "unlock_fingerprint",
            "defaultValue": "",
            "desc": "【可选】指纹在门锁上录入，录入完成后由硬件分配id，发生指纹解锁后，门锁将上报该dp点。请用记录型上报\n【可选】指纹也可通过远程下发",
            "editPermission": true,
            "extContent": "{\"trigger\":\"direct\"}",
            "iconname": "icon-dp_lock",
            "id": 1,
            "mode": "ro",
            "name": "Fingerprint Unlock",
            "property": {
                "max": 999,
                "min": 0,
                "scale": 0,
                "step": 1,
                "type": "value"
            },
            "subType": "value",
            "trigger": "direct",
            "type": "obj"
        },`

(...) all others (...)

`{
            "attr": 0,
            "code": "blacklist_set",
            "defaultValue": "",
            "desc": "离线发卡管理专用",
            "editPermission": false,
            "extContent": "",
            "iconname": "icon-dp_i",
            "id": 61,
            "mode": "rw",
            "name": "Blacklist Set",
            "property": {
                "maxlen": 128,
                "type": "raw"
            },
            "subType": "raw",
            "type": "raw"
        }`
        
It helps some kind of way? If it helps, I can send it here.

I still not able to do a simple thing like getting battery percentage, even when I post a message directly in the broker, but I'll get there at some point, I let you know any updates.

cloudbr34k84, 2024-12-02:
when im back from holidays ill restart this project, but we need someone who knows how these things work.

DarkAztek680, 2024-12-06:
> Can you try with: `[1, 'state', tuya.valueConverter.trueFalse],` and add `e.binary('state', ea.STATE_SET, true, false)` in `exposes` and see if that works?

emimor123, 2025-01-01:
Hi, any advance on this locks?

irzayev, 2025-02-02:
Got _TYST12_srptwvak from ali, any one did it work in z2m? I just need to get the status

Keele520, 2025-03-05:
Hi,

I'm using Tuya ZigBee Smart Door Lock TY0A01 (_TYST12_szmugcn3), wish it work in z2m too.

here https://github.com/Koenkk/zigbee2mqtt/issues/4478#issuecomment-1807222064 is the similar topic.

If I understand correctly, it not working is because communication between tuya gateway and door lock is encrypted, and z2m unable to decrypt it.

And sniff the zigbee signal know that the gateway sends a 0x21 and 0x22 "unknow" command to the device, not sure it is related or not, looking for help/information.

boemekeld, 2025-03-13:
@almeidafb I'm in the same situation here, with a NovaDigital model (manufacturerName: '_TYST12_cmwaswx7').

The best I managed, using TY0A01.js as an external converter, was capturing the LQI and the status (lock/unlock). However, I must admit that I'm not entirely sure about the status reading, as it doesn't occur when unlocking the door but rather periodically.

[image]

[TY0A01.txt](https://github.com/user-attachments/files/19237273/TY0A01.txt)

```info 2025-03-13 18:37:42z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0x0ceff6fffe744068', payload '{"action":"unlock","action_source":105,"action_user":101,"battery":null,"linkquality":120,"state":"LOCK","update":{"installed_version":-1,"latest_version":-1,"state":null}}'
info 2025-03-13 18:37:42z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0x0ceff6fffe744068/action', payload 'unlock'
info 2025-03-13 18:37:44z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0x0ceff6fffe744068', payload '{"action":"lock","battery":null,"linkquality":120,"state":"LOCK","update":{"installed_version":-1,"latest_version":-1,"state":null}}'
info 2025-03-13 18:37:44z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0x0ceff6fffe744068/action', payload 'lock'
info 2025-03-13 18:40:02z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0x0ceff6fffe744068', payload '{"action":"unlock","action_source":105,"action_user":101,"battery":null,"linkquality":87,"state":"LOCK","update":{"installed_version":-1,"latest_version":-1,"state":null}}'
info 2025-03-13 18:40:02z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0x0ceff6fffe744068/action', payload 'unlock'
info 2025-03-13 18:40:04z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0x0ceff6fffe744068', payload '{"action":"lock","battery":null,"linkquality":87,"state":"LOCK","update":{"installed_version":-1,"latest_version":-1,"state":null}}'
info 2025-03-13 18:40:04z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0x0ceff6fffe744068/action', payload 'lock'
info 2025-03-13 18:42:17z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0x0ceff6fffe744068', payload '{"action":"unlock","action_source":105,"action_user":101,"battery":null,"linkquality":105,"state":"LOCK","update":{"installed_version":-1,"latest_version":-1,"state":null}}'
info 2025-03-13 18:42:17z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0x0ceff6fffe744068/action', payload 'unlock'
info 2025-03-13 18:42:19z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0x0ceff6fffe744068', payload '{"action":"lock","battery":null,"linkquality":105,"state":"LOCK","update":{"installed_version":-1,"latest_version":-1,"state":null}}'
info 2025-03-13 18:42:19z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0x0ceff6fffe744068/action', payload 'lock'

almeidafb, 2025-03-19:
@Koenkk , is this the same as issue #4478 ? Should we continue the discussion there?

closed 2025-03-20 by Koenkk