Koenkk/zigbee2mqtt #22516: Ember driver warings

major328, 2024-05-07:
### What happened?

After switched to ember driver I got this messages:
"[2024-05-04 20:51:28] warning: 	zh:ember:ezsp: Received network/route error ROUTE_ERROR_MANY_TO_ONE_ROUTE_FAILURE for "14594".
[2024-05-04 20:51:43] warning: 	zh:ember:ezsp: Received network/route error ROUTE_ERROR_MANY_TO_ONE_ROUTE_FAILURE for "17127"."

### What did you expect to happen?

information how to fix this

### How to reproduce it (minimal and precise)

switched from ezsp to ember driver

### Zigbee2MQTT version

 1.37.0 

### Adapter firmware version

7.4.1.0 build 0

### Adapter

Sonoff ZBDongle-E

### Setup

Add-on on HomeAssistant

rinne, 2024-05-07:
Same to me.  Also random devices emit intermittently the following:

```
[2024-05-07 12:35:49] warning:  zh:ember:ezsp: Received network/route error ROUTE_ERROR_SOURCE_ROUTE_FAILURE for "30294".
[2024-05-07 12:35:51] warning:  zh:ember:ezsp: Received network/route error ROUTE_ERROR_SOURCE_ROUTE_FAILURE for "30294".
[2024-05-07 12:40:39] warning:  zh:ember:ezsp: Received network/route error ROUTE_ERROR_SOURCE_ROUTE_FAILURE for "52510".
[2024-05-07 12:40:39] warning:  zh:ember:ezsp: Received network/route error ROUTE_ERROR_SOURCE_ROUTE_FAILURE for "52510".
[2024-05-07 12:40:39] warning:  zh:ember:ezsp: Received network/route error ROUTE_ERROR_SOURCE_ROUTE_FAILURE for "52510".
[2024-05-07 12:40:39] warning:  zh:ember:ezsp: Received network/route error ROUTE_ERROR_SOURCE_ROUTE_FAILURE for "52510".
[2024-05-07 12:40:49] warning:  zh:ember:ezsp: Received network/route error ROUTE_ERROR_SOURCE_ROUTE_FAILURE for "30294".
[2024-05-07 12:40:50] warning:  zh:ember:ezsp: Received network/route error ROUTE_ERROR_SOURCE_ROUTE_FAILURE for "30294".
[2024-05-07 12:40:52] warning:  zh:ember:ezsp: Received network/route error ROUTE_ERROR_SOURCE_ROUTE_FAILURE for "30294".
[2024-05-07 12:45:30] warning:  zh:ember:ezsp: Received network/route error ROUTE_ERROR_SOURCE_ROUTE_FAILURE for "12985".
[2024-05-07 12:45:30] warning:  zh:ember:ezsp: Received network/route error ROUTE_ERROR_MANY_TO_ONE_ROUTE_FAILURE for "12985".
[2024-05-07 12:45:30] warning:  zh:ember:ezsp: Received network/route error ROUTE_ERROR_MANY_TO_ONE_ROUTE_FAILURE for "12985".
[2024-05-07 12:45:30] warning:  zh:ember:ezsp: Received network/route error ROUTE_ERROR_MANY_TO_ONE_ROUTE_FAILURE for "12985".
[2024-05-07 12:45:30] warning:  zh:ember:ezsp: Received network/route error ROUTE_ERROR_MANY_TO_ONE_ROUTE_FAILURE for "12985".
[2024-05-07 12:45:39] warning:  zh:ember:ezsp: Received network/route error ROUTE_ERROR_SOURCE_ROUTE_FAILURE for "52510".
```

BTW, it's annoying that the device address is in base-10 on these messages but hex in UI.

Many to one for me seems to be related to network scan.

```
[2024-05-07 12:46:32] info:     z2m: Starting network scan (includeRoutes 'false')
[2024-05-07 12:46:45] warning:  zh:ember:ezsp: Received network/route error ROUTE_ERROR_MANY_TO_ONE_ROUTE_FAILURE for "41626".
[2024-05-07 12:46:54] info:     z2m: Network scan finished
```

I have the same versions and same zb-stick as @major328

LookInIs, 2024-05-07:
Zigbee2MQTT version [1.37.0] 
Coordinator type EmberZNet, Sonoff ZBDongle-E
Coordinator revision 7.4.1 [GA]
Add-on on HomeAssistant

**Network**
**Total 27**

**By device type**
- End devices: 20
- Router: 7
- By power source
- Mains (single phase): 13
- Battery: 13
- DC Source: 1

**By vendor**
- LUMI: 7
- _TZ3000_dlp6yvs8: 4
- _TZ3000_gvn91tmx: 2
- Slacky-DIY: 2
- IKEA of Sweden: 2
- _TZ3000_qp7x8u3a: 1
- _TZ3000_hafsqare: 1
- _TZ3000_wkai4ga5: 1
- _TZ3000_jl7qyupf: 1
- _TZ3000_ji4araar: 1
- _TZE200_znbl8dj5: 1
- _TZ3000_qewo8dlz: 1
- _TZE200_yvx5lh6k: 1
- _TZ3000_mx3vgyea: 1
- _TZ3210_95txyzbx: 1

**By model**
- TS0012: 5
- lumi.weather: 3
- TS0013: 2
- TS0011: 2
- TS0601: 2
- TS011F: 2
- Watermeter_TLSR8258: 2
- TS0044: 1
- lumi.sensor_motion.aq2: 1
- lumi.light.aqcn02: 1
- TRADFRI remote control: 1
- TRADFRI transformer 30W: 1
- TS000F: 1
- TS0503B: 1
- lumi.sensor_magnet.aq2: 1
- lumi.remote.b1acn01: 1

after switching to ember, I have a lot of messages of this type in the logs:
```
[2024-05-07 15:26:54] warning: 	zh:ember:ezsp: Received network/route error ROUTE_ERROR_MANY_TO_ONE_ROUTE_FAILURE for "40134".
[2024-05-07 15:26:56] warning: 	zh:ember:ezsp: Received network/route error ROUTE_ERROR_MANY_TO_ONE_ROUTE_FAILURE for "40134".
[2024-05-07 15:26:56] warning: 	zh:ember:ezsp: Received network/route error ROUTE_ERROR_MANY_TO_ONE_ROUTE_FAILURE for "40134".
[2024-05-07 15:31:17] warning: 	zh:ember:ezsp: Received network/route error ROUTE_ERROR_MANY_TO_ONE_ROUTE_FAILURE for "52347".
[2024-05-07 15:34:24] warning: 	zh:ember:ezsp: Received network/route error ROUTE_ERROR_NON_TREE_LINK_FAILURE for "52347".
[2024-05-07 15:36:26] warning: 	zh:ember:ezsp: Received network/route error ROUTE_ERROR_MANY_TO_ONE_ROUTE_FAILURE for "1445".
[2024-05-07 15:36:41] warning: 	zh:ember:ezsp: Received network/route error ROUTE_ERROR_MANY_TO_ONE_ROUTE_FAILURE for "1445".
[2024-05-07 15:38:10] warning: 	zh:ember:ezsp: Received network/route error ROUTE_ERROR_MANY_TO_ONE_ROUTE_FAILURE for "62905".
[2024-05-07 15:40:24] warning: 	zh:ember:ezsp: Received network/route error ROUTE_ERROR_MANY_TO_ONE_ROUTE_FAILURE for "52347".
[2024-05-07 15:41:26] warning: 	zh:ember:ezsp: Received network/route error ROUTE_ERROR_MANY_TO_ONE_ROUTE_FAILURE for "1445".
[2024-05-07 15:41:34] warning: 	zh:ember:ezsp: Received network/route error ROUTE_ERROR_MANY_TO_ONE_ROUTE_FAILURE for "52347".
[2024-05-07 15:41:36] warning: 	zh:ember:ezsp: Received network/route error ROUTE_ERROR_MANY_TO_ONE_ROUTE_FAILURE for "52347".
[2024-05-07 15:41:41] warning: 	zh:ember:ezsp: Received network/route error ROUTE_ERROR_MANY_TO_ONE_ROUTE_FAILURE for "1445".
[2024-05-07 15:41:55] warning: 	z2m: Failed to ping 'Лампа' (attempt 1/2, ZCL command 0x00158d0007fff5f4/1 genBasic.read(["zclVersion"], {"timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed ({"target":65222,"apsFrame":{"profileId":260,"clusterId":0,"sourceEndpoint":1,"destinationEndpoint":1,"options":4416,"groupId":0,"sequence":222},"zclSequence":86,"commandIdentifier":1} timed out after 10000ms))
[2024-05-07 15:44:01] warning: 	zh:ember:ezsp: Received network/route error ROUTE_ERROR_MANY_TO_ONE_ROUTE_FAILURE for "29052".
[2024-05-07 15:44:01] warning: 	zh:ember:ezsp: Received network/route error ROUTE_ERROR_MANY_TO_ONE_ROUTE_FAILURE for "29052".
[2024-05-07 15:44:01] warning: 	zh:ember:ezsp: Received network/route error ROUTE_ERROR_MANY_TO_ONE_ROUTE_FAILURE for "29052".
[2024-05-07 15:44:02] warning: 	zh:ember:ezsp: Received network/route error ROUTE_ERROR_MANY_TO_ONE_ROUTE_FAILURE for "29052".
[2024-05-07 15:44:57] warning: 	zh:ember:ezsp: Received network/route error ROUTE_ERROR_MANY_TO_ONE_ROUTE_FAILURE for "52347".
[2024-05-07 15:45:00] warning: 	zh:ember:ezsp: Received network/route error ROUTE_ERROR_MANY_TO_ONE_ROUTE_FAILURE for "52347".
[2024-05-07 15:45:00] warning: 	zh:ember:ezsp: Received network/route error ROUTE_ERROR_MANY_TO_ONE_ROUTE_FAILURE for "52347".
[2024-05-07 15:45:00] warning: 	zh:ember:ezsp: Received network/route error ROUTE_ERROR_MANY_TO_ONE_ROUTE_FAILURE for "52347".
[2024-05-07 15:46:26] warning: 	zh:ember:ezsp: Received network/route error ROUTE_ERROR_MANY_TO_ONE_ROUTE_FAILURE for "1445".
[2024-05-07 15:47:03] warning: 	zh:ember:ezsp: Received network/route error ROUTE_ERROR_MANY_TO_ONE_ROUTE_FAILURE for "40134".
```

I also noticed that the stability of the network has become worse, the reaction time is longer.
I read somewhere that this behavior may be related to the peculiarities of the TRADFRI transformer 30W (router)

thewan056, 2024-05-07:
> * TS011F: 2

This all began when I changed from ezsp to ember.

I started noticing this error after my Tuya Smart plug (with power monitoring by polling) stopped reporting power. Instead of showing any values, anything related to power (Power, Current, Voltage, Energy) shows "N/A" in zb2mqtt. However, the other functions, such as on/off state, power outage memory and indicator mode for said plug in zb2mqtt worked fine. I could remotely control the plug, but there was no power monitoring.

After removing and attempting to readd the smart plug to my network, the ROUTE_ERROR_MANY_TO_ONE_ROUTE_FAILURE appeared, although the plug got added to my zb network just fine, leading me to believe it has something to do with the smart plug. I attempted to troubleshoot by removing the smartplug, then instead of permit join all, i selected routers that were closer to the smart plug. I started getting another error that the device was blacklisted. Then I took the smart plug, moved it closer to my coordinator and permit join on the coordinator only. The ROUTE_ERROR_MANY_TO_ONE_ROUTE_FAILURE appeared, the smart plug joined my network and the power values still show N/A.

I switched back to ezsp. The errors stopped appearing, and my smartplug now works normally with the power reporting values instead of N/A. 

I conclude that ember has some kind of compatibility issue with this smartplug. 
https://www.zigbee2mqtt.io/devices/BSD29_1.html#tuya-bsd29_1

Zigbee Model
TS011F
Zigbee Manufacturer
_TZ3000_okaz9tjs
Description
Smart plug (with power monitoring by polling)

Nerivec, 2024-05-07:
@thewan056 Can you open a new issue for that device with missing power monitoring and provide a `debug` log of the pairing procedure?

rinne, 2024-05-08:
> For network/route errors: a few of these are normal, it's just the network living it's life. You can also get several during z2m start or during a join (again, just the network making up it's mind 😉). It can however indicate a problem when you start seeing lots of these during normal operation, especially if from the same device, or from devices in the same area (it can mean a router is not doing its job properly).

In my network of 39 switches, temperature sensors, smoke alarms and some end-device-switches, I've now used Ember for 24 hours and it seems to work better than ezsp ever did and all devices are reachable and they seem to work.

I see 2900 ROUTE_ERROR_SOURCE_ROUTE_FAILURE messages in the log for 26 different devices.  They seem to occur in bursts of 3 to 6 messages all for same device.  Each burst takes a second or two and the bursts take place every couple minutes.

```
[2024-05-08 10:09:55] warning:  zh:ember:ezsp: Received network/route error ROUTE_ERROR_SOURCE_ROUTE_FAILURE for "30294".
[2024-05-08 10:09:55] warning:  zh:ember:ezsp: Received network/route error ROUTE_ERROR_SOURCE_ROUTE_FAILURE for "30294".
[2024-05-08 10:09:57] warning:  zh:ember:ezsp: Received network/route error ROUTE_ERROR_SOURCE_ROUTE_FAILURE for "30294".
[2024-05-08 10:09:59] warning:  zh:ember:ezsp: Received network/route error ROUTE_ERROR_SOURCE_ROUTE_FAILURE for "30294".
[2024-05-08 10:10:52] info:     zh:ember: [NCP COUNTERS] 3645,731,3176,1379,545,107,59,28,1540,588,0,2,86,0,0,111,0,0,0,0,0,294,0,0,0,0,0,0,2,0,0,0,32,0,0,0,0,0,0,0,0
[2024-05-08 10:10:52] info:     zh:ember: [ASH COUNTERS] 16465,4727,591,4136,0,0,0,0,95278,4392,4392,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0
[2024-05-08 10:11:57] warning:  zh:ember:ezsp: Received network/route error ROUTE_ERROR_SOURCE_ROUTE_FAILURE for "52510".
[2024-05-08 10:11:57] warning:  zh:ember:ezsp: Received network/route error ROUTE_ERROR_SOURCE_ROUTE_FAILURE for "52510".
[2024-05-08 10:11:57] warning:  zh:ember:ezsp: Received network/route error ROUTE_ERROR_SOURCE_ROUTE_FAILURE for "52510".
[2024-05-08 10:11:57] warning:  zh:ember:ezsp: Received network/route error ROUTE_ERROR_SOURCE_ROUTE_FAILURE for "52510".
[2024-05-08 10:11:57] warning:  zh:ember:ezsp: Received network/route error ROUTE_ERROR_SOURCE_ROUTE_FAILURE for "52510".
[2024-05-08 10:11:57] warning:  zh:ember:ezsp: Received network/route error ROUTE_ERROR_SOURCE_ROUTE_FAILURE for "52510".
[2024-05-08 10:11:57] warning:  zh:ember:ezsp: Received network/route error ROUTE_ERROR_SOURCE_ROUTE_FAILURE for "52510".
[2024-05-08 10:11:57] warning:  zh:ember:ezsp: Received network/route error ROUTE_ERROR_SOURCE_ROUTE_FAILURE for "52510".
[2024-05-08 10:12:17] warning:  zh:ember:ezsp: Received network/route error ROUTE_ERROR_SOURCE_ROUTE_FAILURE for "29306".
[2024-05-08 10:12:17] warning:  zh:ember:ezsp: Received network/route error ROUTE_ERROR_SOURCE_ROUTE_FAILURE for "29306".
[2024-05-08 10:12:17] warning:  zh:ember:ezsp: Received network/route error ROUTE_ERROR_SOURCE_ROUTE_FAILURE for "29306".
[2024-05-08 10:12:17] warning:  zh:ember:ezsp: Received network/route error ROUTE_ERROR_SOURCE_ROUTE_FAILURE for "29306".
[2024-05-08 10:12:19] warning:  zh:ember:ezsp: Received network/route error ROUTE_ERROR_SOURCE_ROUTE_FAILURE for "29306".
[2024-05-08 10:12:20] warning:  zh:ember:ezsp: Received network/route error ROUTE_ERROR_SOURCE_ROUTE_FAILURE for "29306".
[2024-05-08 10:16:57] warning:  zh:ember:ezsp: Received network/route error ROUTE_ERROR_SOURCE_ROUTE_FAILURE for "52510".
[2024-05-08 10:16:57] warning:  zh:ember:ezsp: Received network/route error ROUTE_ERROR_SOURCE_ROUTE_FAILURE for "52510".
[2024-05-08 10:16:58] warning:  zh:ember:ezsp: Received network/route error ROUTE_ERROR_SOURCE_ROUTE_FAILURE for "52510".
[2024-05-08 10:16:58] warning:  zh:ember:ezsp: Received network/route error ROUTE_ERROR_SOURCE_ROUTE_FAILURE for "52510".
[2024-05-08 10:17:17] warning:  zh:ember:ezsp: Received network/route error ROUTE_ERROR_SOURCE_ROUTE_FAILURE for "29306".
[2024-05-08 10:17:18] warning:  zh:ember:ezsp: Received network/route error ROUTE_ERROR_SOURCE_ROUTE_FAILURE for "29306".
[2024-05-08 10:17:20] warning:  zh:ember:ezsp: Received network/route error ROUTE_ERROR_SOURCE_ROUTE_FAILURE for "29306".
```

rinne, 2024-05-08:
The majority of the messages are for random end device sensors like TS0201 but also router devices like TS011F_plug_1 are there. The odd one for ZBMINIL2 end-device single wire relay also occurs.

Nerivec, 2024-05-08:
@rinne Do you have the complete log file for those ~24h? I'll take a look, see if I can spot something.

piellex, 2024-05-08:
Same problem, I updated to the ember driver, flashed the ncp-uart-hw-v7.4.1.0-zbdonglee-115200 firmware (because in my case if I use subsequent zigbee2mqtt firmware it doesn't start at all). With normal ezsp driver, no problem, except the warnings to update the driver to ember

rinne, 2024-05-08:
> @rinne Do you have the complete log file for those ~24h? I'll take a look, see if I can spot something.

Indeed I have.

[tr-ember.log](https://github.com/Koenkk/zigbee2mqtt/files/15252768/tr-ember.log)

Nerivec, 2024-05-08:
Just to be clear: Network/route errors are not necessarily a "problem". They happened with `ezsp` too, you just didn't see them because `ezsp` doesn't make use of them. `ember` does because it can be useful to troubleshoot your own network. Like I said, it can indicate a problem if you receive many (doesn't mean your network doesn't work though, just likely that one or more of your routers isn't doing its job quite right, or possibly getting some interferences on the 2.4GHz band). _It already was only a `warning`, will be lowered to `info` in next release to avoid panic._

@rinne Your network relies heavily on Tuya routers, which are know to be subpar in a lot of cases. Maybe try to add a "good" router next time you need a new device. _If you want recommendations, check the zigbee2mqtt discord. Definitely avoid Aqara though._

rinne, 2024-05-08:
> @rinne Your network relies heavily on Tuya routers, which are know to be subpar in a lot of cases. Maybe try to add a "good" router next time you need a new device.

But they have so nice form factor and cost next to nothing.  I do see your point though :).

morhimi, 2024-05-09:
> BTW, it's annoying that the device address is in base-10 on these messages but hex in UI.

@Nerivec any chance to change the device id in the log to hex? (or add hex in addition to decimal)

Nerivec, 2024-05-09:
https://github.com/nurikk/zigbee2mqtt-frontend/issues/1971

thewan056, 2024-05-10:
I will try to find time to reproduce this because this involves making my parents house zigbee network "unreliable" temporarily so that I can get logs. At that time I was just thinking of making the network stable again (it was being less responsive than usual) so the only thing on my mind back then was getting back to normal first.

Also for those that criticize me and others buying tuya or tuya based products rebranded by others, we have **no choice**. For example the smart plug socket stated above. Unlike US or EU plug sockets, UK plug sockets for zigbee are **really** hard to find. On top of that where we live, the only other options for zigbee UK plug sockets  besides tuya and tuya rebrands are Sonoff and Ikea. Both Sonoff and Ikea plugs have impractical shape and size to fit in to certain areas of the house. And the Sonoff in particular is known to be not reliable and lacking functionality like restoring state after a power outage. And yes I have both Ikea and Sonoff sockets so I know their advantages and disadvantages first hand. 

Many other smart products that are zigbee based only tuya and their rebrands exist in our market, both locally and online internationally. As a 3rd world developing country, we can only hope that we get the newer versions of products, like the newer ikea socket that has a more compact shape, that may or may never reach our shores. So we have no choice and make do with what we have. If we encounter problems we report them and if it is not fixable, then fine. We are not complaining, thats just the reality we live with.

RaceNJason, 2024-05-14:
Router: SLZB-06M Updated to latest driver which switches from EZSP to ember
Made appropriate changes to HA
All BSD01 (recognized as BSD29_1) showed they disconnected from the network and offline. Unplugging/replugging them in did not fix it. So I reset one to readd it to the network. Zigbee2MQTT reported it could not add it (no matter how many times I tried). My second Zigbee2MQTT network, ran off a SLZB-06 router and zstack driver, continue to have no problems with this smart switch. Since I feel forced to continue with the ember driver and not revert back because, according to Zigbee2MQTT, it will no longer work in the next update...I pulled all my BSD01 plugs in the upstairs network. I'm hoping this gets fixed since there isn't much option (i.e. stop using Zigbee2MQTT or the BSD01 plugs, I of course chose to stop using the plugs for now).

Nerivec, 2024-05-14:
@RaceNJason Please create a new issue and provide a debug log of the pairing process with ember.

I'm closing this to avoid unrelated issues getting lost in this "generic" thread.

Dimmers81, 2024-06-05:
Yep same issue after updating to Ember

MairusuPawa, 2024-06-20:
Impacted as well

* Skyconnect dongle running the latest firmware, on USB2
* Migrated from EZSP to Ember
* Latest HomeAssistant on Odroid N2, latest Z2M

ncodee, 2024-07-07:
Same issue here after migrating to ember from ezsp.

latez, 2024-07-09:
Another affected over here - 
SBC: Rasberry Pi 3b+ 
Adapter: TubeZB EFR32 MGM21 Ethernet/USB Serial Coordinator
GBL metadata: NabuCasaMetadata(metadata_version=1, sdk_version='4.4.3', ezsp_version='7.4.2.0',

tandarra, 2024-07-25:
How can we figure out which device is "41954"? ROUTE_ERROR_MANY_TO_ONE_ROUTE_FAILURE for "41954".

RaceNJason, 2024-07-29:
> How can we figure out which device is "41954"? ROUTE_ERROR_MANY_TO_ONE_ROUTE_FAILURE for "41954".

Convert 41954 to hexadecimal (A3E2) and then search for it in your IEEE Address list in the devices page of your Zigbee2MQTT installation

samfrjn11, 2024-08-01:
Same issue, disappointing to see thread is closed

Dimmers81, 2024-08-01:
FYI / PSA: Two months ago, they released firmware version 7.4.3. I personally updated to this, and the ROUTE_ERROR_MANY_TO_ONE_ROUTE_FAILURE  has gone away. I suggest giving this a try while backing up your existing setup.
https://github.com/darkxst/silabs-firmware-builder/tree/main/firmware_builds/zbdonglee

thewan056, 2024-08-03:
> FYI / PSA: Two months ago, they released firmware version 7.4.3. I personally updated to this, and the ROUTE_ERROR_MANY_TO_ONE_ROUTE_FAILURE has gone away. I suggest giving this a try while backing [...]

Try monitoring your logs longer. It does seem to reduce it abit, but sit there longer at the zb2mqtt page and it comes back

antpop, 2024-08-08:
Has anyone resolved this issue? I cannot connect ledvance light bulb after forced delete, it joined z2m, but interview failed. Is there anything I can try? Already turned on and off the addon and ha, tried switching sonoff dongle-e firmware to previous releases and even going back to ezsp, but nothing seems to work. I’m thinking about uninstaling z2m addon and repairing every device again, but I’m afraid that it would just brake other devices…. 
Should I open new issue if this one is closed? Maybe there already is a new one?

samfrjn11, 2024-08-09:
> Has anyone resolved this issue? I cannot connect ledvance light bulb after forced delete, it joined z2m, but interview failed. Is there anything I can try? Already turned on and off the addon and [...]

Your issue seems to be unrelated to this topic. Just keep retrying, I had a similar issue months ago with a Philips Hue lightbulb. Make sure to try to reset the bulb itself as well. Might help you!

samfrjn11, 2024-08-09:
> FYI / PSA: Two months ago, they released firmware version 7.4.3. I personally updated to this, and the ROUTE_ERROR_MANY_TO_ONE_ROUTE_FAILURE has gone away. I suggest giving this a try while backing [...]

I'm actually running 7.4.3 and experiencing these issues. 
However: I was recently having issues with another integration which was updated in HA Core, however my instance seemed to be still running an older version of that integration. So, eventhough my Core version was at latest 2024.7.4, the integration was not up to date.

I ran `ha > core rebuild` and it fixed my integration issues. I also no longer received any z2m/herdsman/ember errors like `ROUTE_ERROR_MANY_TO_ONE_ROUTE_FAILURE `

Hope it helps for anyone else as well.

So, in summary:
in console, run `(ha) core rebuild`

antpop, 2024-08-10:
> Your issue seems to be unrelated to this topic. Just keep retrying, I had a similar issue months ago with a Philips Hue lightbulb. Make sure to try to reset the bulb itself as well. Might help you!

Thanks! Unfortunately it didn't work. I bought another zigbee device and after pairing it I could finally pair the broken one!

10der, 2024-08-11:
> I ran `ha > core rebuild` and it fixed my integration issues. 

What does this have to do with the Z2M?

hardwareadictos, 2024-09-04:
Same error but trying to update some Hue Ambiance Lightstrips when trying to update via OTA:

```
network/route error ROUTE_ERROR_MANY_TO_ONE_ROUTE_FAILURE for "46249".
```

I'm on ember also,  Dongle-E, 7.4.3

senna1992, 2024-09-15:
HA yellow, same issue here.

according to my latest debuging,
it only affected devices routing their traffic over atleast 2 routers before entering the coordinator.

Affected devices:
develco SMSZB-120
Heiman HS2SK
Merten MEG2380-0325 / Schneider Electric EKO09716

Maybe associated devices:
eWeLight ZB-CL01 / Lonsoho ZB-RGBCW

All of the affected devices could possibly at some point access the route between zb-cl01 and the coordinator.

senna1992, 2024-09-19:
since i removed the lonsoho (tuya) ledstripe drivers i did not recieve the issue anymore. took a few hours for the devices to reroute before the issue stopped appearing

CoderJoeyMoment, 2024-10-06:
Same here:

EmberZNet 7.4.1 [GA] 
slzb-06m

End devices: 86
Router: 43

major328, 2024-10-07:
From yesterday to today 4 ZBMINI-L2 stopped communicating. I can see them
on the devices list, but I receive "ROUTE_ERROR_SOURCE_ROUTE_FAILURE"
messages. All others are OK. I did nothing around z2m or zigbee, there was
only a restart of the whole system due to a HACS thermostat integration
update.

CoderJoeyMoment ***@***.***> ezt írta (időpont: 2024. okt.
6., V, 16:08):

senna1992, 2024-10-07:
Sounds like they routed using one out of the 5 and since it broke, others didn't catch up connectivity from some other source.

Is it possible that there are not many other router (ac powered) zb devices around?

major328, 2024-10-07:
around these units (means inside the garage where these units are
installed) I have 2 other AC powered routers working properly. One ZBMINI
and one Xiaomi Zigbee plug, and near I have other 2 ZBMINIs. This lost
group was seen by Z2M 12 hours before, which is the time when I restarted
the HomeAssistant.

senna1992 ***@***.***> ezt írta (időpont: 2024. okt. 7., H,
9:02):

major328, 2024-10-07:
> But as I know these units has no routing functionality. ZBMINI-L2 is only
> an phase powered switch, not a router , but maybe I'm wrong.
>
> senna1992 ***@***.***> ezt írta (időpont: 2024. okt. 7., H,
> 9:02):

I've switched off the AC power and after a half minute switched on and all
are communicating again

mbrucher, 2024-10-25:
I found the same issue, mainly on the SPM01V2 power meters. It's super frustrating, they do work, my network work, but having thousands of error messages that we can't discard in bulk is frustrating because the web UI is basically useless until we discard them.
Probably would be better as an info, and maybe even a debug if it's for network debugging purposes.

k-zakhariy, 2024-11-02:
> Same issue here, using `ember` (`7.4.1[GA]`) on the `SLZB-06M` coordinator.

have you found a solution ?

RuneNyhuus, 2024-12-04:
I also have 1000 of these errors

zh:ember:ezsp: Received network/route error ROUTE_ERROR_SOURCE_ROUTE_FAILURE for "8868".

Is there really no solution for this ?
I have only connected 10 LEDVance Zigbee Plugs, and my z2m is a mess with all these errors.

Kzibi, 2024-12-09:
+1
It is realy disapointing.
HA cannot be assistant cause off z2m is untrusted. So my interested on smart home going down

Dam0s, 2024-12-18:
Moved from ezsp Conbee II to ember Sonoff Dongle E, took me two day of re-configuring 100+ devices and I wish I hadn't now. I get these ROUTE_ERROR_SOURCE_ROUTE_FAILURE for....., all the time.
My main issue is the whole system going offline twice now (migrated about a month ago) but I've not been home at the time to diagnose, welcome any articles on how you troubleshoot capturing logs after restart if anyone can advise.
Other issue is just general unreliability of router devices, either to on/off actions for example. It's rare but it didn't happend before.
Other annoying issue is with PIR sensors:
Aqara P1: Unsupported, they pair but the detected/undetected is completly unreliable
Ikea E1525 and Sonoff: Work most of the time but often drop off and need re-pairing.
I don't know if the aboive error is related to this, just looking to troubeshoort each issue at a time.

yoch, 2024-12-23:
Same issue here using ember and Z2M 1.42.0 (zigbee-herdsman 2.1.9) with a SONOFF ZDongle.

#### Adapter version info
`{"ezsp":13,"revision":"7.4.5 [GA]","build":0,"major":7,"minor":4,"patch":5,"special":0,"type":170}`)

#### Log example
```
[2024-12-23 18:43:18] info: 	zh:ember:ezsp: Received network/route error ROUTE_ERROR_MANY_TO_ONE_ROUTE_FAILURE for "55395".
[2024-12-23 18:43:18] info: 	zh:ember:ezsp: Received network/route error ROUTE_ERROR_MANY_TO_ONE_ROUTE_FAILURE for "34762".
[2024-12-23 18:43:18] info: 	zh:ember:ezsp: Received network/route error ROUTE_ERROR_MANY_TO_ONE_ROUTE_FAILURE for "13352".
[2024-12-23 18:43:18] info: 	zh:ember:ezsp: Received network/route error ROUTE_ERROR_MANY_TO_ONE_ROUTE_FAILURE for "34762".
[2024-12-23 18:43:18] info: 	zh:ember:ezsp: Received network/route error ROUTE_ERROR_MANY_TO_ONE_ROUTE_FAILURE for "13352".
[2024-12-23 18:43:18] info: 	zh:ember:ezsp: Received network/route error ROUTE_ERROR_MANY_TO_ONE_ROUTE_FAILURE for "55395".
[2024-12-23 18:43:19] info: 	zh:ember:ezsp: Received network/route error ROUTE_ERROR_MANY_TO_ONE_ROUTE_FAILURE for "34762".
[2024-12-23 18:43:19] info: 	zh:ember:ezsp: Received network/route error ROUTE_ERROR_MANY_TO_ONE_ROUTE_FAILURE for "55037".
[2024-12-23 18:43:19] info: 	zh:ember:ezsp: Received network/route error ROUTE_ERROR_MANY_TO_ONE_ROUTE_FAILURE for "13352".
[2024-12-23 18:43:19] info: 	zh:ember:ezsp: Received network/route error ROUTE_ERROR_MANY_TO_ONE_ROUTE_FAILURE for "20612".
[2024-12-23 18:43:20] info: 	zh:ember:ezsp: Received network/route error ROUTE_ERROR_MANY_TO_ONE_ROUTE_FAILURE for "13352".
[2024-12-23 18:43:20] info: 	zh:ember:ezsp: Received network/route error ROUTE_ERROR_MANY_TO_ONE_ROUTE_FAILURE for "34762".
[2024-12-23 18:43:25] info: 	zh:ember:ezsp: Received network/route error ROUTE_ERROR_MANY_TO_ONE_ROUTE_FAILURE for "34762".
[2024-12-23 18:43:26] info: 	zh:ember:ezsp: Received network/route error ROUTE_ERROR_MANY_TO_ONE_ROUTE_FAILURE for "55037".
[2024-12-23 18:43:26] info: 	zh:ember:ezsp: Received network/route error ROUTE_ERROR_SOURCE_ROUTE_FAILURE for "13352".
[2024-12-23 18:43:26] info: 	zh:ember:ezsp: Received network/route error ROUTE_ERROR_MANY_TO_ONE_ROUTE_FAILURE for "20612".
[2024-12-23 18:43:26] info: 	zh:ember:ezsp: Received network/route error ROUTE_ERROR_MANY_TO_ONE_ROUTE_FAILURE for "55395".
[2024-12-23 18:43:26] info: 	zh:ember:ezsp: Received network/route error ROUTE_ERROR_MANY_TO_ONE_ROUTE_FAILURE for "20612".
[2024-12-23 18:43:26] info: 	zh:ember:ezsp: Received network/route error ROUTE_ERROR_MANY_TO_ONE_ROUTE_FAILURE for "13352".
[2024-12-23 18:43:26] info: 	zh:ember:ezsp: Received network/route error ROUTE_ERROR_MANY_TO_ONE_ROUTE_FAILURE for "34762".
[2024-12-23 18:43:26] info: 	zh:ember:ezsp: Received network/route error ROUTE_ERROR_MANY_TO_ONE_ROUTE_FAILURE for "34762".
[2024-12-23 18:43:26] info: 	zh:ember:ezsp: Received network/route error ROUTE_ERROR_SOURCE_ROUTE_FAILURE for "55395".
[2024-12-23 18:43:27] info: 	zh:ember:ezsp: Received network/route error ROUTE_ERROR_MANY_TO_ONE_ROUTE_FAILURE for "13352".
[2024-12-23 18:43:27] info: 	zh:ember:ezsp: Received network/route error ROUTE_ERROR_SOURCE_ROUTE_FAILURE for "13352".
[2024-12-23 18:43:27] info: 	zh:ember:ezsp: Received network/route error ROUTE_ERROR_MANY_TO_ONE_ROUTE_FAILURE for "34762".
[2024-12-23 18:43:27] info: 	zh:ember:ezsp: Received network/route error ROUTE_ERROR_MANY_TO_ONE_ROUTE_FAILURE for "55037".
[2024-12-23 18:43:28] info: 	zh:ember:ezsp: Received network/route error ROUTE_ERROR_SOURCE_ROUTE_FAILURE for "13352".
[2024-12-23 18:43:28] info: 	zh:ember:ezsp: Received network/route error ROUTE_ERROR_SOURCE_ROUTE_FAILURE for "55395".
[2024-12-23 18:43:29] info: 	zh:ember:ezsp: Received network/route error ROUTE_ERROR_MANY_TO_ONE_ROUTE_FAILURE for "55395".
[2024-12-23 18:43:29] info: 	zh:ember:ezsp: Received network/route error ROUTE_ERROR_MANY_TO_ONE_ROUTE_FAILURE for "34762".
[2024-12-23 18:43:29] info: 	zh:ember:ezsp: Received network/route error ROUTE_ERROR_MANY_TO_ONE_ROUTE_FAILURE for "55037".
[2024-12-23 18:43:29] info: 	zh:ember:ezsp: Received network/route error ROUTE_ERROR_SOURCE_ROUTE_FAILURE for "20612".
[2024-12-23 18:43:29] info: 	zh:ember:ezsp: Received network/route error ROUTE_ERROR_SOURCE_ROUTE_FAILURE for "13352".
[2024-12-23 18:43:30] info: 	zh:ember:ezsp: Received network/route error ROUTE_ERROR_SOURCE_ROUTE_FAILURE for "34762".
[2024-12-23 18:43:39] info: 	zh:ember:ezsp: Received network/route error ROUTE_ERROR_SOURCE_ROUTE_FAILURE for "34762".
[2024-12-23 18:43:39] info: 	zh:ember:ezsp: Received network/route error ROUTE_ERROR_MANY_TO_ONE_ROUTE_FAILURE for "34762".
[2024-12-23 18:43:39] info: 	zh:ember:ezsp: Received network/route error ROUTE_ERROR_MANY_TO_ONE_ROUTE_FAILURE for "13352".
[2024-12-23 18:43:39] info: 	zh:ember:ezsp: Received network/route error ROUTE_ERROR_MANY_TO_ONE_ROUTE_FAILURE for "34762".
[2024-12-23 18:43:39] info: 	zh:ember:ezsp: Received network/route error ROUTE_ERROR_MANY_TO_ONE_ROUTE_FAILURE for "20612".
[2024-12-23 18:43:40] info: 	zh:ember:ezsp: Received network/route error ROUTE_ERROR_SOURCE_ROUTE_FAILURE for "13352".
[2024-12-23 18:43:40] info: 	zh:ember:ezsp: Received network/route error ROUTE_ERROR_MANY_TO_ONE_ROUTE_FAILURE for "34762".
[2024-12-23 18:43:40] info: 	zh:ember:ezsp: Received network/route error ROUTE_ERROR_SOURCE_ROUTE_FAILURE for "55395".
[... cut ...]
[2024-12-23 18:46:01] info: 	zh:ember:ezsp: Received network/route error ROUTE_ERROR_MANY_TO_ONE_ROUTE_FAILURE for "34762".
[2024-12-23 18:46:01] info: 	zh:ember:ezsp: Received network/route error ROUTE_ERROR_MANY_TO_ONE_ROUTE_FAILURE for "13352".
[2024-12-23 18:46:01] info: 	zh:ember:ezsp: Received network/route error ROUTE_ERROR_MANY_TO_ONE_ROUTE_FAILURE for "34762".
[2024-12-23 18:46:01] info: 	zh:ember:ezsp: Received network/route error ROUTE_ERROR_SOURCE_ROUTE_FAILURE for "34762".
[2024-12-23 18:46:02] info: 	zh:ember:ezsp: Received network/route error ROUTE_ERROR_MANY_TO_ONE_ROUTE_FAILURE for "34762".

```

Dam0s, 2024-12-27:
What device were you getting ROUTE_ERROR_SOURCE_ROUTE_FAILURE  error? I hadn't seen it until this week

It started happening on a Sonoff ZBMINI:
[2024-12-27   15:12:59] info:     zh:ember:ezsp:   Received network/route error ROUTE_ERROR_SOURCE_ROUTE_FAILURE for   "51369".
[2024-12-27 15:13:01]   error:    z2m: Publish 'set' 'state' to   'ES_FanSw' failed: 'Error: ZCL command 0x00124b0024c16233/1 genOnOff.on({},   {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"reservedBits":0,"writeUndiv":false})   failed (Delivery failed for '51369'.)'

I moved back to ezsp and it still did not work, I just recieved the failed message above. I checked the device and it would no re-pair it was as if it had crashed. I have to unwire it to kill the power and once re-powered it started working again.

I'm going to stay on ezsp for a couple of weeks as I've not used this before.

Kzibi, 2024-12-27:
Sonoff Dongle P with latest fw.
I think it doesent metter.  Ember is faulty

sat-fan2019, 2024-12-28:
I have the same problem with my dongle-e and ember. For me it looks like that ember has a less antenna gain or in general the Donngle-P as a better one. I would like to a move to my old  sonoff-P  stick. How can I do that without loosing my configuration ?

Sim0cYz, 2024-12-28:
I'm having the same issue (Received network/route error ROUTE_TOO_MANY_TO_ONE_FAILURE) but also ROUTE_ERROR_ADDRESS_CONFLICT with devices being kicked off the network. I have those 4 Sonoff ZBMINI that work fine for a while, then they are constantly being kicked off and disappear from Z2M. 
All the other devices (different brands, Vimar, Tuya, Philips Hue, IKEA...) work just fine. 
Zigbee2MQTT 1.42.0
Coordinator ZBDongle-E ember 7.4.4

Dam0s, 2024-12-30:
I've not seen ROUTE_ERROR_ADDRESS_CONFLICT at all and my 4 x ZBMINI's work fine. It was just that one time, when I got ROUTE_ERROR_SOURCE_ROUTE_FAILURE and had to physically kill power to the ZBMINI before being able to re-pair it.

Sim0cYz, 2024-12-30:
I figured out that those address issues were caused by Vimar devices. Removing them from my zigbee network fixed every weird behavior...

Dam0s, 2024-12-30:
> I have the same problem with my dongle-e and ember. For me it looks like that ember has a less antenna gain or in general the Donngle-P as a better one. I would like to a move to my old sonoff-P stick. How can I do that without loosing my configuration ?

You won't loose your config if you replace your dongle but you will need to re-pair everything. As you re-pair you'll see your devices return with their orginal name. Ideally, test it out with another raspberry pi (or whatever HW you have spare) and restore from a backup of your current HA.

sylarm27, 2024-12-30:
I have Dongle-e with ember and I see a lot " error ROUTE_ERROR_MANY_TO_ONE_ROUTE_FAILURE" and "ROUTE_ERROR_SOURCE_ROUTE_FAILURE", I have re-paired all network several times, and it comes again. Is it the coordinator issue? Replace the coordinator for what? to solve the issue...

Barry40, 2025-01-06:
Same here.....  A lot of ROUTE_ERROR_MANY_TO_ONE_ROUTE_FAILURE for "#ID#" in our log.
Firmware 7.4.4 Ember

senna1992, 2025-01-06:
@Barry40 please check your device ID and use z2m map view to see through which device it is routing.

ID might need to be translated from decimal to hex to identify more easily

Modem-Tones, 2025-01-08:
I'm new to z2m and just set up my first network this week with a ZBDongle-E. I'm getting hundreds of these warnings through normal operation of my network, and it might be the cause of poor performance I've occasionally noticed.

I see ROUTE_ERROR_MANY_TO_ONE_ROUTE_FAILURE associated with all AC powered device types in my network: Hue Bulbs, Inovelli Blue 2-1 switches, and Third Reality plugs. Those should each be reliable routers from my understanding. Enabling Adaptive Lighting seems to trigger the warnings more frequently even when AL is only targeting Zigbee groups. I've also seen ROUTE_ERROR_SOURCE_ROUTE_FAILURE thrown a few times.

When I select a specific problematic device in the z2m map, I am not sure what to look for. One such device which threw both "ROUTE_ERROR_MANY_TO_ONE_ROUTE_FAILURE" and "ROUTE_ERROR_SOURCE_ROUTE_FAILURE" shows a direct connection to the controller. This is an Inovelli Blue switch named "Kitchen Recessed Lights Switch" in my image.

[image]

My network is made up of Phillips Hue Bulbs (a few types), Inovelli Switches, Third Reality plugs, and 14 battery powered SOMFY shade end devices:

By vendor
Signify Netherlands B.V.: 27
SOMFY: 14
Inovelli: 11
Third Reality, Inc: 4

Zigbee2MQTT 2.0.0
Coordinator ZBDongle-E ember 7.4.5

senna1992, 2025-01-08:
It might be that the issue lies within Routing a routing device?

Thus exceeding the capacity per message indirectly?

I see your device routing to entrance chandelier which itself is routing to some battery powered devices.

Beppuz81, 2025-01-10:
My situation is similar to yours. The Zigbee network is unstable and slow. I receive thousands of network/route errors.

I use a Sonoff Zigbee Dongle E flashed with the latest firmware available (ncp-uart-hw-v7.4.5.0-zbdonglee-115200.gbl) with Zigbee2MQTT 2.0. 

I hope someone help us.

sylarm27, 2025-01-10:
In my case the situation drastically get better when I replace 2 old module switch by 2 **zbminiR2**, now all my devices prefer this new ones, and I all most don't have errors. Seems Tuya devices have some particularities...

senna1992, 2025-01-10:
I second that tuya related concern.
My issues got better after removing some tuya  zigbee based RGB light strip drivers.

Sim0cYz, 2025-01-10:
I was pointing my finger at one ultra cheap Tuya TS0501B in my network, but as far as I removed all my Vimar devices (roller shutters and light switches), everything returned acting normal and blazing fast. So I am not sure it's something with Tuya devices, I now own several (mmWave sensors, power plugs...) mixed with Sonoff ZBMini and it's more than okay..

[15 more replies]