Koenkk/zigbee2mqtt #22517: Aqara switch do action enum differently [QBKG41LM, Aqara E1 2-Gang with neutral]

pawelbudynek, 2024-05-07:
### What happened?

The switch is pairing without any issues, you can switch it to decoupled mode, but then HA is not receiving any actions when you click the switch.
I never had this issue with previous versions of Aqara Wall Switches :(

### What did you expect to happen?

I expect to receive enum action according to the documentation: https://www.zigbee2mqtt.io/devices/QBKG41LM.html

### How to reproduce it (minimal and precise)

1. Pair the device
2. Switch device to decoupled mode.
3. Press button on the device

### Zigbee2MQTT version

1.37.0-1

### Adapter firmware version

-

### Adapter

Conbee II

### Setup

Pi 4b, Addon Z2M on Home Assitant Supervised

adynis, 2024-05-08:
Maybe seems similar to 21679 ? (which is a loong topic, but shortly: it was fixed BUT also a coordinator firmware udpate was applied)

pawelbudynek, 2024-05-09:
@adynis, you were right! I updated conbee with new firmware and it works now 👍 
I still don't understand why this happen, I have 70 Zigbee devices and they all were working perfectly fine with old firmware.
Thank you.

closed 2024-05-09 by pawelbudynek