Koenkk/zigbee2mqtt #22518: Unable to join more than 45 devices

Patricen1, 2024-05-07:
Hello,

For some reason I do have a Tuya TS011F smart plug (brand new) that went offline within Zigbee2mqtt frontend.
I deleted the device from zigbee2mqtt, pressed "permit join" and long pressed on plug button until the LED blinks. After 3 minutes, LED stops blinking, probably timeout, but device is not associated.
Same issue with another TS011F smart plug that I could never associate to Zigbee2mqtt.

Any help?

Thanks

Notes:
- I'm using Zigbee2mqtt 1.37.0 but the issue was already present using V1.36.0 or V1.36.1
- I do have already 13 Tuya TS011F smart plugs connected, i'm then sure to know what is the accociation process.
- I tried to associate the device close to the coordinator (ConBee2/RaspBee2)
- There are 45 devices connected
- Is there a way to factory reset the smart plugs?
- Is there a cache file where the plugs could be "remembered" within Zigbee2mqtt while the device does not appear on devices list?
- Blocklist and passlist are empty, 
- Edit:
I also tried to allow device to join using specific routers else than coordinator, no improvement.
I do not know how count how many devices are connected to coordinator else than counting links on map.
- Edit2:
Tried with other device type (temperature sensor), not possible to join the network, it looks then like it is a Z2M issue.

_Originally posted by @Patricen1 in https://github.com/Koenkk/zigbee2mqtt/discussions/22461_

olleveque, 2024-05-07:
Hi 
I have similar issue with my configuration. I'm using hacked lidl gateway and if I record more device others start to disconnect. I was expecting that Z2M updates brings more reliability in the losses of devices. It seems that now I have more to worry with ember firmware to update where I found no clue to do so. 
Best regards

Nerivec, 2024-05-07:
You may be hitting a firmware limit. How many routers do you have and how many end devices?

@olleveque Silabs is freezing development for all Series 0/1 (older hardware, which I believe that gateway is using). You will have to stay on `ezsp` until you move to a new, more recent, adapter. _Please note that you may experience problems with newer devices since that gateway can only handle old Zigbee revisions because of its outdated firmware._ I'd say, with the new, more powerful, MG24-based adapters starting to come out (z2m docs will be updated soon), you might want to start looking into these; we have some very good early feedback, with networks that can reach over 300 devices!

Patricen1, 2024-05-07:
Hello,

I do have lots of routers, at least 15 routers and 30 end devices. 3 more routers are Ikea recessed lights that are not powered all day long.
Is there a way updating the firmware of conbee2 in case it is out of date?

Thanks!

Patricen1, 2024-05-13:
Hello,
I finally updated Conbee2 to last firmware 0x26780700, but issue remains the same. Any clue?
Best regards

Patrice

Patricen1, 2024-05-17:
Finally got the solution, I replaced the coordinator with Sonoff ZBDongle-E but no progress with TS011F smart plugs. I then found out a post and realised those plugs are pretty crappy and with the last firmware, they have to be at less than 50cm from the coordinator to get paired.

closed 2024-05-17 by Patricen1