Koenkk/zigbee2mqtt #22527: z2m: Failed to call 'OnEvent' 'stop' (TypeError: Cannot read properties of undefined (reading 'getDevices')

eswenson1, 2024-05-07:
### What happened?

I have been using another Zigbee USB Dongle (ThirdReality) and I tried switching to a smlight SL-ZB-06 Zigbee adapter.  When I try to start Zigbee2MQTT in HAOS, I get this:
```
[10:37:06] INFO: Preparing to start...
[10:37:08] INFO: Socat not enabled
[10:37:15] INFO: Starting Zigbee2MQTT...
[2024-05-07 10:37:22] info: 	z2m: Logging to console, file (filename: log.log)
[2024-05-07 10:37:31] error: 	z2m: Failed to call 'OnEvent' 'stop' (TypeError: Cannot read properties of undefined (reading 'getDevices')
    at Zigbee.devices (/app/lib/zigbee.ts:329:30)
    at OnEvent.stop (/app/lib/extension/onEvent.ts:37:42)
    at Controller.callExtensions (/app/lib/controller.ts:336:42)
    at Controller.stop (/app/lib/controller.ts:208:9)
    at stop (/app/index.js:111:5)
    at process.handleQuit (/app/index.js:117:9))
TypeError: Cannot read properties of undefined (reading 'getDeviceByIeeeAddr')
    at Zigbee.resolveDevice (/app/lib/zigbee.ts:250:42)
    at Zigbee.resolveEntity (/app/lib/zigbee.ts:277:45)
    at /app/lib/state.ts:36:66
    at Array.filter (<anonymous>)
    at State.stop (/app/lib/state.ts:36:14)
    at Controller.stop (/app/lib/controller.ts:212:20)
    at stop (/app/index.js:111:5)
    at process.handleQuit (/app/index.js:117:9)
 ```

I'm running the latest firmware on the SL-ZB06 and can access its web interface fine at its URL.  The config for this adapter is:
```
data_path: /config/zigbee2mqtt
socat:
  enabled: false
  master: pty,raw,echo=0,link=/tmp/ttyZ2M,mode=777
  slave: tcp-listen:8485,keepalive,nodelay,reuseaddr,keepidle=1,keepintvl=1,keepcnt=5
  options: "-d -d"
  log: false
mqtt: {}
serial:
  port: tcp://192.168.0.127:6638
  baudrate: 115200
```
Note: the SL-ZB06 documentation says to use:
```
adapter: Unknown
```
and I've tried that and:
```
adapter: none
```
And in either case, I get errors during the saving of the configuration indicative of the config's not matching the required schema.

Sometimes, when I try to restart zigbee2mqtt, the output looks a bit different:
```
[10:45:14] INFO: Preparing to start...
[10:45:16] INFO: Socat not enabled
[10:45:23] INFO: Starting Zigbee2MQTT...
[2024-05-07 10:45:31] info: 	z2m: Logging to console, file (filename: log.log)
[2024-05-07 10:45:41] info: 	z2m: Starting Zigbee2MQTT version 1.37.0 (commit #unknown)
[2024-05-07 10:45:41] info: 	z2m: Starting zigbee-herdsman (0.45.0)
[2024-05-07 10:45:43] info: 	zh:zigate: Connected to ZiGate adapter successfully.
[2024-05-07 10:45:48] error: 	z2m: Error while starting zigbee-herdsman
[2024-05-07 10:45:48] error: 	z2m: Failed to start zigbee
[2024-05-07 10:45:48] error: 	z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
[2024-05-07 10:45:48] error: 	z2m: Exiting...
[2024-05-07 10:45:48] error: 	z2m: Error: failed to connect to zigate adapter [object Object] after 5000ms
    at ZiGateAdapter.start (/app/node_modules/zigbee-herdsman/src/adapter/zigate/adapter/zigateAdapter.ts:95:19)
    at runNextTicks (node:internal/process/task_queues:60:5)
    at listOnTimeout (node:internal/timers:538:9)
    at processTimers (node:internal/timers:512:7)
    at Controller.start (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:124:29)
    at Zigbee.start (/app/lib/zigbee.ts:62:27)
    at Controller.start (/app/lib/controller.ts:108:27)
    at start (/app/index.js:107:5)
```
Note the reference to ZIGate here.  Is the SL-ZB06 really using a zigate adapter?  Or are there some vestiges from when I was using the other (ZiGate, ThirdReality USB adapter)?

### What did you expect to happen?

I expected zigbee2mqtt to come up just fine.

### How to reproduce it (minimal and precise)

Update zigbee2mqtt configuration as specified above, and restart zigbee2mqtt.

### Zigbee2MQTT version

1.37.0-1

### Adapter firmware version

cam

### Adapter

smlight SLZB-06

### Setup

HAOS on RPI3

eswenson1, 2024-05-07:
I read somewhere that perhaps the ezsp driver is the one to use with the SLZB-06.  When I try that, I get this error:
```
[11:00:09] INFO: Preparing to start...
[11:00:11] INFO: Socat not enabled
[11:00:18] INFO: Starting Zigbee2MQTT...
[2024-05-07 11:00:24] info: 	z2m: Logging to console, file (filename: log.log)
[2024-05-07 11:00:34] info: 	z2m: Starting Zigbee2MQTT version 1.37.0 (commit #unknown)
[2024-05-07 11:00:34] info: 	z2m: Starting zigbee-herdsman (0.45.0)
[2024-05-07 11:00:36] warning: 	zh:ezsp: Deprecated driver 'ezsp' currently in use, 'ember' will become the officially supported EmberZNet driver in next release. If using Zigbee2MQTT see https://github.com/Koenkk/zigbee2mqtt/discussions/21462
[2024-05-07 11:00:46] error: 	zh:ezsp:uart: --> Error: Error: {"sequence":-1} after 10000ms
Error: Failure to connect
    at SerialDriver.resetForReconnect (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:347:19)
    at SerialDriver.emit (node:events:517:28)
    at /app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/uart.ts:348:22
    at Queue.execute (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:35:20)
    at Socket.<anonymous> (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/uart.ts:154:17)
    ```
So that appears not to be right either.

eswenson1, 2024-05-07:
And trying the `ember` adapter gives this:
```
[11:11:19] INFO: Preparing to start...
[11:11:21] INFO: Socat not enabled
[11:11:28] INFO: Starting Zigbee2MQTT...
[2024-05-07 11:11:35] info: 	z2m: Logging to console, file (filename: log.log)
[2024-05-07 11:11:44] info: 	z2m: Starting Zigbee2MQTT version 1.37.0 (commit #unknown)
[2024-05-07 11:11:44] info: 	z2m: Starting zigbee-herdsman (0.45.0)
[2024-05-07 11:11:45] info: 	zh:ember: ======== Ember Adapter Starting ========
[2024-05-07 11:11:45] info: 	zh:ember:ezsp: ======== EZSP starting ========
[2024-05-07 11:11:45] info: 	zh:ember:uart:ash: ======== ASH NCP reset ========
[2024-05-07 11:11:45] info: 	zh:ember:uart:ash: Socket ready
[2024-05-07 11:11:45] info: 	zh:ember:uart:ash: ======== ASH starting ========
[2024-05-07 11:11:47] info: 	zh:ember:uart:ash: ======== ASH NCP reset ========
[2024-05-07 11:11:47] info: 	zh:ember:uart:ash: ======== ASH starting ========
[2024-05-07 11:11:50] info: 	zh:ember:uart:ash: ======== ASH NCP reset ========
[2024-05-07 11:11:50] info: 	zh:ember:uart:ash: ======== ASH starting ========
[2024-05-07 11:11:52] info: 	zh:ember:uart:ash: ======== ASH NCP reset ========
[2024-05-07 11:11:52] info: 	zh:ember:uart:ash: ======== ASH starting ========
[2024-05-07 11:11:55] info: 	zh:ember:uart:ash: ======== ASH NCP reset ========
[2024-05-07 11:11:55] info: 	zh:ember:uart:ash: ======== ASH starting ========
[2024-05-07 11:11:57] error: 	z2m: Error while starting zigbee-herdsman
[2024-05-07 11:11:57] error: 	z2m: Failed to start zigbee
[2024-05-07 11:11:57] error: 	z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
[2024-05-07 11:11:57] error: 	z2m: Exiting...
[2024-05-07 11:11:58] error: 	z2m: Error: Failed to start EZSP layer with status=HOST_FATAL_ERROR.
    at EmberAdapter.initEzsp (/app/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:798:19)
    at EmberAdapter.start (/app/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:2660:24)
    at Controller.start (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:124:29)
    at Zigbee.start (/app/lib/zigbee.ts:62:27)
    at Controller.start (/app/lib/controller.ts:108:27)
    at start (/app/index.js:107:5)
```

dinhchinh82, 2024-05-08:
@eswenson1 please take a look about the guide below if you want to try "ember"
https://github.com/Koenkk/zigbee2mqtt/discussions/21462

If you are using SL-ZB-06, it will not be compatible with ember because SL-ZB-06 is CC2652 based zigbee radio.

eswenson1, 2024-05-08:
Thanks. Didn’t realize that. In any case, see the above errors when I use the ezsp driver.

closed 2024-12-05 by github-actions[bot]

xobix, 2025-02-10:
Same issue with Z2M v2.2.1  and CC2652 dongle on HASSIO

it restarts until the watchdog can't take it anymore...

Just one time it write this error stacktrace before restarting :
`[2025-02-09 23:11:59] [31merror[39m: 	z2m: Failed to call 'OnEvent' 'stop' (TypeError: Cannot read properties of undefined (reading 'getDevicesIterator')
    at Zigbee.devicesIterator (/app/lib/zigbee.ts:353:44)
    at devicesIterator.next (<anonymous>)
    at OnEvent.stop (/app/lib/extension/onEvent.ts:36:20)
    at Controller.callExtensions (/app/lib/controller.ts:368:17)
    at Controller.stop (/app/lib/controller.ts:233:9)
    at stop (/app/index.js:172:5)
    at process.handleQuit (/app/index.js:180:13))`
Maybe there is a link ?