Koenkk/zigbee2mqtt #22528: IKEA TREDANSEN cellular blind battery level > 100% (IKEA E2103)

eekamouze, 2024-05-07:
### What happened?

Since upgrade to z2m 1.37.0, the cover shows values around 180% battery level while last was 90% just before the upgrade

### What did you expect to happen?

keep values around previous ones (90%)

### Zigbee2MQTT version

1.37.0

### Adapter firmware version

6.10.3.0 build 297

### Adapter

Sonoff ZBDongle-E (EFR32MG21)

### Setup

Docker on RPI4

Koenkk, 2024-05-07:
Could you provide the data/database.db entry of this device?

eekamouze, 2024-05-07:
`{"id":25,"type":"EndDevice","ieeeAddr":"0x8cf681fffe34fd1a","nwkAddr":29632,"manufId":4476,"manufName":"IKEA of Sweden","powerSource":"Battery","modelId":"TREDANSEN block-out cellul blind","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":514,"inClusterList":[0,1,3,4,5,32,258,4096,64636],"outClusterList":[25,4096],"clusters":{"genBasic":{"attributes":{"modelId":"TREDANSEN block-out cellul blind","manufacturerName":"IKEA of Sweden","powerSource":3,"zclVersion":3,"appVersion":33,"stackVersion":99,"hwVersion":1,"dateCode":"20201028","swBuildId":"24.4.13"}},"genPollCtrl":{"attributes":{"checkinInterval":172800}},"genPowerCfg":{"attributes":{"batteryPercentageRemaining":180,"batteryVoltage":80}},"closuresWindowCovering":{"attributes":{"currentPositionLiftPercentage":0}}},"binds":[{"cluster":32,"type":"endpoint","deviceIeeeAddress":"0xe0798dfffef8c209","endpointID":1},{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0xe0798dfffef8c209","endpointID":1},{"cluster":258,"type":"endpoint","deviceIeeeAddress":"0xe0798dfffef8c209","endpointID":1}],"configuredReportings":[{"cluster":1,"attrId":33,"minRepIntval":3600,"maxRepIntval":62000,"repChange":0},{"cluster":258,"attrId":8,"minRepIntval":1,"maxRepIntval":62000,"repChange":1},{"cluster":1,"attrId":33,"minRepIntval":3600,"maxRepIntval":62000,"repChange":0,"manufacturerCode":null},{"cluster":258,"attrId":8,"minRepIntval":1,"maxRepIntval":62000,"repChange":1,"manufacturerCode":null}],"meta":{}}},"appVersion":33,"stackVersion":99,"hwVersion":1,"dateCode":"20201028","swBuildId":"24.4.13","zclVersion":3,"interviewCompleted":true,"meta":{"configured":-1281376332},"lastSeen":1715108283343,"checkinInterval":43200}`

Koenkk, 2024-05-08:
Do you have any clue what `swBuildId` it was before the update? (it's in the logs once the update finished)

eekamouze, 2024-05-09:
Not sure what you mean by "before the update". If it is before the update of the device, I've no clue since the last update was a long time ago and I don't have logs anymore.

Koenkk, 2024-05-10:
I see, fixed the issue!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

closed 2024-05-10 by Koenkk