Koenkk/zigbee2mqtt #22530: Homeassistant Yellow Ember

Mar1usW3, 2024-05-07:
### What happened?

I recently switched to ember but every first start the following error occurs. Second start is fine.

### Zigbee2MQTT version

1.37.0

### Adapter firmware version

7.4.2

### Adapter

HA Yellow

### Setup

Addon in HA

### Debug log

```
[2024-05-07 20:58:22] info: 	z2m: Logging to console, file (filename: log.log)
[2024-05-07 20:58:26] debug: 	z2m: Loaded state from file /config/zigbee2mqtt/state.json
[2024-05-07 20:58:26] info: 	z2m: Starting Zigbee2MQTT version 1.37.0 (commit #unknown)
[2024-05-07 20:58:26] info: 	z2m: Starting zigbee-herdsman (0.45.0)
[2024-05-07 20:58:26] debug: 	z2m: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null,"delay":null,"disableLED":false},"backupPath":"/config/zigbee2mqtt/coordinator_backup.json","databaseBackupPath":"/config/zigbee2mqtt/database.db.backup","databasePath":"/config/zigbee2mqtt/database.db","network":{"channelList":[15],"extendedPanID":[0,87,174,108,124,189,189,1],"networkKey":"HIDDEN","panID":20910},"serialPort":{"adapter":"ember","baudRate":115200,"path":"/dev/ttyAMA1","rtscts":true}}'
[2024-05-07 20:58:26] debug: 	zh:adapter: Failed to validate path: 'Error: spawn udevadm ENOENT'
[2024-05-07 20:58:26] debug: 	zh:ember: Using delay=5.
[2024-05-07 20:58:26] debug: 	zh:controller: Starting with options '{"network":{"networkKeyDistribute":false,"networkKey":[10,50,250,203,143,196,207,222,241,222,211,80,171,24,131,183],"panID":20910,"extendedPanID":[0,87,174,108,124,189,189,1],"channelList":[15]},"serialPort":{"baudRate":115200,"rtscts":true,"path":"/dev/ttyAMA1","adapter":"ember"},"databasePath":"/config/zigbee2mqtt/database.db","databaseBackupPath":"/config/zigbee2mqtt/database.db.backup","backupPath":"/config/zigbee2mqtt/coordinator_backup.json","adapter":{"disableLED":false,"concurrent":null,"delay":null}}'
[2024-05-07 20:58:26] info: 	zh:ember: ======== Ember Adapter Starting ========
[2024-05-07 20:58:26] debug: 	zh:ember: Starting EZSP with stack configuration: "default".
[2024-05-07 20:58:26] info: 	zh:ember:ezsp: ======== EZSP starting ========
[2024-05-07 20:58:26] info: 	zh:ember:uart:ash: ======== ASH NCP reset ========
[2024-05-07 20:58:26] debug: 	zh:ember:uart:ash: Opening serial port with {"path":"/dev/ttyAMA1","baudRate":115200,"rtscts":true,"autoOpen":false,"parity":"none","stopBits":1,"xon":false,"xoff":false}
[2024-05-07 20:58:26] info: 	zh:ember:uart:ash: Serial port opened
[2024-05-07 20:58:26] info: 	zh:ember:uart:ash: ======== ASH starting ========
[2024-05-07 20:58:26] debug: 	zh:ember:uart:ash: ---> [FRAME type=RST]
[2024-05-07 20:58:26] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 0/2500
[2024-05-07 20:58:26] debug: 	zh:ember:uart:ash: <--- [FRAME type=ERROR]
[2024-05-07 20:58:26] error: 	zh:ember:uart:ash: Received ERROR from NCP while connecting, with code=RESET_WATCHDOG.
[2024-05-07 20:58:26] error: 	zh:ember:uart:ash: ASH disconnected | NCP status: ASH_NCP_FATAL_ERROR
[2024-05-07 20:58:26] debug: 	zh:ember:uart:ash: Error while parsing received frame in NOT_CONNECTED state (flags=0), status=ASH_NCP_FATAL_ERROR.
[2024-05-07 20:58:26] info: 	zh:ember:uart:ash: ======== ASH NCP reset ========
[2024-05-07 20:58:26] info: 	zh:ember:uart:ash: ======== ASH starting ========
[2024-05-07 20:58:26] debug: 	zh:ember:uart:ash: ---> [FRAME type=RST]
[2024-05-07 20:58:26] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 0/2500
[2024-05-07 20:58:27] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 100/2500
[2024-05-07 20:58:27] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 200/2500
[2024-05-07 20:58:27] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 300/2500
[2024-05-07 20:58:27] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 400/2500
[2024-05-07 20:58:27] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 500/2500
[2024-05-07 20:58:27] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 600/2500
[2024-05-07 20:58:27] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 700/2500
[2024-05-07 20:58:27] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 800/2500
[2024-05-07 20:58:27] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 900/2500
[2024-05-07 20:58:27] debug: 	zh:ember:uart:ash: <--- [FRAME type=RSTACK]
[2024-05-07 20:58:27] info: 	zh:ember:uart:ash: ======== ASH connected ========
[2024-05-07 20:58:27] info: 	zh:ember:uart:ash: ======== ASH started ========
[2024-05-07 20:58:27] info: 	zh:ember:ezsp: ======== EZSP started ========
[2024-05-07 20:58:27] debug: 	zh:ember:ezsp: ===> [FRAME: ID=0:"VERSION" Seq=0 Len=4]
[2024-05-07 20:58:27] debug: 	zh:ember:uart:ash: ---> [FRAME type=ACK frmRx=0]
[2024-05-07 20:58:27] debug: 	zh:ember:uart:ash: ---> [FRAME type=DATA frmTx=0 frmRx=0]
[2024-05-07 20:58:29] warning: 	zh:ember:uart:ash: Frame(s) in progress cancelled in [1ac1020b0a527e]
[2024-05-07 20:58:29] debug: 	zh:ember:uart:ash: <--- [FRAME type=RSTACK]
[2024-05-07 20:58:29] error: 	zh:ember:uart:ash: Received unexpected reset from NCP, with reason=RESET_SOFTWARE.
[2024-05-07 20:58:29] error: 	zh:ember:uart:ash: ASH disconnected: ASH_ERROR_NCP_RESET | NCP status: ASH_NCP_FATAL_ERROR
[2024-05-07 20:58:29] debug: 	zh:ember:uart:ash: Error while parsing received frame in NOT_CONNECTED state (flags=0), status=HOST_FATAL_ERROR.
[2024-05-07 20:58:29] debug: 	zh:ember:uart:ash: Error while parsing received frame in NOT_CONNECTED state (flags=0), status=HOST_FATAL_ERROR.
[2024-05-07 20:58:30] debug: 	zh:ember:uart:ash: Error while parsing received frame in NOT_CONNECTED state (flags=0), status=HOST_FATAL_ERROR.
[2024-05-07 20:58:33] debug: 	zh:ember:uart:ash: Error while parsing received frame in NOT_CONNECTED state (flags=0), status=HOST_FATAL_ERROR.
[2024-05-07 20:58:37] debug: 	zh:ember:uart:ash: Error while parsing received frame in NOT_CONNECTED state (flags=0), status=HOST_FATAL_ERROR.
[2024-05-07 20:58:40] debug: 	zh:ember:uart:ash: Error while parsing received frame in NOT_CONNECTED state (flags=0), status=HOST_FATAL_ERROR.
[2024-05-07 20:58:42] debug: 	zh:ember:ezsp: =x=> [FRAME: ID=0:"VERSION" Seq=0 Len=4] Error: Error: timed out after 14400ms
[2024-05-07 20:58:42] error: 	zh:ember:ezsp: ERROR Transaction failure; status=undefined. Last: [FRAME: ID=0:"VERSION" Seq=0 Len=4].
[2024-05-07 20:58:42] error: 	zh:ember: !!! NCP FATAL ERROR reason=undefined. ATTEMPTING RESET... !!!
[2024-05-07 20:58:42] debug: 	zh:ember:queue: Dispatching stopped; queue=0 priorityQueue=0
[2024-05-07 20:58:42] info: 	zh:ember:uart:ash: ASH COUNTERS since last clear:
[2024-05-07 20:58:42] info: 	zh:ember:uart:ash:   Total frames: RX=2, TX=3
[2024-05-07 20:58:42] info: 	zh:ember:uart:ash:   Cancelled   : RX=1, TX=0
[2024-05-07 20:58:42] info: 	zh:ember:uart:ash:   DATA frames : RX=0, TX=1
[2024-05-07 20:58:42] info: 	zh:ember:uart:ash:   DATA bytes  : RX=0, TX=4
[2024-05-07 20:58:42] info: 	zh:ember:uart:ash:   Retry frames: RX=0, TX=0
[2024-05-07 20:58:42] info: 	zh:ember:uart:ash:   ACK frames  : RX=0, TX=1
[2024-05-07 20:58:42] info: 	zh:ember:uart:ash:   NAK frames  : RX=0, TX=0
[2024-05-07 20:58:42] info: 	zh:ember:uart:ash:   nRdy frames : RX=0, TX=0
[2024-05-07 20:58:42] info: 	zh:ember:uart:ash:   CRC errors      : RX=0
[2024-05-07 20:58:42] info: 	zh:ember:uart:ash:   Comm errors     : RX=0
[2024-05-07 20:58:42] info: 	zh:ember:uart:ash:   Length < minimum: RX=0
[2024-05-07 20:58:42] info: 	zh:ember:uart:ash:   Length > maximum: RX=0
[2024-05-07 20:58:42] info: 	zh:ember:uart:ash:   Bad controls    : RX=0
[2024-05-07 20:58:42] info: 	zh:ember:uart:ash:   Bad lengths     : RX=0
[2024-05-07 20:58:42] info: 	zh:ember:uart:ash:   Bad ACK numbers : RX=0
[2024-05-07 20:58:42] info: 	zh:ember:uart:ash:   Out of buffers  : RX=0
[2024-05-07 20:58:42] info: 	zh:ember:uart:ash:   Retry dupes     : RX=0
[2024-05-07 20:58:42] info: 	zh:ember:uart:ash:   Out of sequence : RX=0
[2024-05-07 20:58:42] info: 	zh:ember:uart:ash:   ACK timeouts    : RX=0
[2024-05-07 20:58:42] error: 	z2m: Error while starting zigbee-herdsman
[2024-05-07 20:58:42] error: 	z2m: Failed to start zigbee
[2024-05-07 20:58:42] error: 	z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
[2024-05-07 20:58:42] error: 	z2m: Exiting...
[2024-05-07 20:58:42] error: 	z2m: Error: 
    at Ezsp.ezspVersion (/app/node_modules/zigbee-herdsman/src/adapter/ember/ezsp/ezsp.ts:1301:19)
    at EmberAdapter.emberVersion (/app/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:1642:66)
    at EmberAdapter.initEzsp (/app/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:802:9)
    at EmberAdapter.start (/app/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:2660:24)
    at Controller.start (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:124:29)
    at Zigbee.start (/app/lib/zigbee.ts:62:27)
    at Controller.start (/app/lib/controller.ts:108:27)
    at start (/app/index.js:107:5)

```

Nerivec, 2024-05-07:
Try the "no hardware flow control" firmware https://github.com/darkxst/silabs-firmware-builder/raw/ember-nohw/firmware_builds/yellow/ncp-uart-hw-v7.4.1.0-yellow-115200.gbl (make sure to set `rtscts: false` in your config)
It should solve some of these.
There is a potential fix for the rest in dev (it's linked to some virtualized envs, and got worse in last couple of months with HA in particular, might require a fix elsewhere than z2m too...). I'm keeping an eye on it.

Jopinder, 2024-05-07:
Same exact issue on a newly installed Yellow with no devices added.
No ZHA or multiprotocol running. If I change the adapter to "ezsp" z2m starts fine.

Flashed firmware mentioned above and set rtscts: false. z2m starts fine with adapter "ember" now :)

Before flash:
```
[22:52:31] INFO: Preparing to start...
[22:52:32] INFO: Socat not enabled
[22:52:34] INFO: Starting Zigbee2MQTT...
[2024-05-07 22:52:37] info: 	z2m: Logging to console, file (filename: log.log)
[2024-05-07 22:52:41] info: 	z2m: Starting Zigbee2MQTT version 1.37.0 (commit #unknown)
[2024-05-07 22:52:41] info: 	z2m: Starting zigbee-herdsman (0.45.0)
[2024-05-07 22:52:41] info: 	zh:ember: ======== Ember Adapter Starting ========
[2024-05-07 22:52:41] info: 	zh:ember:ezsp: ======== EZSP starting ========
[2024-05-07 22:52:41] info: 	zh:ember:uart:ash: ======== ASH NCP reset ========
[2024-05-07 22:52:41] info: 	zh:ember:uart:ash: Serial port opened
[2024-05-07 22:52:41] info: 	zh:ember:uart:ash: ======== ASH starting ========
[2024-05-07 22:52:41] error: 	zh:ember:uart:ash: Received ERROR from NCP while connecting, with code=RESET_WATCHDOG.
[2024-05-07 22:52:41] error: 	zh:ember:uart:ash: ASH disconnected | NCP status: ASH_NCP_FATAL_ERROR
[2024-05-07 22:52:41] info: 	zh:ember:uart:ash: ======== ASH NCP reset ========
[2024-05-07 22:52:41] info: 	zh:ember:uart:ash: ======== ASH starting ========
[2024-05-07 22:52:42] info: 	zh:ember:uart:ash: ======== ASH connected ========
[2024-05-07 22:52:42] info: 	zh:ember:uart:ash: ======== ASH started ========
[2024-05-07 22:52:42] info: 	zh:ember:ezsp: ======== EZSP started ========
[2024-05-07 22:52:43] warning: 	zh:ember:uart:ash: Frame(s) in progress cancelled in [1ac1020b0a527e]
[2024-05-07 22:52:43] error: 	zh:ember:uart:ash: Received unexpected reset from NCP, with reason=RESET_SOFTWARE.
[2024-05-07 22:52:43] error: 	zh:ember:uart:ash: ASH disconnected: ASH_ERROR_NCP_RESET | NCP status: ASH_NCP_FATAL_ERROR
[2024-05-07 22:52:57] error: 	zh:ember:ezsp: ERROR Transaction failure; status=undefined. Last: [FRAME: ID=0:"VERSION" Seq=0 Len=4].
[2024-05-07 22:52:57] error: 	zh:ember: !!! NCP FATAL ERROR reason=undefined. ATTEMPTING RESET... !!!
[2024-05-07 22:52:57] info: 	zh:ember:uart:ash: ASH COUNTERS since last clear:
[2024-05-07 22:52:57] info: 	zh:ember:uart:ash:   Total frames: RX=2, TX=3
[2024-05-07 22:52:57] info: 	zh:ember:uart:ash:   Cancelled   : RX=1, TX=0
[2024-05-07 22:52:57] info: 	zh:ember:uart:ash:   DATA frames : RX=0, TX=1
[2024-05-07 22:52:57] info: 	zh:ember:uart:ash:   DATA bytes  : RX=0, TX=4
[2024-05-07 22:52:57] info: 	zh:ember:uart:ash:   Retry frames: RX=0, TX=0
[2024-05-07 22:52:57] info: 	zh:ember:uart:ash:   ACK frames  : RX=0, TX=1
[2024-05-07 22:52:57] info: 	zh:ember:uart:ash:   NAK frames  : RX=0, TX=0
[2024-05-07 22:52:57] info: 	zh:ember:uart:ash:   nRdy frames : RX=0, TX=0
[2024-05-07 22:52:57] info: 	zh:ember:uart:ash:   CRC errors      : RX=0
[2024-05-07 22:52:57] info: 	zh:ember:uart:ash:   Comm errors     : RX=0
[2024-05-07 22:52:57] info: 	zh:ember:uart:ash:   Length < minimum: RX=0
[2024-05-07 22:52:57] info: 	zh:ember:uart:ash:   Length > maximum: RX=0
[2024-05-07 22:52:57] info: 	zh:ember:uart:ash:   Bad controls    : RX=0
[2024-05-07 22:52:57] info: 	zh:ember:uart:ash:   Bad lengths     : RX=0
[2024-05-07 22:52:57] info: 	zh:ember:uart:ash:   Bad ACK numbers : RX=0
[2024-05-07 22:52:57] info: 	zh:ember:uart:ash:   Out of buffers  : RX=0
[2024-05-07 22:52:57] info: 	zh:ember:uart:ash:   Retry dupes     : RX=0
[2024-05-07 22:52:57] info: 	zh:ember:uart:ash:   Out of sequence : RX=0
[2024-05-07 22:52:57] info: 	zh:ember:uart:ash:   ACK timeouts    : RX=0
[2024-05-07 22:52:57] error: 	z2m: Error while starting zigbee-herdsman
[2024-05-07 22:52:57] error: 	z2m: Failed to start zigbee
[2024-05-07 22:52:57] error: 	z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
[2024-05-07 22:52:57] error: 	z2m: Exiting...
[2024-05-07 22:52:57] error: 	z2m: Error: 
    at Ezsp.ezspVersion (/app/node_modules/zigbee-herdsman/src/adapter/ember/ezsp/ezsp.ts:1301:19)
    at EmberAdapter.emberVersion (/app/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:1642:66)
    at EmberAdapter.initEzsp (/app/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:802:9)
    at EmberAdapter.start (/app/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:2660:24)
    at Controller.start (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:124:29)
    at Zigbee.start (/app/lib/zigbee.ts:62:27)
    at Controller.start (/app/lib/controller.ts:108:27)
    at start (/app/index.js:107:5)
```

After flash:
```
[23:01:44] INFO: Preparing to start...
[23:01:45] INFO: Socat not enabled
[23:01:47] INFO: Starting Zigbee2MQTT...
[2024-05-07 23:01:50] info: 	z2m: Logging to console, file (filename: log.log)
[2024-05-07 23:01:54] info: 	z2m: Starting Zigbee2MQTT version 1.37.0 (commit #unknown)
[2024-05-07 23:01:54] info: 	z2m: Starting zigbee-herdsman (0.45.0)
[2024-05-07 23:01:54] info: 	zh:ember: ======== Ember Adapter Starting ========
[2024-05-07 23:01:54] info: 	zh:ember:ezsp: ======== EZSP starting ========
[2024-05-07 23:01:54] info: 	zh:ember:uart:ash: ======== ASH NCP reset ========
[2024-05-07 23:01:54] info: 	zh:ember:uart:ash: RTS/CTS config is off, enabling software flow control.
[2024-05-07 23:01:54] info: 	zh:ember:uart:ash: Serial port opened
[... cut ...]
[2024-05-07 23:01:56] info: 	z2m: Coordinator firmware version: '{"meta":{"build":0,"ezsp":13,"major":7,"minor":4,"patch":1,"revision":"7.4.1 [GA]","special":0,"type":170},"type":"EmberZNet"}'
[2024-05-07 23:01:56] info: 	z2m: Currently 0 devices are joined:
[2024-05-07 23:01:56] info: 	z2m: Zigbee: disabling joining new devices.
[2024-05-07 23:01:56] info: 	z2m: Connecting to MQTT server at mqtt://core-mosquitto:1883
[2024-05-07 23:01:56] info: 	z2m: Connected to MQTT server
[2024-05-07 23:01:56] info: 	z2m: Started frontend on port 8099
[2024-05-07 23:01:56] info: 	z2m: Zigbee2MQTT started!
```

Mar1usW3, 2024-05-08:
> Try the "no hardware flow control" firmware https://github.com/darkxst/silabs-firmware-builder/raw/ember-nohw/firmware_builds/yellow/ncp-uart-hw-v7.4.1.0-yellow-115200.gbl (make sure to set `rtscts: [...]

flashed the "no hardware flow control" firmware but 7.4.2.0 version and set rtscts to flase and looking good now

closed 2024-05-08 by Mar1usW3

Nerivec, 2024-05-09:
@Mar1usW3 Do you mind testing a firmware with hardware flow control on your Yellow with 1.37.1 and provide a debug log of the startup sequence?
_Make sure to adjust the `rtscts` setting accordingly._

Mar1usW3, 2024-05-10:
> @Mar1usW3 Do you mind testing a firmware with hardware flow control on your Yellow with 1.37.1 and provide a debug log of the startup sequence?
> 
> _Make sure to adjust the `rtscts` setting accordingly._

Can do it. So rtscts needs to be true?

Nerivec, 2024-05-10:
Yes, with hardware flow control, true, without, false.

Mar1usW3, 2024-05-10:
so i flashed the yellow with https://github.com/darkxst/silabs-firmware-builder/raw/main/firmware_builds/yellow/ncp-uart-hw-v7.4.2.0-yellow-115200.gbl and rtscts to true. It is now starting but with some errors.
I only copied the logfile till MQTT connect. If you need more let me know

```
[2024-05-10 13:04:13] info: 	z2m: Logging to console, file (filename: log.log)
[2024-05-10 13:04:13] debug: 	z2m: Loaded state from file /config/zigbee2mqtt/state.json
[2024-05-10 13:04:13] info: 	z2m: Starting Zigbee2MQTT version 1.37.1 (commit #unknown)
[2024-05-10 13:04:13] info: 	z2m: Starting zigbee-herdsman (0.46.6)
[2024-05-10 13:04:13] debug: 	z2m: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null,"delay":null,"disableLED":false},"backupPath":"/config/zigbee2mqtt/coordinator_backup.json","databaseBackupPath":"/config/zigbee2mqtt/database.db.backup","databasePath":"/config/zigbee2mqtt/database.db","network":{"channelList":[15],"extendedPanID":[xXx],"networkKey":"HIDDEN","panID":20910},"serialPort":{"adapter":"ember","baudRate":115200,"path":"/dev/ttyAMA1","rtscts":true}}'
[2024-05-10 13:04:13] debug: 	zh:adapter: Failed to validate path: 'Error: spawn udevadm ENOENT'
[2024-05-10 13:04:13] debug: 	zh:ember: Using delay=5.
[2024-05-10 13:04:13] debug: 	zh:controller: Starting with options '{"network":{"networkKeyDistribute":false,"networkKey":[10,50,250,203,143,196,207,222,241,222,211,80,171,24,131,183],"panID":20910,"extendedPanID":[xXx],"channelList":[15]},"serialPort":{"baudRate":115200,"rtscts":true,"path":"/dev/ttyAMA1","adapter":"ember"},"databasePath":"/config/zigbee2mqtt/database.db","databaseBackupPath":"/config/zigbee2mqtt/database.db.backup","backupPath":"/config/zigbee2mqtt/coordinator_backup.json","adapter":{"disableLED":false,"concurrent":null,"delay":null}}'
[2024-05-10 13:04:13] info: 	zh:ember: ======== Ember Adapter Starting ========
[2024-05-10 13:04:13] debug: 	zh:ember: Starting EZSP with stack configuration: "default".
[2024-05-10 13:04:13] info: 	zh:ember:ezsp: ======== EZSP starting ========
[2024-05-10 13:04:13] info: 	zh:ember:uart:ash: ======== ASH NCP reset ========
[2024-05-10 13:04:13] debug: 	zh:ember:uart:ash: Opening serial port with {"path":"/dev/ttyAMA1","baudRate":115200,"rtscts":true,"autoOpen":false,"parity":"none","stopBits":1,"xon":false,"xoff":false}
[2024-05-10 13:04:13] info: 	zh:ember:uart:ash: Serial port opened
[2024-05-10 13:04:13] info: 	zh:ember:uart:ash: ======== ASH starting ========
[2024-05-10 13:04:13] debug: 	zh:ember:uart:ash: ---> [FRAME type=RST]
[2024-05-10 13:04:13] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 0/2500
[2024-05-10 13:04:13] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 100/2500
[2024-05-10 13:04:13] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 200/2500
[2024-05-10 13:04:13] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 300/2500
[2024-05-10 13:04:14] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 400/2500
[2024-05-10 13:04:14] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 500/2500
[2024-05-10 13:04:14] debug: 	zh:ember:uart:ash: <--- [FRAME type=ERROR]
[2024-05-10 13:04:14] error: 	zh:ember:uart:ash: Received ERROR from NCP while connecting, with code=RESET_WATCHDOG.
[2024-05-10 13:04:14] error: 	zh:ember:uart:ash: ASH disconnected | NCP status: ASH_NCP_FATAL_ERROR
[2024-05-10 13:04:14] error: 	zh:ember:uart:ash: Error while parsing received frame, status=ASH_NCP_FATAL_ERROR.
[2024-05-10 13:04:14] info: 	zh:ember:uart:ash: ======== ASH NCP reset ========
[2024-05-10 13:04:14] info: 	zh:ember:uart:ash: ======== ASH starting ========
[2024-05-10 13:04:14] debug: 	zh:ember:uart:ash: ---> [FRAME type=RST]
[2024-05-10 13:04:14] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 0/2500
[2024-05-10 13:04:14] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 100/2500
[2024-05-10 13:04:14] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 200/2500
[2024-05-10 13:04:14] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 300/2500
[2024-05-10 13:04:14] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 400/2500
[2024-05-10 13:04:14] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 500/2500
[2024-05-10 13:04:14] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 600/2500
[2024-05-10 13:04:14] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 700/2500
[2024-05-10 13:04:15] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 800/2500
[2024-05-10 13:04:15] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 900/2500
[2024-05-10 13:04:15] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 1000/2500
[2024-05-10 13:04:15] debug: 	zh:ember:uart:ash: <--- [FRAME type=RSTACK]
[2024-05-10 13:04:15] info: 	zh:ember:uart:ash: ======== ASH connected ========
[2024-05-10 13:04:15] debug: 	zh:ember:uart:ash: <--- [FRAME type=NAK]
[2024-05-10 13:04:15] debug: 	zh:ember:uart:ash: <--- [FRAME type=NAK ackNum=0]
[2024-05-10 13:04:15] debug: 	zh:ember:uart:ash: ---> [FRAME type=ACK frmRx=0]
[2024-05-10 13:04:15] info: 	zh:ember:uart:ash: ======== ASH started ========
[2024-05-10 13:04:15] info: 	zh:ember:ezsp: ======== EZSP started ========
[2024-05-10 13:04:15] debug: 	zh:ember:ezsp: ===> [FRAME: ID=0:"VERSION" Seq=0 Len=4]
[2024-05-10 13:04:15] debug: 	zh:ember:uart:ash: ---> [FRAME type=DATA frmTx=0 frmRx=0]
[2024-05-10 13:04:15] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA]
[2024-05-10 13:04:15] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=1]
[... cut ...]
[2024-05-10 13:04:16] debug: 	zh:ember:ezsp: ezspIncomingMessageHandler(): callback called with: [type=BROADCAST_LOOPBACK], [apsFrame={"profileId":41440,"clusterId":33,"sourceEndpoint":242,"destinationEndpoint":242,"options":256,"groupId":0,"sequence":238}], [lastHopLqi=255], [lastHopRssi=0], [sender=0], [bindingIndex=255], [addressIndex=255], [messageContents=1902020a0000]
[2024-05-10 13:04:16] info: 	z2m: Connected to MQTT server
[2024-05-10 13:04:16] debug: 	z2m: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload '{"state":"online"}'
```

Nerivec, 2024-05-10:
Thanks, that's just what I need. Appreciated!
The hardware flow control firmware has a messier start than the software flow control one at the moment, so the errors at the top are expected; I wanted to make sure that with the new fixes it started nonetheless. You can revert to no hardware flow control if you want to get rid of those startup errors entirely 😉

swind88, 2025-07-26:
Hello, I'm not able to use the HA yellow with MQTT, I guess it's the same issue described here. Can you explain me how to flash the firmware?