Koenkk/zigbee2mqtt #22534: Aqara Roller Shade Driver E1 state unknown

Alfy1080, 2024-05-08:
### What happened?

After upgrading Z2M to v1.37.0, the state of my Aqara Roller Shade Drivers E1 is now Unknown in Home Assistant, due to which in Google Home the devices appear to be offline. The roller shades are still controllable via voice command from Google Assistant, though every time i issue a voice command i get an error saying that the feature is not supported for the devices (opening, closing, setting a certain percentage) and in the Google Home UI the devices appear offline and show no command available. The roller shades are also controllable via Home Assistant without any issue there. This only affects Google Home.

### What did you expect to happen?

I expected that the roller shades would appear online in Google Home, the controls to be available in the Google Home UI and that there would be no error when issuing voice commands via Google Assistant 

### How to reproduce it (minimal and precise)

Pair a roller shade driver E1 to Z2M in Home Assistant > publish the device to Google Home via Nabu Casa cloud > Check the device state in Google Home UI/Issue a command for the roller shade via Google Assistant.

### Zigbee2MQTT version

1.37.0

### Adapter firmware version

20230507

### Adapter

Sonoff ZB Dongle-P (CC2652P)

### Setup

Home Assistant Supervised installation on Debian 12 running on generic x86 hardware

### Debug log

[log.log](https://github.com/Koenkk/zigbee2mqtt/files/15244030/log.log)

pantherale0, 2024-05-08:
similar problem here, although after upgrading my blinds are always "closed" which was the state they were before upgrading. Changing the settings to invert the position also does not help and the state remains closed. As you can see below the state is closed, but the position is actually open.

[image]

mamrai1, 2024-05-08:
The problem is fixed in latest dev update

Koenkk, 2024-05-08:
Duplicate of https://github.com/Koenkk/zigbee2mqtt/issues/22387

closed 2024-05-08 by Koenkk