Koenkk/zigbee2mqtt #22537: Entity ID name does not change

m-landowski, 2024-05-08:
### What happened?

I would like to change the Entity ID for the newly added device in Z2m (HA add-on).

IN z2m set: 
[image]

[image]

 

### How to reproduce it (minimal and precise)

Please Add New Device in Z2M , and Set Friendly Name 

### Zigbee2MQTT version

 1.37.0

### Adapter firmware version

0x26580700

### Adapter

ConBee2/RaspBee2

### Setup

ADD On HA

dyvyak, 2024-05-08:
same problem here. also on official android application is missing icon for renaming devices

Koenkk, 2024-05-08:
It's fixed in https://github.com/nurikk/zigbee2mqtt-frontend/pull/2010, there will be a hotfix release in the coming days.

closed 2024-05-08 by Koenkk