Koenkk/zigbee2mqtt #22540: zh:ember:ezsp: Received network/route error ROUTE_ERROR_MANY_TO_ONE_ROUTE_FAILURE

prodoff, 2024-05-08:
### What happened?

Depuis la mise à jour de la clé et le passage en z2m 1.37 et ember, j'ai de nombreux warnings

### How to reproduce it (minimal and precise)

Just run Z2M...

### Zigbee2MQTT version

1.37.0-1

### Adapter firmware version

7.4.3 [GA]

### Adapter

sonoff dongle_E

### Setup

addon zigbee2mqtt 1.37.0-1 / NUC Hp prodesk G400 (I5-8Go)/ HA2024.5.2

### Debug log

[2024-05-08 10:51:08] warning: 	z2m: Failed to ping 'plug zb 006' (attempt 1/2, ZCL command 0xa4c1380fee2a7965/1 genBasic.read(["zclVersion"], {"timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Delivery failed for {"profileId":260,"clusterId":0,"sourceEndpoint":1,"destinationEndpoint":1,"options":4416,"groupId":0,"sequence":109}))
[2024-05-08 10:52:56] error: 	z2m: Publish 'set' 'state' to 'Salle de bain' failed: 'Error: ZCL command 0xa4c138dacecbb74c/1 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed ({"target":61966,"apsFrame":{"profileId":260,"clusterId":6,"sourceEndpoint":1,"destinationEndpoint":1,"options":4416,"groupId":0,"sequence":180},"zclSequence":7,"commandIdentifier":11} timed out after 10000ms)'
[2024-05-08 10:54:38] warning: 	zh:ember:ezsp: Received network/route error ROUTE_ERROR_MANY_TO_ONE_ROUTE_FAILURE for "52460".
[2024-05-08 10:54:38] warning: 	zh:ember:ezsp: Received network/route error ROUTE_ERROR_MANY_TO_ONE_ROUTE_FAILURE for "52460".
[2024-05-08 10:56:02] warning: 	zh:ember:ezsp: Received network/route error ROUTE_ERROR_MANY_TO_ONE_ROUTE_FAILURE for "31066".
[2024-05-08 10:57:42] warning: 	zh:ember:ezsp: Received network/route error ROUTE_ERROR_MANY_TO_ONE_ROUTE_FAILURE for "32585".
[2024-05-08 10:58:33] warning: 	zh:ember:ezsp: Received network/route error ROUTE_ERROR_MANY_TO_ONE_ROUTE_FAILURE for "19414".
[2024-05-08 10:58:49] warning: 	zh:ember:ezsp: Received network/route error ROUTE_ERROR_MANY_TO_ONE_ROUTE_FAILURE for "31066".
[2024-05-08 10:58:52] warning: 	zh:ember:ezsp: Received network/route error ROUTE_ERROR_MANY_TO_ONE_ROUTE_FAILURE for "61859".
[2024-05-08 11:01:29] error: 	z2m: Publish 'set' 'state' to 'Fingerbot 1' failed: 'Error: ZCL command 0xa4c1388f2d3731ce/1 genOnOff.on({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed ({"target":60046,"apsFrame":{"profileId":260,"clusterId":6,"sourceEndpoint":1,"destinationEndpoint":1,"options":4416,"groupId":0,"sequence":208},"zclSequence":102,"commandIdentifier":11} timed out after 10000ms)'
[2024-05-08 11:01:32] warning: 	zh:ember:ezsp: Received network/route error ROUTE_ERROR_MANY_TO_ONE_ROUTE_FAILURE for "31066".
[2024-05-08 11:02:02] warning: 	zh:ember:ezsp: Received network/route error ROUTE_ERROR_MANY_TO_ONE_ROUTE_FAILURE for "38485".
[2024-05-08 11:02:57] warning: 	zh:ember:ezsp: Received network/route error ROUTE_ERROR_MANY_TO_ONE_ROUTE_FAILURE for "36914".
[2024-05-08 11:07:45] warning: 	zh:ember:ezsp: Received network/route error ROUTE_ERROR_MANY_TO_ONE_ROUTE_FAILURE for "33903".
[2024-05-08 11:10:51] warning: 	zh:ember:ezsp: Received network/route error ROUTE_ERROR_MANY_TO_ONE_ROUTE_FAILURE for "35716".
[2024-05-08 11:11:02] warning: 	zh:ember:ezsp: Received network/route error ROUTE_ERROR_MANY_TO_ONE_ROUTE_FAILURE for "52460".
[2024-05-08 11:11:22] info: 	zh:ember: [NCP COUNTERS] 17193,644,5852,3479,1380,358,9,2,3069,2107,0,20,97,595,569,301,0,0,0,0,0,632,0,0,0,2489,0,0,0,0,0,0,28,0,0,0,0,0,0,0,0
[2024-05-08 11:11:22] info: 	zh:ember: [ASH COUNTERS] 56378,10842,2180,8661,0,0,0,0,210106,8710,8709,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0
[2024-05-08 11:12:22] warning: 	zh:ember:ezsp: Received network/route error ROUTE_ERROR_MANY_TO_ONE_ROUTE_FAILURE for "31066".
[2024-05-08 11:13:20] warning: 	zh:ember:ezsp: Received network/route error ROUTE_ERROR_MANY_TO_ONE_ROUTE_FAILURE for "61966".
[2024-05-08 11:15:34] warning: 	zh:ember:ezsp: Received network/route error ROUTE_ERROR_MANY_TO_ONE_ROUTE_FAILURE for "19414".
[2024-05-08 11:16:41] warning: 	zh:ember:ezsp: Received network/route error ROUTE_ERROR_MANY_TO_ONE_ROUTE_FAILURE for "52460".
[2024-05-08 11:17:13] warning: 	zh:ember:ezsp: Received network/route error ROUTE_ERROR_MANY_TO_ONE_ROUTE_FAILURE for "26490".
[2024-05-08 11:18:20] warning: 	zh:ember:ezsp: Received network/route error ROUTE_ERROR_MANY_TO_ONE_ROUTE_FAILURE for "38485".
[2024-05-08 11:18:24] warning: 	zh:ember:ezsp: Received network/route error ROUTE_ERROR_MANY_TO_ONE_ROUTE_FAILURE for "5612".
[2024-05-08 11:18:24] warning: 	zh:ember:ezsp: Received network/route error ROUTE_ERROR_MANY_TO_ONE_ROUTE_FAILURE for "5612".
[2024-05-08 11:18:35] warning: 	zh:ember:ezsp: Received network/route error ROUTE_ERROR_MANY_TO_ONE_ROUTE_FAILURE for "33903".
[2024-05-08 11:21:02] warning: 	zh:ember:ezsp: Received network/route error ROUTE_ERROR_MANY_TO_ONE_ROUTE_FAILURE for "31066".
[2024-05-08 11:23:35] warning: 	zh:ember:ezsp: Received network/route error ROUTE_ERROR_MANY_TO_ONE_ROUTE_FAILURE for "19414".
[2024-05-08 11:23:49] warning: 	zh:ember:ezsp: Received network/route error ROUTE_ERROR_MANY_TO_ONE_ROUTE_FAILURE for "38485".
[2024-05-08 11:24:29] warning: 	zh:ember:ezsp: Received network/route error ROUTE_ERROR_MANY_TO_ONE_ROUTE_FAILURE for "5612".
[2024-05-08 11:26:28] warning: 	zh:ember:ezsp: Received network/route error ROUTE_ERROR_MANY_TO_ONE_ROUTE_FAILURE for "38485".
[2024-05-08 11:26:40] error: 	z2m: Publish 'set' 'state' to 'Relais frisquet' failed: 'Error: ZCL command 0xa4c13822d6bc0e62/1 genOnOff.on({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed ({"target":52460,"apsFrame":{"profileId":260,"clusterId":6,"sourceEndpoint":1,"destinationEndpoint":1,"options":4416,"groupId":0,"sequence":37},"zclSequence":134,"commandIdentifier":11} timed out after 10000ms)'

Ciqsky, 2024-05-09:
me too, same problem

Nerivec, 2024-05-10:
Please see https://github.com/Koenkk/zigbee2mqtt/issues/22516

closed 2024-05-10 by Nerivec