Koenkk/zigbee2mqtt #22543: [New device support]: Smart water level meter TLC2206 zigbee

TonioClos, 2024-05-08:
### Link

https://es.aliexpress.com/item/1005006395402636.html?gatewayAdapt=glo2esp

### Database entry

{"id":66,"type":"EndDevice","ieeeAddr":"0x0ceff6fffe4bdda2","nwkAddr":39698,"manufId":4098,"manufName":"_TZE200_lvkk0hdg","powerSource":"Battery","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,4,5,61184],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"�ME-\u0019%NE-\u0019aNE-j�NE-j\u000fPE-j�KE-j�KE-\u0019>LE-i","65506":31,"65508":0,"65534":0,"appVersion":65,"modelId":"TS0601","manufacturerName":"_TZE200_lvkk0hdg","powerSource":3,"zclVersion":3,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":65,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":-708457359},"lastSeen":1706198663760,"defaultSendRequestWhen":"immediate"}

### Comments

I tried following the guideline to add a new device and using the external definition another guy did in this post #21015 but anything seems to work now. I had it at least supported but without any funtion working or showing in HA, I had another device configured following the guideline for tuya devices in [https://www.zigbee2mqtt.io/advanced/support-new-devices/02_support_new_tuya_devices.html](url) and at least shows the data I need but not all the data.
I have two questions in here: 
- Is there any problem in the external definition I'm not seeing? at least before this last update of Z2M, I had the sensor showing me the linkqualy.
- I don´t know if multiple integrations of tuya devices causes them to collide and just the first works, the fingerprint it's different in the two but in the other seems to be working fine.

Sorry if I made mistakes with my english and for been new to programming, maybe many things can be found easier than I think. Thank you so much in advance with any help given by the community and everyone n_n

### External definition

```shell
//BAsically a combination of the code that it's working on the other device, the oficial documentation of Z2M and
//the post mentioned in the comments

const {} = require('zigbee-herdsman-converters/lib/modernExtend');
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
    // Since a lot of TuYa devices use the same modelID, but use different datapoints
    // it's necessary to provide a fingerprint instead of a zigbeeModel
    fingerprint: [
        {
            // The model ID from: Device with modelID 'TS0601' is not supported
            // You may need to add \u0000 at the end of the name in some cases
            modelID: 'TS0601',
            // The manufacturer name from: Device with modelID 'TS0601' is not supported.
            manufacturerName: '_TZE200_lvkk0hdg',
        },
    ],
    model: 'TYZS3-IPEX',
    vendor: 'TuYa',
    whiteLabel: [
        {
          vendor: 'EPTTECH',
          model: 'TLC2206-ZB',
        },
      ],
description: 'EPTTECH Tank Level Monitor Zigbee',
fromZigbee: [tuya.fz.datapoints],
toZigbee: [tuya.tz.datapoints],
onEvent: tuya.onEventSetTime,
configure: tuya.configureMagicPacket,
exposes: [
  e.numeric('liquid_level_percent', ea.STATE)
    .withUnit('%')
    .withDescription('Liquid level percentage'),
  e.numeric('liquid_depth', ea.STATE)
    .withUnit('m')
    .withDescription('Liquid Depth'),
  e.enum('liquid_state', ea.STATE, ['low', 'normal', 'high'])
    .withDescription('Liquid State'),
  // =============== Found but non functional ===============
  /*
  e.numeric('max_set', ea.STATE_SET)
    .withUnit('%')
    .withDescription('Liquid max percentage')
    .withValueMin(0)
    .withValueMax(100)
    .withValueStep(1),
  e.numeric('mini_set', ea.STATE_SET)
    .withUnit('%')
    .withDescription('Liquid minimal percentage')
    .withValueMin(0)
    .withValueMax(100)
    .withValueStep(1),
  e.numeric('installation_height', ea.STATE_SET)
    .withUnit('mm')
    .withDescription('Height from sensor to tank bottom')
    .withValueMin(100)
    .withValueMax(2500)
    .withValueStep(1),
  e.numeric('liquid_depth_max', ea.STATE_SET)
    .withUnit('mm')
    .withDescription('Height from sensor to liquid level')
    .withValueMin(100)
    .withValueMax(1000)
    .withValueStep(1),
  */
],
meta: {
  tuyaDatapoints: [
    [1, 'liquid_state', tuya.valueConverterBasic.lookup({'low': 1, 'normal': 0, 'high': 2})],
    [2, 'liquid_depth', tuya.valueConverter.divideBy100],
    [22, 'liquid_level_percent', tuya.valueConverter.raw],
    // =============== Found but non functional ===============
    /*
    [7, 'max_set', tuya.valueConverter.raw],
    [8, 'mini_set', tuya.valueConverter.raw],
    [19, 'installation_height', tuya.valueConverter.raw],
    [21, 'liquid_depth_max', tuya.valueConverter.raw],
    */
  ],
},
};

module.exports = definition;
```

TonioClos, 2024-05-15:
@Koenkk any opportunity to at least know why is it not even showing as supported even with the code that the documentation shows?

Koenkk, 2024-05-16:
Probably the external converter fails to load due to a syntax error, this will be logged at z2m startup. Look for `Failed to load external converter file`

TonioClos, 2024-05-31:
This was completed on the original post, that reactivated and a really nice guy completed the code and works fine.
Here the [link](https://github.com/Koenkk/zigbee2mqtt/issues/21015) and mention the post #21015.
thanks @Koenkk for helping me and @gregorobreza for the code :D

RobLeighton67, 2024-09-12:
Used the referenced converter and all working fine after addition, but the device is very spammy - Zigbee msg every 5s or so - I can debounce it in Z2M to lower the MQTT msg load but I'd prefer the device to have a minimum reporting time of say 30s ?
Can a converter be tweaked to configure this behaviour with Tuya devices, @gregorobreza  ?

ivanfmartinez, 2024-09-12:
> Used the referenced converter and all working fine after addition, but the device is very spammy - Zigbee msg every 5s or so - I can debounce it in Z2M to lower the MQTT msg load but I'd prefer the [...]

I have solved my SPAMMING problem with little changes on Z2M, In my case it worked without side effects

https://github.com/Koenkk/zigbee2mqtt/issues/17984#issuecomment-2267638647changes on z2m, In my case it worked without side effects

https://github.com/Koenkk/zigbee2mqtt/issues/17984#issuecomment-2267638647

Maybe @Koenkk can give some feedback about how will be a good definitive solution for the many cases of this SPAMMING devices

Koenkk, 2024-09-13:
Tuya doesn't allow changing the reporting interval, so we cannot reduce the spamming of this device.

ivanfmartinez, 2024-09-13:
> Tuya doesn't allow changing the reporting interval, so we cannot reduce the spamming of this device.

But at least we can ignore some messages in this cases that the real changes are not so frequent. And with this we dont have a lot of mqtt messages and huge data storage on homeassistant.

DrTrotty, 2024-09-26:
I can't get the device to stop spamming. If I set the debunce value, it simply doesn't change anything, maybe only after a few minutes. How did you do? @ivanfmartinez

ivanfmartinez, 2024-09-27:
> I can't get the device to stop spamming. If I set the debunce value, it simply doesn't change anything, maybe only after a few minutes. How did you do? @ivanfmartinez

@DrTrotty  debouce does not work for this devices that keep sending messages, debounce needs that device stay some time without sending message.

I made this changes to just drop some messages received from the device : 

https://github.com/Koenkk/zigbee2mqtt/issues/17984#issuecomment-2267638647

DrTrotty, 2024-09-27:
Thanks. I looked. Unfortunately, this is beyond my abilities...

DrTrotty, 2024-10-02:
Thank you. I am happy!

closed 2025-01-04 by github-actions[bot]

anlek, 2025-01-13:
Any updates on this?