Koenkk/zigbee2mqtt #22544: Warnings with 1.37.0-1

raveit65, 2024-05-08:
### What happened?

I see weird warnings in z2m log after a new start of z2m.
```
[2024-05-08 18:28:57] info: 	z2m: Licht Schlafz Boxen (0xf0d1b80000187efe): AC25697 - LEDVANCE SMART+ CLASSIC MULTICOLOUR 60 10W E27 (Router)
[2024-05-08 18:28:57] info: 	z2m: Thermo Schlafz (0x00158d0001ff9101): SPZB0001 - Eurotronic Spirit Zigbee wireless heater thermostat (EndDevice)
[2024-05-08 18:28:57] info: 	z2m: Zigbee: disabling joining new devices.
[2024-05-08 18:28:57] info: 	z2m: Connecting to MQTT server at mqtt://core-mosquitto:1883
[2024-05-08 18:28:57] info: 	z2m: Connected to MQTT server
[2024-05-08 18:28:58] info: 	z2m: Started frontend on port 8099
[2024-05-08 18:28:58] info: 	z2m: Zigbee2MQTT started!
[2024-05-08 18:29:12] error: 	zh:zstack:znp: Error while parsing to ZpiObject 'Error: CommandID '159' from subsystem '5' not found
    at Function.fromUnpiFrame (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/znp/zpiObject.ts:60:19)
    at Znp.onUnpiParsed (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/znp/znp.ts:93:38)
    at Parser.emit (node:events:517:28)
    at Parser.parseNext (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/unpi/parser.ts:45:26)
    at Parser.parseNext (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/unpi/parser.ts:51:22)
    at Parser._transform (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/unpi/parser.ts:19:14)
    at Parser.Transform._write (node:internal/streams/transform:175:8)
    at writeOrBuffer (node:internal/streams/writable:392:12)
    at _write (node:internal/streams/writable:333:10)
    at Parser.Writable.write (node:internal/streams/writable:337:10)'
[2024-05-08 18:30:31] info: 	z2m: Starting network scan (includeRoutes 'false')
[2024-05-08 18:31:19] info: 	z2m: Network scan finished
```
z2m seems to be working but i missed a clean log withot warnings which i don't understand.
Maybe this is evil?

### What did you expect to happen?

No warning in z2m log like it was with previous version.

### How to reproduce it (minimal and precise)

Start or restart z2m on a hp ThinClint_t630

### Zigbee2MQTT version

1.37.0-1

### Adapter firmware version

13.4.0(zbbrdgpro)

### Adapter

sonoff.zbbridge.pro

### Setup

plain on x86_64 , HP ThinClint_t630

### Debug log

[2024-05-08 18:28:57] info: 	z2m: Licht Schlafz Boxen (0xf0d1b80000187efe): AC25697 - LEDVANCE SMART+ CLASSIC MULTICOLOUR 60 10W E27 (Router)
[2024-05-08 18:28:57] info: 	z2m: Thermo Schlafz (0x00158d0001ff9101): SPZB0001 - Eurotronic Spirit Zigbee wireless heater thermostat (EndDevice)
[2024-05-08 18:28:57] info: 	z2m: Zigbee: disabling joining new devices.
[2024-05-08 18:28:57] info: 	z2m: Connecting to MQTT server at mqtt://core-mosquitto:1883
[2024-05-08 18:28:57] info: 	z2m: Connected to MQTT server
[2024-05-08 18:28:58] info: 	z2m: Started frontend on port 8099
[2024-05-08 18:28:58] info: 	z2m: Zigbee2MQTT started!
[2024-05-08 18:29:12] error: 	zh:zstack:znp: Error while parsing to ZpiObject 'Error: CommandID '159' from subsystem '5' not found
    at Function.fromUnpiFrame (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/znp/zpiObject.ts:60:19)
    at Znp.onUnpiParsed (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/znp/znp.ts:93:38)
    at Parser.emit (node:events:517:28)
    at Parser.parseNext (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/unpi/parser.ts:45:26)
    at Parser.parseNext (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/unpi/parser.ts:51:22)
    at Parser._transform (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/unpi/parser.ts:19:14)
    at Parser.Transform._write (node:internal/streams/transform:175:8)
    at writeOrBuffer (node:internal/streams/writable:392:12)
    at _write (node:internal/streams/writable:333:10)
    at Parser.Writable.write (node:internal/streams/writable:337:10)'
[2024-05-08 18:30:31] info: 	z2m: Starting network scan (includeRoutes 'false')
[2024-05-08 18:31:19] info: 	z2m: Network scan finished

Trashwarez, 2024-05-09:
Same here. 

Error "error: 	zh:zstack:znp: Error while parsing to ZpiObject 'Error: CommandID '159' from subsystem '5' not found" 

pops up after reboot on a raspberry pi 3 with bookworm, nodejs 20.13 and a SONOFF ZBDongle-P with firmware 20230507.

Flash of prior firmware versions did not help as well as the deletion of "data/ coordinator_backup.json".

Had to rollback to 1.36.1. Everything works fine again.

RobertClarke64, 2024-05-10:
I have this issue too. But for me I only get this issue when I allow new devices to join and start the pairing sequence on a device. Once the error happens, I can't even try to pair new devices - nothing at all appears in the logs. I have to unplug the coordinator and restart Z2M, only then I can try to pair a device again, but it just gives me this error again.

|||
|----------------|----------------------|
|**Z2M Version**| 1.37.1 |
|**Coordinator**|  SONOFF ZBDongle-P |
|**Coordinator Firmware** | 20210708 |
|**No. Devices**| 15 (6 routers, 9 end devices) |

**Logs:**

```
[2024-05-10 11:29:46] info: 	z2m: Zigbee2MQTT started!
[2024-05-10 11:30:02] info: 	z2m: Zigbee: allowing new devices to join.
[2024-05-10 11:30:28] error: 	zh:zstack:znp: Error while parsing to ZpiObject 'Error: CommandID '159' from subsystem '5' not found
    at Function.fromUnpiFrame (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/znp/zpiObject.ts:60:19)
    at Znp.onUnpiParsed (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/znp/znp.ts:93:38)
    at Parser.emit (node:events:517:28)
    at Parser.parseNext (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/unpi/parser.ts:45:26)
    at Parser._transform (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/unpi/parser.ts:19:14)
    at Parser.Transform._write (node:internal/streams/transform:175:8)
    at writeOrBuffer (node:internal/streams/writable:392:12)
    at _write (node:internal/streams/writable:333:10)
    at Parser.Writable.write (node:internal/streams/writable:337:10)
    at SerialPort.ondata (node:internal/streams/readable:809:22)'
```

anthonym21, 2024-05-16:
My whole network is basically not working with this issue the only error in my log

Koenkk, 2024-05-19:
Does this issue also occur with the following firmware? https://github.com/Koenkk/Z-Stack-firmware/discussions/496

anthonym21, 2024-05-19:
> Koenkk/Z-Stack-firmware#496

I'll try tonight, for now I was able to work around it by using TCP instead
of websockets on the broker

Trashwarez, 2024-05-19:
> Does this issue also occur with the following firmware? [Koenkk/Z-Stack-firmware#496](https://github.com/Koenkk/Z-Stack-firmware/discussions/496)

Looks promising!

As stated above, the error popped up in my logs after reboot.

I rebooted 3 times now and can confirm that I don't have this error anymore on Zigbee2MQTT 1.37.1 when using the above posted firmware (20240315). I will check it again tomorrow and comment here if there are some issues.

harvindhillon, 2024-05-26:
I am on the latest 20240316 firmware for P7 and latest Zigbee2MQTT 1.37.1, I still get this error on start up

daangel27, 2024-05-29:
Hello,

Same error using : 
- Z2M version : 1.37.1
- Coordinator: Sonoff ZBDongle-P
- Coordinator FW : zStack3x0 version 20240315

**Logs:** 

_[2024-05-29 09:56:19] info: 	z2m: Started frontend on port 8099
[2024-05-29 09:56:20] info: 	z2m: Zigbee2MQTT started!
[2024-05-29 09:56:29] error: 	zh:zstack:znp: Error while parsing to ZpiObject 'Error: CommandID '159' from subsystem '5' not found
    at Function.fromUnpiFrame (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/znp/zpiObject.ts:60:19)
    at Znp.onUnpiParsed (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/znp/znp.ts:93:38)
    at Parser.emit (node:events:517:28)
    at Parser.parseNext (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/unpi/parser.ts:45:26)
    at Parser._transform (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/unpi/parser.ts:19:14)
    at Parser.Transform._write (node:internal/streams/transform:175:8)
    at writeOrBuffer (node:internal/streams/writable:392:12)
    at _write (node:internal/streams/writable:333:10)
    at Parser.Writable.write (node:internal/streams/writable:337:10)
    at SerialPort.ondata (node:internal/streams/readable:809:22)'_

harvindhillon, 2024-06-04:
I switched to the SMLIGHT SZLB-06 and get the same error as well on the latest test firmware 20240315

[2024-06-04 22:42:33] error:    zh:zstack:znp: Error while parsing to ZpiObject 'Error: CommandID '159' from subsystem '5' not found
    at Function.fromUnpiFrame (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/znp/zpiObject.ts:60:19)
    at Znp.onUnpiParsed (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/znp/znp.ts:93:38)
    at Parser.emit (node:events:517:28)
    at Parser.parseNext (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/unpi/parser.ts:45:26)
    at Parser._transform (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/unpi/parser.ts:19:14)
    at Parser.Transform._write (node:internal/streams/transform:175:8)
    at writeOrBuffer (node:internal/streams/writable:392:12)
    at _write (node:internal/streams/writable:333:10)
    at Parser.Writable.write (node:internal/streams/writable:337:10)
    at Socket.ondata (node:internal/streams/readable:809:22)'

Koenkk, 2024-06-05:
I'm going to start working on a 7.40 firmware, let's see if the issue still occurs with that one.

sanyafifa, 2024-06-30:
`zh:zstack:znp: Error while parsing to ZpiObject 'Error: CommandID '254' from subsystem '5' not found at Function.fromUnpiFrame (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/z-stack/znp/zpiObject.ts:60:19) at Znp.onUnpiParsed (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/z-stack/znp/znp.ts:93:38) at Parser.emit (node:events:517:28) at Parser.parseNext (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/z-stack/unpi/parser.ts:45:26) at Parser._transform (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/z-stack/unpi/parser.ts:19:14) at Parser.Transform._write (node:internal/streams/transform:175:8) at writeOrBuffer (node:internal/streams/writable:392:12) at _write (node:internal/streams/writable:333:10) at Parser.Writable.write (node:internal/streams/writable:337:10) at SerialPort.ondata (node:internal/streams/readable:809:22)'`

[image]

Trashwarez, 2024-07-02:
Last working version is 1.36.1.

`[2024-07-02 23:16:14] error:    zh:zstack:znp: Error while parsing to ZpiObject 'Error: CommandID '159' from subsystem '5' not found
    at Function.fromUnpiFrame (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/z-stack/znp/zpiObject.ts:60:19)
    at Znp.onUnpiParsed (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/z-stack/znp/znp.ts:93:38)
    at Parser.emit (node:events:519:28)
    at Parser.parseNext (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/z-stack/unpi/parser.ts:45:26)
    at Parser.parseNext (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/z-stack/unpi/parser.ts:51:22)
    at Parser._transform (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/z-stack/unpi/parser.ts:19:14)
    at Parser.Transform._write (node:internal/streams/transform:171:8)
    at writeOrBuffer (node:internal/streams/writable:564:12)
    at _write (node:internal/streams/writable:493:10)
    at Parser.Writable.write (node:internal/streams/writable:502:10)'
[2024-07-02 23:16:14] error:    zh:zstack:znp: Error while parsing to ZpiObject 'Error: CommandID '159' from subsystem '5' not found
    at Function.fromUnpiFrame (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/z-stack/znp/zpiObject.ts:60:19)
    at Znp.onUnpiParsed (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/z-stack/znp/znp.ts:93:38)
    at Parser.emit (node:events:519:28)
    at Parser.parseNext (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/z-stack/unpi/parser.ts:45:26)
    at Parser._transform (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/z-stack/unpi/parser.ts:19:14)
    at Parser.Transform._write (node:internal/streams/transform:171:8)
    at writeOrBuffer (node:internal/streams/writable:564:12)
    at _write (node:internal/streams/writable:493:10)
    at Parser.Writable.write (node:internal/streams/writable:502:10)
    at SerialPort.ondata (node:internal/streams/readable:1007:22)'
[2024-07-02 23:16:14] error:    zh:zstack:znp: Error while parsing to ZpiObject 'Error: CommandID '159' from subsystem '5' not found
    at Function.fromUnpiFrame (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/z-stack/znp/zpiObject.ts:60:19)
    at Znp.onUnpiParsed (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/z-stack/znp/znp.ts:93:38)
    at Parser.emit (node:events:519:28)
    at Parser.parseNext (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/z-stack/unpi/parser.ts:45:26)
    at Parser.parseNext (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/z-stack/unpi/parser.ts:51:22)
    at Parser._transform (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/z-stack/unpi/parser.ts:19:14)
    at Parser.Transform._write (node:internal/streams/transform:171:8)
    at writeOrBuffer (node:internal/streams/writable:564:12)
    at _write (node:internal/streams/writable:493:10)
    at Parser.Writable.write (node:internal/streams/writable:502:10)'
[2024-07-02 23:16:14] error:    zh:zstack:znp: Error while parsing to ZpiObject 'Error: CommandID '159' from subsystem '5' not found
    at Function.fromUnpiFrame (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/z-stack/znp/zpiObject.ts:60:19)
    at Znp.onUnpiParsed (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/z-stack/znp/znp.ts:93:38)
    at Parser.emit (node:events:519:28)
    at Parser.parseNext (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/z-stack/unpi/parser.ts:45:26)
    at Parser.parseNext (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/z-stack/unpi/parser.ts:51:22)
    at Parser.parseNext (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/z-stack/unpi/parser.ts:51:22)
    at Parser._transform (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/z-stack/unpi/parser.ts:19:14)
    at Parser.Transform._write (node:internal/streams/transform:171:8)
    at writeOrBuffer (node:internal/streams/writable:564:12)
    at _write (node:internal/streams/writable:493:10)'
[2024-07-02 23:16:14] error:    zh:zstack:znp: Error while parsing to ZpiObject 'Error: CommandID '159' from subsystem '5' not found
    at Function.fromUnpiFrame (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/z-stack/znp/zpiObject.ts:60:19)
    at Znp.onUnpiParsed (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/z-stack/znp/znp.ts:93:38)
    at Parser.emit (node:events:519:28)
    at Parser.parseNext (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/z-stack/unpi/parser.ts:45:26)
    at Parser.parseNext (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/z-stack/unpi/parser.ts:51:22)
    at Parser.parseNext (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/z-stack/unpi/parser.ts:51:22)
    at Parser.parseNext (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/z-stack/unpi/parser.ts:51:22)
    at Parser._transform (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/z-stack/unpi/parser.ts:19:14)
    at Parser.Transform._write (node:internal/streams/transform:171:8)
    at writeOrBuffer (node:internal/streams/writable:564:12)'
[2024-07-02 23:16:14] error:    zh:zstack:znp: Error while parsing to ZpiObject 'Error: CommandID '159' from subsystem '5' not found
    at Function.fromUnpiFrame (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/z-stack/znp/zpiObject.ts:60:19)
    at Znp.onUnpiParsed (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/z-stack/znp/znp.ts:93:38)
[... cut ...]
    at Parser.parseNext (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/z-stack/unpi/parser.ts:51:22)
    at Parser.parseNext (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/z-stack/unpi/parser.ts:51:22)
    at Parser.parseNext (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/z-stack/unpi/parser.ts:51:22)
    at Parser.parseNext (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/z-stack/unpi/parser.ts:51:22)
    at Parser._transform (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/z-stack/unpi/parser.ts:19:14)'`

Trashwarez, 2024-08-03:
Just giving a heads-up to encourage others here to check by themselves. Upper issues are gone on my system since version 1.39.1.

mpuff, 2024-08-08:
i am using slzb06p7 with latest firmware 20240716 and zigbee2mqtt 1.39.1 in docker.
after a reboot, I get the error and no devices are in the gui, after a map scan, everything is there anymore and looks like working.

BOlaerts, 2024-08-12:
I am using the generic x86-64 HA install with Zigbee2MQTT add-on (version 1.39.1-1).
Log when (re)starting the add-on:
[08:20:46] INFO: Preparing to start...
[08:20:47] INFO: Socat not enabled
[08:20:48] INFO: Starting Zigbee2MQTT...
Starting Zigbee2MQTT without watchdog.
[2024-08-12 08:21:16] error: 	zh:zstack:znp: Error while parsing to ZpiObject 'Error: CommandID '159' from subsystem '5' not found
    at Function.fromUnpiFrame (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/znp/zpiObject.ts:66:19)
    at Znp.onUnpiParsed (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/znp/znp.ts:88:38)
    at Parser.emit (node:events:517:28)
    at Parser.parseNext (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/unpi/parser.ts:46:26)
    at Parser._transform (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/unpi/parser.ts:20:14)
    at Parser.Transform._write (node:internal/streams/transform:175:8)
    at writeOrBuffer (node:internal/streams/writable:392:12)
    at _write (node:internal/streams/writable:333:10)
    at Parser.Writable.write (node:internal/streams/writable:337:10)
    at SerialPort.ondata (node:internal/streams/readable:809:22)'

Zigbee devices are working normally in HA, but the add-on gui is not operational anymore, adding new devices impossible for the moment.

BOlaerts, 2024-09-20:
Error dissapeared I think with the release of version 1.40.1-0, but now it's back...
Current version: 1.40.1-1

Koenkk, 2024-09-20:
Will be fixed in the next z2m release (https://github.com/Koenkk/zigbee-herdsman/commit/c77db3067ddabc1d047fe7e94929d6116bd3a15d)

closed 2024-09-20 by Koenkk