Koenkk/zigbee2mqtt #22545: Error while starting zigbee-herdsman

alexandrechoske, 2024-05-08:
### What happened?

Failed to init port with error Error: Error Resource temporarily unavailable Cannot lock port
Zigbee2MQTT:error 2024-05-08 13:55:58: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2024-05-08 13:55:58: Failed to start zigbee
Zigbee2MQTT:error 2024-05-08 13:55:58: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions

### What did you expect to happen?

Work properly

### How to reproduce it (minimal and precise)

[image]

/dev/serial/by-id/usb-ITEAD_SONOFF_Zigbee_3.0_USB_Dongle_Plus_V2_20240123191300-if00

Zigbee dongle plugged. ( also tried with cc2351 )

### Zigbee2MQTT version

1.36.1-1

### Adapter firmware version

20240123191300

### Adapter

ZBDongle-e

### Setup

ITEAD_SONOFF_Zigbee_3.0_USB_Dongle_Plus_V2

### Debug log

```
======== ASH stopped ========
Failed to init port with error Error: Error Resource temporarily unavailable Cannot lock port
Zigbee2MQTT:error 2024-05-08 13:55:58: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2024-05-08 13:55:58: Failed to start zigbee
Zigbee2MQTT:error 2024-05-08 13:55:58: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
Zigbee2MQTT:error 2024-05-08 13:55:58: Exiting...
Zigbee2MQTT:error 2024-05-08 13:55:58: Error: Failed to start EZSP layer with status=HOST_FATAL_ERROR.
    at EmberAdapter.initEzsp (/app/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:804:19)
    at EmberAdapter.start (/app/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:2661:24)
    at Controller.start (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:132:29)
    at Zigbee.start (/app/lib/zigbee.ts:62:27)
    at Controller.start (/app/lib/controller.ts:109:27)
    at start (/app/index.js:107:5)
[13:57:49] INFO: Preparing to start...
[13:57:49] INFO: Socat not enabled
[13:57:50] INFO: Starting Zigbee2MQTT...
Zigbee2MQTT:info  2024-05-08 13:57:52: Logging to console and directory: '/config/zigbee2mqtt/log/2024-05-08.13-57-52' filename: log.txt
Zigbee2MQTT:info  2024-05-08 13:57:52: Starting Zigbee2MQTT version 1.36.1 (commit #unknown)
Zigbee2MQTT:info  2024-05-08 13:57:52: Starting zigbee-herdsman (0.40.3)
Zigbee2MQTT:error 2024-05-08 13:57:52: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2024-05-08 13:57:52: Failed to start zigbee
Zigbee2MQTT:error 2024-05-08 13:57:52: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
Zigbee2MQTT:error 2024-05-08 13:57:52: Exiting...
Zigbee2MQTT:error 2024-05-08 13:57:52: Error: Error Resource temporarily unavailable Cannot lock port
```

Also tried with latest release as well and same problem starting zigbee-herdsman.

[image]

chickyd3v, 2024-05-09:
Second to this issue. Same SONOFF Zigbee 3.0 USB Dongle Plus V2, same "Error: Failed to start EZSP layer with status=HOST_FATAL_ERROR" message.

krusse, 2024-05-09:
I also had this issue.

Thanks @chickyd3v for the suggestion to downgrade. I didn't go back as far as you (1.36.1 for docker, ncp-uart-hw-v7.3.1.0 for the dongle), and it is back working as expected.

alexandrechoske, 2024-05-09:
How to downgrade the zigbee docker version guys?

krusse, 2024-05-09:
> How to downgrade the zigbee docker version guys?

In your docker-compose:

image: koenkk/zigbee2mqtt:1.36.1

instead of

image: koenkk/zigbee2mqtt

jdwEmSys, 2024-05-09:
I changed to Zigbee2MQTT version 1.36.1. The module is programmed with version: NabuCasa_SkyConnect_EZSP_v7.1.1.0_ncp-uart-hw_115200.gbl
But I get still the same error?
@krusse: where dit you find this Firmware: “ncp-uart-hw-v7.3.1.0”

nict41, 2024-05-10:
I managed to get mine working:
Update Z2M to 1.37.1-1
Reflash Dongle with 7.3.1.0 build 176 via web flasher (https://dialedin.com.au/blog/sonoff-zbdongle-e-rcp-firmware)
Reboot HA host

krusse, 2024-05-10:
> @krusse: where did you find this Firmware: “ncp-uart-hw-v7.3.1.0”

@jdwEmSys I used this web flasher: [https://darkxst.github.io/silabs-firmware-builder/](https://darkxst.github.io/silabs-firmware-builder/). However, I have the SONOFF Zigbee 3.0 USB Dongle Plus V2, not the skyconnect.

roadjet, 2024-05-11:
can you upload external firmware? i always get an error. can only choose from the 4 options

alexandrechoske, 2024-05-16:
> I just reinstall the same version and it work now

Same version of Z2M?
I am still with no solutions here, tried everything.
Next step is to leave Proxmox to HASOS directly to see if works properly.

djiesr, 2024-05-16:
Yes, same version With Sonoff 3.0

I have many problems With z2m and zha in ha 2024.5 , i use noth. Nut
before,  I try to unplug the usb ,Turn off ha , restart, reconnect the usb
in Amother port, restart z2m and it work to.

nxtkittWi, 2024-06-11:
Your unplug, shut down the computer, restart, and plug into diff USB port worked for me!

djiesr, 2024-06-12:
I had to remove the multinework of my skyconnect. That was the cause of the
problem.

akhosha2, 2024-06-14:
> Your unplug, shut down the computer, restart, and plug into diff USB port worked for me!

I tried this a couple times with no luck. Next step is to try to downgrade the firmware.

akhosha2, 2024-06-14:
> > Your unplug, shut down the computer, restart, and plug into diff USB port worked for me!
> 
> I tried this a couple times with no luck. Next step is to try to downgrade the firmware.

Reverting the dongle firmware to ncp-uart-hw_6.10.3_115200.gbl worked for me.

alexandrechoske, 2024-06-18:
I just gave up on this and installed ZHA and is working 100%.

sbuchanYVR, 2024-06-18:
ya, I am about to give up as well.  It seems that the SiliconLabs chipset support is unstable and 'experimental'.  I took a brand new RP5 last nite and a brand new test install of HA core.. and tried again .. the same issue.  I borrowed a friend's TI Chipset dongle .. worked.

schollex, 2024-07-30:
I had a running setup with Z2M 1.39.0 wth ZBDongle-E/Sonoff Zigbee 3.0 USB Dongle Plus V2 connected via USB-extension-cable.
I decided to swich from ezsp to ember. But this did not work, regardless I was trying different firmware-versions. Rollback to ezsp was also not possible.
After removing the USB-extension-cable, ember worked for me. My running setup on RPI with Docker version 26.1.3, build b72abbb:

Zigbee2MQTT-Version
    [1.39.0](https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.39.0) commit: [0326926](https://github.com/Koenkk/zigbee2mqtt/commit/0326926)

Coordinator-Typ
    EmberZNet

Coordinator-Version
    7.4.3 [GA]

Frontend-Version
    0.7.2

Zigbee Herdsman Konverter Version
    19.62.0

Zigbee Herdsman Version
    0.50.1

Remark:
lsusb as well as /dev/serial/by-id show different device-names for my zigbee-dongle depending whether usb-extension-cable is used or not. Correct is QinHeng Electronics SONOFF Zigbee 3.0 USB Dongle Plus V2 respectively usb-ITEAD_SONOFF_Zigbee_3.0_USB_Dongle_Plus_V2_XXX-if00 where XXX is adapter-specific (serialnumber?)

terrencestella, 2024-08-24:
hi all,

I used to have the Sonoff ZBDongle-P version, but I broke the antenna by accident... So I bought a new one from Amazon, the Sonoff ZBDongle-E this time.

My expectation was that it would just be updating the names in my docker config, but boy was I wrong......

Have been busy since last night, but now I am reaching out for help.

I have HA running on an old Dell Optiplex, along with mosquitto and zigbee2mqtt, which worked fine!

But now, I am constantly getting this:

[2024-08-24 09:26:07] info: 	z2m: Logging to console, file (filename: log.log)
[2024-08-24 09:26:07] info: 	z2m: Starting Zigbee2MQTT version 1.39.1 (commit #e132316)
[2024-08-24 09:26:07] info: 	z2m: Starting zigbee-herdsman (0.55.3)
[2024-08-24 09:26:08] info: 	zh:ember: Using default stack config.
[2024-08-24 09:26:08] info: 	zh:ember: ======== Ember Adapter Starting ========
[2024-08-24 09:26:08] info: 	zh:ember:ezsp: ======== EZSP starting ========
[2024-08-24 09:26:08] info: 	zh:ember:uart:ash: ======== ASH Adapter reset ========
[2024-08-24 09:26:08] info: 	zh:ember:uart:ash: RTS/CTS config is off, enabling software flow control.
[2024-08-24 09:26:08] info: 	zh:ember:uart:ash: Serial port opened
[2024-08-24 09:26:08] info: 	zh:ember:uart:ash: ======== ASH starting ========
[2024-08-24 09:26:09] info: 	zh:ember:uart:ash: ======== ASH connected ========
[2024-08-24 09:26:09] info: 	zh:ember:uart:ash: ======== ASH started ========
[2024-08-24 09:26:09] info: 	zh:ember:ezsp: ======== EZSP started ========
[2024-08-24 09:26:09] info: 	zh:ember: Adapter EZSP protocol version (13) lower than Host. Switched.
[2024-08-24 09:26:09] info: 	zh:ember: Adapter version info: {"ezsp":13,"revision":"7.4.2 [GA]","build":0,"major":7,"minor":4,"patch":2,"special":0,"type":170}
[2024-08-24 09:26:09] error: 	z2m: Error while starting zigbee-herdsman
[2024-08-24 09:26:09] error: 	z2m: Failed to start zigbee
[2024-08-24 09:26:09] error: 	z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
[2024-08-24 09:26:09] error: 	z2m: Exiting...
[2024-08-24 09:26:09] error: 	z2m: Error: [BACKUP] Current backup file is not for EmberZNet stack.
    at EmberAdapter.getStoredBackup (/app/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:1196:23)
    at EmberAdapter.initTrustCenter (/app/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:988:37)
    at runNextTicks (node:internal/process/task_queues:60:5)
    at processImmediate (node:internal/timers:447:9)
    at EmberAdapter.initEzsp (/app/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:781:18)
    at EmberAdapter.start (/app/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:1758:24)
    at Controller.start (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:127:29)
    at Zigbee.start (/app/lib/zigbee.ts:63:27)
    at Controller.start (/app/lib/controller.ts:139:27)
    at start (/app/index.js:154:5)

I already reflashed my stick to Zigbee (EZSP) 7.4.2.0 build 0, but still, cant get it to work. I also updated the adapter to ember, also did not work. I reflashed it once more with Zigbee (EZSP) 7.4.2.0 build 0, also doesn't work. Rebooted the system, did not work.

What can I do to get it working?

jgsaez9, 2024-11-13:
+1 same error

yajmi, 2024-11-29:
@shinebrillant  I'm facing the exact same issue as you, did you resolve yours ?

shinebrillant, 2024-11-30:
Not yet, I bought SLZB-06M, it has not arrived yet, I hope to be able to use Zigbee2MQTT.

I also found that after removing Connect ZBT-1, the setting option still appears in HomeAssistant's Settings/System/Hardware.

yajmi, 2024-11-30:
@shinebrillant  I was able to make it work with ZBT-1 adapter, you just need to flash your zigbee adapter to the last available framework. the web flasher ( https://connectzbt1.home-assistant.io/firmware-update/ ) being currently unavailable, I had to use the addon silabs_flasher here: https://github.com/home-assistant/addons/blob/master/silabs_flasher/DOCS.md, it's can be found on HA Add-on store, to make it visible you have to opt-in to "Advanced Mode" which can be changed on your user profile page.

shinebrillant, 2024-12-01:
I successfully installed Silicon Labs Flasher, but what should I fill in firmware_url?
Do I still need to install Zigbee2MQTT addons?
Can you elaborate? Thanks.

Vladimir-Ovechenko, 2024-12-20:
@shinebrillant I had the same problem and struggled with it all week and finally managed to solve it.
Here are my settings in Zigbee2MQTT
And I'm not sure if Silicon Labs Flasher helped me
But all in all, it finally works for my Sonoff_Zigbee_3.0_USB_Dongle_Plus_V2
[image]
[image]

Sjorsa, 2025-01-16:
I had the same error. First mistake I made was not having a configuration.yaml for Z2M (I use docker). Then it still didn't work: I had to update the adapter to the ember firmware. I followed this video: https://www.youtube.com/watch?v=x1QeNPi6tK8

FLEKI1, 2025-01-31:
For me, I solved the problem by starting the system without the Sonoff stick plugged in, starting HA, then inserting the stick, copying the device information, and starting Z2M. If you start the system with the USB plugged in, it blocks the port of the serial address. I hope i can Help few guys of you! the Struggle is there.

cafedelmar01, 2025-02-17:
> > How to downgrade the zigbee docker version guys?
> 
> In your docker-compose:
> 
> image: koenkk/zigbee2mqtt:1.36.1
> 
> instead of
> 
> image: koenkk/zigbee2mqtt

Thanks! Needs this one, to revert back to an older image! Thanks!

SurfiBird81, 2025-03-04:
I still cant use the stick and get the error message. [2025-03-04 20:47:52] error: 	z2m: Error while starting zigbee-herdsman
[2025-03-04 20:47:52] error: 	z2m: Failed to start zigbee-herdsman
[2025-03-04 20:47:52] error: 	z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start_crashes-runtime.html for possible solutions
[2025-03-04 20:47:52] error: 	z2m: Exiting...
[2025-03-04 20:47:52] error: 	z2m: Error: Failed to start EZSP layer with status=HOST_FATAL_ERROR.
    at EmberAdapter.initEzsp (/app/node_modules/.pnpm/zigbee-herdsman@3.2.7/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:670:19)
    at EmberAdapter.start (/app/node_modules/.pnpm/zigbee-herdsman@3.2.7/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:1536:24)
    at Controller.start (/app/node_modules/.pnpm/zigbee-herdsman@3.2.7/node_modules/zigbee-herdsman/src/controller/controller.ts:136:29)
    at Zigbee.start (/app/lib/zigbee.ts:69:27)
    at Controller.start (/app/lib/controller.ts:142:13)
    at start (/app/index.js:161:5)
Are there any further recommendations? Latest Firmware, ZHA uninastalled, ember...all done

halohalomietku, 2025-03-05:
Had the same issue with Sonoff_Zigbee_3.0_USB_Dongle_Plus_V2. Ultimately I changed as below to obsolete ezsp and now running fine, however in log I see warning: 	zh:ezsp: 'ezsp' driver is deprecated and will only remain to provide support for older firmware (pre 7.4.x). Migration to 'ember' is recommended. If using Zigbee2MQTT see https://github.com/Koenkk/zigbee2mqtt/discussions/21462

adapter: ezsp
port: >-
  /dev/serial/by-id/usb-ITEAD_SONOFF_Zigbee_3.0_USB_Dongle_Plus_V2_20240103185705-if00

but at least for now is working with ZigBee2MQTT 2.1.3 😃

ShundraL, 2025-03-07:
Proper adapter setting solved my issue.

[image]

SurfiBird81, 2025-03-07:
I am using the P-Dongle and zstack works much.better. Strange....Also tried ember and ezsp without success ...

closed 2025-05-15 by github-actions[bot]

miknai, 2025-10-26:
The config below works for me. I have flashed my dongle to the latest firmware, and also disabled the ZHA.

```
data_path: /config/zigbee2mqtt
socat:
  enabled: false
  master: pty,raw,echo=0,link=/tmp/ttyZ2M,mode=777
  slave: tcp-listen:8485,keepalive,nodelay,reuseaddr,keepidle=1,keepintvl=1,keepcnt=5
  options: "-d -d"
  log: false
mqtt:
  user: <user>
  password: <password>
  server: mqtt://core-mosquitto:1883
serial:
  port: >-
    /dev/serial/by-id/usb-Itead_Sonoff_Zigbee_3.0_USB_Dongle_Plus_V2_92b49c2539f4ef11967bbf1b6d9880ab-if00-port0
  adapter: ember
  baudrate: 115200
```