Koenkk/zigbee2mqtt #22550: Multiple E1C-NB7 Leaving Shortly After Joining

BradleyShaner, 2024-05-08:
### What happened?

I've got 3 Sengled E1C-NB7 devices that previously worked great. Yesterday, I noticed one of them was unavailable so I tried to re-pair it. It joined, but then it left after a few seconds. During the time it was connected I could control it from the UI by switching it on and off just fine. So I ordered two new ones and they're both doing the same exact thing. I then joined a new Leak Detector TS0207 and it works just fine, staying connected and reporting leak status.

These three are the Sengled devices:
0xb0ce1814036c3e14
0xb0ce1814036c4a25
0xb0ce1814036c3e14

### What did you expect to happen?

Device to join and stay joined.

### How to reproduce it (minimal and precise)

Enable "Permit join" and they will continuously join and leave until the timer runs out.

### Zigbee2MQTT version

1.37.0 

### Adapter firmware version

20221226

### Adapter

SLZB-06

### Setup

HassOS in a Proxmox container

### Debug log

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/15254660/log.txt)

WayneManion, 2024-05-20:
I have several of these devices. They worked well until recently. In my case they appear to stay joined, but they become very lazy about reporting the power/energy values. Like I only get a value if I hit the "get value" icon in the UI.

BradleyShaner, 2024-05-21:
I'm going to close this for now. I just tried it again and it works, and stays connected. If you're having reporting issues you may want to adjust these attributes. A freshly joined plug doesn't report properly but after making these changes it works great.
[image]

closed 2024-05-21 by BradleyShaner