Koenkk/zigbee2mqtt #22551: [Bug]  Last Seen reporting even when device is not connected to network!?

Ltek, 2024-05-08:
### What happened?

"Last Seen" is does not work for this by design ( cant find any docs on it) or does not work due to a bug.
... After 24 hrs of testing.  Z2M will report it as 'seen' and still not communicate with it... I have a sensor that has not changed any parameters in several days -and- Z2M Map shows its not even connected (no line to coordinate or any device) yet it was "Last Seen' 2 hours ago?

[image]

[image]

[image]

### What did you expect to happen?

report correctly

### How to reproduce it (minimal and precise)

turn it on in Settings

### Zigbee2MQTT version

1.37

### Adapter firmware version

EmberZNet Coordinator revision 7.4.3 [GA]

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus-E

### Setup

Windows, Homeseer

Nerivec, 2024-05-10:
Last seen should update in four cases: received a network address update, received a device announce, device (re)joined, or any kind of status update. Something likely updated in the background for that device which triggered the last seen update. After that, it would have gone to sleep (battery-powered).

Can you provide some logs from that period and the IEEE address of the device in question? I'll take a look just in case.

Ltek, 2024-11-09:
still needs fixing

Ltek, 2025-01-11:
not stale... still fresh

Ltek, 2025-03-13:
bad bot

closed 2025-05-21 by github-actions[bot]