Koenkk/zigbee2mqtt #22554: Gledopto GL-C-002P reported as RGBCCT instead of CCT to Home Assistant

ReX1983, 2024-05-08:
### What happened?

2 days ago, an OTA update (firmware 20551203) caused my Gledopto GL-C-002P to stop turning the light on. The device was present and responsive in zigbee2mqtt but it would not turn on the light anymore.

Up until this update, the device has been configured as CCT and has been online for at least 1 year.

After resetting it, I was able to re-pair it. zigbee2mqtt recognized the device (restoring the friendly name I gave it earlier) and the device is now also reported to Home Assistant.

The problem I am facing is that it is reported as RGB+CCT (GL-C-008P) in Home Assistant, while it is correctly identified as CCT (GL-C-006P) in zigbee2mqtt (following the [guide here](https://www.zigbee2mqtt.io/devices/GL-C-008P.html))

I have not found a way to correct the model in Home Assistant and as such, now Home Assistant presents me with options to change the color, which fails in zigbe2mqtt.

Any way to fix it? 

### What did you expect to happen?

When in CCT mode, the GL-C-002P is reported as CCT (GL-C-006P) in Home Assistant.

### How to reproduce it (minimal and precise)

Update firmware to 20551203
Re-pair the device (hold reset)

### Zigbee2MQTT version

 1.37.0

### Adapter firmware version

6.10.3.0 build 297

### Adapter

Sonoff Dongle Plus ZBDongle E (EFR32MG21 + CH9102F)

### Setup

Add-on on Home Assistant

diegocjorge, 2024-05-11:
You'll need to manually revert the mode to CCT on the controller (indicated by a green light), because each time you update the firmware it resets the mode back to RGB+CCT.

If Z2M continues to identify it as GL-C-008P, simply restart Z2M. It should then recognize the change.

ReX1983, 2024-05-11:
I did that, Z2M recognises it correctly, but Home Assistant still reports it wrongly, irrespective how many times I restart either Z2M or Home Assistant. Is there a way to force a model update in Home Assistant?

diegocjorge, 2024-05-11:
I've never encountered this issue, but here are a few things you could try:
1) Attempt to adjust the brightness and temperature via the Z2M interface to see if that triggers an update in HA;
2) Navigate to Integrations > MQTT. Click on the three dots and select "reload";
3) If the above steps don't yield any changes, you might need to install the MQTT Explorer addon (available at this GitHub repository: https://github.com/GollumDom/addon-repository) and manually remove the retained topic. I typically resort to this when a device has been removed from Z2M, but continues to appear in HA.

ReX1983, 2024-05-12:
Thanks, somehow the model got fixed on its own. The only thing that happened in between was a Home Assistant core update and its associated reboot. I have noted down your strategies in case this happens again.

closed 2024-05-12 by ReX1983