Koenkk/zigbee2mqtt #22556: Zen Thermostat support incomplete, multiple bugs. Works in ZHA, not in z2mqtt.

Stratoco, 2024-05-09:
### What happened?

OTB does not support setting temperature, nor does it allow the HA climate entity to set temperature. The following tweaks made it semi-work, but this thermostat is fully supported in ZHA so this is a z2mqtt issue. 

The following had to be added, as this attribute was not automatically added: 
[image]

Publishing an mqtt message to zigbee2mqtt/IEEE Address/set with the payload {"occupied_cooling_setpoint": ""} made them show up in HA, however it shows both heating and cooling on the HA card despite being set to "cool" mode. 

[image]

Additionally, setting temperature works roughly 30% of the time, at best. This device is fully and flawlessly supported in ZHA, but I am trying to bring everything under one umbrella. I would be happy to work with someone to test out tweaks/fixes or to provide further data. Thank you! 

### What did you expect to happen?

The device would work like it does with ZHA

### How to reproduce it (minimal and precise)

Purchase Zen Thermostat on ebay ($35). Connect to Z2MQTT. 

### Zigbee2MQTT version

1.37.0

### Adapter firmware version

20210708

### Adapter

Sonoff_Zigbee_3.0_USB_Dongle_Plus

### Setup

Z2M on HASS VM on x76

Stratoco, 2024-05-09:
Thank you to @andynbaker for his tips here, which got me from not working at all to semi-working. 

https://github.com/Koenkk/zigbee2mqtt/issues/4320#issuecomment-1169102978

frasderp, 2024-10-04:
Any advances on this one? I am having the same issues, can not set temperatures in zigbee2mqtt, but works fine in ZHA.

closed 2025-01-04 by github-actions[bot]

worldgod13, 2025-01-08:
Does anyone have a fix? I just migrated to z2m from zha where zen worked perfectly. Now, not so much.

Stratoco, 2025-01-16:
No. The latest update seems to have killed it entirely. I was able to set temps previously, although it was very buggy. Now nothing happens at all.

worldgod13, 2025-01-16:
Well, as an action for automation I can set/change the temp by Climate: Set Target Temperature, choosing entity climate.zen and setting the min and max to the same with hvac mode set to heat. 

This works for me for now as I do not manually change the temp very often. I am sure it will work for cooling too but won't find that out until April or May.