Koenkk/zigbee2mqtt #22557: Dongle crashing with ZCL error: SRSP - AF - dataRequest after 6000ms

lucasteligioridis, 2024-05-09:
### What happened?

After a few days, generally like 2-4 days my zigbee dongle crashes with the error in the title.

Now I have already changed/done a few things which I'll list here so its clear and explain further:
- Have changed USB cables with long ones
- Changed from a USB dongle to a network dongle, so USB issues are no longer even in question
- Moved the zigbee dongle to another part of my house to see if it was something to do with interference

At this point I'm actually at a loss as to what the issue is! For what its worth this issue wasn't always a thing, I cant remember but it feels like it happened to me about 5-6 months ago and I have no idea what changed. Is there a chance a device on my mesh is causing this crash of my stick or something z2m is indirectly doing?

And to also clarify it looks like whatever this issue is crashes my stick and not z2m, z2m functions as per normal and actually can't boot correctly until I power reset the dongle as it's in a crashed/stuck state.

I have no turned on debug logging and will keep it that way so when it crashes maybe I can provide something of more detail.

Do you have any idea or have any suggestions on how to figure out what the hell is happening? Fact that this same issue is occurring on 2 separate zigbee sticks that were USB and network adapters highlights to me that its nothing to do with the stick and now its something to do with a device or z2m causing the crash. Any help is greatly appreciated please! I'm at a loss here.

### What did you expect to happen?

The dongle not to crash and prevent me needing to restart it.

### How to reproduce it (minimal and precise)

I have no idea how to reproduce this other than just waiting for a few days for the inevitable crash.

### Zigbee2MQTT version

1.37.0

### Adapter firmware version

20230507

### Adapter

TubesZB CC2652 P7

### Setup

Running in a container as standalone in Kubernetes / also in Unraid, both fail in the same way.

### Debug log

N/A Will provide debug logs when it crashes again.

Koenkk, 2024-05-09:
Should be fixed with the new fw: https://github.com/Koenkk/Z-Stack-firmware/discussions/496

lucasteligioridis, 2024-05-09:
Oh interesting! 

Is the issue for all people on this firmware version? Is that firmware ready now or still in beta?

lucasteligioridis, 2024-05-10:
Ah read the issue, I loaded this firmware on my stick. Seems like it is related, if anything I'll make any comments or feedback over in that thread as it seems to be a common issue. Will close this one for now and keep discussions over there. Can re-open this if we need to, but does seem related to the firmware and potentially a mix of devices.

closed 2024-05-10 by lucasteligioridis