Koenkk/zigbee2mqtt #22561: Can't generate helm releases with '+' in the version.

jlpedrosa, 2024-05-09:
### What happened?

This is just a heads up. The current version of helm releases fails to run the helm release chart process if the version contains a '+' simbol (which is a valid sem versioning). 

I tracked down the problem an open a PR to fix it, they don't seem very responsive: https://github.com/helm/chart-releaser/pull/406

Charts with '-' symbols work ok in the interim.

### What did you expect to happen?

Not a zigbee2mqtt issue.

### How to reproduce it (minimal and precise)

Not a zigbee2mqtt issue.

### Zigbee2MQTT version

1.37.0

### Adapter firmware version

20230507

### Adapter

any

### Setup

any

### Debug log

NA

Koenkk, 2024-05-09:
There was also a mistake in the branch name, fixed it in https://github.com/Koenkk/zigbee2mqtt/commit/34f0332564b91c0d866a667233caefa4c6da911c

I guess we should hold back https://github.com/Koenkk/zigbee2mqtt.io/pull/2738 for now?

jlpedrosa, 2024-05-09:
I don't think is a problem with the cadence of releases that you have (I don't think we well need to release hot fixes). Imo is fine, I'm just tracking the issue here, and let you know.

jlpedrosa, 2024-05-09:
If possible, I'd do a PR to enable the workflow release of the helm chart via "click" on the UI and ask you to trigger it if you are up to it.

Koenkk, 2024-05-10:
@jlpedrosa can you check if the chart is published, then I can merge https://github.com/Koenkk/zigbee2mqtt.io/pull/2743

closed 2024-12-07 by github-actions[bot]