Koenkk/zigbee2mqtt #22562: Innr plug SP240 1of2 OTA-updates failed and cannot be restarted

Picard16, 2024-05-09:
### What happened?

After I updated my two Innr plugs from 1.4.3 to 1.6.22, one of them failed, is still on 1.4.3 and I cannot restart the update in OTA update board in HA/zigbe2mqtt as nothing happend when I hit the button:
[image]

When I look into the zigbee2mqtt/state.json file, both plugs seemed to be updated:
[image]

[image]

I tried to edit the state.json, but nothing changed then, and my changes have been overwritten again.

(this behavior was already the case before I updated my ZBDongleE from EZSP and 7.3.1 to ember and 7.4.3)

### What did you expect to happen?

...that I can restart the failed update.

### How to reproduce it (minimal and precise)

I can try to restart it as often as I like.

### Zigbee2MQTT version

1.37.0

### Adapter firmware version

7.4.3 EmberZNet

### Adapter

Sonoff ZBDongleE

### Setup

HA add-on on RasPi5

### Debug log

Info 2024-05-09 13:29:41 Checking if update available for 'Innr Steckdose EG'
Info 2024-05-09 13:29:42 No update available for 'Innr Steckdose EG'

I just saw that the other plug does not respond although its online and I can switch it (but this might be another issue):

Info 2024-05-09 13:29:24 Checking if update available for 'Innr Steckdose DECT '
Error 2024-05-09 13:30:24 Failed to check if update available for 'Innr Steckdose DECT ' (OTA: Device didn't respond to OTA request)

fir3drag0n, 2024-08-22:
I do have the same problem, one adapter stays at 1.4.3. Did you figure out a solution?

Picard16, 2024-08-24:
No, unfortunately not. But it "healed" itself with newest firmware release 1.7.22.
It recognized a newer release and then it worked again.

closed 2025-01-04 by github-actions[bot]