Koenkk/zigbee2mqtt #22563: Error while parsing to NpiFrame

nicolapejo, 2024-05-09:
### What happened?

The docker crashes and, even after rebooting, the devices no longer respond

### Zigbee2MQTT version

1.37.0

### Adapter firmware version

6.10.3.0 build 297

### Adapter

ITEAD_SONOFF_Zigbee_3.0_USB_Dongle_Plus_V2

### Setup

Docker in a Unraid system

### Debug log

Error: zdoRequest>request error
    at Driver.zdoRequest (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/driver.ts:724:23)
    at EZSPAdapter.nodeDescriptorInternal (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/adapter/ezspAdapter.ts:390:28)
    at /app/node_modules/zigbee-herdsman/src/adapter/ezsp/adapter/ezspAdapter.ts:380:32
    at Queue.execute (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:35:20)
    at EZSPAdapter.handleDeviceJoin (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/adapter/ezspAdapter.ts:147:13)
[2024-05-09 13:19:38] error:    zh:ezsp:uart: -!- break waiting (4)
[2024-05-09 13:19:38] error:    zh:ezsp:uart: Can't send DATA frame (3,6,0): 3b00010500
[2024-05-09 13:19:47] error:    zh:ezsp:uart: --> Error: Error: {"sequence":4} after 4000ms
[2024-05-09 13:19:47] error:    zh:ezsp:uart: -!- break rewaiting (4)
[2024-05-09 13:19:47] error:    zh:ezsp:uart: Can't resend DATA frame (3,6,1): 3b00010500
[2024-05-09 13:19:47] error:    zh:ezsp:ezsp: Watchdog heartbeat timeout Error: Failure send nop:{"type":"Buffer","data":[59,0,1,5,0]}
    at /app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:686:23
    at Queue.execute (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:35:20)
    at Ezsp.watchdogHandler (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:795:13)
[2024-05-09 13:19:55] error:    zh:ezsp:uart: --> Error: Error: {"sequence":1} after 4000ms
[2024-05-09 13:19:55] error:    zh:ezsp:uart: -!- break waiting (1)
[2024-05-09 13:19:55] error:    zh:ezsp:uart: Can't send DATA frame (0,1,0): 00000004
[2024-05-09 13:19:59] error:    zh:ezsp:uart: --> Error: Error: {"sequence":1} after 4000ms
[2024-05-09 13:19:59] error:    zh:ezsp:uart: -!- break rewaiting (1)
[2024-05-09 13:19:59] error:    zh:ezsp:uart: Can't resend DATA frame (0,1,1): 00000004
[2024-05-09 13:20:16] error:    zh:ezsp:uart: --> Error: Error: {"sequence":6} after 4000ms
[2024-05-09 13:20:16] error:    zh:ezsp:uart: -!- break waiting (6)
[2024-05-09 13:20:16] error:    zh:ezsp:uart: Can't send DATA frame (5,5,0): 250001630003
[2024-05-09 13:21:20] error:    zh:ezsp:uart: --> Error: Error: {"sequence":1} after 4000ms
[2024-05-09 13:21:20] error:    zh:ezsp:uart: -!- break waiting (1)
[2024-05-09 13:21:20] error:    zh:ezsp:uart: Can't send DATA frame (0,5,0): 3000010500
[2024-05-09 13:21:28] error:    zh:ezsp:uart: --> Error: Error: {"sequence":1} after 4000ms
[2024-05-09 13:21:28] error:    zh:ezsp:uart: -!- break rewaiting (1)
[2024-05-09 13:21:28] error:    zh:ezsp:uart: Can't resend DATA frame (0,5,1): 3000010500
[2024-05-09 13:21:28] error:    zh:ezsp:ezsp: Watchdog heartbeat timeout Error: Failure send nop:{"type":"Buffer","data":[48,0,1,5,0]}
    at /app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:686:23
    at Queue.execute (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:35:20)
    at Ezsp.watchdogHandler (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:795:13)
[2024-05-09 13:21:29] error:    zh:ezsp:uart: <-- CRC error: 9c148f6d633f9874f0cf638bfc5836a5ebc9de6f8fffc5ed577e|ed57|f562
[2024-05-09 13:21:29] error:    zh:ezsp:uart: Error while parsing to NpiFrame 'Error: <-- CRC error: 9c148f6d633f9874f0cf638bfc5836a5ebc9de6f8fffc5ed577e|ed57|f562
    at Frame.checkCRC (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/frame.ts:87:19)
    at SerialDriver.onParsed (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/uart.ts:169:19)
    at Parser.emit (node:events:517:28)
    at Parser._transform (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/parser.ts:49:26)
    at Parser.Transform._write (node:internal/streams/transform:175:8)
    at writeOrBuffer (node:internal/streams/writable:392:12)
    at _write (node:internal/streams/writable:333:10)
    at Parser.Writable.write (node:internal/streams/writable:337:10)
    at SerialPort.ondata (node:internal/streams/readable:809:22)
    at SerialPort.emit (node:events:517:28)'
[2024-05-09 13:21:29] error:    zh:ezsp:uart: <-- CRC error: f0cf638bfc5836a5ebc9de6f8fffc55e577e|5e57|489f
[2024-05-09 13:21:29] error:    zh:ezsp:uart: Error while parsing to NpiFrame 'Error: <-- CRC error: f0cf638bfc5836a5ebc9de6f8fffc55e577e|5e57|489f
[... 191 lines cut ...]
    at Parser.Transform._write (node:internal/streams/transform:175:8)
    at writeOrBuffer (node:internal/streams/writable:392:12)
    at _write (node:internal/streams/writable:333:10)
    at Parser.Writable.write (node:internal/streams/writable:337:10)
    at SerialPort.ondata (node:internal/streams/readable:809:22)
    at SerialPort.emit (node:events:517:28)'
[2024-05-09 13:31:00] error: 	zh:ezsp:uart: <-- CRC error: 0c71b1a9112a15b65894a524ab5593499cf28f6d633f9874f0cfe78bfcd936a5ebc9de6f7471b1a9112a15b65894a524ab5593499cf59b6003629874f0cf368bfc1336a5ebc9de6f8fffc582847e|8284|4d82
[2024-05-09 13:31:00] error: 	zh:ezsp:uart: Error while parsing to NpiFrame 'Error: <-- CRC error: 0c71b1a9112a15b65894a524ab5593499cf28f6d633f9874f0cfe78bfcd936a5ebc9de6f7471b1a9112a15b65894a524ab5593499cf59b6003629874f0cf368bfc1336a5ebc9de6f8fffc582847e|8284|4d82
    at Frame.checkCRC (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/frame.ts:87:19)
    at SerialDriver.onParsed (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/uart.ts:169:19)
    at Parser.emit (node:events:517:28)
    at Parser._transform (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/parser.ts:49:26)
    at Parser.Transform._write (node:internal/streams/transform:175:8)
    at writeOrBuffer (node:internal/streams/writable:392:12)
    at _write (node:internal/streams/writable:333:10)
    at Parser.Writable.write (node:internal/streams/writable:337:10)
    at SerialPort.ondata (node:internal/streams/readable:809:22)
    at SerialPort.emit (node:events:517:28)'
[2024-05-09 13:31:08] error: 	zh:ezsp:uart: --> Error: Error: {"sequence":1} after 4000ms
[2024-05-09 13:31:08] error: 	zh:ezsp:uart: -!- break waiting (1)
[2024-05-09 13:31:08] error: 	zh:ezsp:uart: Can't send DATA frame (0,1,0): 00000004
[2024-05-09 13:31:12] error: 	zh:ezsp:uart: --> Error: Error: {"sequence":1} after 4000ms
[2024-05-09 13:31:12] error: 	zh:ezsp:uart: -!- break rewaiting (1)
[2024-05-09 13:31:12] error: 	zh:ezsp:uart: Can't resend DATA frame (0,1,1): 00000004
[2024-05-09 13:37:25] error: 	zhc: Failed to apply calibration to 'illuminance_lux': 'illuminance_lux_calibration' is not a number, got string ()

nicolapejo, 2024-05-10:
UPDATE:  Today it started working again, but the following errors are reported on the log. Maybe I need to change the firmware on my dongle?

[2024-05-10 18:17:35] error:    zh:ezsp:ezsp: Frame changeSourceRouteHandler parsing error: RangeError: Attempt to access memory outside buffer bounds
    at new NodeError (node:internal/errors:405:5)
    at boundsError (node:internal/buffer:86:11)
    at Buffer.readUInt16LE (node:internal/buffer:245:5)
    at Buffer.readUIntLE (node:internal/buffer:182:17)
    at Function.deserialize (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/types/basic.ts:19:67)
    at new EZSPFrameData (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:180:54)
    at /app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:154:23
    at Array.every (<anonymous>)
    at Function.createFrame (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:148:15)
    at Ezsp.onFrameReceived (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:439:35)
[2024-05-10 18:17:35] error:    zh:ezsp:ezsp: Unparsed frame 0xc4. Skipped
[2024-05-10 18:25:25] error:    zhc: Failed to apply calibration to 'illuminance_lux': 'illuminance_lux_calibration' is not a number, got string ()
[2024-05-10 18:35:26] error:    zhc: Failed to apply calibration to 'illuminance_lux': 'illuminance_lux_calibration' is not a number, got string ()
[2024-05-10 18:46:45] error:    zhc: Failed to apply calibration to 'illuminance_lux': 'illuminance_lux_calibration' is not a number, got string ()
[2024-05-10 18:46:55] error:    zhc: Failed to apply calibration to 'illuminance_lux': 'illuminance_lux_calibration' is not a number, got string ()
[2024-05-10 18:47:05] error:    zhc: Failed to apply calibration to 'illuminance_lux': 'illuminance_lux_calibration' is not a number, got string ()
[2024-05-10 18:47:15] error:    zhc: Failed to apply calibration to 'illuminance_lux': 'illuminance_lux_calibration' is not a number, got string ()
[2024-05-10 18:47:25] error:    zhc: Failed to apply calibration to 'illuminance_lux': 'illuminance_lux_calibration' is not a number, got string ()
[2024-05-10 18:47:35] error:    zhc: Failed to apply calibration to 'illuminance_lux': 'illuminance_lux_calibration' is not a number, got string ()
[2024-05-10 18:57:36] error:    zhc: Failed to apply calibration to 'illuminance_lux': 'illuminance_lux_calibration' is not a number, got string ()
[2024-05-10 18:58:33] warning:  zh:ezsp: Deprecated driver 'ezsp' currently in use, 'ember' will become the officially supported EmberZNet driver in next release. If using Zigbee2MQTT see https://github.com/Koenkk/zigbee2mqtt/discussions/21462

gianlucaf81, 2024-05-12:
same here, zigbee2mqtt on lxc proxmox, Xiaomi Gateway 3.

`Info 2024-05-12 20:01:17Starting network scan (includeRoutes 'false')
Info 2024-05-12 20:01:42Network scan finished
Error 2024-05-12 20:04:41<-- CRC error: 759eb157546f15b65894a524ab5593499c566b0476889874f0cf858bfc704f059eb157546f15b65894a524ab5593496f86d91a227e|1a22|f50f
Error 2024-05-12 20:04:41Error while parsing to NpiFrame 'Error: <-- CRC error: 759eb157546f15b65894a524ab5593499c566b0476889874f0cf858bfc704f059eb157546f15b65894a524ab5593496f86d91a227e|1a22|f50f at Frame.checkCRC (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/frame.ts:87:19) at SerialDriver.onParsed (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/uart.ts:169:19) at Parser.emit (node:events:517:28) at Parser._transform (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/parser.ts:49:26) at Parser.Transform._write (node:internal/streams/transform:175:8) at writeOrBuffer (node:internal/streams/writable:392:12) at _write (node:internal/streams/writable:333:10) at Parser.Writable.write (node:internal/streams/writable:337:10) at Socket.ondata (node:internal/streams/readable:809:22) at Socket.emit (node:events:517:28)'
Error 2024-05-12 20:04:41<-- CRC error: 7d9eb157546f15b65894a524ab5593c9de6f86d9f2f27e|f2f2|aa5f
Error 2024-05-12 20:04:41Error while parsing to NpiFrame 'Error: <-- CRC error: 7d9eb157546f15b65894a524ab5593c9de6f86d9f2f27e|f2f2|aa5f at Frame.checkCRC (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/frame.ts:87:19) at SerialDriver.onParsed (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/uart.ts:169:19) at Parser.emit (node:events:517:28) at Parser._transform (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/parser.ts:49:26) at Parser.Transform._write (node:internal/streams/transform:175:8) at writeOrBuffer (node:internal/streams/writable:392:12) at _write (node:internal/streams/writable:333:10) at Parser.Writable.write (node:internal/streams/writable:337:10) at Socket.ondata (node:internal/streams/readable:809:22) at Socket.emit (node:events:517:28)'
Error 2024-05-12 20:04:41<-- CRC error: 0d9eb157546f15b65894a524ab5593499c576b0476889874159eb157546f15b65894a524ab5593490476889874f0cf858bfc704fa5ebc9de259eb157546f15b65894a524ab5593497d9eb157546f15b65894a524ab55939eb157546f15b65894a524ab5593499c1d9eb157546f15b65894a524ab559349d988587e|8858|4f31
Error 2024-05-12 20:04:41Error while parsing to NpiFrame 'Error: <-- CRC error: 0d9eb157546f15b65894a524ab5593499c576b0476889874159eb157546f15b65894a524ab5593490476889874f0cf858bfc704fa5ebc9de259eb157546f15b65894a524ab5593497d9eb157546f15b65894a524ab55939eb157546f15b65894a524ab5593499c1d9eb157546f15b65894a524ab559349d988587e|8858|4f31 at Frame.checkCRC (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/frame.ts:87:19) at SerialDriver.onParsed (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/uart.ts:169:19) at Parser.emit (node:events:517:28) at Parser._transform (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/parser.ts:49:26) at Parser.Transform._write (node:internal/streams/transform:175:8) at writeOrBuffer (node:internal/streams/writable:392:12) at _write (node:internal/streams/writable:333:10) at Parser.Writable.write (node:internal/streams/writable:337:10) at Socket.ondata (node:internal/streams/readable:809:22) at Socket.emit (node:events:517:28)'
Error 2024-05-12 20:04:41<-- CRC error: 2d9eb157546f15b65894a524de6f86d9b50d7e|b50d|371e
Error 2024-05-12 20:04:41Error while parsing to NpiFrame 'Error: <-- CRC error: 2d9eb157546f15b65894a524de6f86d9b50d7e|b50d|371e at Frame.checkCRC (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/frame.ts:87:19) at SerialDriver.onParsed (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/uart.ts:169:19) at Parser.emit (node:events:517:28) at Parser._transform (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/parser.ts:49:26) at Parser.Transform._write (node:internal/streams/transform:175:8) at writeOrBuffer (node:internal/streams/writable:392:12) at _write (node:internal/streams/writable:333:10) at Parser.Writable.write (node:internal/streams/writable:337:10) at Socket.ondata (node:internal/streams/readable:809:22) at Socket.emit (node:events:517:28)'
Error 2024-05-12 20:04:41<-- CRC error: 359eb157546f15b65894a524ab559349de6f86d9da967e|da96|885b
Error 2024-05-12 20:04:41Error while parsing to NpiFrame 'Error: <-- CRC error: 359eb157546f15b65894a524ab559349de6f86d9da967e|da96|885b at Frame.checkCRC (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/frame.ts:87:19) at SerialDriver.onParsed (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/uart.ts:169:19) at Parser.emit (node:events:517:28) at Parser._transform (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/parser.ts:49:26) at Parser.Transform._write (node:internal/streams/transform:175:8) at writeOrBuffer (node:internal/streams/writable:392:12) at _write (node:internal/streams/writable:333:10) at Parser.Writable.write (node:internal/streams/writable:337:10) at Socket.ondata (node:internal/streams/readable:809:22) at Socket.emit (node:events:517:28)'
[... cut ...]  (node:internal/streams/readable:809:22) at Socket.emit (node:events:517:28)'
Error 2024-05-12 20:13:18Publish 'set' 'state' to 'Tapparella Studio' failed: 'Error: ZCL command 0x00047400011a04ed/1 closuresWindowCovering.stop({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (sendZclFrameToEndpointInternal error)'`

osmeest, 2024-05-13:
I had something similar with Z2M 1.37.0 and 1.37.1 using HomeAssistant Addon with Sonoff Zigbee Dongle Plus (/dev/serial/by-id/usb-ITEAD_SONOFF_Zigbee_3.0_USB_Dongle_Plus_V2_20221201102927-if00).

Also, since installation of 1.37.0, it mentioned that EZSP is deprecated and would be replaced by Ember but in the Z2M documentation, ZB Dongle P is not mentioned anymore. Only the ZB Dongle E is mentioned. Does that mean support goes away ?

gianlucaf81, 2024-05-13:
> I had something similar with Z2M 1.37.0 and 1.37.1 using HomeAssistant Addon with Sonoff Zigbee Dongle Plus (/dev/serial/by-id/usb-ITEAD_SONOFF_Zigbee_3.0_USB_Dongle_Plus_V2_20221201102927-if00).
> 
> Since adapter type is said to be optional, I tried with "auto" instead of "ezsp" in the serial configuration. But then Z2M wouldn't start at all !
> 
> I restored 1.36.1 from backup and it worked.
> 
> I tried reinstalling 1.37.1 over my working restore of 1.36.1 and it also worked.

wait wait, what did you do? a downgrade or something else?

osmeest, 2024-05-13:
I restored from the (partial) backup that was made when 1.37.0 got installed.

osmeest, 2024-05-13:
To be complete, I updated my dongle firmware to [7.4.2](https://github.com/darkxst/silabs-firmware-builder/blob/main/firmware_builds/zbdonglee/ncp-uart-hw-v7.4.2.0-zbdonglee-115200.gbl) using [NabuCasa Silabs universal flasher](https://github.com/NabuCasa/universal-silabs-flasher) and restarted Z2M 1.37.1 with Ember driver. All working OK until now.

gianlucaf81, 2024-05-13:
> To be complete, I updated my dongle firmware to [7.4.2](https://github.com/darkxst/silabs-firmware-builder/blob/main/firmware_builds/zbdonglee/ncp-uart-hw-v7.4.2.0-zbdonglee-115200.gbl) using [...]

I suspect this is the key part of the problem and the resolution. 😢

nicolapejo, 2024-05-17:
The problem seems to have been solved, I have updated the zbdongle-e with the silabs firmware flasher (https://darkxst.github.io/silabs-firmware-builder/) to version 7.3.1.0 build 176

closed 2024-05-17 by nicolapejo

bdeluca, 2024-08-29:
I have the same issue on 7.3.1.0