Koenkk/zigbee2mqtt #22566: Home Assistant name can't be reset to [nothing]

BoneheadFraggle, 2024-05-09:
### What happened?

After testing the option to change "Home Assistant name" (which works fine), the "Home Assistant name" can't be set to [nothing] so it seems like once you have entered a Home Assistant name, you always need to change that manually after changing the Friendly name in the Rename device dialog to change the name in Home Assistant..

### What did you expect to happen?

I expected to be allowed to delete the manually set "Home Assistant name", and that it then would revert to using the Friendly Name.

### How to reproduce it (minimal and precise)

Set a "Home Assistant name", save, then try to delete it.

### Zigbee2MQTT version

1.37.0-1

### Adapter firmware version

20220219

### Adapter

Sonoff Zigbee 3.0

### Setup

Add-on onHome Assistant OS

tackin, 2024-05-21:
Would like to add:
Using "Rename device" button in z2m to change the "Friendly name" with the option "Update Home Assistant entity ID" enabled will not overwrite the manually set "Home Assistant name" anymore. Tested on z2m v.1.37.1

Skirmisher, 2024-06-03:
> Would like to add: Using "Rename device" button in z2m to change the "Friendly name" with the option "Update Home Assistant entity ID" enabled will not overwrite the manually set "Home Assistant name" anymore. Tested on z2m v.1.37.1

This is still the case as of Z2M 1.38.0, and is what's currently getting at me, since I'm [re-pairing my entire network](https://github.com/Koenkk/zigbee2mqtt/discussions/22919) and have a number of automations that refer to entity IDs matching the device "friendly name". It gets a bit confusing, because on the Home Assistant side, the term "friendly name" is sometimes used to refer to the plain-language device name (Z2M's "Home Assistant name"), while Z2M's "friendly name" is the device's MQTT topic. The latter makes more sense as an entity ID, but right now the former is what's being used unconditionally instead...

closed 2024-12-26 by github-actions[bot]