Koenkk/zigbee2mqtt #22573: "error: z2m: Failed to configure TypeError: func is not a function" with new devices added. Version: 1.37.1-1

Aughmen, 2024-05-10:
### What happened?

error: z2m: Failed to configure TypeError: func is not a function

### What did you expect to happen?

Nothing

### How to reproduce it (minimal and precise)

With this new firmware  1.37.1-1 and previous if you add new devices you will have always this error...I also tried to remove and add them again without success

### Zigbee2MQTT version

 1.37.1-1

### Adapter firmware version

7.4.2 [GA]

### Adapter

EmberZNet usb-ITEAD_SONOFF_Zigbee_3.0_USB_Dongle_Plus_V2_20231220144601-if00

### Setup

Raspberry 5 with Hassio

Juugonomics, 2024-05-10:
I  have the same issue

Koenkk, 2024-05-10:
Could you provide the debug log of this?

See [this](https://www.zigbee2mqtt.io/guide/usage/debug.html) on how to enable debug logging.

Aughmen, 2024-05-10:
[2024-05-11 00:42:06] debug: 	zh:ember:ezsp: ===> [FRAME: ID=52:"SEND_UNICAST" Seq=139 Len=26]
[2024-05-11 00:42:06] debug: 	zh:ember:uart:ash: ---> [FRAME type=DATA frmTx=3 frmRx=5]
[2024-05-11 00:42:06] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA]
[2024-05-11 00:42:06] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=3]
[2024-05-11 00:42:06] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=3 frmNum=5] Added to rxQueue
[2024-05-11 00:42:06] debug: 	zh:ember:uart:ash: ---> [FRAME type=ACK frmRx=6]
[2024-05-11 00:42:06] debug: 	zh:ember:ezsp: Skipping async callback while waiting for response to command.
[2024-05-11 00:42:06] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA]
[2024-05-11 00:42:06] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=4]
[2024-05-11 00:42:06] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=4 frmNum=6] Added to rxQueue
[2024-05-11 00:42:06] debug: 	zh:ember:uart:ash: ---> [FRAME type=ACK frmRx=7]
[2024-05-11 00:42:06] debug: 	zh:ember:ezsp: Skipping async callback while waiting for response to command.
[2024-05-11 00:42:06] debug: 	zh:ember:ezsp: <=== [FRAME: ID=52:"SEND_UNICAST" Seq=139 Len=7]
[2024-05-11 00:42:06] debug: 	zh:ember:ezsp: ~~~> [SENT type=DIRECT apsSequence=47 messageTag=89 status=SUCCESS]
[2024-05-11 00:42:06] debug: 	zh:ember:ezsp: <=== [FRAME: ID=63:"MESSAGE_SENT_HANDLER" Seq=138 Len=27]
[2024-05-11 00:42:06] debug: 	zh:ember:ezsp: ezspMessageSentHandler(): callback called with: [type=DIRECT], [indexOrDestination=8393], [apsFrame={"profileId":260,"clusterId":1030,"sourceEndpoint":1,"destinationEndpoint":1,"options":4352,"groupId":0,"sequence":46}], [messageTag=88], [status=SUCCESS], [messageContents=181c0b0100]
[2024-05-11 00:42:06] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA]
[2024-05-11 00:42:06] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=4]
[2024-05-11 00:42:06] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=4 frmNum=7] Added to rxQueue
[2024-05-11 00:42:06] debug: 	zh:ember:uart:ash: ---> [FRAME type=ACK frmRx=0]
[2024-05-11 00:42:06] debug: 	zh:ember:ezsp: <=== [FRAME: ID=69:"INCOMING_MESSAGE_HANDLER" Seq=139 Len=33]
[2024-05-11 00:42:06] debug: 	zh:ember:ezsp: ezspIncomingMessageHandler(): callback called with: [type=UNICAST], [apsFrame={"profileId":260,"clusterId":1030,"sourceEndpoint":1,"destinationEndpoint":1,"options":320,"groupId":0,"sequence":174}], [lastHopLqi=172], [lastHopRssi=199], [sender=8393], [bindingIndex=255], [addressIndex=255], [messageContents=081d012200002002]
[2024-05-11 00:42:06] debug: 	zh:controller: Received payload: clusterID=1030, address=8393, groupID=0, endpoint=1, destinationEndpoint=1, wasBroadcast=false, linkQuality=172, frame={"header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":false,"reservedBits":0},"manufacturerCode":null,"transactionSequenceNumber":29,"commandIdentifier":1},"payload":[{"attrId":34,"status":0,"dataType":32,"attrData":2}],"command":{"ID":1,"name":"readRsp","parameters":[{"name":"attrId","type":33},{"name":"status","type":32},{"name":"dataType","type":32,"conditions":[{"type":"statusEquals","value":0}]},{"name":"attrData","type":1000,"conditions":[{"type":"statusEquals","value":0}]}]}}
[2024-05-11 00:42:06] debug: 	zh:controller:endpoint: ZCL command 0x0ceff6fffe16912b/1 msOccupancySensing.defaultRsp({"cmdId":1,"statusCode":0}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":29,"writeUndiv":false})
[2024-05-11 00:42:06] debug: 	zh:ember:queue: Status queue=1 priorityQueue=0.
[2024-05-11 00:42:06] debug: 	z2m: Received Zigbee message from 'Sensore Mov LabR', type 'readResponse', cluster 'msOccupancySensing', data '{"ultrasonicUToOThreshold":2}' from endpoint 1 with groupID 0
[2024-05-11 00:42:06] error: 	z2m: Failed to configure 'Sensore Mov LabR', attempt 1 (TypeError: func is not a function
    at Object.configure (/app/node_modules/zigbee-herdsman-converters/src/index.ts:172:27)
    at processTicksAndRejections (node:internal/process/task_queues:95:5)
    at Configure.configure (/app/lib/extension/configure.ts:121:13)
    at Immediate.<anonymous> (/app/lib/extension/configure.ts:74:17))

Aughmen, 2024-05-10:
I don't know if it's good in this way

Aughmen, 2024-05-10:
[2024-05-11 00:47:32] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA]
[2024-05-11 00:47:32] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=4]
[2024-05-11 00:47:32] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=4 frmNum=6] Added to rxQueue
[2024-05-11 00:47:33] debug: 	zh:ember:uart:ash: ---> [FRAME type=ACK frmRx=7]
[2024-05-11 00:47:33] debug: 	zh:ember:ezsp: <=== [FRAME: ID=69:"INCOMING_MESSAGE_HANDLER" Seq=123 Len=43]
[2024-05-11 00:47:33] debug: 	zh:ember:ezsp: ezspIncomingMessageHandler(): callback called with: [type=UNICAST], [apsFrame={"profileId":260,"clusterId":25,"sourceEndpoint":1,"destinationEndpoint":1,"options":320,"groupId":0,"sequence":182}], [lastHopLqi=196], [lastHopRssi=205], [sender=55955], [bindingIndex=255], [addressIndex=255], [messageContents=01890302211010000342410006aa0000401400]
[2024-05-11 00:47:33] debug: 	zh:controller: Received payload: clusterID=25, address=55955, groupID=0, endpoint=1, destinationEndpoint=1, wasBroadcast=false, linkQuality=196, frame={"header":{"frameControl":{"frameType":1,"manufacturerSpecific":false,"direction":0,"disableDefaultResponse":false,"reservedBits":0},"manufacturerCode":null,"transactionSequenceNumber":137,"commandIdentifier":3},"payload":{"fieldControl":2,"manufacturerCode":4129,"imageType":16,"fileVersion":4276739,"fileOffset":43526,"maximumDataSize":64},"command":{"ID":3,"response":5,"parameters":[{"name":"fieldControl","type":32},{"name":"manufacturerCode","type":33},{"name":"imageType","type":33},{"name":"fileVersion","type":35},{"name":"fileOffset","type":35},{"name":"maximumDataSize","type":32}],"name":"imageBlockRequest"}}
[2024-05-11 00:47:33] debug: 	zh:controller: Skipping command 'imageBlockRequest' because it is missing from the lookup
[2024-05-11 00:47:33] debug: 	zhc:ota:common: Request offsets: fileOffset=43526 pageOffset=0                 dataSize=64
[2024-05-11 00:47:33] debug: 	zhc:ota:common: Payload offsets: start=43526 end=43590 dataSize=64
[2024-05-11 00:47:33] debug: 	zh:controller:endpoint: CommandResponse 0x000474000026569b/1 genOta.imageBlockResponse({"status":0,"manufacturerCode":4129,"imageType":16,"fileVersion":4276739,"fileOffset":43526,"dataSize":64,"data":{"type":"Buffer","data":[124,62,175,39,204,0,178,213,140,149,33,74,172,149,162,196,238,90,159,161,49,231,103,22,162,162,186,151,70,124,76,119,70,95,49,31,173,135,76,118,116,93,131,149,228,32,183,88,214,142,239,93,36,194,153,166,63,67,218,235,18,61,107,23]}}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false})
[2024-05-11 00:47:33] debug: 	zh:ember:queue: Status queue=0 priorityQueue=0.
[2024-05-11 00:47:33] debug: 	zh:ember: ~~~> [ZCL to=55955 apsFrame={"profileId":260,"clusterId":25,"sourceEndpoint":1,"destinationEndpoint":1,"options":4352,"groupId":0,"sequence":0} header={"frameControl":{"reservedBits":0,"frameType":1,"direction":1,"disableDefaultResponse":true,"manufacturerSpecific":false},"manufacturerCode":null,"transactionSequenceNumber":137,"commandIdentifier":5}]

Aughmen, 2024-05-10:
There is also this error:
error: 	zh:ember:uart:ash: Received frame with CRC error

Aughmen, 2024-05-10:
This is my log with errors continuously that repeat from devices added after last update of Zigbee2Mqtt and the update of device that arrived at 99 % and restarted over and over again

Info 2024-05-11 01:14:10Configuring 'Sensore Sala ContrattiR'
Error 2024-05-11 01:14:10Failed to configure 'Sensore Sala ContrattiR', attempt 3 (TypeError: func is not a function at Object.configure (/app/node_modules/zigbee-herdsman-converters/src/index.ts:172:27) at Configure.configure (/app/lib/extension/configure.ts:121:13) at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:174:17))
Info 2024-05-11 01:14:21Configuring 'Sensore Mov LabR'
Error 2024-05-11 01:14:21Failed to configure 'Sensore Mov LabR', attempt 3 (TypeError: func is not a function at Object.configure (/app/node_modules/zigbee-herdsman-converters/src/index.ts:172:27) at Configure.configure (/app/lib/extension/configure.ts:121:13) at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:174:17))
Info 2024-05-11 01:14:27Update of 'Faretti Laboratorio Nord' at 0.00%
Info 2024-05-11 01:14:31Configuring 'Sensore Mov Lab2'
Error 2024-05-11 01:14:31Failed to configure 'Sensore Mov Lab2', attempt 3 (TypeError: func is not a function at Object.configure (/app/node_modules/zigbee-herdsman-converters/src/index.ts:172:27) at Configure.configure (/app/lib/extension/configure.ts:121:13) at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:174:17))

Koenkk, 2024-05-12:
What device is `Sensore Mov Lab2`, `'Sensore Mov LabR` and `Sensore Sala ContrattiR`?

Aughmen, 2024-05-12:
> What device is `Sensore Mov Lab2`, `'Sensore Mov LabR` and `Sensore Sala ContrattiR`?

They are movement Sensor from Sonoff SNZB-06P

radaiko, 2024-05-13:
I received today my first Sonoff SNZB-06P and got the same error messages in the log.

axelboe, 2024-05-15:
I received the same sensor today and have the same error.

Sensor firmware is 1.0.6 which leads to an error on the OTA screen: Firmware on '0x0cae5ffffec13371' (SNZB-06P) is newer than latest firmware online.

Koenkk, 2024-05-19:
Fixed!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

closed 2024-05-19 by Koenkk

nicola-spreafico, 2024-05-20:
Same error for me, just received my first SNZB-06P today and got this error during pairing.
```Failed to configure '0x0ceff6fffe3b0b63', attempt 4 (TypeError: func is not a function at Object.configure (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/src/index.ts:172:27) at processTicksAndRejections (node:internal/process/task_queues:95:5) at Configure.configure (/opt/zigbee2mqtt/lib/extension/configure.ts:121:13) at Configure.onMQTTMessage (/opt/zigbee2mqtt/lib/extension/configure.ts:55:21) at EventEmitter.wrappedCallback (/opt/zigbee2mqtt/lib/eventBus.ts:174:17))```

https://www.zigbee2mqtt.io/devices/SNZB-06P.html

My setup
```Versione Zigbee2MQTT [1.37.1](https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.37.1) commit: [d043fd56]
Tipo Coordinator EmberZNet
Revisione Coordinator 7.4.1 [GA]
Indirizzo IEEE Coordinatore 0x943469fffe71ac61
Versione frontend 0.6.167 | 19.37.2 | 0.46.6```

I understand that a fix is already in motion and it is available on dev branch, I look forward on having this on production branch

linuxmaniac, 2024-05-21:
Is there anything to do apart of using the latest-dev branch? I still see this error:
```
2024-05-21 20:28:44Failed to configure 'sensor04', attempt 1 (Error: ZCL command 0x54ef441000a71304/1 genPowerCfg.read(["batteryVoltage"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 33803 - 1 - 19 - 1 - 1 after 10000ms) at Object.start (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:63:23) at ZStackAdapter.sendZclFrameToEndpointInternal (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:498:47) at processTicksAndRejections (node:internal/process/task_queues:95:5) at Queue.execute (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:35:20) at Request.send (/app/node_modules/zigbee-herdsman/src/controller/helpers/request.ts:79:20) at Endpoint.zclCommand (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:767:28) at Endpoint.read (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:402:29) at configure (/app/node_modules/zigbee-herdsman-converters/src/devices/lumi.ts:1316:13) at Object.configure (/app/node_modules/zigbee-herdsman-converters/src/index.ts:173:21) at Configure.configure (/app/lib/extension/configure.ts:121:13))
```

```
Zigbee2MQTT version
1.37.1-dev commit: [d484405](https://github.com/Koenkk/zigbee2mqtt/commit/d484405)
Coordinator type zStack3x0
Coordinator revision 20240315
Coordinator IEEE Address 0x00124b0029e8b649
Frontend version 0.6.168
19.42.0
0.48.1
```

aslabsalbeh, 2024-05-29:
Also having this issue with the Sonoff presence sensor mentioned above.

Is there an easy way to downgrade z2mqtt without relying on restoring HA backup (as that means I'll lose many changes done in the last 2-4 weeks) ?