Koenkk/zigbee2mqtt #22574: Bticino K4570CWI Action always empty string or null, not enter or leave

syntesys87, 2024-05-10:
### What happened?

When I press a button on my Bticino K4570CWI, the action is empty, instead of enter or leave.

### How to reproduce it (minimal and precise)

Click a button on Bticino K4570CWI and check Action in Z2M frontend

### Zigbee2MQTT version

1.37.1 with ember driver

### Adapter firmware version

7.4.2.0 build 0

### Adapter

EZSP v12 SONOFF_Zigbee_3.0_USB_Dongle_Plus_V2

### Setup

HAOS

### Debug log

[log.log](https://github.com/Koenkk/zigbee2mqtt/files/15270757/log.log)

syntesys87, 2024-06-02:
@FabianMangold can you please check it? Thank you.

FabianMangold, 2024-07-17:
Hi @syntesys87,

If i remember well, this is due to HomeAssistant's legacy trigger. Can you check if those are active on your end ?
If so, please try to disable them and let me know if that works.
Thanks,

closed 2025-01-03 by github-actions[bot]