Koenkk/zigbee2mqtt #22575: All motion detection from any Third Reality device doesn't update expose after update to v1.37.1-1

ZetaWaves, 2024-05-10:
### What happened?

After update to v1.37.1-1, no Third Reality device is updating motion detection expose.

### What did you expect to happen?

Motion detection normally works. 

### How to reproduce it (minimal and precise)

Update to v1.37.1-1 (doesn't work)
Revert to previous release (works again)

### Zigbee2MQTT version

v1.37.1-1

### Adapter firmware version

latest

### Adapter

Sonoff P 3.0 

### Setup

Home Assistant Add-on

dieneuser, 2024-05-11:
I have the same problem with a [DIY](https://gitlab.com/homrepo/applications/multisensor-lite) device. All properties work - only occupancy remains on motion in _inactive_ (or _null_) state.
In version 1.37.0 the sensor still works without errors. 
I think there is a mistake in this change: (https://github.com/Koenkk/zigbee-herdsman-converters/issues/7441)

1.37.0:
[image]

1.37.1 and dev branch:
[image]

kris-anderson, 2024-05-12:
Can confirm this issue on all my Third Reality Multi-Function Nightlights (3RSNL02043Z).

Lux and light controls work, but motion sensing no longer works.

Reverting to 1.37.0 gets motion sensing reporting again.

fxfitz, 2024-05-12:
I'm observing the same!

pauldcomanici, 2024-05-12:
I'm noticing something similar with [Aqara P1 motion](https://www.zigbee2mqtt.io/devices/RTCGQ14LM.html#aqara-rtcgq14lm). In my case, the issue is caused by the fact that the illuminance_lux entity is updated after the motion entity resulting in my automation to not work properly.
When I check MQTT Explorer I get:
```
{"battery":100,"detection_interval":20,"device_temperature":26,"illuminance":5,"linkquality":178,"motion_sensitivity":"high","occupancy":true,"power_outage_count":0,"trigger_indicator":false,"update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null,"voltage":3019}
```
in the z2m logs panel I see:
```
info 2024-05-12 08:24:26MQTT publish: topic 'zigbee2mqtt/0.BJ motion', payload '{"battery":100,"detection_interval":20,"device_temperature":26,"illuminance":5,"linkquality":178,"motion_sensitivity":"high","occupancy":true,"power_outage_count":0,"trigger_indicator":false,"update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null,"voltage":3019}'
```
In both cases I have good values, however when my automation runs I notice that motion entity triggers from 'off' to 'on' and the illuminance_lux entity has the old value (not the one as we see from the above logs. I did a fix where I have a new trigger based on illuminance_lux entity with a delay of 1sec to catch this use-case; however will be nice if the entities are updated accordingly.

I'm not 100% sure, but I think this issue started with version 1.36.1

barosan909, 2024-05-13:
Same here. Also the door sensor stopped working at the same time

nholben, 2024-05-13:
Same here for my devices. No motion reporting on 1.37.1, everything else appears to work.

Koenkk, 2024-05-14:
Regarding the third reality devices, could you provide the debug log when motion should be detected?

See [this](https://www.zigbee2mqtt.io/guide/usage/debug.html) on how to enable debug logging.

fxfitz, 2024-05-14:
```
[2024-05-14 14:49:04] debug: 	z2m: Received MQTT message on 'homeassistant/binary_sensor/0xb40ecfd3a00c0000/occupancy/config' with data '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"device":{"identifiers":["zigbee2mqtt_0xb40ecfd3a00c0000"],"manufacturer":"Third Reality","model":"Zigbee multi-function night light (3RSNL02043Z)","name":"Delaney's Hallway - Nightlight","sw_version":"v0.00.72","via_device":"zigbee2mqtt_bridge_0xf082c0fffe4348c4"},"device_class":"motion","object_id":"delaney's_hallway_-_nightlight_occupancy","origin":{"name":"Zigbee2MQTT","sw":"1.37.1","url":"https://www.zigbee2mqtt.io"},"payload_off":false,"payload_on":true,"state_topic":"zigbee2mqtt/Delaney's Hallway - Nightlight","unique_id":"0xb40ecfd3a00c0000_occupancy_zigbee2mqtt","value_template":"{{ value_json.occupancy }}"}'
[2024-05-14 14:49:04] debug: 	z2m: Received MQTT message on 'homeassistant/sensor/0xb40ecfd3a00c0000/illuminance/config' with data '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"device":{"identifiers":["zigbee2mqtt_0xb40ecfd3a00c0000"],"manufacturer":"Third Reality","model":"Zigbee multi-function night light (3RSNL02043Z)","name":"Delaney's Hallway - Nightlight","sw_version":"v0.00.72","via_device":"zigbee2mqtt_bridge_0xf082c0fffe4348c4"},"enabled_by_default":false,"name":"Illuminance","object_id":"delaney's_hallway_-_nightlight_illuminance","origin":{"name":"Zigbee2MQTT","sw":"1.37.1","url":"https://www.zigbee2mqtt.io"},"state_class":"measurement","state_topic":"zigbee2mqtt/Delaney's Hallway - Nightlight","unique_id":"0xb40ecfd3a00c0000_illuminance_zigbee2mqtt","value_template":"{{ value_json.illuminance }}"}'
[2024-05-14 14:49:04] debug: 	z2m: Received MQTT message on 'homeassistant/sensor/0xb40ecfd3a00c0000/illuminance_lux/config' with data '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"device":{"identifiers":["zigbee2mqtt_0xb40ecfd3a00c0000"],"manufacturer":"Third Reality","model":"Zigbee multi-function night light (3RSNL02043Z)","name":"Delaney's Hallway - Nightlight","sw_version":"v0.00.72","via_device":"zigbee2mqtt_bridge_0xf082c0fffe4348c4"},"device_class":"illuminance","enabled_by_default":true,"object_id":"delaney's_hallway_-_nightlight_illuminance_lux","origin":{"name":"Zigbee2MQTT","sw":"1.37.1","url":"https://www.zigbee2mqtt.io"},"state_class":"measurement","state_topic":"zigbee2mqtt/Delaney's Hallway - Nightlight","unique_id":"0xb40ecfd3a00c0000_illuminance_lux_zigbee2mqtt","unit_of_measurement":"lx","value_template":"{{ value_json.illuminance_lux }}"}'
[2024-05-14 14:49:04] debug: 	z2m: Received MQTT message on 'homeassistant/light/0xb40ecfd3a00c0000/light/config' with data '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"brightness":true,"brightness_scale":254,"command_topic":"zigbee2mqtt/Delaney's Hallway - Nightlight/set","device":{"identifiers":["zigbee2mqtt_0xb40ecfd3a00c0000"],"manufacturer":"Third Reality","model":"Zigbee multi-function night light (3RSNL02043Z)","name":"Delaney's Hallway - Nightlight","sw_version":"v0.00.72","via_device":"zigbee2mqtt_bridge_0xf082c0fffe4348c4"},"effect":true,"effect_list":["blink","breathe","okay","channel_change","finish_effect","stop_effect","colorloop","stop_colorloop"],"name":null,"object_id":"delaney's_hallway_-_nightlight","origin":{"name":"Zigbee2MQTT","sw":"1.37.1","url":"https://www.zigbee2mqtt.io"},"schema":"json","state_topic":"zigbee2mqtt/Delaney's Hallway - Nightlight","supported_color_modes":["xy"],"unique_id":"0xb40ecfd3a00c0000_light_zigbee2mqtt"}'
[2024-05-14 14:49:04] debug: 	z2m: Received MQTT message on 'homeassistant/select/0xb40ecfd3a00c0000/effect/config' with data '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"command_topic":"zigbee2mqtt/Delaney's Hallway - Nightlight/set/effect","device":{"identifiers":["zigbee2mqtt_0xb40ecfd3a00c0000"],"manufacturer":"Third Reality","model":"Zigbee multi-function night light (3RSNL02043Z)","name":"Delaney's Hallway - Nightlight","sw_version":"v0.00.72","via_device":"zigbee2mqtt_bridge_0xf082c0fffe4348c4"},"enabled_by_default":false,"icon":"mdi:palette","name":"Effect","object_id":"delaney's_hallway_-_nightlight_effect","options":["blink","breathe","okay","channel_change","finish_effect","stop_effect","colorloop","stop_colorloop"],"origin":{"name":"Zigbee2MQTT","sw":"1.37.1","url":"https://www.zigbee2mqtt.io"},"unique_id":"0xb40ecfd3a00c0000_effect_zigbee2mqtt"}'
[2024-05-14 14:49:04] debug: 	z2m: Received MQTT message on 'homeassistant/sensor/0xb40ecfd3a00c0000/power_on_behavior/config' with data '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"device":{"identifiers":["zigbee2mqtt_0xb40ecfd3a00c0000"],"manufacturer":"Third Reality","model":"Zigbee multi-function night light (3RSNL02043Z)","name":"Delaney's Hallway - Nightlight","sw_version":"v0.00.72","via_device":"zigbee2mqtt_bridge_0xf082c0fffe4348c4"},"enabled_by_default":false,"entity_category":"diagnostic","icon":"mdi:power-settings","name":"Power-on behavior","object_id":"delaney's_hallway_-_nightlight_power_on_behavior","origin":{"name":"Zigbee2MQTT","sw":"1.37.1","url":"https://www.zigbee2mqtt.io"},"state_topic":"zigbee2mqtt/Delaney's Hallway - Nightlight","unique_id":"0xb40ecfd3a00c0000_power_on_behavior_zigbee2mqtt","value_t [... cut ...] ":false,"entity_category":"diagnostic","icon":"mdi:update","name":"Update state","object_id":"delaney's_hallway_-_nightlight_update_state","origin":{"name":"Zigbee2MQTT","sw":"1.37.1","url":"https://www.zigbee2mqtt.io"},"state_topic":"zigbee2mqtt/Delaney's Hallway - Nightlight","unique_id":"0xb40ecfd3a00c0000_update_state_zigbee2mqtt","value_template":"{{ value_json['update']['state'] }}"}'
```

I _think_ this is what you need. Let me know if you need something else, or a more specific log line.

Koenkk, 2024-05-15:
This is something different, can you:

- Start z2m and wait 60 seconds
- Trigger occupancy,  you should see `Received Zigbee message` (if not that's a problem)
- Provide that log here

Koenkk, 2024-05-16:
I think this has already been fixed in https://github.com/Koenkk/zigbee-herdsman-converters/pull/7521, can you check with the latest [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)?

digitalcardboard, 2024-05-19:
+1 on this for a Third Reality 3RSNL02043Z. I thought I had misconfigured it. Trying to sort out how to switch to edge to test.

digitalcardboard, 2024-05-19:
> I think this has already been fixed in [Koenkk/zigbee-herdsman-converters#7521](https://github.com/Koenkk/zigbee-herdsman-converters/pull/7521), can you check with the latest [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)?

Dev branch works, at least for my Third Reality 3RSNL02043Z.

fxfitz, 2024-05-29:
Sounds like we have a verification that it's fixed in the dev branch. When can we expect the fix to be released?

Koenkk, 2024-05-30:
Next release is on Saturday!

closed 2024-05-30 by Koenkk