Koenkk/zigbee2mqtt #22578: PTM 215 ZE does not show action in HA

tluethi71, 2024-05-10:
### What happened?

Hello
I have paired 2 PTM 215 ZE to zigbee2mqtt successfully. It shows the the linkquality and the status of the action (when pressing a button) in zigbee2mqtt but in Home Assistant it only shows the linkquality and no sensor called action. Somehow the action does not come through zigbee2mqtt to HA. Is this a known bug?
Thanks for any help
Br Tom

### What did you expect to happen?

When pressing a button (e.g. press_1, release_1) it should be shown in the according mqtt-device in HomeAssistant. The action is shown in zigbee2mqtt but not in HA.

### How to reproduce it (minimal and precise)

Click a button on the PTM 215 ZE and check what happens in zigbee2mqtt (status, see pic) and then check in HA the according device (see pic).  It does not show the action "press_1" or "release_1" and there is no action sensor.
[image]
[image]

### Zigbee2MQTT version

1.37.1

### Adapter firmware version

0x26720700

### Adapter

ConBee2/RaspBee2

### Setup

Intel NUC 11

### Debug log

{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2024.5.2",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.12.2",
    "docker": true,
    "arch": "x86_64",
    "timezone": "Europe/Zurich",
    "os_name": "Linux",
    "os_version": "6.6.29-haos",
    "supervisor": "2024.05.1",
    "host_os": "Home Assistant OS 12.3",
    "docker_version": "25.0.5",
    "chassis": "embedded",
    "run_as_root": true
  },
  "custom_components": {
    "daily_min_max": {
      "documentation": "https://github.com/philsson/HomeAssistantCustomComponents",
      "version": "1.0",
      "requirements": []
    },
    "proconip_pool_controller": {
      "documentation": "https://github.com/ylabonte/proconip-hass",
      "version": "v1.2.0",
      "requirements": [
        "proconip>=1.3.0"
      ]
    },
    "swisscom_internetbox": {
      "documentation": "https://github.com/schmidtfx/swisscom-internetbox-hass",
      "version": "0.1.1",
[... 355 lines cut ...]
                "model": "Pushbutton transmitter module (PTM 215ZE)",
                "name": "zb2m_taster_pendelleuchte",
                "via_device": "zigbee2mqtt_bridge_0x00212effff0985cb"
              },
              "origin": {
                "name": "Zigbee2MQTT",
                "sw_version": "1.37.0",
                "support_url": "https://www.zigbee2mqtt.io"
              },
              "payload": "release_1",
              "subtype": "release_1",
              "topic": "zigbee2mqtt/zb2m_taster_pendelleuchte/action",
              "type": "action",
              "platform": "mqtt"
            }
          },
          "trigger_key": [
            "device_automation",
            "0x0000000001567624 action_release_1"
          ]
        }
      ]
    }
  }
}

AlexandreKueny, 2024-05-10:
How did you manage to pair it ? I've been trying for several hours without any success :/

tluethi71, 2024-05-10:
It is very tricky, I needed a couple of try.

it is important that you follow exactly the instruction on how to pair pn zigbee2mqtt.io.

1. press a button
2. Press another button
3. Press a buton for 7s
4. Press the same for less than 2 sec
5. Press the same for another 7s
6. Press the same button again if you operate on channel 11 (if other channel see manu
7.  press another button

you have to enable Elapsed in the settings. Very important.

does it help?

extreme4u, 2024-05-11:
Hello, I've been having issues with the PTM 215 ZE switches since the update as well. They either don't respond at all anymore or only very sporadically. A factory reset and repairing is no longer possible. I'm using the SkyConnect Zigbee dongle from Homeassistant. Fortunately, I had a Zigbee dongle from Deconz available. Relearning with it worked without any problems. So, I assume the update has somehow caused damage here.

dzerrenner, 2024-05-27:
I'm also confused. I plan to integrate a second zigbee network in my setup and bought a new adapter (UZG-01, zstack, CC2652P7). My legacy network runs on Channel 11, my new is planned to run on channel 20. But i'm not able to pair my PTM 215Z to the new network. The reset-and-switch-channel procedure is a bit of a hassle, so at first I thought i failed at the button sequence. But after setting the switch to channel 20 and seeing nothing at all in the logs i switched the switch back to channel 11 and could re-pair it without problems to the old instance. 

Versions: 
  - legacy (old): [1.35.3](https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.35.3) commit: [fe0742a6](https://github.com/Koenkk/zigbee2mqtt/commit/fe0742a6), zigbee-herdsman (0.33.8)
  - new: Zigbee2MQTT version [1.37.1](https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.35.3) commit: [cda867a](https://github.com/Koenkk/zigbee2mqtt/commit/cda867a), zigbee-herdsman (0.46.6)

dzerrenner, 2024-05-27:
> How did you manage to pair it ? I've been trying for several hours without any success :/

There is another way to do this if you want to use one of 4 predefined channels (11, 15, 20 or 25) *and* have the possibility to press more than one button at once:

 - Reset the switch: press all 4 buttons and press and hold the energy bar for 7+ seconds
 - Select channel: Press one of the 4 buttons corresponding to the desired channel (top right: 11, top left 15, bottom left: 20, bottom right: 25) and press and hold the energy bar for 7+ seconds
 - Confirm Channel: Press top right and bottom left together and press the energy bar. 
 
There should be a log message for commissioning after the second step. The third step seems to store the new channel in the switch.

These steps are taken from the manual for the PTM *216* Z, but this worked for me on my PTM 215 Z. I'm unable to find a manual for the PTM 215 Z.

tluethi71, 2024-06-02:
still have the problem. In the zigbee2mqtt I can see in the status the following: 

{
    "linkquality": 127,
    "action": "release_1",
    "elapsed": 230
}

But it does not come through HA... I only see the linkquality 127 in the device..

robertjanquist, 2024-08-23:
Starting from version `1.37.0` my Friends of Hue switch (PTM215Z) is not working anymore.

When I press and release any button in version `1.37.0` I get this debug log:

``` log
[2024-08-23 20:42:27] debug: 	zh:deconz:frameparser: GP data notification
[2024-08-23 20:42:27] debug: 	zh:deconz:frameparser: GP_DATA_INDICATION - src id: 0x174ce96 cmd id: 0x12 frameCounter: 0x170a
[2024-08-23 20:42:27] debug: 	zh:controller:greenpower: Received unhandled command '0x12' from '24432278'
[2024-08-23 20:42:27] debug: 	zh:controller: Data is from unknown device with address '24432278', skipping...
[2024-08-23 20:42:27] debug: 	zh:deconz:frameparser: GP data notification
[2024-08-23 20:42:27] debug: 	zh:deconz:frameparser: GP data notification
[2024-08-23 20:42:27] debug: 	zh:deconz:frameparser: DEVICE_STATE changed: 10101010
[2024-08-23 20:42:27] debug: 	zh:deconz:driver: networkstate: 2 apsDataConfirm: 0 apsDataIndication: 1 configChanged: 0 apsRequestFreeSlots: 1
[2024-08-23 20:42:27] debug: 	zh:deconz:driver: query aps data indication
[2024-08-23 20:42:27] debug: 	zh:deconz:driver: DATA_INDICATION - sending read data request - SeqNr. 34
[2024-08-23 20:42:27] debug: 	zh:deconz:frameparser: DATA_INDICATION RESPONSE - seqNr. 34 srcAddr: 0xc988 destAddr: 0x0 profile id: 0xa1e0 cluster id: 0x21 lqi: 255
[2024-08-23 20:42:27] debug: 	zh:deconz:frameparser: response payload: 17,115,0,136,84,150,206,116,1,10,23,0,0,18,0,136,201,219
[2024-08-23 20:42:27] debug: 	zh:controller:greenpower: Received unhandled command '0x12' from '51592'
[2024-08-23 20:42:27] debug: 	zh:deconz:driver: networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
[2024-08-23 20:42:27] debug: 	zh:controller: Data is from unknown device with address '51592', skipping...
[2024-08-23 20:42:27] debug: 	zh:deconz:frameparser: GP data notification
[2024-08-23 20:42:27] debug: 	zh:deconz:frameparser: GP_DATA_INDICATION - src id: 0x174ce96 cmd id: 0x16 frameCounter: 0x170b
[2024-08-23 20:42:27] debug: 	zh:controller:greenpower: Received unhandled command '0x16' from '24432278'
[2024-08-23 20:42:27] debug: 	zh:controller: Data is from unknown device with address '24432278', skipping...
[2024-08-23 20:42:27] debug: 	zh:deconz:frameparser: GP data notification
[2024-08-23 20:42:27] debug: 	zh:deconz:frameparser: GP data notification
[2024-08-23 20:42:27] debug: 	zh:deconz:frameparser: DEVICE_STATE changed: 10101010
[2024-08-23 20:42:27] debug: 	zh:deconz:driver: networkstate: 2 apsDataConfirm: 0 apsDataIndication: 1 configChanged: 0 apsRequestFreeSlots: 1
[2024-08-23 20:42:27] debug: 	zh:deconz:driver: query aps data indication
[2024-08-23 20:42:27] debug: 	zh:deconz:driver: DATA_INDICATION - sending read data request - SeqNr. 35
[2024-08-23 20:42:27] debug: 	zh:deconz:frameparser: DATA_INDICATION RESPONSE - seqNr. 35 srcAddr: 0xc988 destAddr: 0x0 profile id: 0xa1e0 cluster id: 0x21 lqi: 255
[2024-08-23 20:42:27] debug: 	zh:deconz:frameparser: response payload: 17,116,0,136,84,150,206,116,1,11,23,0,0,22,0,136,201,218
[2024-08-23 20:42:27] debug: 	zh:controller:greenpower: Received unhandled command '0x16' from '51592'
[2024-08-23 20:42:27] debug: 	zh:deconz:driver: networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
[2024-08-23 20:42:27] debug: 	zh:controller: Data is from unknown device with address '51592', skipping...
```

When downgrading to version `1.36.1` it all works.

Does anyone know what is changed in version `1.37.0`?

coendaar, 2024-09-03:
@robertjanquist  seems to me you have the same issue I have: #23814 
Did you get it resolved?

robertjanquist, 2024-09-03:
@coendaar I did not have the time to dive into this issue. So currently I did not upgrade to version `1.37.0` or higher.

Koenkk, 2024-09-05:
Let's continue in https://github.com/Koenkk/zigbee2mqtt/issues/23814

closed 2024-09-05 by Koenkk

tluethi71, 2024-09-14:
Solved:

1. you have to set `legacy_triggers: true` in the configuration.yaml in zigbee2mqtt
2. you have to create an automation to trigger the state `action`. in my following automation I am only triggering 2  buttons (3 and 4)

`alias: "Taster: Buero Lightstrip"
description: ""
trigger:
  - platform: state
    entity_id:
      - sensor.zb2m_taster_buero_tom_action
    to: press_3
    id: "3"
  - platform: state
    entity_id:
      - sensor.zb2m_taster_buero_tom_action
    to: press_4
    id: "4"
condition: []
action:
  - choose:
      - conditions:
          - condition: trigger
            id:
              - "3"
        sequence:
          - action: light.turn_off
            metadata: {}
            data: {}
            target:
              entity_id: light.zb2m_lightstrip_buero_tom
      - conditions:
          - condition: trigger
            id:
              - "4"
        sequence:
          - action: light.turn_on
            metadata: {}
            data: {}
            target:
              entity_id: light.zb2m_lightstrip_buero_tom
mode: single`