Koenkk/zigbee2mqtt #22581: [Feature request]: 希望能对低信号设备拒绝连接保证设备通信可靠性,It is hoped that the rejection of connection to low-signal equipment can guarantee the reliability of equipment communication

qq547176052, 2024-05-10:
### Is your feature request related to a problem? Please describe

希望能对低信号设备拒绝连接保证设备通信可靠性
It is hoped that the rejection of connection to low-signal equipment can guarantee the reliability of equipment communication

### Describe the solution you'd like

设置信号值低到某个值就拒绝连接
Set the signal value below a certain value to reject the connection

### Describe alternatives you've considered

没有
no

### Additional context

没有
no

GSzabados, 2024-05-10:
Sorry for the question, but why would you force this? I have devices which report 0 at LQI and they still work flawlessly.

qq547176052, 2024-05-11:
在协调器拒绝信号差的设备连接可以让这些设备连接到路由器而不是连接到协调器
The coordinator rejects poor-signal device connections so that the devices connect to the router instead of the coordinator

GSzabados, 2024-05-11:
You have two options to do this:

Either select the device which you want to connect to. Or select All to connect to either the coordinator or any other repeater.

If you select Coordinator it will connect to the coordinator.

[image]

But as I remember your coordinator has to support this functionality as well.

Anyhow the Zigbee mesh should be self healing, and each device should choose the best possible route. Maybe it is down to your device (device's firmware) why it connects to the coordinator.

qq547176052, 2024-05-11:
[image]
信号很好也会出现控制失败 而且不会自动恢复
Control failures occur when the signal is good and do not automatically recover
[image]

GSzabados, 2024-05-11:
You must have some other problem. Your device has an LQI of 189. That is a great value.

qq547176052, 2024-05-11:
[image]
串口直接连接linux开发板上的docker没有其他硬件
The serial port connects directly to Docker on the Linux Development Board without additional hardware
实在无法判断是什么地方的问题
It is impossible to tell where the problem lies

qq547176052, 2024-05-11:
[image]
1次控制 会发送3次数据这样正常吗
1 control, will send 3 data is this normal

GSzabados, 2024-05-11:
> It is impossible to tell where the problem lies

What is this board and how is it built exactly? My wild guess would be, that you have some serious interference issue. That Zigbee antenna should face the other direction, away from that USB-C port, and any USB port.

Have you heard about USB 3.0 interference on Zigbee?

https://www.youtube.com/watch?v=tHqZhNcFEvA

qq547176052, 2024-05-11:
开发版是:Orange Pi Zero3
板子上没有usb 3.0
这是开发板资料
http://www.orangepi.cn/html/hardWare/computerAndMicrocontrollers/details/Orange-Pi-Zero-3.html

Development: Orange Pi Zero3

There is no USB 3.0 on the board

This is the development board information
http://www.orangepi.cn/html/hardWare/computerAndMicrocontrollers/details/Orange-Pi-Zero-3.html

qq547176052, 2024-05-11:
1次控制会返回3条mqtt消息这是正常的吗?
我全新使用docker安装运行zigbee2mqtt程序
删除ha
重新安装mqtt服务器
还是没有得到解决

Is it normal for 3 MQTT messages to be returned from a control?

I am using the new Docker installation to run the zigbee2mqtt program

Delete ha

Reinstall the MQTT server

It still hasn't been resolved
[image]

GSzabados, 2024-05-11:
I think that is normal. I see 3 messages in MQTT Explorer as well.

jack9603301, 2024-05-12:
mqtt should not produce duplicate messages

closed 2024-12-09 by github-actions[bot]