Koenkk/zigbee2mqtt #22582: After flash update to ember. Z2M configurationnot excepting ember

canedje, 2024-05-10:
### What happened?

I just flashed my Sonoff ZBDongle-E to latest 7.4.2.0 build. Using this tutorial:
https://smarthomescene.com/guides/how-to-use-ember-driver-with-efr32mg21-zigbee-adapters/?fbclid=IwZXh0bgNhZW0CMTAAAR3p8Cy-teqA7gg_Fn2xUkTDArGSTlTZRORNU7pgXl7__Rvuf9xVL1Dgtiw_aem_ASu41kv-OnrMVTE3Y8okCTbiHubgfS9SgUDMZtypAozlPWAd-tXbooK2_XYw36NmxHlptQOmm8gjzewfoWFZGqNV

Flashing worked fine.
But after changing mij configuration file adapter to ember.
My Z2M docker didn't start up with a CRC error.
[image]
I did change the adapter type back to ezsp and Z2M did start up again.
At about it gives indeed EZSP instead of embe. It gives the right build versionr:
[image]

### What did you expect to happen?

I Expected ember as type

### How to reproduce it (minimal and precise)

See above. Just change adapter in ember instead of EZSP

### Zigbee2MQTT version

1.37.1 commit: ea39d86

### Adapter firmware version

7.4.2.0 build 0

### Adapter

Sonoff ZBDongle-E 

### Setup

Z2M as a docker under Unraid

wastez, 2024-05-10:
I think that not the crc error was the reason why it wasn’t starting. 
In the error below a old backup file triggered the error.

canedje, 2024-05-10:
ahhhh okey.
Thanks. It could be possible because I did make a new backup file when running as EZSP.
After that I did retry the ember adapter

wastez, 2024-05-10:
@canedje 
The backup is stored in home folder of z2m.
In case you use it as home assistant addon it is in /config/zigbee2mqtt.

Nerivec, 2024-05-14:
An old backup file prevents ember from starting to avoid overwriting stuff without your knowledge and gives the error you posted that the version is not supported. If you move it out of the folder, or rename/delete it, then ember should start.
_PS: A CRC error is common on start, it's just noise on the port, there is a retry mechanism just for that._

Please see https://www.zigbee2mqtt.io/guide/adapters/emberznet.html#logging-specific-to-this-stack for ember-specific logging details.

closed 2024-05-14 by Nerivec