Koenkk/zigbee2mqtt #22584: Crash - DatabaseEntry with ID already exists

Nezz, 2024-05-10:
### What happened?

I'm trying to reset three Tradfri lights on the same chandelier. When I do this, Z2M crashes.
```
[2024-05-10 21:52:36] warning: 	z2m: Device '0x980c33fffef167d8' left the network
Error: DatabaseEntry with ID '19' already exists
    at Database.insert (/app/node_modules/zigbee-herdsman/src/controller/database.ts:38:19)
    at Device.undelete (/app/node_modules/zigbee-herdsman/src/controller/model/device.ts:439:25)
    at Controller.onDeviceJoined (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:573:20)
[21:52:56] INFO: Preparing to start...
[21:52:57] INFO: Socat not enabled
[21:53:02] INFO: Starting Zigbee2MQTT...
```

### What did you expect to happen?

Three lights to get paired, Z2M not to crash

### How to reproduce it (minimal and precise)

1. Attach three lights to the same chandelier (e.g. https://www.zigbee2mqtt.io/devices/LED2201G8.html or https://www.zigbee2mqtt.io/devices/LED1732G11.html)
2. Reset them using the wall switch using this method: https://www.youtube.com/watch?v=mJm9YpPrGzk

### Zigbee2MQTT version

1.37.1

### Adapter firmware version

221226

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus TI CC2652P + CP2102(N)

### Setup

Add-on on HAOS on x86-64

### Debug log

See logs above. Crash logs are only visible on HAOS logs.

Koenkk, 2024-05-12:
Pushed a fix, can you check if it works?

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

Nezz, 2024-10-19:
Pairing multiple lights at once works now 🎉

closed 2024-10-19 by Nezz