Koenkk/zigbee2mqtt #22589: Z2M not starting after update to 1.37.1-1

Aaroneisele55, 2024-05-11:
### What happened?

I did update the HA Addon to 1.37.1-1, then it ididn't start naymore.
I restored the backup I had (from the previous version of the addon), but it still doesn't start

### What did you expect to happen?

It should have started normally

### Zigbee2MQTT version

1.37.0-1

### Adapter firmware version

?, can't access Z2M interface

### Adapter

Sonoff ZBDONGLE-P

### Setup

Add-On on Home Assistant OS on a Raspberry Pi 4

### Debug log

[log.log](https://github.com/Koenkk/zigbee2mqtt/files/15281780/log.log)

MKMZ, 2024-05-11:
I have the same issue also on Sonoff ZBDONGLE-P. Logs say that Checksum check has failed.

vsetotje, 2024-05-11:
same here sonoff dongle stopped working

KrX3D, 2024-05-12:
exactly the same here

Aaroneisele55, 2024-05-12:
I have now switched back to ZHA because I can't depend on the unstable project Z2M is for me with constant breaking changes without **ANY** notifications or warnings...

jee67, 2024-05-13:
Same problem here, anyone found a solution?

vsetotje, 2024-05-13:
guess we need to wait for update . Strange thing is that i restored from backup, but it didn't help

Aaroneisele55, 2024-05-13:
> guess we need to wait for update . Strange thing is that i restored from backup, but it didn't help

I think that it corrupts some of the Z2M data files (/config/zigbee2mqtt/) so unless you have a backup of those, you can't undo it...

jee67, 2024-05-13:
I removed z2m completely and reinstalled the latest version. I had to reconnect almost all devices, but by using a backup of configuration.yaml they were recognized instantly.

stefangehn, 2024-05-13:
From the log it looks like the zigbee communication part works just fine.
However the configure mqtt broker does not seem to be reachable from z2m:

Relevant log Excerpt:
```
[2024-05-11 08:01:02] info: 	z2m: Connecting to MQTT server at mqtt://core-mosquitto:1883
[2024-05-11 08:01:02] debug: 	z2m: Using MQTT login with username: addons
[2024-05-11 08:01:06] error: 	z2m: MQTT error: getaddrinfo ENOTFOUND core-mosquitto
[2024-05-11 08:01:06] error: 	z2m: MQTT failed to connect, exiting...
```

z2m does not manage the MQTT broker at all so either the configure hostname is wrong (`ENOTFOUND core-mosquitto` more or less means that the hostname `core-mosquitto` could not be resolved into an IP address) or the host or container normally available under that name is currently not running.

I don't know who maintains that HA-Addon but as far as I know it is *not* part of z2m.

So TL;DR: looks like a setup issue or a bug in the addon code that configures the MQTT broker.

wastez, 2024-05-13:
> I have now switched back to ZHA because I can't depend on the unstable project Z2M is for me with constant breaking changes without **ANY** notifications or warnings...

I don’t see the problem. You don’t have to update and if something isn’t working as it should you can revert to the last version. 
The coders aren’t payed for there work, they do it in there free time and problems can appear if the code will be extended and in the moment there are a lot main changes in the core of z2m, so revert to the last version and wait a few weeks/month. 
But it’s up to you….

I’m happy that they are working hard to get the code to the latest state.

Pferdebockwurst, 2024-05-15:
This should be fixed in the development branch because of [this commit](https://github.com/Koenkk/zigbee2mqtt/commit/a0c9a1769f7e25673672e3a312b2e4bd3812ff32).

@Koenkk Can you confirm and maybe elaborate the problem? The only change I see is the dependency to mqtt, which you lowered to an even older version as 1.36.1 was using.

Koenkk, 2024-05-15:
@Pferdebockwurst something is broken in the latest MQTT library, still investigating this.

massimoa01, 2024-05-17:
Hi if your specify the mqtt protocol to Server configuration in mqtt section it works. I mean change server: x.y.w.z:1883 in server: mqtt://x.y.w.z:1883

Tonio16, 2024-05-17:
Hello

I have something similar. I just updated my sonoff dongle 3.0 P since I had some troubles with some bulbs: no update of state, error ZLC (or ZCL?) command.
Before to update my firmware, I was able to use zigbee2mqtt. Now it is not working. A firmware donwgrade did not help.
Here the log at start.
`> zigbee2mqtt@1.37.1 start
> node index.js

[2024-05-17 21:12:45] info:     z2m: Logging to console, file (filename: log.log)
[2024-05-17 21:12:46] info:     z2m: Starting Zigbee2MQTT version 1.37.1 (commit #cda867a3)
[2024-05-17 21:12:46] info:     z2m: Starting zigbee-herdsman (0.46.6)
[2024-05-17 21:12:46] info:     zh:zstack:znp: Opening SerialPort with {"path":"/dev/serial/by-id/usb-ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus_e622bd522f1aec118d0a76e5f01c6278-if00-port0","baudRate":115200,"rtscts":false,"autoOpen":false}
[2024-05-17 21:12:46] error:    z2m: Error while starting zigbee-herdsman
[2024-05-17 21:12:46] error:    z2m: Failed to start zigbee
[2024-05-17 21:12:46] error:    z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
[2024-05-17 21:12:46] error:    z2m: Exiting...
[2024-05-17 21:12:46] error:    z2m: Error: Error Resource temporarily unavailable Cannot lock port`

Antoine

Pferdebockwurst, 2024-05-17:
@Tonio16 I think your problem is not related to this issue. Some other process might be using the port of your coordinator. Maybe it's z2m already running (a second instance) or something else.

Tonio16, 2024-05-17:
> @Tonio16 I think your problem is not related to this issue. Some other process might be using the port of your coordinator. Maybe it's z2m already running (a second instance) or something else.
> 
> Is your coordinator connected at all? Maybe something went wrong when updating the firmware. What's the output of `ls -l /dev/serial/by-id`?

Hello

No, zigbee2mqtt is not running, frontend is not available.
My coordinator is connected,  the ls command I already tested it, output is OK. 

Antoine

Pferdebockwurst, 2024-05-17:
What's the output?

Tonio16, 2024-05-18:
> What's the output?

This one
> total 0
> lrwxrwxrwx 1 root root 13 May 17 20:06 usb-0658_0200-if00 -> ../../ttyACM0
> lrwxrwxrwx 1 root root 13 May 17 20:44 usb-ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus_e622bd522f1aec118d0a76e5f01c6278-if00-port0 -> ../../ttyUSB0

Pferdebockwurst, 2024-05-18:
Ok, this looks good. Have you tried flashing the stick again? Maybe something went wrong. Are you sure you took the right firmware image?

Tonio16, 2024-05-19:
BAsed on flash programmer 2, it's fine:
>Initiate access to target: COM3 using 2-pin cJTAG.
>Reading file: F:/bibliotheques/Mes documents/jeedom/sonoff/CC1352P2_CC2652P_launchpad_coordinator_20240315.hex.
>Start flash erase ...
>Erase finished successfully.
>Start flash programming ...
>Programming finished successfully.
>Start flash verify ...
>Skip verification of unassigned page: 22.
>Skip verification of unassigned page: 23.
>Skip verification of unassigned page: 24.
>Skip verification of unassigned page: 25.
>Skip verification of unassigned page: 26.
>Skip verification of unassigned page: 27.
>Skip verification of unassigned page: 28.
>Skip verification of unassigned page: 29.
>Skip verification of unassigned page: 30.
>Skip verification of unassigned page: 31.
>Skip verification of unassigned page: 32.
>Skip verification of unassigned page: 33.
>Skip verification of unassigned page: 34.
>Skip verification of unassigned page: 35.
>Skip verification of unassigned page: 36.
>Skip verification of unassigned page: 37.
>Skip verification of unassigned page: 38.
>Skip verification of unassigned page: 39.
>Skip verification of unassigned page: 40.
>Skip verification of unassigned page: 41.
>Skip verification of unassigned page: 42.
>Page: 0 verified OK.
>Page: 1 verified OK.
>Page: 2 verified OK.
>Page: 3 verified OK.
>Page: 4 verified OK.
>Page: 5 verified OK.
>Page: 6 verified OK.
>Page: 7 verified OK.
>Page: 8 verified OK.
>Page: 9 verified OK.
>Page: 10 verified OK.
>Page: 11 verified OK.
>Page: 12 verified OK.
>Page: 13 verified OK.
>Page: 14 verified OK.
>Page: 15 verified OK.
>Page: 16 verified OK.
>Page: 17 verified OK.
>Page: 18 verified OK.
>Page: 19 verified OK.
>Page: 20 verified OK.
>Page: 21 verified OK.
>Page: 43 verified OK.
>Verification finished successfully.
>Reset target ...
>Reset of target successful.

lux73, 2024-05-22:
> Hi if your specify the mqtt protocol to Server configuration in mqtt section it works. I mean change server: x.y.w.z:1883 in server: mqtt://x.y.w.z:1883

this should be the default config entry...   don't know why people not follow the defaults:

[image]

if someone not follow the Specs, sooner or later it will be problematic ^^

jackyohh, 2024-05-24:
WORKAROUND:
i changed domain name into FQDN (mqtt:server --> server.myside.com)
Now it's working fine.

bedee80, 2024-06-01:
Had the same issue with 1.37.1, just updated to z2m 1.38, now all works fine again

Rosengren74, 2024-06-17:
Get the same error:
[2024-06-17 10:01:02] info: 	z2m: Connecting to MQTT server at mqtt://core-mosquitto:1883
[2024-06-17 10:01:02] debug: 	z2m: Using MQTT login with username: addons
[2024-06-17 10:01:06] error: 	z2m: MQTT error: getaddrinfo ENOTFOUND core-mosquitto
[2024-06-17 10:01:06] error: 	z2m: MQTT failed to connect, exiting...

But change the server adress to "server: mqtt://xxx.xxx.xxx.xxx/:1883" (change xxx to local ipadress), and then it´s work.

Thanks to @lux73

vsetotje, 2024-06-17:
> Get the same error: [2024-06-17 10:01:02] info: z2m: Connecting to MQTT server at mqtt://core-mosquitto:1883 [2024-06-17 10:01:02] debug: z2m: Using MQTT login with username: addons [2024-06-17 [...]

not sure regarding you specific case, but update solved everything for me and i didn't tinker with settings at all.

KalleKal, 2024-11-10:
> Get the same error: [2024-06-17 10:01:02] info: z2m: Connecting to MQTT server at mqtt://core-mosquitto:1883 [2024-06-17 10:01:02] debug: z2m: Using MQTT login with username: addons [2024-06-17 [...]

Had the same issue. On my side the mistake was, that for some reason the "autostart at boottime"-flag was disabled on my mosquitto MQTT broker. After starting it, the error at Z2M vanished.

tkarami, 2024-11-16:
> Hi if your specify the mqtt protocol to Server configuration in mqtt section it works. I mean change server: x.y.w.z:1883 in server: mqtt://x.y.w.z:1883

you just helped me solve my issue. thanks :)

closed 2025-01-23 by github-actions[bot]