Koenkk/zigbee2mqtt #22590: After a restart in the network zigbee2mqtt does not reconnect to the adapter

kikejherrera, 2024-05-11:
### What happened?

I have the SLZB-06M adapter connected via lan configured with ember. When there is a loss of connectivity because of a network reboot for example, even for a short time, the zigbee2mqtt addon of HomeAssistant disconnects, but the add-on does not stop. A manual restart of the add-on is necessary to retry the connection to the adapter. 

### What did you expect to happen?

Extend the time for zigbee2mqtt to disconnect if this is the normal operation, but at least periodically try the connection again or some automatic restart option.

### How to reproduce it (minimal and precise)

- The system is working
- I restart the network
- The addon is still running but the controller is disconnected and does not communicate.
- I stop the addon
- I start the addon

### Zigbee2MQTT version

1.37.1

### Adapter firmware version

7.4.3 [GA]

### Adapter

SLZB-06M

### Setup

Add-on on HomeAssistant in Intel NUC, HaOS system, connected via lan and remote adapter also connected via lan too.

### Debug log

[log.log](https://github.com/Koenkk/zigbee2mqtt/files/15283043/log.log)

RuneNyhuus, 2024-05-17:
I have the same issues on my slzb6m too..
If they lose Network connection for a few sec, Zigbee2mqtt(hass addon) just crashing and not retrying the connection to the adapter?
Is there a way to tell zigbee2mqtt just keep looking for the tcp address of the Zigbee adaptor?

dinhchinh82, 2024-06-02:
> I have the same issues on my slzb6m too.. If they lose Network connection for a few sec, Zigbee2mqtt(hass addon) just crashing and not retrying the connection to the adapter? Is there a way to tell zigbee2mqtt just keep looking for the tcp address of the Zigbee adaptor?

I got exactly same issue with the latest z2m 1.38.0 now. 
We hope it could be fixed soon. Thank you.

dinhchinh82, 2024-06-04:
@Nerivec @Koenkk Could you take a look on this issue? This error is very annoyed when the Zigbee coordinator connects via TCP and encounters problems, the Zigbee2mqtt addon does not automatically reconnect the TCP coordinator. It requires a manual restart of Zigbee2mqtt to function normally again.

Koenkk, 2024-06-04:
In the log I see that Z2M stops: `Adapter disconnected, stopping`, but you see that the addon itself doesn't stop?

Nerivec, 2024-06-04:
@kikejherrera The problem of the add-on not stopping when the adapter disconnects should be fixed in 1.38.0.

kikejherrera, 2024-06-04:
> In the log I see that Z2M stops: `Adapter disconnected, stopping`, but you see that the addon itself doesn't stop?

exactly

kikejherrera, 2024-06-04:
> @kikejherrera The problem of the add-on not stopping when the adapter disconnects should be fixed in 1.38.0.

Version 1.38.0-1 installed
I disconnected, waited a bit and reconnected. This is the log. The plugin remains active but the frontend is not working (502 bad gateway).

[log.log](https://github.com/user-attachments/files/15569426/log.log)

kikejherrera, 2024-06-04:
> @kikejherrera The problem of the add-on not stopping when the adapter disconnects should be fixed in 1.38.0.

I manually restart the plugin and this is the log. But if I don't notice because some device doesn't work, it can be like this for hours. When restarting manually there are devices that don't connect again, but I understand that this is a different problem.

[log.log](https://github.com/user-attachments/files/15569500/log.log)

Koenkk, 2024-06-04:
For some reason I don't see the `Adapter disconnected` in the log anymore?

dinhchinh82, 2024-06-05:
@Koenkk yes, in the log the last message is `Adapter disconnected, stopping` but the addon still in running state as below:

[image]
It seems the addon is in the freezed state and unable to reconnect to the adapter again.

Here is the log:
> Info 2024-06-05 07:36:20 Out of sequence : RX=0
Info 2024-06-05 07:36:20 ACK timeouts : RX=0
Info 2024-06-05 07:36:20======== ASH stopped ========
Error 2024-06-05 07:36:20Failed to init port with error Error: connect EHOSTUNREACH 192.168.86.27:8888
Error 2024-06-05 07:36:20Failed to reset and init NCP. Error: Failed to start EZSP layer with status=HOST_FATAL_ERROR.
Info 2024-06-05 07:36:20ASH COUNTERS since last clear:
Info 2024-06-05 07:36:20 Total frames: RX=0, TX=0
Info 2024-06-05 07:36:20 Cancelled : RX=0, TX=0
Info 2024-06-05 07:36:20 DATA frames : RX=0, TX=0
Info 2024-06-05 07:36:20 DATA bytes : RX=0, TX=0
Info 2024-06-05 07:36:20 Retry frames: RX=0, TX=0
Info 2024-06-05 07:36:20 ACK frames : RX=0, TX=0
Info 2024-06-05 07:36:20 NAK frames : RX=0, TX=0
Info 2024-06-05 07:36:20 nRdy frames : RX=0, TX=0
Info 2024-06-05 07:36:20 CRC errors : RX=0
Info 2024-06-05 07:36:20 Comm errors : RX=0
Info 2024-06-05 07:36:20 Length < minimum: RX=0
Info 2024-06-05 07:36:20 Length > maximum: RX=0
Info 2024-06-05 07:36:20 Bad controls : RX=0
Info 2024-06-05 07:36:20 Bad lengths : RX=0
Info 2024-06-05 07:36:20 Bad ACK numbers : RX=0
Info 2024-06-05 07:36:20 Out of buffers : RX=0
Info 2024-06-05 07:36:20 Retry dupes : RX=0
Info 2024-06-05 07:36:20 Out of sequence : RX=0
Info 2024-06-05 07:36:20 ACK timeouts : RX=0
Info 2024-06-05 07:36:20======== ASH stopped ========
Info 2024-06-05 07:36:20======== EZSP stopped ========
Info 2024-06-05 07:36:20======== Ember Adapter Stopped ========
Error 2024-06-05 07:36:20Adapter disconnected, stopping

dinhchinh82, 2024-06-05:
> And if you do a manual restart of the HA addon it starts working again? Can you try to disconnect another time to see if the issue always occurs?

@Koenkk I confirm that if restarting the z2m HA addon manually, it will be working again. I already tested this case about 10 times.

dinhchinh82, 2024-06-05:
@Koenkk @Nerivec I just set up a fresh environment of z2m 1.38.0-1 and everything work normally. The addon is able to reconnect the TCP adapter again. I will keep testing this case more carefully.

Thank you very much for your quick response.

> Info 2024-06-05 08:07:26======== ASH stopped ========
Info 2024-06-05 08:07:26======== EZSP stopped ========
Info 2024-06-05 08:07:26======== Ember Adapter Stopped ========
Error 2024-06-05 08:07:26Adapter disconnected, stopping
Info 2024-06-05 08:09:58MQTT publish: topic 'homeassistant/binary_sensor/122105103981011015010911311611650_0x28dba7fffe501035/connection_state/config', payload '{"device":{"hw_version":"EmberZNet 7.4.3 [GA]","identifiers":["zigbee2mqtt_bridge_0x28dba7fffe501035"],"manufacturer":"Zigbee2MQTT","model":"Bridge","name":"Zigbee2MQTT Bridge","sw_version":"1.38.0"},"device_class":"connectivity","entity_category":"diagnostic","name":"Connection state","object_id":"zigbee2mqtt_bridge_connection_state","origin":{"name":"Zigbee2MQTT","sw":"1.38.0","url":"https://www.zigbee2mqtt.io"},"payload_off":"offline","payload_on":"online","state_topic":"zigbee2mqtt2/bridge/state","unique_id":"bridge_0x28dba7fffe501035_connection_state_zigbee2mqtt2","value_template":"{{ value }}"}'

This is the connection sensor:
[image]

dinhchinh82, 2024-06-05:
@Koenkk @Nerivec after about 4 hours running from last success, the z2m addon got freezed again and unable to reconnect :(
Please check the screenshot and the log below:
[image]

[image]

> [2024-06-05 21:18:54] info: 	zh:ember: [NCP COUNTERS] 0,542,70,0,0,0,0,0,0,0,0,0,61,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0
[2024-06-05 21:18:54] info: 	zh:ember: [ASH COUNTERS] 451,98,48,49,0,0,0,0,462,50,49,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0
[2024-06-05 21:22:30] info: 	zh:ember:uart:ash: Port error: Error: read ETIMEDOUT
[2024-06-05 21:22:30] error: 	zh:ember: !!! NCP FATAL ERROR reason=ERROR_SERIAL_INIT. ATTEMPTING RESET... !!!
[2024-06-05 21:22:30] info: 	zh:ember:queue: Request dispatching stopped; queue=0 priorityQueue=0
[2024-06-05 21:22:30] info: 	zh:ember:uart:ash: ASH COUNTERS since last clear:
[2024-06-05 21:22:30] info: 	zh:ember:uart:ash:   Total frames: RX=0, TX=0
[2024-06-05 21:22:30] info: 	zh:ember:uart:ash:   Cancelled   : RX=0, TX=0
[2024-06-05 21:22:30] info: 	zh:ember:uart:ash:   DATA frames : RX=0, TX=0
[2024-06-05 21:22:30] info: 	zh:ember:uart:ash:   DATA bytes  : RX=0, TX=0
[2024-06-05 21:22:30] info: 	zh:ember:uart:ash:   Retry frames: RX=0, TX=0
[2024-06-05 21:22:30] info: 	zh:ember:uart:ash:   ACK frames  : RX=0, TX=0
[2024-06-05 21:22:30] info: 	zh:ember:uart:ash:   NAK frames  : RX=0, TX=0
[2024-06-05 21:22:30] info: 	zh:ember:uart:ash:   nRdy frames : RX=0, TX=0
[2024-06-05 21:22:30] info: 	zh:ember:uart:ash:   CRC errors      : RX=0
[2024-06-05 21:22:30] info: 	zh:ember:uart:ash:   Comm errors     : RX=0
[2024-06-05 21:22:30] info: 	zh:ember:uart:ash:   Length < minimum: RX=0
[2024-06-05 21:22:30] info: 	zh:ember:uart:ash:   Length > maximum: RX=0
[2024-06-05 21:22:30] info: 	zh:ember:uart:ash:   Bad controls    : RX=0
[2024-06-05 21:22:30] info: 	zh:ember:uart:ash:   Bad lengths     : RX=0
[2024-06-05 21:22:30] info: 	zh:ember:uart:ash:   Bad ACK numbers : RX=0
[2024-06-05 21:22:30] info: 	zh:ember:uart:ash:   Out of buffers  : RX=0
[2024-06-05 21:22:30] info: 	zh:ember:uart:ash:   Retry dupes     : RX=0
[2024-06-05 21:22:30] info: 	zh:ember:uart:ash:   Out of sequence : RX=0
[2024-06-05 21:22:30] info: 	zh:ember:uart:ash:   ACK timeouts    : RX=0
[2024-06-05 21:22:30] info: 	zh:ember:uart:ash: ======== ASH stopped ========
[2024-06-05 21:22:30] info: 	zh:ember:ezsp: ======== EZSP stopped ========
[2024-06-05 21:22:30] info: 	zh:ember: ======== Ember Adapter Stopped ========
[2024-06-05 21:22:30] info: 	zh:ember:uart:ash: Port closed. Error? true
[2024-06-05 21:22:30] info: 	zh:ember: ======== Ember Adapter Starting ========
[2024-06-05 21:22:30] info: 	zh:ember:ezsp: ======== EZSP starting ========
[2024-06-05 21:22:30] info: 	zh:ember:uart:ash: ======== ASH NCP reset ========
[2024-06-05 21:22:33] info: 	zh:ember:uart:ash: ASH COUNTERS since last clear:
[2024-06-05 21:22:33] info: 	zh:ember:uart:ash:   Total frames: RX=0, TX=0
[2024-06-05 21:22:33] info: 	zh:ember:uart:ash:   Cancelled   : RX=0, TX=0
[2024-06-05 21:22:33] info: 	zh:ember:uart:ash:   DATA frames : RX=0, TX=0
[2024-06-05 21:22:33] info: 	zh:ember:uart:ash:   DATA bytes  : RX=0, TX=0
[2024-06-05 21:22:33] info: 	zh:ember:uart:ash:   Retry frames: RX=0, TX=0
[2024-06-05 21:22:33] info: 	zh:ember:uart:ash:   ACK frames  : RX=0, TX=0
[2024-06-05 21:22:33] info: 	zh:ember:uart:ash:   NAK frames  : RX=0, TX=0
[2024-06-05 21:22:33] info: 	zh:ember:uart:ash:   nRdy frames : RX=0, TX=0
[2024-06-05 21:22:33] info: 	zh:ember:uart:ash:   CRC errors      : RX=0
[2024-06-05 21:22:33] info: 	zh:ember:uart:ash:   Comm errors     : RX=0
[2024-06-05 21:22:33] info: 	zh:ember:uart:ash:   Length < minimum: RX=0
[2024-06-05 21:22:33] info: 	zh:ember:uart:ash:   Length > maximum: RX=0
[2024-06-05 21:22:33] info: 	zh:ember:uart:ash:   Bad controls    : RX=0
[2024-06-05 21:22:33] info: 	zh:ember:uart:ash:   Bad lengths     : RX=0
[2024-06-05 21:22:33] info: 	zh:ember:uart:ash:   Bad ACK numbers : RX=0
[2024-06-05 21:22:33] info: 	zh:ember:uart:ash:   Out of buffers  : RX=0
[2024-06-05 21:22:33] info: 	zh:ember:uart:ash:   Retry dupes     : RX=0
[2024-06-05 21:22:33] info: 	zh:ember:uart:ash:   Out of sequence : RX=0
[2024-06-05 21:22:33] info: 	zh:ember:uart:ash:   ACK timeouts    : RX=0
[2024-06-05 21:22:33] info: 	zh:ember:uart:ash: ======== ASH stopped ========
[2024-06-05 21:22:33] error: 	zh:ember:uart:ash: Failed to init port with error Error: connect ENETUNREACH 192.168.86.27:8888
[2024-06-05 21:22:33] error: 	zh:ember: Failed to reset and init NCP. Error: Failed to start EZSP layer with status=HOST_FATAL_ERROR.

dinhchinh82, 2024-06-06:
After restarting the z2m addon manually to make it work, the problem come again after several hours. 
[log.log](https://github.com/user-attachments/files/15598275/log.log)

passbe, 2024-06-13:
I'm having a similar issue. 

Start zigbee2mqtt inside a container with configuration connecting to a SLZB-06 and another MQTT container. Everything comes up fine, devices check in, Home Assistant can see and report on everything. Then after about a minute of working it fails with the following:

```
[2024-06-13 03:20:30] info:     zh:zstack:znp: Socket error
[2024-06-13 03:20:30] info:     zh:zstack:znp: Port closed
[2024-06-13 03:20:30] debug:    zh:controller: Adapter disconnected
[2024-06-13 03:20:30] info:     zh:zstack:znp: closing
[2024-06-13 03:20:30] error:    z2m: Adapter disconnected, stopping
[2024-06-13 03:20:30] debug:    z2m: Saving state to file /app/data/state.json
[2024-06-13 03:20:30] info:     z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload '{"state":"offline"}'
[2024-06-13 03:20:30] info:     z2m:mqtt: Disconnecting from MQTT server
[2024-06-13 03:20:30] info:     z2m: Stopping zigbee-herdsman...
[2024-06-13 03:20:30] debug:    zh:controller:database: Writing database to '/app/data/database.db'
[2024-06-13 03:20:30] info:     z2m: Stopped zigbee-herdsman
[2024-06-13 03:20:30] info:     z2m: Stopped Zigbee2MQTT
```
Checked firewall logs and no other network issues at present. The logs inside the SLZB-06 just indicate a disconnection:
```
[13:19:31] taskZB | New client: 10.0.0.10 id: 0
[13:20:30] taskZB | Client disconnected, id: 0
```

My configuration:

```
homeassistant: true
permit_join: false
frontend: true
availability: true
mqtt:
  base_topic: zigbee2mqtt
  server: mqtt://mosquitto:1883
  user: XXX
  password: XXX
  include_device_information: true
serial:
  port: tcp://10.0.0.185:6638
advanced:
  transmit_power: 20
  channel: 25
  legacy_availability_payload: false
  last_seen: ISO_8601_local
  legacy_api: false
  homeassistant_legacy_entity_attributes: false
  log_level: debug
device_options:
  legacy: false
devices:
  --cut off--
```

Running latest version on SLZB-06 at version v2.3.6.

passbe, 2024-06-20:
After a lot of debugging I believe this is not a zigbee2mqtt issue in my case. It seems with an upgrade of podman (5.1.1) the new rootless networking layer (pasta 2024_06_07.8a83b53) is causing the issue as reverting back to slirp4netns the issue resolves itself. I've been unable to determine why this is occurring.

dinhchinh82, 2024-06-20:
> After a lot of debugging I believe this is not a zigbee2mqtt issue in my case. It seems with an upgrade of podman (5.1.1) the new rootless networking layer (pasta 2024_06_07.8a83b53) is causing the [...]

could you share the command to revert?

passbe, 2024-06-20:
I updated the line ```default_rootless_network_cmd = "slirp4netns"``` inside my ```/etc/containers/containers.conf``` file. Restart everything and the issue was resolved.

dinhchinh82, 2024-06-20:
> I updated the line `default_rootless_network_cmd = "slirp4netns"` inside my `/etc/containers/containers.conf` file. Restart everything and the issue was resolved.

@passbe So I guess you are using docker version now? I just confused how to fix it using HA z2m addons.

Kuechenfenster, 2024-07-17:
is there a update for HA users yet in how to solve this issue?
I run 2 06M sticks, one has the issue the other one not...

leroyloren, 2024-07-26:
HA addon: 1.39.0
Adapter: UZG-01 (XZG 20240707)

tonight the router automatically updated (thus disconnecting from the network) and thus the add-on did not reconnect.

in the morning it was necessary to press "Run" on the add-on and everything continues to work.

[log.txt](https://github.com/user-attachments/files/16388095/log.txt)

kalpik, 2024-07-26:
Yep, this happens to me as well, every time I reboot my router, Z2M crashes, and then I have to manually restart it.

ezfrag2021, 2024-08-07:
Same for me. After a power cut which caused the zigbee adaptor to restart, the Z2M addon crashes and does not restart and I have to press the start button in order to get it working again.

crysxd, 2024-08-26:
Same here, router restart causes z2m to stop and even after the adapter is available again it doesn't restart itself

ezfrag2021, 2024-08-26:
As a workaround until it's fixed I added an automation with the trigger as the state of connection of the zigbee adapter. When it changes from connected to disconnected the "start add-on" action fires to restart z2mqtt and repeats until the state changes back to connected.

crysxd, 2024-08-27:
@ezfrag2021 That's such an obvious solution. Quick manual for others:

- Go to "Devices & services" and create a new "Ping (ICMP)" device for the ethernet Zigbee adapter
- Create a new automation, when state of the ping device changes to "connected", start z2m

```
alias: WHEN slzb06 becomes available THEN start zigbee2mqtt
description: ""
trigger:
  - platform: state
    entity_id:
      - binary_sensor.slzb_06
    to: "on"
    for:
      hours: 0
      minutes: 0
      seconds: 15
condition: []
action:
  - action: hassio.addon_start
    metadata: {}
    data:
      addon: 45df7312_zigbee2mqtt
mode: single
```

ezfrag2021, 2024-08-27:
@crysxd That's a much more elegant solution than mine. And I never knew about how to create a Ping device before so you taught me something new. Thanks!

asagent7, 2024-09-03:
> Restart everything and the issue was resolved

@passbe what all do you recommend restarting?

sbrivio-rh, 2024-11-21:
This is now fixed in passt [2024_11_21.238c69f](https://archives.passt.top/passt-user/20241121164535.555fc41a@elisabeth/).

See also https://github.com/containers/podman/discussions/24572.

sidey79, 2024-12-03:
I have the same issue, after a short Network disruption , zigbee2mqtt doesn't recover.

I haven't understood the solution with the ping device. What device is this and how does it restart the process or docker Container?

sbrivio-rh, 2024-12-03:
> I have the same issue, after a short Network disruption , zigbee2mqtt doesn't recover.
> 
> I haven't understood the solution with the ping device.

If you're using Podman 5.x, chances are you simply need to update pasta(1) (package is called `passt`) to the latest version, because it contains a proper fix for that.

sidey79, 2024-12-04:
@sbrivio-rh 

I use portainer which is using docker-compose.
Why can this fixed from the host which is running the container solution?

sbrivio-rh, 2024-12-04:
> I use portainer which is using docker-compose.

Ah, no, sorry, then it's another issue. I was referring to the issue described at https://github.com/Koenkk/zigbee2mqtt/issues/22590#issuecomment-2179847056 and https://github.com/Koenkk/zigbee2mqtt/issues/22590#issuecomment-2477049374, but it looks like that's not the same as originally reported on this ticket.

dedors, 2024-12-12:
Not sure if the exact same problem, as I have a remote SLZB06 connected via Wireguard to my HA. I occasionally find the addon not running anymore, after there was connection loss.

sidey79, 2024-12-12:
@dedors
This ist the problem i have too.

vgisbert, 2024-12-21:
I have the same issue, after a short Network disruption, zigbee2mqtt doesn't recover.

jenshenk, 2025-01-04:
Great to have found this thread! I have yet another variation with this issue and it is very confusing. Form my experience I started suspecting that it is something in the mqtt autodiscovery by home assistant which is creating the issue. 

The setup: 
I'm running HA core (2024.12.5) as docker on one server (Ubuntu 22.04.5 LTS) and have a dedicated little android box running the latest zigbee2mqtt as docker (debian 12)

The issue in my case: 
- After fully restarting the server home assistant is running on, the issue shows, the zigbee2mqtt bridge and all devices are unavailable. However I also noticed something pretty interesting that points at potentially the mqtt integration from home assistant leading to the issue. (Maybe interesting info: Until earlier today I was still running ha 2024.7.4 and while the x2b bridge was being discovered, connected devices did not - upgrading to 2024.12.5 solved this, but also shows there might be something connected to the mqtt integration)   
- The interesting part: When the issue shows and I remove all z2m entries from the mqtt integration they eventually get repopulated but the z2m bridge still shows as disconnected
- If I fully restart the box with zigbee2mqtt the issue remains. I had it once where 2-3 restarts solved it, but this seems not really trustworthy
- What seems to always solve the problem is logging into the box with zigbee2mqtt and stopping and starting the z2m container - it's pretty confusing that this fixes it always but a full restart of the box does not

When it isn't happening: 
If I only restart the home assistant docker, restart home assistant via the gui or restart the z2m docker everything works as expected 

Hope this can be fixed since I wanted to fully move from zha to z2m now and this pretty much creates a road block :/

jenshenk, 2025-01-05:
A potential idea how to fix this, maybe it's stupid and z2m is already doing this but maybe it is useful: 
Due to all the server restarts I was just reminded about other devices that work via the mqtt integration having issues after a full server restart and also how I usually fix this. I have a bunch of tuya devices that I flashed with openbeken and some of them always reconnect without issues but a handful tend to stay unavailable, interestingly which show this issue is a bit random. I used to just restart them but found the more elegant way is to log in and manually trigger the home assistant discovery. Then after a few seconds they are shown as available again. Maybe this could be something z2m could do automatically, too, after loosing connection with the mqtt server?

dinhchinh82, 2025-03-08:
still waiting for the solution to fix this issue completely.

kalpik, 2025-03-11:
Yep, this is still an issue.

finpolar, 2025-03-19:
I have a bit same type of problem with slzb-06. The coordinator is connected over the network to a VPS with zigbee2mqtt, mosquitto and HA container in the Docker network. After restarting the VPS, Docker, or just the zigbee2mqtt container, the coordinator must be manually restarted every time, otherwise zigbee2mqtt cannot establish a connection to the coordinator and the container will not start.

GroteGehaktBal, 2025-04-04:
I have the same issue

ezfrag2021, 2025-08-17:
If you scroll up you will find a workaround which has worked fine for me.

[workaround](https://github.com/Koenkk/zigbee2mqtt/issues/22590#issuecomment-2311595131)

sidey79, 2025-08-17:
Loosing the network connection to the adapter, causes a crash at my installation:

```
2025-08-17T10:35:27.257035750Z [2025-08-17 12:35:27] debug: 	zh:deconz:driver: Opening TCP socket with 10.2.11.90:45108
2025-08-17T10:35:27.273829937Z [2025-08-17 12:35:27] error: 	zh:deconz:driver: Socket error Error: connect ECONNREFUSED 10.2.11.90:45108
2025-08-17T10:35:27.275473167Z [2025-08-17 12:35:27] debug: 	zh:deconz:driver: Error: Error while opening socket
2025-08-17T10:35:27.276500669Z [2025-08-17 12:35:27] info: 	zh:deconz:driver: Port close: state: undefined, reason: true
2025-08-17T10:35:27.328979198Z 
2025-08-17T10:35:27.329010400Z /app/node_modules/.pnpm/zigbee-herdsman@5.0.4/node_modules/zigbee-herdsman/src/adapter/deconz/driver/driver.ts:827
2025-08-17T10:35:27.329015932Z         this.emitStateEvent(DriverEvent.Disconnected);
2025-08-17T10:35:27.329019538Z              ^
2025-08-17T10:35:27.329022803Z TypeError: this.emitStateEvent is not a function
2025-08-17T10:35:27.329026351Z     at Socket.onPortClose (/app/node_modules/.pnpm/zigbee-herdsman@5.0.4/node_modules/zigbee-herdsman/src/adapter/deconz/driver/driver.ts:827:14)
2025-08-17T10:35:27.329042399Z     at Object.onceWrapper (node:events:633:26)
2025-08-17T10:35:27.329046064Z     at Socket.emit (node:events:518:28)
2025-08-17T10:35:27.329049103Z     at TCP.<anonymous> (node:net:351:12)
```

benedikt-bartscher, 2025-08-17:
Sadly the workaround only works when using z2m as HAOS Addon :/

Koenkk, 2025-09-17:
@JoshuaDodds for this we have the [watchdog feature](https://www.zigbee2mqtt.io/guide/installation/15_watchdog.html)

JoshuaDodds, 2025-09-22:
> [@JoshuaDodds](https://github.com/JoshuaDodds) for this we have the [watchdog feature](https://www.zigbee2mqtt.io/guide/installation/15_watchdog.html)

Very cool!  Thanks for pointing it out.  Works a charm!

JoshuaDodds, 2025-09-24:
> > I just set mine up so that it tries every minute to restart for 10 minutes before exiting.
> > Or do you want to see how I put that in a kubernetes manifest?
> 
> [@JoshuaDodds](https://github.com/JoshuaDodds) yes. That's great if you could share your actual configuration on docker file or z2m addon's configuration json file. Thank you.

```
    spec:
      hostNetwork: false
      hostname: zigbee2mqtt-cc2652p
      serviceAccountName: default
      dnsPolicy: ClusterFirst
      enableServiceLinks: true
      containers:
        - name: zigbee2mqtt-cc2652p
          image: "koenkk/zigbee2mqtt:1.41.0"
          imagePullPolicy: IfNotPresent
          securityContext:
            privileged: false
          env:
            - name: "TZ"
              value: "Europe/Amsterdam"
            - name: "ZIGBEE2MQTT_DATA"
              value: "/data"
            - name: Z2M_WATCHDOG
              value: "1,1,1,1,1,1,1,1,1,1"  # 10 retries, 1 minute apart
```

Gruntoneer, 2025-10-23:
[image]

Setup the following automation. That is tested and working.

user514, 2025-11-17:
As soon as my router reboots or I have a power outage, my smlight-06 coordinator connected to that router via ethernet, disconnects for 20 seconds, and that makes zigbee2mqtt STOP, and does not restart after connection with router and coordinator is reestablished. So, for those who are more confortable with yaml, and want to add a telegram, advice, etc... to know if the zigbee network is re-established, I use this workaround:

First, create a sensor that has the online or offline state of the zigbee2mqtt:

```
- name: "zigbee2mqtt_online_or_offline"
  state_topic: "zigbee2mqtt/bridge/state"
  value_template: "{{ value_json.state }}"
  icon: mdi:router-wireless
```

...this will create the sensor: 'sensor.zigbee2mqtt_online_or_offline' with a value of 'online' or 'offline', of course.

Then, an automation that check if the state changes or if 14 minutes has passed. (in case something fail, fot it to keep checking not only triggered by bridge state change):

```
- id: 'zigbee2mqtt_check_online_offline_6777273677112'
  alias: Zigbee2mqtt check online offline
  mode: single
  trigger:
    - platform: state
      entity_id: sensor.zigbee2mqtt_online_or_offline
    - platform: time_pattern 
      minutes: "/14"
  condition:
    - condition: template
      value_template: "{{ is_state('sensor.zigbee2mqtt_online_or_offline', 'offline') }}"
  action:
    ### TELEGRAMs:
    - service: telegram_bot.send_message
      data:
        target: 'XXXXXXXXX' ### (CHANGE THIS TO YOUR TELEGRAM ID IN CASE YOU WANT TO BE ADVISED)
        message: "Zigbee2MQTT has stopped at {{ now().strftime('%H:%M') }}. In 2 minutes I will try to restart it..."
    - delay: 00:02:00
    - service: hassio.addon_start
      data:
        addon: "45df7312_zigbee2mqtt"
```

...in the last line you could see the addon 'slug' which is the name of the container, etc. At least in my HAOS. You could check that name with the command line, but easiest is to check on your addon. Now it is called 'hostname', like that:

[image]

I have another automation that sends a telegram when the status of the sensor pass from offline to online. You will receive that, two minutes after zigbee network has gone down, telling that the connection is ok again. I'm not including it here, because is very similar to the previous automation. Just remove the 14 minutes trigger and the addon_start sections and in platform state add the usual 'from: offline' and a 'to: online' lines...

note: ...be careful with the underscore instead of a dash, at the slug name. A better way to be sure, is to enter this command at ssh terminal of your installation, you will see the correct slug name there:

`# ha addons list | grep zigbee2mqtt`