Koenkk/zigbee2mqtt #22594: zigbee2mqtt wants /dev/ttyUSB0 but port: tcp://xx.yy.zz.aa:6638 is configured

Hilbert70, 2024-05-11:
### What happened?

I moved to SLZB-06P7 over ethernet and removed the zigbee stick on /dev/ttyUSB0 from the configuration.
After removing the stick and rebooting, the docker container gives the error that it cannot find /dev/ttyUSB0

configuration.yml
```
homeassistant: true
permit_join: false
mqtt:
  base_topic: zigbee2mqtt
  server: mqtt://192.168.178.10
serial:
  port: tcp://192.168.178.19:6638
  baudrate: 115200
  adapter: zstack
  disable_led: false
advanced:
  transmit_power: 20
  rtscts: false
  channel: 25
  log_syslog:
    app_name: Zigbee2MQTT
    eol: /n
    host: 192.168.178.10
    localhost: localhost
    path: /dev/log
    pid: process.pid
    port: 123
    protocol: tcp4
    type: '5424'
  log_directory: /app/data/log/
  log_file: log-%TIMESTAMP%.txt
  log_level: warn
frontend:
  port: 8080
```

### What did you expect to happen?

I expected to happen that it starts with this configuration:
```
serial:
  port: tcp://192.168.178.19:6638
  baudrate: 115200
  adapter: zstack
  disable_led: false
```

### How to reproduce it (minimal and precise)

unplug the device on /dev/ttyUSB0 and it will not start, portainer gives error 500
docker start zigbee2mqtt returns with the error that it can not find /dev/ttyUSB0
plugging back the zigbee stick makes it working again
```27da098e5218099578173c9e55be1e34502cddab5f2af6d84cd42c2c46b695c9
AppArmorProfile
Args
0 /sbin/tini
1 --
2 node
3 index.js
Config
AttachStderr true
AttachStdin false
AttachStdout true
Cmd
0 /sbin/tini
1 --
2 node
3 index.js
Domainname
Entrypoint [ docker-entrypoint.sh ]
Env
0 TZ=Europe/Amsterdam
1 PATH=/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin
2 NODE_ENV=production
ExposedPorts
8080/tcp { }
Hostname 594466654c8e
Image koenkk/zigbee2mqtt:latest
Labels { }
OnBuild
OpenStdin false
StdinOnce false
Tty false
User
Volumes
/app/data
/run/udev
WorkingDir /app
Created 2024-04-29T21:52:04.22099825Z
Driver zfs
ExecIDs```

/app/data -> /opt/zigbee2mqtt/data
/run/udev -> /run/udev

### Zigbee2MQTT version

1.36.1

### Adapter firmware version

SLZB-06P7 Firmware version:	20240316	v2.1.5.dev

### Adapter

SLZB-06P7

### Setup

standalone docker container with zigbee2mqtt

### Debug log

`docker start zigbee2mqtt`
Error response from daemon: error gathering device information while adding custom device "/dev/ttyUSB0": no such file or directory
Error: failed to start containers: zigbee2mqtt

BeardedTinker, 2024-05-12:
Did you recreate the container after removing the USB stick without -d (--device) in docker or you just restarted it?
If you've just restarted it, docker is looking to map the stick as you've initially set that mapping.

Hilbert70, 2024-05-12:
That is a good one, completely overlooked that one!
```
Devices
0
CgroupPermissions rwm
PathInContainer /dev/ttyUSB0
PathOnHost /dev/ttyUSB0

```
But I found these.
At the "Runtime & Resources" tab.

Removed that entry and recreated and it is working! 

Thanks a lot!

Hilbert70, 2024-05-12:
At the "Runtime & Resources" tab in portainer.

Removed that entry and recreated and it is working!

Or removing the `-d` as @BeardedTinker said!

closed 2024-05-12 by Hilbert70