Koenkk/zigbee2mqtt #22607: Get log back to website as before

theDiverDK, 2024-05-12:
### What happened?

After the changes that no long show the log on the website, I can't figure out how to get it back :(

I have this in config file:

advanced:
  log_level: info
  log_debug_to_mqtt_frontend: true

I have tried 'log level: debug' as well.

But on the website, I only get log entries, if i select debug, but then I get to much info.

I would just like to have the same info as before, where I can see the events as they happen.

Anyone who can help?

Thanks  
  

### Zigbee2MQTT version

1.37.1 commit: ea39d86

### Adapter firmware version

Coordinator type zStack3x0 Coordinator revision 20230507

### Adapter

Sonoff Zigbee Sensor | Zigbee 3.0

### Setup

Zigbee2mqtt running on Pi4

LaurentChardin, 2024-05-12:
You might want to have a look here : https://github.com/Koenkk/zigbee2mqtt/issues/22522#issuecomment-2104619552

theDiverDK, 2024-05-13:
Thanks LaurentChardin

closed 2024-11-11 by theDiverDK