Koenkk/zigbee2mqtt #22609: Problems with 1.37.0 and 1.37.1

pedroke, 2024-05-13:
### What happened?

Hi guys,
I'm not sure if it's any kind of local issue. But since 1.37.0 I'm having a lot of issues with my network. It started as terrible delays after updating to 1.37.0 and some of my devices dropped the network and then when I updated to 1.37.1 I thought my issues were gone, but it worked only few days and now the network again is unstable. My routers were dropping connections and now when I tried to re-pair them, I'm not able. I'm getting:

Info 2024-05-13 08:47:21Interview for '0xccccccfffeeab05b' started
Info 2024-05-13 08:47:21Device '0xccccccfffeeab05b' joined
Info 2024-05-13 08:47:21Starting interview of '0xccccccfffeeab05b'
Error 2024-05-13 08:47:42Interview failed for '0xccccccfffeeab05b with error 'Error: Interview failed because can not get active endpoints ('0xccccccfffeeab05b')'
Error 2024-05-13 08:47:42Failed to interview '0xccccccfffeeab05b', device has not successfully been paired

In this specific case it is coming from PLUG-001SPB2..
I'm not sure what's going on, I tried restarting bot Rpi and z2m multiple times but except of old aqara devices which are directly paired to coordinator, everything else is having issues. E.g.:

Error 2024-05-13 08:52:09Default response to 0xa4c138f8118bf555 failed

Am I the only one which such terrible issues? Before update to 1.37.0 everything worked absolutely fine.

### Zigbee2MQTT version

1.37.1

### Adapter firmware version

zStack3x0 - 20230507

### Adapter

electrolama zig-a-zig

### Setup

Home assistant

### Debug log

Info 2024-05-13 08:47:21Interview for '0xccccccfffeeab05b' started
Info 2024-05-13 08:47:21Device '0xccccccfffeeab05b' joined
Info 2024-05-13 08:47:21Starting interview of '0xccccccfffeeab05b'
Error 2024-05-13 08:47:42Interview failed for '0xccccccfffeeab05b with error 'Error: Interview failed because can not get active endpoints ('0xccccccfffeeab05b')'
Error 2024-05-13 08:47:42Failed to interview '0xccccccfffeeab05b', device has not successfully been paired

angrycatmeowmeow, 2024-05-13:
Not just you #22452

pedroke, 2024-05-15:
I'm closing, because for some reason I managed to pair the routers again. I'm not sure what helped though.. maybe multiple restarts of coordinator, some network related changes or just luck.. but anyway.. routers are now back in the network. But still I have a feeling that it is less stable than before 1.37.0 .. although I cannot find any reason or explanation for that. Sorry.

closed 2024-05-15 by pedroke