Koenkk/zigbee2mqtt #22610: Ubisys D1 errors with `Cluster with name 'manuSpecificUbisysDeviceSetup' does not exist`

adriaandotcom, 2024-05-13:
### What happened?

When trying to set config with (`zigbee2mqtt/0x001fee0000008009/set`) this payload:

```json
{
    "configure_device_setup": {
        "input_action_templates": [
            {
                "//_comment": "will automatically use endpoint 1",
                "type": "dimmer_double"
            }
        ]
    }
}
```

I get:

```
[2024-05-13 06:36:52] info: 	z2m: Zigbee: allowing new devices to join.
[2024-05-13 06:36:55] info: 	zh:controller: Interview for '0x001fee0000008009' started
[2024-05-13 06:36:55] info: 	z2m: Device '0x001fee0000008009' joined
[2024-05-13 06:36:55] info: 	z2m: Starting interview of '0x001fee0000008009'
[2024-05-13 06:36:57] info: 	zh:controller: Succesfully interviewed '0x001fee0000008009'
[2024-05-13 06:36:57] info: 	z2m: Successfully interviewed '0x001fee0000008009', device has successfully been paired
[2024-05-13 06:36:57] info: 	z2m: Device '0x001fee0000008009' is supported, identified as: Ubisys Universal dimmer D1 (D1)
[2024-05-13 06:36:57] info: 	z2m: Configuring '0x001fee0000008009'
[2024-05-13 06:36:58] info: 	z2m: Successfully configured '0x001fee0000008009'
[2024-05-13 06:38:02] info: 	z2m: Zigbee: disabling joining new devices.
[2024-05-13 06:38:56] info: 	z2m: Changed config for device 0x001fee0000008009
[2024-05-13 06:38:58] info: 	z2m: Changed config for device 0x001fee0000008009
[2024-05-13 06:39:38] error: 	z2m: Publish 'set' 'configure_device_setup' to '0x001fee0000008009' failed: 'Error: Cluster with name 'manuSpecificUbisysDeviceSetup' does not exist'
[2024-05-13 06:39:59] info: 	z2m: Checking if update available for '0x001fee0000008009'
[2024-05-13 06:40:00] info: 	z2m: No update available for '0x001fee0000008009'
[2024-05-13 06:40:27] error: 	z2m: Publish 'set' 'configure_device_setup' to '0x001fee0000008009' failed: 'Error: Cluster with name 'manuSpecificUbisysDeviceSetup' does not exist'
[2024-05-13 06:44:40] error: 	z2m: Publish 'set' 'configure_device_setup' to '0x001fee0000008009' failed: 'Error: Cluster with name 'manuSpecificUbisysDeviceSetup' does not exist'
[2024-05-13 06:53:27] info: 	z2m: Successfully changed options
[2024-05-13 06:57:08] error: 	z2m: Publish 'set' 'configure_device_setup' to '0x001fee0000008009' failed: 'Error: Cluster with name 'manuSpecificUbisysDeviceSetup' does not exist'
```

It didn't happen for 2 other D1 devices I have.

### What did you expect to happen?

It would allow me to use `configure_device_setup` without erorrs.

### How to reproduce it (minimal and precise)

- Reset Ubisys D1
- Update to latest [version 2.4.0](https://www.ubisys.de/en/support/firmware/change-logs-d1/)
- Remove from z2m
- Reset Ubisys D1
- Adopt again
- Sent `configure_device_setup` command

### Zigbee2MQTT version

1.37.1

### Adapter firmware version

0x26580700

### Adapter

ConBee II

### Setup

Add-on on Home Assistant OS on Intel NUC

### Debug log

```
[2024-05-13 06:36:52] info: 	z2m: Zigbee: allowing new devices to join.
[2024-05-13 06:36:55] info: 	zh:controller: Interview for '0x001fee0000008009' started
[2024-05-13 06:36:55] info: 	z2m: Device '0x001fee0000008009' joined
[2024-05-13 06:36:55] info: 	z2m: Starting interview of '0x001fee0000008009'
[2024-05-13 06:36:57] info: 	zh:controller: Succesfully interviewed '0x001fee0000008009'
[2024-05-13 06:36:57] info: 	z2m: Successfully interviewed '0x001fee0000008009', device has successfully been paired
[2024-05-13 06:36:57] info: 	z2m: Device '0x001fee0000008009' is supported, identified as: Ubisys Universal dimmer D1 (D1)
[2024-05-13 06:36:57] info: 	z2m: Configuring '0x001fee0000008009'
[2024-05-13 06:36:58] info: 	z2m: Successfully configured '0x001fee0000008009'
[2024-05-13 06:38:02] info: 	z2m: Zigbee: disabling joining new devices.
[2024-05-13 06:38:56] info: 	z2m: Changed config for device 0x001fee0000008009
[2024-05-13 06:38:58] info: 	z2m: Changed config for device 0x001fee0000008009
[2024-05-13 06:39:38] error: 	z2m: Publish 'set' 'configure_device_setup' to '0x001fee0000008009' failed: 'Error: Cluster with name 'manuSpecificUbisysDeviceSetup' does not exist'
[2024-05-13 06:39:59] info: 	z2m: Checking if update available for '0x001fee0000008009'
[2024-05-13 06:40:00] info: 	z2m: No update available for '0x001fee0000008009'
[2024-05-13 06:40:27] error: 	z2m: Publish 'set' 'configure_device_setup' to '0x001fee0000008009' failed: 'Error: Cluster with name 'manuSpecificUbisysDeviceSetup' does not exist'
[2024-05-13 06:44:40] error: 	z2m: Publish 'set' 'configure_device_setup' to '0x001fee0000008009' failed: 'Error: Cluster with name 'manuSpecificUbisysDeviceSetup' does not exist'
[2024-05-13 06:53:27] info: 	z2m: Successfully changed options
[2024-05-13 06:57:08] error: 	z2m: Publish 'set' 'configure_device_setup' to '0x001fee0000008009' failed: 'Error: Cluster with name 'manuSpecificUbisysDeviceSetup' does not exist'
```

sjorge, 2024-05-13:
@Koenkk could this be the race again were the customClusters are not yet added ?

The D1 should have it, as it has the correct extend: https://github.com/Koenkk/zigbee-herdsman-converters/blob/f5ed0bfb5508a28efe507fff8d2578f39c0db4f1/src/devices/ubisys.ts#L832

Koenkk, 2024-05-14:
@sjorge I don't think so, it's happening even 20 secs after z2m startup.

adriaandotcom, 2024-05-17:
Can I help with more debugging info? I have ~30 Ubisys devices lying around :)

When running `generated_external_definition` I get:

```js
const {deviceEndpoints, identify, onOff, electricityMeter, commandsOnOff, commandsLevelCtrl, commandsColorCtrl} = require('zigbee-herdsman-converters/lib/modernExtend');

const definition = {
    zigbeeModel: ['D1 (5503)'],
    model: 'D1 (5503)',
    vendor: 'ubisys',
    description: 'Automatically generated definition',
    extend: [deviceEndpoints({"endpoints":{"1":1,"2":2,"3":3,"4":4,"232":232}}), identify(), onOff({"powerOnBehavior":false}), electricityMeter(), commandsOnOff({"endpointNames":["2","3"]}), commandsLevelCtrl({"endpointNames":["2","3"]}), commandsColorCtrl({"endpointNames":["2","3"]})],
    meta: {"multiEndpoint":true},
};

module.exports = definition;
```

Both `model` and `zigbeeModel` have the ` (5503)` in them, but I don't see that in [src/devices/ubisys.ts#L764](https://github.com/Koenkk/zigbee-herdsman-converters/blob/010319fc669fbdc2d7ef9ad3aa12ee4f3c540a0e/src/devices/ubisys.ts#L764). Not sure if relevant.

sjorge, 2024-05-17:
It matches on this line https://github.com/Koenkk/zigbee-herdsman-converters/blob/010319fc669fbdc2d7ef9ad3aa12ee4f3c540a0e/src/devices/ubisys.ts#L763

The D1 and R0 are the only devices I don't have and I can't replicate with my S1/S2/J1 which get the clusters extended the same way.

I'm not sure what is wrong, my guess would be the modernExtend somehow not adding the cluster.

@Koenkk is there and easy way to check on a running instance?

sjorge, 2024-05-17:
I'm able to replicate on latest dev, it was fine right after i merged the change though 🤷

I won't have time this weekend to dig further, but should have some time on monday.

I don't immediately see what would break it. I wonder if it somehow has to do with not setting a
manufacturer code for that cluster. The other ubisys ones do set it

sjorge, 2024-05-17:
@Koenkk I have a theory:

https://github.com/Koenkk/zigbee-herdsman-converters/blob/010319fc669fbdc2d7ef9ad3aa12ee4f3c540a0e/src/devices/ubisys.ts#L314

We pass `{}` as the 3rd argument, I believe it's calling:
https://github.com/Koenkk/zigbee-herdsman/blob/a2bede94b93fbbb35236d71aeeada1a8533268ba/src/zspec/zcl/utils.ts#L232-L236

So the 3rd argument would be `customClusters` ... so we're passing a empty list of customClusters, so that is why the call doesn't find it? (I'm not usual dev setup so I can't test this)

sjorge, 2024-05-18:
@adriaandotcom Seems my theory from yesterday was correct, https://github.com/Koenkk/zigbee-herdsman-converters/pull/7532 should fix this.

closed 2024-05-18 by Koenkk

adriaandotcom, 2024-05-20:
Amazing. Can I test with the dev branch?

sjorge, 2024-05-20:
It got merged into zhc, and there has been a zhc release, so I think it should be in dev already.