Koenkk/zigbee2mqtt #22611: Legrand Wireless Home Away 064873 not sending actions in Z2M

Loic691, 2024-05-13:
### What happened?

Legrand Home away not sending any action when pushing button.
The device is yet ok in Z2M and last seen is refreshed when I push button.
I have set a tiny group
[image]

### What did you expect to happen?

Have the actions information in Z2M and so on in HA

### How to reproduce it (minimal and precise)

just click on button

### Zigbee2MQTT version

1.37.1

### Adapter firmware version

7.4.1 [GA]

### Adapter

Sonoff E dongle

### Setup

Add-on on HA

### Debug log

No log are generated

Loic691, 2024-05-13:
ping @Nerivec to set the ember tag

Nerivec, 2024-05-13:
Please provide a debug log of the triggering of an action with ember so I can see what is going on.

Loic691, 2024-05-18:
Here's the log
My wireless device home away Legrand is 51090
```
[2024-05-18 14:35:48] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA]
[2024-05-18 14:35:48] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=3]
[2024-05-18 14:35:48] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=3 frmNum=5] Added to rxQueue
[2024-05-18 14:35:48] debug: 	zh:ember:uart:ash: ---> [FRAME type=ACK frmRx=6]
[2024-05-18 14:35:48] debug: 	zh:ember:ezsp: <=== [FRAME: ID=63:"MESSAGE_SENT_HANDLER" Seq=58 Len=22]
[2024-05-18 14:35:48] debug: 	zh:ember:ezsp: ezspMessageSentHandler(): callback called with: [type=MULTICAST], [indexOrDestination=65535], [apsFrame={"profileId":260,"clusterId":5,"sourceEndpoint":1,"destinationEndpoint":255,"options":0,"groupId":65526,"sequence":203}], [messageTag=255], [status=SUCCESS], [messageContents=]
[2024-05-18 14:35:48] debug: 	z2m: Retrieving state of 'Bouton Départ Arrivée' after reconnect
[2024-05-18 14:35:48] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA]
[2024-05-18 14:35:48] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=3]
[2024-05-18 14:35:48] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=3 frmNum=6] Added to rxQueue
[2024-05-18 14:35:48] debug: 	zh:ember:uart:ash: ---> [FRAME type=ACK frmRx=7]
[2024-05-18 14:35:48] debug: 	zh:ember:ezsp: <=== [FRAME: ID=63:"MESSAGE_SENT_HANDLER" Seq=58 Len=22]
[2024-05-18 14:35:48] debug: 	zh:ember:ezsp: ezspMessageSentHandler(): callback called with: [type=MULTICAST], [indexOrDestination=65535], [apsFrame={"profileId":260,"clusterId":5,"sourceEndpoint":1,"destinationEndpoint":255,"options":0,"groupId":65526,"sequence":204}], [messageTag=255], [status=SUCCESS], [messageContents=]
[2024-05-18 14:35:49] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA]
[2024-05-18 14:35:49] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=3]
[2024-05-18 14:35:49] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=3 frmNum=7] Added to rxQueue
[2024-05-18 14:35:49] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA]
[2024-05-18 14:35:49] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=3]
[2024-05-18 14:35:49] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=3 frmNum=0] Added to rxQueue
[2024-05-18 14:35:49] debug: 	zh:ember:uart:ash: ---> [FRAME type=ACK frmRx=1]
[2024-05-18 14:35:49] debug: 	zh:ember:ezsp: <=== [FRAME: ID=196:"INCOMING_NETWORK_STATUS_HANDLER" Seq=58 Len=8]
[2024-05-18 14:35:49] debug: 	zh:ember:ezsp: ezspIncomingNetworkStatusHandler(): callback called with: [errorCode=ROUTE_ERROR_SOURCE_ROUTE_FAILURE], [target=51090]
[2024-05-18 14:35:49] info: 	zh:ember:ezsp: Received network/route error ROUTE_ERROR_SOURCE_ROUTE_FAILURE for "51090".
[2024-05-18 14:35:49] debug: 	zh:ember:ezsp: <=== [FRAME: ID=128:"INCOMING_ROUTE_ERROR_HANDLER" Seq=58 Len=8]
[2024-05-18 14:35:49] debug: 	zh:ember:ezsp: ezspIncomingRouteErrorHandler(): callback called with: [status=SOURCE_ROUTE_FAILURE], [target=51090]
[2024-05-18 14:35:50] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA]
[2024-05-18 14:35:50] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=3]
[2024-05-18 14:35:50] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=3 frmNum=1] Added to rxQueue
[2024-05-18 14:35:50] debug: 	zh:ember:uart:ash: ---> [FRAME type=ACK frmRx=2]
[2024-05-18 14:35:50] debug: 	zh:ember:ezsp: <=== [FRAME: ID=63:"MESSAGE_SENT_HANDLER" Seq=58 Len=29]
[2024-05-18 14:35:50] debug: 	zh:ember:ezsp: ezspMessageSentHandler(): callback called with: [type=DIRECT], [indexOrDestination=51090], [apsFrame={"profileId":260,"clusterId":1,"sourceEndpoint":1,"destinationEndpoint":1,"options":4416,"groupId":0,"sequence":228}], [messageTag=5], [status=DELIVERY_FAILED], [messageContents=142110b7002000]
[2024-05-18 14:35:50] debug: 	zh:controller:requestqueue: Request Queue (0x0004740000ab283c/1): send failed, expires in 86294.516 seconds
[2024-05-18 14:35:50] debug: 	zh:ember: ~~~> [ZCL to=51090 apsFrame={"profileId":260,"clusterId":1,"sourceEndpoint":1,"destinationEndpoint":1,"options":4416,"groupId":0,"sequence":0} header={"frameControl":{"reservedBits":0,"frameType":0,"direction":0,"disableDefaultResponse":true,"manufacturerSpecific":true},"manufacturerCode":4129,"transactionSequenceNumber":185,"commandIdentifier":0}]
[2024-05-18 14:35:50] debug: 	zh:ember:ezsp: ===> [FRAME: ID=52:"SEND_UNICAST" Seq=59 Len=28]
[2024-05-18 14:35:50] debug: 	zh:ember:uart:ash: ---> [FRAME type=DATA frmTx=3 frmRx=2]
[2024-05-18 14:35:51] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA]
[2024-05-18 14:35:51] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=4]
[2024-05-18 14:35:51] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=4 frmNum=2] Added to rxQueue
[2024-05-18 14:35:51] debug: 	zh:ember:uart:ash: ---> [FRAME type=ACK frmRx=3]
[2024-05-18 14:35:51] debug: 	zh:ember:ezsp: <=== [FRAME: ID=52:"SEND_UNICAST" Seq=59 Len=7]
[2024-05-18 14:35:51] debug: 	zh:ember:ezsp: ~~~> [SENT type=DIRECT apsSequence=229 messageTag=6 status=SUCCESS]
[2024-05-18 14:35:51] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA]
[2024-05-18 14:35:51] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=4]
[2024-05-18 14:35:51] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=4 frmNum=3] Added to rxQueue
[2024-05-18 14:35:51] debug: 	zh:ember:uart:ash: ---> [FRAME type=ACK frmRx=4]
[2024-05-18 14:35:51] debug: 	zh:ember:ezsp: <=== [FRAME: ID=196:"INCOMING_NETWORK_STATUS_HANDLER" Seq=59 Len=8]
[... cut ...]
[2024-05-18 14:35:55] debug: 	zh:ember:ezsp: ezspMessageSentHandler(): callback called with: [type=DIRECT], [indexOrDestination=51090], [apsFrame={"profileId":260,"clusterId":1,"sourceEndpoint":1,"destinationEndpoint":1,"options":4416,"groupId":0,"sequence":229}], [messageTag=6], [status=DELIVERY_FAILED], [messageContents=142110b9002000]
[2024-05-18 14:35:55] debug: 	zh:controller:endpoint: Request Queue (0x0004740000ab283c/1): queue request (transaction failed)
[2024-05-18 14:35:55] debug: 	zh:controller:requestqueue: Request Queue (0x0004740000ab283c/1): Sending when active. Expires: 1716122146172
```

Loic691, 2024-05-21:
Here's a complete log for 12h since starting with firmware 7.4.3

My issues in 7.4.1 before updating to 7.4.1 :

- No action when pushed button on 064873 scene home away Legrand wireless
- OTA update fails on double gangs remote wireless 067774
- No polloing control on all these device and one of my simple gangs remote switch 067773 (other polling / wakeup is OK each hour)
- Cover remote switch works but with big latency randomly : 067647

Since yesterday and firmware update to 7.4.3 :
- Same but with loosing action on 067774

Loic691, 2024-05-21:
Update of my last post with multipled joined Legrand device

closed 2024-12-25 by github-actions[bot]