Koenkk/zigbee2mqtt #22612: Bosch Radiator thermostat II modes somehow broken after 1.37.1

ytugarev, 2024-05-13:
### What happened?

Hi,
I started to get 

> 2024-05-13 11:52:59.353 ERROR (MainThread) [homeassistant.components.mqtt.climate] Invalid modes mode: manual
2024-05-13 11:52:59.368 ERROR (MainThread) [homeassistant.components.mqtt.climate] Invalid modes mode: manual
2024-05-13 11:52:59.941 ERROR (MainThread) [homeassistant.components.mqtt.climate] Invalid modes mode: manual
2024-05-13 11:53:26.819 ERROR (MainThread) [homeassistant.components.mqtt.climate] Invalid modes mode: manual

after upgrading to  1.37.1

### What did you expect to happen?

Work as usual

### How to reproduce it (minimal and precise)

Every time I refresh  "System Mode" on my RBSH-TRV0-ZB-EU error appears in HA log
[image]

### Zigbee2MQTT version

1.37.1

### Adapter firmware version

20230507

### Adapter

CC2652P

### Setup

All Docker

### Debug log

[log.log](https://github.com/Koenkk/zigbee2mqtt/files/15292780/log.log)

myrrdin81, 2024-05-13:
Same here... I've gone back one version(Docker) and everything is fine again...[https://github.com/Koenkk/zigbee2mqtt/discussions/22585](url)

ytugarev, 2024-05-13:
Oh, I have not gotten cooling setpoint so far :) but looks really similar

Bkrickl, 2024-05-14:
I reverted back to 1.36.1-1 because i have the same issue.
My problem is that my Heating Automation (Advanced Heating Control) was not working anymore, showing the same error in the log when setting the occupied heating point.

ytugarev, 2024-05-15:
I have a script which alters "Remote Temp" by an event of other temp sensor change or every 15 min. Have not noticed any disruptions.

ytugarev, 2024-05-16:
@burmistrzak nope, same error with Zigbee2MQTT version
1.37.1-dev commit: [2963397](https://github.com/Koenkk/zigbee2mqtt/commit/2963397)

`2024-05-16 16:10:43.567 ERROR (MainThread) [homeassistant.components.mqtt.climate] Invalid modes mode: manual
2024-05-16 16:10:45.915 ERROR (MainThread) [homeassistant.components.mqtt.climate] Invalid modes mode: manual
2024-05-16 16:10:45.918 ERROR (MainThread) [homeassistant.components.mqtt.climate] Invalid modes mode: manual
2024-05-16 16:10:46.423 ERROR (MainThread) [homeassistant.components.mqtt.climate] Invalid modes mode: manual
2024-05-16 16:10:58.126 ERROR (MainThread) [homeassistant.components.mqtt.climate] Invalid modes mode: manual
2024-05-16 16:11:54.243 ERROR (MainThread) [homeassistant.components.mqtt.climate] Invalid modes mode: manual
2024-05-16 16:12:49.785 ERROR (MainThread) [homeassistant.components.mqtt.climate] Invalid modes mode: manual
2024-05-16 16:12:49.823 ERROR (MainThread) [homeassistant.components.mqtt.climate] Invalid modes mode: manual
2024-05-16 16:12:52.527 ERROR (MainThread) [homeassistant.components.mqtt.climate] Invalid modes mode: manual`

ytugarev, 2024-05-16:
@burmistrzak good news - after few initial errors ( I updated just container with z2m w/o HA reboot) there are no log errors anymore. I also restarted HA - seems to work ok. 
[image]

closed 2024-08-16 by ytugarev