Koenkk/zigbee2mqtt #22615: BTcino K4003C/L4003C/N4003C/NT4003C OTA doesn't work

raelix, 2024-05-13:
### What happened?

I tried to upgrade my `BTcino K4003C` devices but the OTA upgrade doesn't work it says `OTA: Device didn't respond to OTA request`.
I tried turning off the power and instantly sending the OTA command, I also tried to delete it from z2mqtt and send the OTA command once re-associated but nothing to do.

### What did you expect to happen?

The upgrade to run successfully.

### How to reproduce it (minimal and precise)

Just click on the update firmware and it doesn't work. 

### Zigbee2MQTT version

1.37.1

### Adapter firmware version

7.4.1

### Adapter

EZSP ember

### Setup

home-assistant add-on containerized

### Debug log

`Update of 'switch kitchen' failed (OTA: Device didn't respond to OTA request)`

langas6, 2024-05-14:
From first hand experience, unless you're missing something with the current setup, don't OTA update. I had a working K4003CWi, which was sending the correct "on" "off" commands from the 2 buttons on the switch. Now they both just send "toggle" as the action. I've been spending about 2 weeks off and on trying to get the switch to respond properly again, but nothing i've found will work.

It's not even showing in z2m as the correct device anymore. So if yours is showing as a K4003C, leave the firmware and enjoy a working unit.

Nerivec, 2024-05-15:
From what @langas6 is saying, it looks like this device may have breaking changes depending on the version. So, indeed, if yours works, I'd say don't update. As for the error, do you have the debug log file from around the time you tried to update?

@langas6 If you create an issue describing the problem with a debug log file attached, of you toggling the device every which way it can be (and isn't currently working), I'm sure someone from converters can help you figure out what changed and get it supported again eventually.

langas6, 2024-05-16:
> From what @langas6 is saying, it looks like this device may have breaking changes depending on the version. So, indeed, if yours works, I'd say don't update. As for the error, do you have the debug [...]

This would be fantastic, i'm away until monday next week, so will do some testing with debug logs and post up a new issue. Thank you for the response on this one. It's been causing me a lot of issues (doing testing to get the "wife factor" higher) and need it to work the correct way again.

Thanks so much for the reply!!

langas6, 2024-11-13:
Realised I've replied to the wrong thread here is the link to the reply:https://github.com/Koenkk/zigbee2mqtt/issues/22703#issuecomment-2333233308

raelix, 2025-01-13:
Well, the issue is still present. It could be a good idea to keep them updated like any other device

closed 2025-03-22 by github-actions[bot]