Koenkk/zigbee2mqtt #22616: IKEA Badring E2202 not be paired successfully

MiAutomations, 2024-05-13:
### What happened?

When I try to pair the sensor with zigbee2MQtt It appears as not supported, bit in the list of compatibility devices  this device is already apear as supported

### What did you expect to happen?

The device supported once is already in the list of devices supported 

### How to reproduce it (minimal and precise)

Try to add the device over the GUI of Z2M and this error still apear

### Zigbee2MQTT version

1.37.1 

### Adapter firmware version

 20221226

### Adapter

Sonoff Dongle Plus P

### Setup

Mini PC running HAOS

### Debug log

[image]

Mettchen, 2024-05-13:
Please add your herdsman debug logging too.

To enable herdsman debug logging, see https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging

Have you already tried multiple times and done a reboot?

MiAutomations, 2024-05-13:
> Please add your herdsman debug logging too.
> 
> To enable herdsman debug logging, see https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging
> 
> Have you already tried multiple times and done a reboot?

Hello, Yes I already try several time remove and try again pair the sensor without success also the restart to the hots and also to the Z2M and HA performed 

The log with debug level: 

[image]

Mettchen, 2024-05-13:
Please also add your zigbee herdsman logs - those are the z2m logs.

pedroke, 2024-05-13:
Seems to be the same root cause as my issue https://github.com/Koenkk/zigbee2mqtt/issues/22609
Also: "cannot get active endpoints"... something got very broken since 1.37.0... :(

MiAutomations, 2024-05-13:
> Seems to be the same root cause as my issue #22609 Also: "cannot get active endpoints"... something got very broken since 1.37.0... :(

Your are probably right, I restore the version zigbee2mqtt_1.36.1-1 and now the sensor is working well 

And by the way I'm also facing a several issues and unstable ZigBee network  1.37.0 also

MiAutomations, 2024-05-14:
> herdsman

Hello Sorry what you mean with herdsman logs?

MiAutomations, 2024-05-14:
By the way I'm now facing a issue that all my zigbee network is "missing" after the restore that I performed in order to try solve this issue with the IKEA Sensor 

[image]

I restarted the Zigbee addon and now I have again some connections but I'm missing a lot of connections between the devices

[image]

What can I do in order to try fix this ? 

Thank you in advance for your support

Mettchen, 2024-05-14:
@MiAutomations check my previous reply for zigbee herdsman logs. 

> To enable herdsman debug logging, see https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging

Does the connection get established if you try to toggle one of the mains powered devices that's currently not in the map? 
I don't know if the map is persistent through a reboot.

MiAutomations, 2024-05-14:
> @MiAutomations check my previous reply for zigbee herdsman logs.
>
>
> Does the connection get established if you try to toggle one of the mains powered devices that's currently not in the map? I don't [...]

Please let me know if is this kind of log that you are mentioned, by the way after the restart of the addon I'm facing the same issue all the devices are unlinked ... pass a hour or something like that I loosed all the connections between  the devices that are previously   established after the restart of the addon 

 

    at Endpoint.zclCommand (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:760:28)
Zigbee2MQTT:info  2024-05-14 14:28:13: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'state' to 'Luz Casa de Banho Pequena' failed: 'Error: ZCL command 0x680ae2fffe01d293/2 genOnOff.on({}, {\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":false,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (SRSP - AF - dataRequest after 6000ms)'","meta":{"friendly_name":"Luz Casa de Banho Pequena"},"type":"zigbee_publish_error"}'
Zigbee2MQTT:debug 2024-05-14 14:28:16: Received MQTT message on 'zigbee2mqtt/Casa de banho Grande /right/set' with data 'ON'
Zigbee2MQTT:debug 2024-05-14 14:28:16: Publishing 'set' 'state' to 'Casa de banho Grande '
Zigbee2MQTT:debug 2024-05-14 14:28:16: Received MQTT message on 'zigbee2mqtt/Casa de banho Grande /left/set' with data 'ON'
Zigbee2MQTT:debug 2024-05-14 14:28:16: Publishing 'set' 'state' to 'Casa de banho Grande '
Zigbee2MQTT:debug 2024-05-14 14:28:16: Received MQTT message on 'zigbee2mqtt/Casa de banho Grande /left/set' with data 'ON'
Zigbee2MQTT:debug 2024-05-14 14:28:16: Publishing 'set' 'state' to 'Casa de banho Grande '
Zigbee2MQTT:debug 2024-05-14 14:28:17: Received MQTT message on 'zigbee2mqtt/Casa de banho Grande /right/set' with data 'ON'
Zigbee2MQTT:debug 2024-05-14 14:28:17: Publishing 'set' 'state' to 'Casa de banho Grande '
Zigbee2MQTT:error 2024-05-14 14:28:19: Publish 'set' 'state' to 'Luz Casa de Banho Pequena' failed: 'Error: ZCL command 0x680ae2fffe01d293/1 genOnOff.on({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (SRSP - AF - dataRequest after 6000ms)'
Zigbee2MQTT:debug 2024-05-14 14:28:19: Error: ZCL command 0x680ae2fffe01d293/1 genOnOff.on({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (SRSP - AF - dataRequest after 6000ms)
    at Object.start (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:63:23)
    at /app/node_modules/zigbee-herdsman/src/adapter/z-stack/znp/znp.ts:312:45
    at Queue.execute (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:35:26)
    at Znp.request (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/znp/znp.ts:300:27)
    at ZStackAdapter.dataRequest (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:1005:24)
    at ZStackAdapter.sendZclFrameToEndpointInternal (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:387:46)
    at /app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:357:25
    at Queue.execute (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:35:26)
    at Request.send (/app/node_modules/zigbee-herdsman/src/controller/helpers/request.ts:79:20)
    at Endpoint.zclCommand (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:760:28)
Zigbee2MQTT:info  2024-05-14 14:28:19: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'state' to 'Luz Casa de Banho Pequena' failed: 'Error: ZCL command 0x680ae2fffe01d293/1 genOnOff.on({}, {\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":false,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (SRSP - AF - dataRequest after 6000ms)'","meta":{"friendly_name":"Luz Casa de Banho Pequena"},"type":"zigbee_publish_error"}'
Zigbee2MQTT:error 2024-05-14 14:28:25: Publish 'set' 'state' to 'Casa de banho Grande ' failed: 'Error: ZCL command 0x54ef4410000f4e5a/2 genOnOff.on({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (SRSP - AF - dataRequest after 6000ms)'
Zigbee2MQTT:debug 2024-05-14 14:28:25: Error: ZCL command 0x54ef4410000f4e5a/2 genOnOff.on({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (SRSP - AF - dataRequest after 6000ms)
    at Object.start (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:63:23)
    at /app/node_modules/zigbee-herdsman/src/adapter/z-stack/znp/znp.ts:312:45
[... cut ...]
Zigbee2MQTT:info  2024-05-14 14:29:01: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'state' to 'Luz Casa de Banho Pequena' failed: 'Error: ZCL command 0x680ae2fffe01d293/2 genOnOff.on({}, {\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":false,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (SRSP - AF - dataRequest after 6000ms)'","meta":{"friendly_name":"Luz Casa de Banho Pequena"},"type":"zigbee_publish_error"}'

closed 2024-05-15 by MiAutomations

reesechris, 2024-07-04:
Same issue here with 1.39.0-1. Any recommendations/updates on this? :/

### Zigbee2MQTT version
1.39.0-1

### Adapter
Sonoff Dongle Plus P

### Debug log
[image]

michamichamicha, 2025-05-03:
Same here
Zigbee2MQTT version 2.3.0

michamichamicha, 2025-05-06:
Found a solution: after engaging pairing mode on both z2m & the device (pressing the link button 4 times within 5 seconds) keep triggering the moisture detection (place wet finger over de contact) and press the link button every 2 seconds.