Koenkk/zigbee2mqtt #22618: ZG2819S-RGBW new device support

chris-1243, 2024-05-13:
### Link

https://www.sunricher.com/4-groups-rgb-zigbee-remote-sr-zg2819s-rgb.html

### Database entry

{"id":19,"type":"EndDevice","ieeeAddr":"0x000d6f001668b62f","nwkAddr":34591,"manufId":4644,"manufName":"Sunricher","powerSource":"Battery","modelId":"ZG2819S-RGBW","epList":[1,2,3,4],"endpoints":{"1":{"profId":260,"epId":1,"devId":261,"inClusterList":[0,1,3,2821,4096],"outClusterList":[3,4,5,6,8,25,768,4096],"clusters":{"genPowerCfg":{"attributes":{"batteryPercentageRemaining":100}},"genBasic":{"attributes":{"appVersion":0,"stackVersion":0,"hwVersion":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x0017880109aeb2b3","endpointID":11},{"cluster":8,"type":"endpoint","deviceIeeeAddress":"0x0017880109aeb2b3","endpointID":11},{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b0026d15e0e","endpointID":1}],"configuredReportings":[],"meta":{}},"2":{"profId":260,"epId":2,"devId":261,"inClusterList":[0,1,3,2821,4096],"outClusterList":[3,4,5,6,8,25,768,4096],"clusters":{"genBasic":{"attributes":{"appVersion":0,"stackVersion":0,"hwVersion":0}},"genPowerCfg":{"attributes":{"batteryPercentageRemaining":100}}},"binds":[],"configuredReportings":[],"meta":{}},"3":{"profId":260,"epId":3,"devId":261,"inClusterList":[0,1,3,2821,4096],"outClusterList":[3,4,5,6,8,25,768,4096],"clusters":{"genBasic":{"attributes":{"appVersion":0,"stackVersion":0,"hwVersion":0}},"genPowerCfg":{"attributes":{"batteryPercentageRemaining":100}}},"binds":[],"configuredReportings":[],"meta":{}},"4":{"profId":260,"epId":4,"devId":261,"inClusterList":[0,1,3,2821,4096],"outClusterList":[3,4,5,6,8,25,768,4096],"clusters":{"genBasic":{"attributes":{"appVersion":0,"stackVersion":0,"hwVersion":0}},"genPowerCfg":{"attributes":{"batteryPercentageRemaining":100}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":0,"stackVersion":0,"hwVersion":0,"dateCode":"NULL","swBuildId":"2.2.3_r12","zclVersion":2,"interviewCompleted":true,"meta":{"configured":1709916092},"lastSeen":1715615954838}

### Comments

I have made an external converter and I have a picture as well. I am just stuck how to make the pull request. Any help would be really appreciated.
 
[image]

### External definition

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
//const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    fingerprint: [{softwareBuildID: '2.2.3_r12'}],
    zigbeeModel: ['ZG2819S-RGBW'],
    model: 'ZB-5028',
    vendor: 'Sunricher',
    description: 'Zigbee handheld remote RGBW 4 channels',
    fromZigbee: [fz.battery, fz.command_on, fz.command_off, fz.command_step, fz.command_move, fz.command_stop, fz.command_recall,
        fz.command_move_hue, fz.command_move_to_color, fz.command_move_to_color_temp],
    exposes: [e.battery(), e.action(['on', 'off', 'brightness_step_up', 'brightness_step_down', 'brightness_move_up',
        'brightness_move_down', 'brightness_stop', 'recall_*', 'hue_move', 'color_temperature_move',
        'color_move', 'hue_stop'])],
    toZigbee: [],
    meta: {multiEndpoint: true},
    endpoint: (device) => {    
        return {ep1: 1, ep2: 2, ep3: 3, ep4: 4};
        },
};

module.exports = definition;
```

closed 2024-06-25 by chris-1243

Mertoid78, 2025-02-02:
I can't connect the device to a zigbee hub. I hold the on button, the LED lights up, I tap s1, and nothing

chris-1243, 2025-02-03:
@Mertoid78 Please check the user manual provided below

[Sunricher - SR-ZG2819S-RGB_UG.pdf](https://github.com/user-attachments/files/18639842/Sunricher.-.SR-ZG2819S-RGB_UG.pdf)

Mertoid78, 2025-02-03:
Thanks for the answer! I tried this instructions. When I try to reset the remote (I press the power button, wait for the LED to light up, then press power 5 times), nothing happens. 
Also, if you hold down the power button and then S1, nothing happens either (((

chris-1243, 2025-02-04:
I am sorry. More than the user manual says I can not provide... I have two of them and the pairing or removal is perfectly working. You have to strictly follow the how to pair / factory reset section.

Mertoid78, 2025-02-04:
I do everything according to the instructions! And I’ve already tried it more than once!!! There are suspicions that the remote control is either buggy or the Chinese have messed something up o.O
Is there a sticker with a QR code on the battery compartment cover?

chris-1243, 2025-02-04:
Mine are 2 years old. Your QR code looks more a production date. Did you buy it as a Sunricher, Iluminize or RGB-Genie? Or any other brands ?

Mertoid78, 2025-02-04:
I bought it as SR-ZG2819S. But it arrived without packaging

chris-1243, 2025-02-04:
Either the device is not working properly or you may miss something while trying to remove / pair it

Mertoid78, 2025-02-04:
On all devices (Sunricher, Iluminize or RGB-Genie), the reset combination is the same... Well, I've already tried resetting or connecting to zigbee 100,500 times. I couldn’t manage to put the device into such a mode that the LED would blink. Yes lights up when you press the All on/off button for a long time. And that's all ((

chris-1243, 2025-02-04:
Try to remove the batteries for let's say a minute. Then press and hold the main on/off button until the led turn on. Release it and press it 5 times. You should see the led blinking 3 times quickly which means it has been successfully reset... More than this, I can not help more.

Yes, one more thing. try to stay calm....

chris-1243, 2025-02-04:
I guess your device is not working as it should. I never had a problem with my 2 remotes...
This might happen unfortunately. You are welcome even I was not able to help you more