Koenkk/zigbee2mqtt #22623: [Bug] Osram Sylvania Lightify LED strips - major errors and not working on some extends.

Ltek, 2024-05-14:
### What happened?

I'm seeing major problems with this deice on Z2M. These work 100% perfectly with Deconz... I have 10 [Osram Sylvania Lightify LED strips ](https://www.zigbee2mqtt.io/devices/4052899926110.html) running on Conbee/Deconz, for many years... including all the built-in features (Blink, breath, etc).

Pairing with Z2M seems fine, no errors. 

Map doesn't show any line/link even with a LQI changes between 64 and 72, which is odd since a Bosch motion sensor )battery device) is literally on the other side of stucco exterior wall (!) and is over 200 LQI.

None of the built-in features work, at all. Only the basics like color, brightness, on/off.

[image](https://github.com/Koenkk/zigbee2mqtt/assets/1626802/cd8af08a-24d6-4cf7-a7b8-b7aff51b6608)

First noticed this...

Publish 'set' 'effect' to 'Basement Office RGB' failed: 'Error: ZCL command 0x841826000009f43b/3 genIdentify.triggerEffect({"effectid":1,"effectvariant":0}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Delivery failed for {"profileId":260,"clusterId":3,"sourceEndpoint":1,"destinationEndpoint":3,"options":4160,"groupId":0,"sequence":186})'

then I tried more properties...

`5/13/2024 8:11:23 PM
Publish 'set' 'color' to 'Basement Office RGB' failed: 'Error: ZCL command 0x841826000009f43b/3 lightingColorCtrl.moveToColor({"transtime":0,"colorx":11534,"colory":48496}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Delivery failed for {"profileId":260,"clusterId":768,"sourceEndpoint":1,"destinationEndpoint":3,"options":4416,"groupId":0,"sequence":243})'

5/13/2024 8:11:18 PM
Publish 'set' 'state' to 'Basement Office RGB' failed: 'Error: ZCL command 0x841826000009f43b/3 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Delivery failed for {"profileId":260,"clusterId":6,"sourceEndpoint":1,"destinationEndpoint":3,"options":4416,"groupId":0,"sequence":242})'

5/13/2024 8:11:13 PM
Publish 'set' 'brightness' to 'Basement Office RGB' failed: 'Error: ZCL command 0x841826000009f43b/3 genLevelCtrl.moveToLevelWithOnOff({"level":254,"transtime":0}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Delivery failed for {"profileId":260,"clusterId":8,"sourceEndpoint":1,"destinationEndpoint":3,"options":4416,"groupId":0,"sequence":241})'

5/13/2024 8:11:08 PM
Publish 'set' 'brightness' to 'Basement Office RGB' failed: 'Error: ZCL command 0x841826000009f43b/3 genLevelCtrl.moveToLevelWithOnOff({"level":5,"transtime":0}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Delivery failed for {"profileId":260,"clusterId":8,"sourceEndpoint":1,"destinationEndpoint":3,"options":4416,"groupId":0,"sequence":240})'

5/13/2024 8:11:03 PM
Publish 'set' 'state' to 'Basement Office RGB' failed: 'Error: ZCL command 0x841826000009f43b/3 genOnOff.on({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Delivery failed for {"profileId":260,"clusterId":6,"sourceEndpoint":1,"destinationEndpoint":3,"options":4416,"groupId":0,"sequence":239})'`

### Zigbee2MQTT version

1.37

### Adapter firmware version

EmberZNet Coordinator revision 7.4.3 [GA]

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus-E

### Setup

Homeseer on Windows

### Debug log

[2024-05-13.10-13-38.zip](https://github.com/Koenkk/zigbee2mqtt/files/15301811/2024-05-13.10-13-38.zip)

Flop2006, 2024-05-18:
In my case with z2m v1.37.1-1, the OSRAM 4052899926110 works via the zigbee2mqtt interface without any problems, but in homeassistant only the entity ‘update’ is shown as available, the light identity is no longer available.

Ltek, 2024-11-17:
bumpers

closed 2025-01-25 by github-actions[bot]