Koenkk/zigbee2mqtt #22624: z2m: Failed to check if update available

hwzolin, 2024-05-14:
### What happened?

[image]
As shown in the image above, when I check update the device, The system gets an exception: "Cannot destructure property 'data' of '(intermediate value)' as it is undefined."

### What did you expect to happen?

The device can be OTA normally.

### How to reproduce it (minimal and precise)

I don't know what's causing it, but this problem comes up every time I update the firmware for the device. The screenshot below is the log of the related exception displayed by zigbee2mqtt.

[image]

### Zigbee2MQTT version

1.37.1-1

### Adapter firmware version

20210708

### Adapter

Sonoff ZBDongle-P

### Setup

Runs as an Add-on on Home Assistant on VMware 17 pro

### Debug log

{"frameType":1,"manufacturerSpecific":false,"direction":0,"disableDefaultResponse":false,"reservedBits":0},"manufacturerCode":null,"transactionSequenceNumber":63,"commandIdentifier":1},"payload":{"fieldControl":0,"manufacturerCode":4659,"imageType":54183,"fileVersion":71},"command":{"ID":1,"response":2,"parameters":[{"name":"fieldControl","type":32},{"name":"manufacturerCode","type":33},{"name":"imageType","type":33},{"name":"fileVersion","type":35}],"name":"queryNextImageRequest"}}
[2024-05-14 11:41:08] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2024-05-14 11:41:08] debug: 	z2m: Received Zigbee message from '0x282c02bfffea0c14', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":71,"imageType":54183,"manufacturerCode":4659}' from endpoint 1 with groupID 0
[2024-05-14 11:41:08] debug: 	zhc:ota:common: Got request '{"fieldControl":0,"manufacturerCode":4659,"imageType":54183,"fileVersion":71}'
[2024-05-14 11:41:08] debug: 	zhc:ota:common: Is new image available for '0x282c02bfffea0c14' (3RSB015BZ), current '{"fieldControl":0,"manufacturerCode":4659,"imageType":54183,"fileVersion":71}'
[2024-05-14 11:41:08] debug: 	zhc:ota: Getting image metadata for '3RSB015BZ'
[2024-05-14 11:41:08] debug: 	z2m: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/check', payload '{"data":{"id":"0x282c02bfffea0c14"},"error":"Failed to check if update available for '0x282c02bfffea0c14' (Cannot destructure property 'data' of '(intermediate value)' as it is undefined.)","status":"error","transaction":"93bvw-7"}'
[2024-05-14 11:41:08] error: 	z2m: Failed to check if update available for '0x282c02bfffea0c14' (Cannot destructure property 'data' of '(intermediate value)' as it is undefined.)
[2024-05-14 11:41:08] debug: 	z2m: TypeError: Cannot destructure property 'data' of '(intermediate value)' as it is undefined.
    at getIndex (/app/node_modules/zigbee-herdsman-converters/src/lib/ota/zigbeeOTA.ts:42:18)
    at processTicksAndRejections (node:internal/process/task_queues:95:5)
    at getImageMeta (/app/node_modules/zigbee-herdsman-converters/src/lib/ota/zigbeeOTA.ts:61:20)
    at Object.isNewImageAvailable (/app/node_modules/zigbee-herdsman-converters/src/lib/ota/common.ts:406:18)
    at Object.isUpdateAvailable (/app/node_modules/zigbee-herdsman-converters/src/lib/ota/common.ts:395:29)
    at OTAUpdate.onMQTTMessage (/app/lib/extension/otaUpdate.ts:209:45)
    at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:174:17)

hwzolin, 2024-05-17:
@Koenkk Can you help me?

majorsl, 2024-05-18:
I'm having this problem with those blinds as well.  OTA reported updates available 1.00.70 -> 1.00.72.

I was able to update 3 of my 11 blinds.  For the others, I try to do a check and get the same error as you.  At one point they were all reporting an update available.

I don't know if Third Reality pulled the update or if something else happened.  When I have some time, I'm going to pull the batteries out of one of them and see if a reset fixed it.  I have try re-pairing and that did not help.

cpettis, 2024-05-18:
I didn’t have the no OTA problem, but after the upgrade the blinds were randomly dropping off the network and were difficult to get to reconnect. I did a local OTA with my phone following the procedure in the video below. It flashed them all back to 1.00.70 and they were all able to rejoin the network. I also reported the issue with 1.00.72 to Third Reality.

https://m.youtube.com/watch?v=kXoXMJdyf78

majorsl, 2024-05-18:
> I didn’t have the no OTA problem, but after the upgrade the blinds were randomly dropping off the network and were difficult to get to reconnect. I did a local OTA with my phone following the [...]

Ooooooo the ones I did get updated are having this kind of issue. I haven't tried what you're suggesting yet, but from the video it seems to only look for the latest version.  Are you able to select a previous firmware version somewhere?

cpettis, 2024-05-18:
No you can’t select the firmware version, but 1.00.70 is the version the app installed.

majorsl, 2024-05-19:
Thanks!  Got 1.00.70 reinstalled and all my problems vanished.  OP, it's probably a good thing you couldn't update to 1.00.72.  Even with the few I did, it seems to really have messed up my network such that some other devices I think were having issues.

3reality-support, 2024-05-20:
> I didn’t have the no OTA problem, but after the upgrade the blinds were randomly dropping off the network and were difficult to get to reconnect. I did a local OTA with my phone following the [...]

Excuse me, after the curtain is offline, do you operate the button on the curtain, or wait for a while, but you can't get back online?

hwzolin, 2024-05-20:
@Koenkk My problem is not limited to curtain, other devices also have this abnormality. I tried the water leak sensor and plug, and both prompted "Cannot destructure property 'data' of '(intermediate value)' as it is undefined."

Slomo17, 2024-05-30:
I am also having the same problem with the Bosch RBSH-TRV0-ZB-EU. It showed, that there is a new firmware availabe, which i tried to install. It failed several times. Now it just shows the error when i try to check for new firmware version.

42Craft, 2024-06-09:
I have the same problem with all my Hue bulbs.

liubotongs, 2024-06-19:
I seem to have the same problem here, but the local path configuration can be OTA normally.
	
Zigbee2MQTT version
1.37.1
	
Debug log	
[2024-06-19 13:31:00] debug: 	z2m: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/check', payload '{"data":{"id":"0x842712fffe16f184"},"error":"Failed to check if update available for '0x842712fffe16f184' (localIndex.concat is not a function)","status":"error","transaction":"jy290-3"}'
[2024-06-19 13:31:00] error: 	z2m: Failed to check if update available for '0x842712fffe16f184' (localIndex.concat is not a function)
[2024-06-19 13:31:00] debug: 	z2m: TypeError: localIndex.concat is not a function
    at getIndex (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/src/lib/ota/zigbeeOTA.ts:54:27)
    at processTicksAndRejections (node:internal/process/task_queues:95:5)
    at getImageMeta (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/src/lib/ota/zigbeeOTA.ts:61:20)
    at Object.isNewImageAvailable (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/src/lib/ota/common.ts:406:18)
    at Object.isUpdateAvailable (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/src/lib/ota/common.ts:395:29)
    at OTAUpdate.onMQTTMessage (/opt/zigbee2mqtt/lib/extension/otaUpdate.ts:209:45)
    at EventEmitter.wrappedCallback (/opt/zigbee2mqtt/lib/eventBus.ts:174:17)

majorsl, 2024-07-21:
> Excuse me, after the curtain is offline, do you operate the button on the curtain, or wait for a while, but you can't get back online?

I know this is old now, but mine would be very erratic in responding to commands with 1.00.72. Sometimes working, sometimes not. Like the other user, I reverted to 1.00.70 and everything has been working perfectly since.  I will not update to 1.00.72 and am concerned about even trying future updates.  I saw in your own forum that there seemed to be something broken in that update.

Aogu181, 2024-10-15:
@liubotongs How do I use a local ota index?

Aogu181, 2024-10-15:
@hwzolin Now you've solved it？

hwzolin, 2024-10-16:
> @hwzolin Now you've solved it？

Well... I remember that I deleted and reinstalled z2m and tried it several times before it succeeded.
But I'm not sure if this problem has been solved.
But the strange thing is that my OTA on ZHA has always been normal. So I don't know if this is a problem with z2m itself.

el-bakkali, 2024-10-30:
im getting the same issue

[image]
[image]

Zigbee2MQTT version
[1.40.2]

Is this a known issue bug ? if so how do we resolve it ?

sbourdette, 2024-11-12:
Same for me on Sonoff  TRVZB 
Unable to upgrade with message : "
Failed to check if update available for 'Radiateur Cuisine' (Cannot destructure property 'data' of '(intermediate value)' as it is undefined.)"
[image]

jbgachot, 2024-11-18:
Don't know about the "concat" related issues, but for those with the
`Cannot destructure property 'data' of '(intermediate value)' as it is undefined.`
Are you running in Docker ?
Check internet connection of the Zigbee2MQTT's process. Or from the container? Are you running your stack with an "internal" network stack?

@hwzolin @Aogu181 @sbourdette

jbgachot, 2024-11-21:
Have you already tried to test your internet connection from inside the container ?
Something like this :
`docker compose exec -ti zigbee2mqtt /bin/sh`
then in the container
`wget https://raw.githubusercontent.com/Koenkk/zigbee-OTA/refs/heads/master/index.json -O /dev/null`
Are you getting an output similar to this ?
```
Connecting to raw.githubusercontent.com (185.199.111.133:443)
saving to '/dev/null'
null                 100% |**************************************************|  414k  0:00:00 ETA
'/dev/null' saved
```

stonie81, 2024-11-21:
Thanks @jbgachot for pointing me in the right direction.
I set up pi-hole some days ago and after that hit me and off course the failed DNS lookup inside the container..
My solution is
```dockerfile
  pihole:
    container_name: pihole
    image: pihole/pihole:latest
    ports:
      - "80:80/tcp"
      - "<IP of the docker host>53:53/tcp"
      - "<IP of the docker host>:53:53/udp"
      - "67:67/udp"
``` 
Thanks for the help! :)

closed 2025-01-29 by github-actions[bot]